stevebuk1/pv_opt #48: Support for DLS-W v2.2 Data Logger / Solarman

huwpower, 2023-12-13:
I have realised that the reason I cannot successfully install PV Opt is because my data logger is incompatible with Solax Modbus. I have a DSL-W v2.2 firmware MW3_15_0501_1.24 which works with the Solarman integration. Is it possible to incorporate support for my datalogger into your project?

fboundy, 2023-12-13:
Hi Huw

I’ve added a template to Issues which lists the info I need to be able to add additional inverters.

The first part should be straightforward: it’s just the various HA entities that are needed to do the calcs. If you can fill that in we should be able to get you running Read Only.

The second part is a bit more vague as every inverter/integration requires different control.

Since you’re running a Solis inverter I know what it can do and which registers need setting. I just need to understand what manual steps you would take to set a timed charge or discharge slot and then I should be able to get something up and running.

Cheers

Francis

huwpower, 2023-12-14:
Yes it is

fboundy, 2023-12-14:
OK - that looks pretty straightforward as it exposes a service to write to the Modbus registers. Should have something for you to try out soon.

fboundy, 2023-12-14:
Can you confirm that you have the following sensors? If their names are slightly different can you please correct them for me? They may have `solis` in the entity_id?

`sensor.battery_voltage`
`sensor.battery_soc`
`sensor.daily_energy_imported`
`sensor.daily_energy_exported`
`sensor.house_load_power`
`sensor.backup_load_power`
`sensor.daily_house_backup_load_consumption`

fboundy, 2023-12-14:
Can you also confirm if you have an entity called something like `storage_control_mode` and if, as I suspect, it simply reports a string such as `Self Use` or `Time of Use`

huwpower, 2023-12-14:
sensor.solis_battery_voltage
sensor.solis_battery_soc
sensor.solis_daily_energy_imported
sensor.solis_daily_energy_exported
sensor.solis_house_load_power
sensor.solis_backup_load_power
sensor.solis_daily_house_backup_load_consumption

sensor.solis_storage_control_mode 

Note the "solis" is actually "Solarman" by default but you can choose to change it.

huwpower, 2023-12-14:
"Time of Use"

fboundy, 2023-12-15:
OK - Have a look at v3.3.0

You'll need to enable beta releases in HACS to see it. 

It should work in Read Only mode if I've got the entities right. It won't write to the inverter yet but it will go through the process and write to the log file what it would have written. I think that's safest until we are happy it's working correctly!

One thing I didn't ask is what tariff you are on. I'm seeing some errors in Eco7 which I haven't unravelled yet but Agile and Flux should be OK.

Let me know how you get on

closed 2024-01-30 by fboundy