stevebuk1/pv_opt #52: Flux charging periods seem to be ignored in favour of peak rate charging

ThomasPrior, 2023-12-14:
**Describe the bug**

Octopus flux customer, cheap rates between 02:00 and 05:00 daily.

PV_OPT seems to be wanting to charge 3500W between 00:00 and 00:30 at a higher rate, rather than wait for the cheap period at 02:00

**Expected behavior**
I would have thought the optimum approach on Flux to be to consume from the grid when the battery is discharged until the cheaper period starts at 02:00.

**Screenshots**

[image]

**Additional context**
Logs: 
[pv_opt.log](https://github.com/fboundy/pv_opt/files/13675172/pv_opt.log)

pv_opt config:

```
# Internal configuration --- DO NOT EDIT ---
pvpy:
  module: pvpy
  global: true

inverters:
  module: inverters
  global: true

pv_opt:
  module: pv_opt
  class: PVOpt
  log: pv_opt_log
  prefix: pvopt
  inverter_type: "SOLIS_SOLAX_MODBUS"
  # inverter_type: "SOLIS_CORE_MODBUS"

  # User configuration ---  EDIT AWAY! ---

  # ========================================
  # Plant parameters
  # ========================================

  # All parameters can be a number or point to an entity UNLESS they start with 'id_'. All of
  # these are required but will be defaulted if not specified

  battery_capacity_Wh: 9400
  # inverter_efficiency_percent: 97 # Default: 97
  # charger_efficiency_percent: 91 # Default: 91
  maximum_dod_percent: 12
  charger_power_watts: 3500
  inverter_power_watts: 5000
  # inverter_loss_watts: 100
  battery_voltage: sensor.solis_battery_voltage
  
  # ========================================
  # Solcast configuration
  # ========================================
  #
  # id_solcast_today: sensor.solcast_pv_forecast_forecast_today
  # id_solcast_tomorrow: sensor.solcast_pv_forecast_forecast_tomorrow

  # ========================================
  # Solar and consumption forecast parameters
  # ========================================
  #
  # Valid options are:
  #
  # Solcast         - the Solcast mid-case forecast [Default]
  # Solcast_p90     - the Solcast high estimate
  # Solcast_p10     - the Solcast high estimate
  #
  # Set this using an "input_select" helper and it can be varied on the fly

  # solar_forecast:
  # - input_select.solar_forecast_source
  #   - Solcast

  # # consumption estimation
  # consumption_history_days:
  #   - 7
  # consumption_margin:
  #   - input_number.solar_opt_consumption_margin
  #  
  # ========================================
  # Octopus account parameters
  # ========================================

  octopus_auto: True # Read tariffs from the Octopus Energy integration. If successful this over-rides the following parameters

  octopus_account: !secret octopus_account
  octopus_api_key: !secret octopus_api_key

  # The following Can be omitted if either of the above options is working correctly:

  octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-A
  octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-A

  # octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
  # octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G
  
  # ===============================================================================================================
  # Brand / Integration Specific Config: SOLIS_SOLAX_MODBUS: https://github.com/wills106/homeassistant-solax-modbus
  # ===============================================================================================================
  #
  # These are the default entities used with the Solis Solax Modbus integration. You can change them here and over-ride the defaults

  battery_voltage: sensor.solis_battery_voltage

  id_consumption:
    - sensor.solis_house_load
    - sensor.solis_bypass_load

  id_battery_soc: sensor.solis_battery_soc
  id_timed_charge_start_hours: number.solis_timed_charge_start_hours
  id_timed_charge_start_minutes: number.solis_timed_charge_start_minutes
  id_timed_charge_end_hours: number.solis_timed_charge_end_hours
  id_timed_charge_end_minutes: number.solis_timed_charge_end_minutes
  id_timed_charge_current: number.solis_timed_charge_current

  id_timed_discharge_start_hours: number.solis_timed_discharge_start_hours
  id_timed_discharge_start_minutes: number.solis_timed_discharge_start_minutes
  id_timed_discharge_end_hours: number.solis_timed_discharge_end_hours
  id_timed_discharge_end_minutes: number.solis_timed_discharge_end_minutes
  id_timed_discharge_current: number.solis_timed_discharge_current

  id_timed_charge_discharge_button: button.solis_update_charge_discharge_times
  id_inverter_mode: select.solis_energy_storage_control_switch

  # ==============================================================================================================
  # Brand / Integration Specific Config: SOLIS_CORE_MODBUS: https://github.com/fboundy/ha_solis_modbus
  # ==============================================================================================================
  #
  # These are the default entities used with the Solis Core Modbus integration. You can change them here and over-ride the defaults

  # modbus_hub: solis
  # modbus_slave: 1
  # battery_voltage: sensor.solis_battery_voltage
  # maximum_dod_percent: sensor.solis_overdischarge_soc
  #
  # id_consumption:
  #   - sensor.solis_house_load_power
  #   - ensor.solis_backup_load_power

  # id_grid_power: sensor.solis_grid_active_power
  # id_inverter_ac_power": sensor.solis_inverter_ac_power
  # id_battery_soc: sensor.solis_battery_soc

  # id_timed_charge_start_hours: sensor.solis_timed_charge_start_hour
  # id_timed_charge_start_minutes: sensor.solis_timed_charge_start_minute
  # id_timed_charge_end_hours: sensor.solis_timed_charge_end_hour
  # id_timed_charge_end_minutes: sensor.solis_timed_charge_end_minute
  # id_timed_charge_current: sensor.solis_timed_charge_current_limit

  # id_timed_discharge_start_hours: sensor.solis_timed_discharge_start_hour
  # id_timed_discharge_start_minutes: sensor.solis_timed_discharge_start_minute
  # id_timed_discharge_end_hours: sensor.solis_timed_discharge_end_hour
  # id_timed_discharge_end_minutes: sensor.solis_timed_discharge_end_minute
  # id_timed_discharge_current: sensor.solis_timed_discharge_current_limit

  # id_timed_charge_discharge_button: button.solis_update_charge_discharge_times
  # id_inverter_mode: select.solis_energy_storage_control_switch

```

fboundy, 2023-12-14:
I found a bug relating to how flux prices were being loaded which is fixed in v 3.2.2

Your price chart should look like this with the Flux tariffs continuing into tomorrow: 
[image]

Just releasing 3.2.2 now so let me know if it makes more sense.

ThomasPrior, 2023-12-14:
Upgaded, I now see the 02:00 to 05:00 period reflected in prediction, thank you!:

[image]

After disabling read-only mode and updating the inverter, I see charge time slot 1 is filled with a period of 16:22 to 16:22. Is this expected?

[image]

fboundy, 2023-12-14:
Yes as charging is currently disabled to allow for discharging during the Saving Session. It should switch back on again 10 minuted before the 02:00 session starts. I try to minimise writing to the inverter until its really needed as the EEPROMs have a limited number of write cycles (apparently)

fboundy, 2023-12-14:
Also good to see that it is maximising the discharge for the Saving session at the cost of less discharge outside of that hour

fboundy, 2023-12-14:
Though for tomorrow it is discharging in two slots at 5000w when it might make more sense to do a continuous 3 hours slot at a lower rate. I haven't played around with the discharge logic very much since I went onto Agile from Flux but I can simulate Flux to see if what it is doing makes sense.

ThomasPrior, 2023-12-15:
The curse of the broken up charging continues! :)

Looks like you're addressing this in #61 but I wanted to report it anyway as extra evidence usually helps.

[image]
[pv_opt.log](https://github.com/fboundy/pv_opt/files/13685377/pv_opt.log)

fboundy, 2023-12-15:
Yes - that's helpful as Flux has different optimisation that Agile. I think 100% SOC should be maintained OK with the current setup but using Backup SOC at 100% until 15:00 would be more elegant. Looks like you are using the Solax integration which should be quite easy to implement it with.

ThomasPrior, 2023-12-16:
Installed 3.3.0 to evaluate, things seem to go a little sideways again with the charging periods.

I didn't grab the logs before downgrading to 3.2.2 so can't give you anything solid to work from. I'll swap back to 3.3.0 and try to replicate tomorrow.

[Screenshot_20231216-022748_Home_Assistant.jpg](https://github.com/fboundy/pv_opt/assets/34111848/384b0936-7850-41ca-a258-e0ab35a84733)

fboundy, 2023-12-16:
Yes - 3.3.0 may have a few bugs in it still.

fboundy, 2023-12-17:
Can you try 3.3.1 and see how you get on?

ThomasPrior, 2023-12-17:
Upgraded to 3.3.1, the charging behaviour has changed for the worse. The only calculated charge period starts at 02:00 as expected with Flux, but ends 30 minutes early at 04:30 with the battery targetting 83% SOC instead of 100%.

Screenshot and logs attached. Expecting a slow week at work this week so will have some more time to test.

[image]
[pv_opt.txt](https://github.com/fboundy/pv_opt/files/13697974/pv_opt.txt)

fboundy, 2023-12-17:
Thanks - can you try playing with the two threshold settings? Dropping them to zero may get you a marginally lower net cost.

It may also change after midnight when it has the next day to look at as well. I’ll add another plot to the dashboard for the next release which shows the forecasts from 12, 8, 4 and 1 hour ahead and is a good QC (shamelessly nicked from PredBat…)

ThomasPrior, 2023-12-18:
I've tried a clean install of 3.3.1 to try to address some issues I've been experiencing there. Unfortunately the functionality has regressed quite some way.

- Optimise Charging, Discharging, Cyclic Charge/Discharge toggles no longer work. I can press them, but they revert to their previous state immediately.
- Pvopt Optimise Frequency Minutes, both Threshold sliders, both Load sliders and Inverter power sliders do not change their values any more and revert to their previous state if changed.
- Charging slots are back to not utilising the full charging period and are not filling the battery.
[pv_opt.txt](https://github.com/fboundy/pv_opt/files/13705519/pv_opt.txt)

If there are any 

[image]

ThomasPrior, 2023-12-18:
Some more entries that might be of interest:

from appdaemon `error.log`:

```
23:20:55 WARNING pv_opt: ------------------------------------------------------------
23:25:28 WARNING pv_opt: ------------------------------------------------------------
23:25:28 WARNING pv_opt: Unexpected error in worker for App pv_opt:
23:25:28 WARNING pv_opt: Worker Ags: {'id': '04ee31e5af714d2ca8f8c419b483c67c', 'name': 'pv_opt', 'objectid': 'de7db8edc21847b6b005873c09ac3a3c', 'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of <pv_opt.PVOpt object at 0x7fd4738e0370>>, 'pin_app': True, 'pin_thread': 1, 'kwargs': {'interval': 300.0, '__thread_id': 'thread-1'}}
23:25:28 WARNING pv_opt: ------------------------------------------------------------
23:25:28 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/pv_opt/pv_opt.py", line 1120, in optimise_time
    self.optimise()
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/pv_opt/pv_opt.py", line 1366, in optimise
    self.inverter.control_discharge(enable=False)
  File "/conf/apps/pv_opt/inverters.py", line 211, in control_discharge
    self.enable_timed_mode()
  File "/conf/apps/pv_opt/inverters.py", line 204, in enable_timed_mode
    self._solis_set_mode_switch(SelfUse=True, Timed=True, GridCharge=True)
  File "/conf/apps/pv_opt/inverters.py", line 389, in _solis_set_mode_switch
    status = self._solis_solax_mode_switch()
AttributeError: 'InverterController' object has no attribute '_solis_solax_mode_switch'

23:25:28 WARNING pv_opt: ------------------------------------------------------------
23:25:57 WARNING pv_opt: ------------------------------------------------------------
23:25:57 WARNING pv_opt: Unexpected error in worker for App pv_opt:
23:25:57 WARNING pv_opt: Worker Ags: {'id': '172b593b33d34adda694b28b15d1fef9', 'name': 'pv_opt', 'objectid': 'de7db8edc21847b6b005873c09ac3a3c', 'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of <pv_opt.PVOpt object at 0x7fd4738e0370>>, 'pin_app': True, 'pin_thread': 1, 'kwargs': {'interval': 300.0, '__thread_id': 'thread-1'}}
23:25:57 WARNING pv_opt: ------------------------------------------------------------
23:25:57 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/pv_opt/pv_opt.py", line 1120, in optimise_time
    self.optimise()
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/pv_opt/pv_opt.py", line 1366, in optimise
    self.inverter.control_discharge(enable=False)
  File "/conf/apps/pv_opt/inverters.py", line 211, in control_discharge
    self.enable_timed_mode()
  File "/conf/apps/pv_opt/inverters.py", line 204, in enable_timed_mode
    self._solis_set_mode_switch(SelfUse=True, Timed=True, GridCharge=True)
  File "/conf/apps/pv_opt/inverters.py", line 389, in _solis_set_mode_switch
    status = self._solis_solax_mode_switch()
AttributeError: 'InverterController' object has no attribute '_solis_solax_mode_switch'

```

From `main.log`:

```
15:09:30 INFO AppDaemon: /conf/apps/pv_opt/config/config.yaml added or modified
15:09:30 WARNING AppDaemon: App 'pvpy' missing 'class' or 'module' entry - ignoring
15:09:30 WARNING AppDaemon: App 'inverters' missing 'class' or 'module' entry - ignoring
15:09:30 INFO AppDaemon: App 'pv_opt' added
15:09:30 INFO AppDaemon: Found 1 active apps
15:09:30 INFO AppDaemon: Found 0 inactive apps
15:09:30 INFO AppDaemon: Found 2 global libraries
15:09:30 INFO AppDaemon: Adding thread 0
15:09:30 INFO AppDaemon: Adding /conf/apps/pv_opt to module import path
15:09:30 INFO AppDaemon: Adding /conf/apps/pv_opt/config to module import path
15:09:30 WARNING AppDaemon: No app description found for: /conf/apps/pv_opt/solis.py - ignoring
15:09:30 INFO AppDaemon: Loading Global Module: /conf/apps/pv_opt/pvpy.py
[... 102 lines cut ...]
15:33:12 INFO AppDaemon: Running on port 5050
15:33:12 INFO HASS: Evaluating startup conditions
15:33:12 INFO HASS: Startup condition met: hass state=RUNNING
15:33:12 INFO HASS: All startup conditions met
15:33:13 INFO AppDaemon: Got initial state from namespace default
15:33:13 INFO AppDaemon: New client Admin Client connected
15:33:14 INFO AppDaemon: Scheduler running in realtime
15:33:14 INFO AppDaemon: Adding /conf/apps to module import path
15:33:14 INFO AppDaemon: Adding /conf/apps/pv_opt to module import path
15:33:14 INFO AppDaemon: Adding /conf/apps/pv_opt/config to module import path
15:33:14 WARNING AppDaemon: No app description found for: /conf/apps/pv_opt/solis.py - ignoring
15:33:14 INFO AppDaemon: Loading Global Module: /conf/apps/pv_opt/pvpy.py
15:33:15 INFO AppDaemon: Loading App Module: /conf/apps/pv_opt/pv_opt.py
15:33:15 WARNING AppDaemon: Unable to find app solis in dependencies for pv_opt
15:33:15 WARNING AppDaemon: Ignoring app pv_opt
[... cut ...] rom module pv_opt
15:33:15 INFO AppDaemon: Calling initialize() for pv_opt
15:33:24 INFO AppDaemon: Client disconnection from Admin Client
15:33:24 INFO AppDaemon: New client Admin Client connected
15:33:27 INFO AppDaemon: Client disconnection from Admin Client
15:33:28 INFO AppDaemon: New client Admin Client connected
15:33:29 INFO AppDaemon: Client disconnection from Admin Client
15:33:29 INFO AppDaemon: New client Admin Client connected
15:33:39 INFO AppDaemon: App initialization complete
```

fboundy, 2023-12-18:
This is a bug I spotted late last night and will be fixed in 3.3.2

fboundy, 2023-12-18:
The first two bullets are related to way the MQTT entities are initialised. I thought this had been fixed in 3.3.1 but I have noticed that sometimes AppDaemon created duplicate entities. Can you please check whether you have any such as `number_pvopt_xxx_2` and also whether they are shown as MQTT or just Number entities.

[cid:8a41b290-c795-4ec1-bf67-4f75b877ef43]

image.png (view on web)<https://github.com/fboundy/pv_opt/assets/34111848/6791e80a-590c-4d43-a329-5d76a5d7029b>

ThomasPrior, 2023-12-18:
Thanks for the pointers regarding MQTT - manually cleared the entries, removed the appdaemon directory and started again.

Unfortunately I'm not able to get this version to load. Log files are attached.

[error.log](https://github.com/fboundy/pv_opt/files/13709096/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/13709097/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/13709098/pv_opt.log)

ThomasPrior, 2023-12-18:
The up side is that 3.2.2 still works, albeit not perfectly. I appreciate the fast turnaround on each update and I want to say that I appreciate the effort.

fboundy, 2023-12-18:
Ok - so I think I can see what’s happening. It’s got into a bit of a loop because all the config entities are changing state and each one triggers a restart.

Just needs a bit of logic to pause the state change handler until all the config items have been reset to default.

closed 2023-12-27 by fboundy