stevebuk1/pv_opt #70: Syntax error reported in pv_opt.py

solarjones, 2023-12-20:
**Describe the bug**
Installation issue -  syntax error reported in pv_opt.py in appdaemon

**To Reproduce**
 File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1693
    "device_class": "current",                elif status["hold_soc"]["active"]:
                                              ^^^^
SyntaxError: invalid syntax

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Appdaemon log:

**s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[22:29:42] INFO: Starting AppDaemon...
2023-12-20 22:29:59.139450 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2023-12-20 22:29:59.151233 INFO AppDaemon: Python version is 3.11.6
2023-12-20 22:29:59.151946 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2023-12-20 22:29:59.158058 INFO AppDaemon: Added log: AppDaemon
2023-12-20 22:29:59.162242 INFO AppDaemon: Added log: Error
2023-12-20 22:29:59.162727 INFO AppDaemon: Added log: Access
2023-12-20 22:29:59.163257 INFO AppDaemon: Added log: Diag
2023-12-20 22:30:00.175148 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2023-12-20 22:30:01.943385 INFO HASS: HASS Plugin Initializing
2023-12-20 22:30:01.945784 WARNING HASS: ha_url not found in HASS configuration - module not initialized
2023-12-20 22:30:01.946719 INFO HASS: HASS Plugin initialization complete
2023-12-20 22:30:01.947356 INFO AppDaemon: Loading Plugin MQTT using class MqttPlugin from module mqttplugin
2023-12-20 22:30:02.460542 INFO MQTT: MQTT Plugin Initializing
2023-12-20 22:30:02.465088 INFO MQTT: Using 'localad/status' as Will Topic
2023-12-20 22:30:02.465522 INFO MQTT: Using 'localad/status' as Birth Topic
2023-12-20 22:30:02.466390 INFO AppDaemon: HTTP is disabled
2023-12-20 22:30:02.574689 INFO HASS: Connected to Home Assistant 2023.12.3
2023-12-20 22:30:02.815610 INFO MQTT: Connected to Broker at URL core-mosquitto:1883
2023-12-20 22:30:02.833793 INFO AppDaemon: App 'pv_opt' added
2023-12-20 22:30:02.852074 INFO AppDaemon: Found 1 active apps
2023-12-20 22:30:02.854388 INFO AppDaemon: Found 0 inactive apps
2023-12-20 22:30:02.855243 INFO AppDaemon: Found 2 global libraries
2023-12-20 22:30:02.855941 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2023-12-20 22:30:02.863263 INFO AppDaemon: Got initial state from namespace mqtt
2023-12-20 22:30:02.864894 INFO MQTT: MQTT Plugin initialization complete
2023-12-20 22:30:03.033157 INFO HASS: Evaluating startup conditions
2023-12-20 22:30:03.093516 INFO HASS: Startup condition met: hass state=RUNNING
2023-12-20 22:30:03.093816 INFO HASS: All startup conditions met
2023-12-20 22:30:08.396670 INFO AppDaemon: Got initial state from namespace default
2023-12-20 22:30:09.906098 INFO AppDaemon: Scheduler running in realtime
2023-12-20 22:30:09.915360 INFO AppDaemon: Adding /homeassistant/appdaemon/apps to module import path
2023-12-20 22:30:09.926168 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pv_opt to module import path
2023-12-20 22:30:09.934003 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pv_opt/config to module import path
2023-12-20 22:30:09.948777 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/pv_opt/solis.py - ignoring
2023-12-20 22:30:09.954318 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py
2023-12-20 22:30:10.164543 WARNING Error: ------------------------------------------------------------
2023-12-20 22:30:10.175562 WARNING Error: Unexpected error loading module: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py:
2023-12-20 22:30:10.178896 WARNING Error: ------------------------------------------------------------
2023-12-20 22:30:10.204897 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1693
    "device_class": "current",                elif status["hold_soc"]["active"]:
                                              ^^^^
SyntaxError: invalid syntax

2023-12-20 22:30:10.222560 WARNING Error: ------------------------------------------------------------
2023-12-20 22:30:10.227979 WARNING AppDaemon: Removing associated apps:
2023-12-20 22:30:10.231052 WARNING AppDaemon: pv_opt
2023-12-20 22:30:10.239272 INFO AppDaemon: Loading Global Module: /homeassistant/appdaemon/apps/pv_opt/pvpy.py
2023-12-20 22:30:16.837590 WARNING AppDaemon: Unable to find app solis in dependencies for pv_opt
2023-12-20 22:30:16.844103 WARNING AppDaemon: Ignoring app pv_opt
2023-12-20 22:30:16.852981 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
2023-12-20 22:30:16.982913 WARNING pv_opt: ------------------------------------------------------------
2023-12-20 22:30:16.984007 WARNING pv_opt: Unexpected error initializing app: pv_opt:
2023-12-20 22:30:16.988236 WARNING pv_opt: ------------------------------------------------------------
2023-12-20 22:30:16.991485 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1035, in check_app_updates
    await self.init_object(app)
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 323, in init_object
    modname = await utils.run_in_executor(self, __import__, app_args["module"])
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1693
    "device_class": "current",                elif status["hold_soc"]["active"]:
                                              ^^^^
SyntaxError: invalid syntax

2023-12-20 22:30:16.999102 WARNING pv_opt: ------------------------------------------------------------
2023-12-20 22:30:17.003469 WARNING AppDaemon: Unable to find module pv_opt - initialize() skipped
2023-12-20 22:30:17.005212 INFO AppDaemon: App initialization complete
2023-12-20 23:20:14.836077 WARNING AppDaemon: Excessive time spent in utility loop: 2386.0ms, 5.0ms in check_app_updates(), 2381.0ms in other**

**Additional context**
It may be user error on my part attempting to set this up, so any help would be greatly appreciated.

fboundy, 2023-12-21:
This is a weird one. Spurious code had got duplicated out of place - probably git related. Fixed in 3.4.1

solarjones, 2023-12-21:
Thanks for sorting that out so fast! I removed the automation and reinstalled/downloaded the latest version 3.4.3, restarted AppDaemon and get a different error now....

2023-12-21 12:19:21.675067 INFO pv_opt: Checking config:
2023-12-21 12:19:21.689109 INFO pv_opt: -----------------------
2023-12-21 12:19:21.705747 WARNING pv_opt:     forced_charge                      = True                                                        True: system default. Not in YAML.
2023-12-21 12:19:21.717969 WARNING pv_opt:     forced_discharge                   = True                                                        True: system default. Not in YAML.
2023-12-21 12:19:21.727570 WARNING pv_opt:     read_only                          = True                                                        True: system default. Not in YAML.
2023-12-21 12:19:21.740293 WARNING pv_opt:     allow_cyclic                       = False                                                      False: system default. Not in YAML.
2023-12-21 12:19:21.753859 WARNING pv_opt:     optimise_frequency_minutes         = 10                                                            10: system default. Not in YAML.
2023-12-21 12:19:21.770879 WARNING pv_opt:     slot_threshold_p                   = 1.0                                                          1.0: system default. Not in YAML.
2023-12-21 12:19:21.783239 WARNING pv_opt:     day_of_week_weighting              = 0.5                                                          0.5: system default. Not in YAML.
2023-12-21 12:19:21.788466 WARNING pv_opt:     pass_threshold_p                   = 4.0                                                          4.0: system default. Not in YAML.
2023-12-21 12:19:21.800459 WARNING pv_opt:     octopus_auto                       = True                                                        True: system default. Not in YAML.
2023-12-21 12:19:21.807217 WARNING pv_opt:     battery_capacity_wh                = 10000                                                      10000: system default. Not in YAML.
2023-12-21 12:19:21.812659 WARNING pv_opt:     inverter_efficiency_percent        = 97                                                            97: system default. Not in YAML.
2023-12-21 12:19:21.820220 WARNING pv_opt:     charger_efficiency_percent         = 91                                                            91: system default. Not in YAML.
2023-12-21 12:19:21.828807 WARNING pv_opt:     charger_power_watts                = 3000                                                        3000: system default. Not in YAML.
2023-12-21 12:19:21.834249 WARNING pv_opt:     inverter_power_watts               = 3600                                                        3600: system default. Not in YAML.
2023-12-21 12:19:21.844975 WARNING pv_opt:     inverter_loss_watts                = 100                                                          100: system default. Not in YAML.
2023-12-21 12:19:21.864409 WARNING pv_opt:     solar_forecast                     = Solcast                                                   Solcast: system default. Not in YAML.
2023-12-21 12:19:21.882539 WARNING pv_opt:     id_solcast_today                   = sensor.solcast_pv_forecast_forecast_today                 2.36665: system default. Not in YAML.
2023-12-21 12:19:21.898519 WARNING pv_opt:     id_solcast_tomorrow                = sensor.solcast_pv_forecast_forecast_tomorrow              1.23335: system default. Not in YAML.
2023-12-21 12:19:21.904050 WARNING pv_opt:     consumption_history_days           = 7                                                              7: system default. Not in YAML.
2023-12-21 12:19:21.908951 WARNING pv_opt:     consumption_margin                 = 10                                                            10: system default. Not in YAML.
2023-12-21 12:19:21.914868 WARNING pv_opt:     consumption_grouping               = mean                                                        mean: system default. Not in YAML.
2023-12-21 12:19:21.921318 WARNING pv_opt:     forced_power_group_tolerance       = 100                                                          100: system default. Not in YAML.
2023-12-21 12:19:21.933773 WARNING pv_opt:     maximum_dod_percent                = number.solis_battery_minimum_soc                            15.0: system default. Not in YAML.
2023-12-21 12:19:21.942235 WARNING pv_opt:     id_battery_soc                     = sensor.solis_battery_soc                                    88.0: system default. Not in YAML.
2023-12-21 12:19:21.955034 WARNING pv_opt:     id_consumption                     = ['sensor.solis_house_load', 'sensor.solis_bypass_load']    800.0: system default. Not in YAML.
2023-12-21 12:19:21.967882 WARNING pv_opt:     id_consumption_today               = sensor.solis_house_load_today                               14.3: system default. Not in YAML.
2023-12-21 12:19:21.974392 WARNING pv_opt:     id_grid_import_today               = sensor.solis_grid_import_today                              20.4: system default. Not in YAML.
2023-12-21 12:19:21.987331 WARNING pv_opt:     id_grid_export_today               = sensor.solis_grid_export_today                               0.0: system default. Not in YAML.
2023-12-21 12:19:22.003585 WARNING pv_opt:     id_grid_import_power               = sensor.solis_grid_import_power                              40.0: system default. Not in YAML.
[... cut ...]
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 320, in _estimate_capacity
    self.hass2df(entity_id=self.config["id_battery_soc"], days=7)
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 235, in hass2df
    df = pd.DataFrame(hist[0]).set_index("last_updated")["state"]
                      ~~~~^^^
TypeError: 'NoneType' object is not subscriptable

2023-12-21 12:19:41.720604 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 12:19:41.728708 INFO AppDaemon: App initialization complete

Thanks again for your help.

solarjones, 2023-12-21:
Very odd, after a restart of HA,  I get different errors...

2023-12-21 13:54:26.517059 INFO pv_opt: Checking config:
2023-12-21 13:54:26.530029 INFO pv_opt: -----------------------
2023-12-21 13:54:26.542731 WARNING pv_opt:     forced_charge                      = True                                                        True: system default. Not in YAML.
2023-12-21 13:54:26.559050 WARNING pv_opt:     forced_discharge                   = True                                                        True: system default. Not in YAML.
2023-12-21 13:54:26.563911 WARNING pv_opt:     read_only                          = True                                                        True: system default. Not in YAML.
2023-12-21 13:54:26.571249 WARNING pv_opt:     allow_cyclic                       = False                                                      False: system default. Not in YAML.
2023-12-21 13:54:26.581179 WARNING pv_opt:     optimise_frequency_minutes         = 10                                                            10: system default. Not in YAML.
2023-12-21 13:54:26.591444 WARNING pv_opt:     slot_threshold_p                   = 1.0                                                          1.0: system default. Not in YAML.
2023-12-21 13:54:26.595978 WARNING pv_opt:     day_of_week_weighting              = 0.5                                                          0.5: system default. Not in YAML.
2023-12-21 13:54:26.607846 WARNING pv_opt:     pass_threshold_p                   = 4.0                                                          4.0: system default. Not in YAML.
2023-12-21 13:54:26.614109 WARNING pv_opt:     octopus_auto                       = True                                                        True: system default. Not in YAML.
2023-12-21 13:54:26.625275 WARNING pv_opt:     battery_capacity_wh                = 10000                                                      10000: system default. Not in YAML.
2023-12-21 13:54:26.632352 WARNING pv_opt:     inverter_efficiency_percent        = 97                                                            97: system default. Not in YAML.
2023-12-21 13:54:26.639833 WARNING pv_opt:     charger_efficiency_percent         = 91                                                            91: system default. Not in YAML.
2023-12-21 13:54:26.649228 WARNING pv_opt:     charger_power_watts                = 3000                                                        3000: system default. Not in YAML.
2023-12-21 13:54:26.656819 WARNING pv_opt:     inverter_power_watts               = 3600                                                        3600: system default. Not in YAML.
2023-12-21 13:54:26.671211 WARNING pv_opt:     inverter_loss_watts                = 100                                                          100: system default. Not in YAML.
2023-12-21 13:54:26.695111 WARNING pv_opt:     solar_forecast                     = Solcast                                                   Solcast: system default. Not in YAML.
2023-12-21 13:54:26.714577 WARNING pv_opt:     id_solcast_today                   = sensor.solcast_pv_forecast_forecast_today                 2.1508: system default. Not in YAML.
2023-12-21 13:54:26.736303 WARNING pv_opt:     id_solcast_tomorrow                = sensor.solcast_pv_forecast_forecast_tomorrow              1.22705: system default. Not in YAML.
2023-12-21 13:54:26.753196 WARNING pv_opt:     consumption_history_days           = 7                                                              7: system default. Not in YAML.
2023-12-21 13:54:26.765889 WARNING pv_opt:     consumption_margin                 = 10                                                            10: system default. Not in YAML.
2023-12-21 13:54:26.784740 WARNING pv_opt:     consumption_grouping               = mean                                                        mean: system default. Not in YAML.
2023-12-21 13:54:26.806815 WARNING pv_opt:     forced_power_group_tolerance       = 100                                                          100: system default. Not in YAML.
2023-12-21 13:54:26.817524 WARNING pv_opt:     maximum_dod_percent                = number.solis_battery_minimum_soc                            15.0: system default. Not in YAML.
2023-12-21 13:54:26.835908 WARNING pv_opt:     id_battery_soc                     = sensor.solis_battery_soc                                    88.0: system default. Not in YAML.
2023-12-21 13:54:26.850121 WARNING pv_opt:     id_consumption                     = ['sensor.solis_house_load', 'sensor.solis_bypass_load']    906.0: system default. Not in YAML.
2023-12-21 13:54:26.866967 WARNING pv_opt:     id_consumption_today               = sensor.solis_house_load_today                               16.1: system default. Not in YAML.
2023-12-21 13:54:26.876552 WARNING pv_opt:     id_grid_import_today               = sensor.solis_grid_import_today                              21.1: system default. Not in YAML.
2023-12-21 13:54:26.900706 WARNING pv_opt:     id_grid_export_today               = sensor.solis_grid_export_today                               0.0: system default. Not in YAML.
2023-12-21 13:54:26.918935 WARNING pv_opt:     id_grid_import_power               = sensor.solis_grid_import_power                             151.0: system default. Not in YAML.
[... cut ...]
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 307, in _estimate_capacity
    self.hass2df(
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 235, in hass2df
    df = pd.DataFrame(hist[0]).set_index("last_updated")["state"]
                      ~~~~^^^
TypeError: 'NoneType' object is not subscriptable

2023-12-21 13:54:37.600624 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 13:54:37.641892 WARNING AppDaemon: Excessive time spent in utility loop: 11369.0ms, 11369.0ms in check_app_updates(), 0.0ms in other

fboundy, 2023-12-21:
I've disabled the function that was causing this error as it's not essential. It looks like it wasn't getting the necessary data from HA so when I re-enable it I will add some logging and error tracing to test this.

solarjones, 2023-12-21:
Ok thanks, just loaded 3.4.4 and get the following...

...
2023-12-21 15:29:04.689729 INFO pv_opt: No upcoming Octopus Saving Events detected or joined:
2023-12-21 15:29:04.694286 INFO pv_opt: Starting Opimisation with discharge disabled
2023-12-21 15:29:04.701436 INFO pv_opt: ---------------------------------------------------
2023-12-21 15:29:04.899951 INFO pv_opt: Solcast forecast loaded OK
2023-12-21 15:29:04.905877 INFO pv_opt: Getting expected consumption data
2023-12-21 15:29:14.924988 WARNING pv_opt: Coroutine (<coroutine object Hass.get_history at 0x7fa048e60940>) took too long (10 seconds), cancelling the task...
2023-12-21 15:29:14.928274 ERROR pv_opt: Unable to get historical consumption from sensor.solis_house_load. 'NoneType' object is not subscriptable
2023-12-21 15:29:21.468465 INFO pv_opt: Initial SOC: 87.0
2023-12-21 15:29:21.476165 INFO pv_opt: Calculating Base flows
2023-12-21 15:29:21.486354 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 15:29:21.491351 WARNING pv_opt: Unexpected error running initialize() for pv_opt
2023-12-21 15:29:21.492425 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 15:29:21.547681 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 3791, in get_loc
    return self._engine.get_loc(casted_key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "index.pyx", line 152, in pandas._libs.index.IndexEngine.get_loc
  File "index.pyx", line 181, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 7080, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 7088, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'consumption'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 294, in initialize
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1282, in optimise
    self.base = self.pv_system.flows(
                ^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 458, in flows
    consumption = static_flows[cols["consumption"]]
                  ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/frame.py", line 3893, in __getitem__
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 3798, in get_loc
    raise KeyError(key) from err
KeyError: 'consumption'

2023-12-21 15:29:21.554135 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 15:29:21.556754 INFO AppDaemon: App initialization complete

fboundy, 2023-12-21:
So it looks like there is no historical data coming from sensor.solis_house_load. Is there data in that entity in Home Assistant for the last 7 days?

solarjones, 2023-12-21:
Yes, and checked the sensor and looks ok to me. 
[image]

solarjones, 2023-12-21:
Brilliant, and thanks again for your help!  

I can't seem to get AppDaemon logging to work at all. Added the config as per your instructions, created the share/log directories but still nothing. Also note when AppDaemon starts it doesn't show INFO AppDaemon: Added log: PV_Opt.  I know appdaemon.config is being read and using secrets ok. Checked permissions on the folders (root with drwxr-xr-x). Resorted to asking ChatGRP and everything seems correct, just not writing any files.

fboundy, 2023-12-21:
Hmm – hard to know exactly what’s happening there. Have you tried a complete restart of the Add-On and HA?

fboundy, 2023-12-21:
OK - try 3.4.5-alpha. You'll need to enable pre-releases in HACS to see it. I'm wondering if this and your logging issues might be related?

solarjones, 2023-12-21:
Loaded 3.4.5....  I will also reinstall AppDaemon & MQTT and try again later to see if that improves the situation.

023-12-21 20:11:44.719205 INFO pv_opt: Exposing config to Home Assistant:
2023-12-21 20:11:44.721499 INFO pv_opt: ----------------------------------
2023-12-21 20:11:44.821291 INFO pv_opt: 
2023-12-21 20:11:44.835804 INFO pv_opt:   Config Item                               HA Entity                                   Current State
2023-12-21 20:11:44.844503 INFO pv_opt:   -----------                               ---------                                   -------------
2023-12-21 20:11:44.853338 INFO pv_opt:   forced_charge                             switch.pvopt_forced_charge                  on
2023-12-21 20:11:44.867053 INFO pv_opt:   forced_discharge                          switch.pvopt_forced_discharge               off
2023-12-21 20:11:44.884608 INFO pv_opt:   read_only                                 switch.pvopt_read_only                      on
2023-12-21 20:11:44.896332 INFO pv_opt:   allow_cyclic                              switch.pvopt_allow_cyclic                   off
2023-12-21 20:11:44.909395 INFO pv_opt:   optimise_frequency_minutes                number.pvopt_optimise_frequency_minutes     10
2023-12-21 20:11:44.919161 INFO pv_opt:   slot_threshold_p                          number.pvopt_slot_threshold_p               1.0
2023-12-21 20:11:44.935128 INFO pv_opt:   day_of_week_weighting                     number.pvopt_day_of_week_weighting          0.5
2023-12-21 20:11:44.943968 INFO pv_opt:   pass_threshold_p                          number.pvopt_pass_threshold_p               4.0
2023-12-21 20:11:44.954876 INFO pv_opt:   battery_capacity_wh                       number.pvopt_battery_capacity_wh            10000
2023-12-21 20:11:44.961044 INFO pv_opt:   inverter_efficiency_percent               number.pvopt_inverter_efficiency_percent    97
2023-12-21 20:11:44.980092 INFO pv_opt:   charger_efficiency_percent                number.pvopt_charger_efficiency_percent     91
2023-12-21 20:11:44.992809 INFO pv_opt:   charger_power_watts                       number.pvopt_charger_power_watts            3000
2023-12-21 20:11:45.005762 INFO pv_opt:   inverter_power_watts                      number.pvopt_inverter_power_watts           3600
2023-12-21 20:11:45.017238 INFO pv_opt:   inverter_loss_watts                       number.pvopt_inverter_loss_watts            100
2023-12-21 20:11:45.030485 INFO pv_opt:   solar_forecast                            select.pvopt_solar_forecast                 Solcast
2023-12-21 20:11:45.043515 INFO pv_opt:   consumption_history_days                  number.pvopt_consumption_history_days       7
2023-12-21 20:11:45.054010 INFO pv_opt:   consumption_margin                        number.pvopt_consumption_margin             10
2023-12-21 20:11:45.066297 INFO pv_opt:   consumption_grouping                      select.pvopt_consumption_grouping           mean
2023-12-21 20:11:45.081905 INFO pv_opt:   forced_power_group_tolerance              number.pvopt_forced_power_group_tolerance   100
2023-12-21 20:11:55.097613 WARNING pv_opt: Coroutine (<coroutine object Hass.get_history at 0x7f23c50ee040>) took too long (10 seconds), cancelling the task...
2023-12-21 20:11:55.103111 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 20:11:55.104264 WARNING pv_opt: Unexpected error in worker for App pv_opt:
2023-12-21 20:11:55.104965 WARNING pv_opt: Worker Ags: {}
2023-12-21 20:11:55.107079 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 20:11:55.112034 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 273, in initialize
    self._estimate_capacity()
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 307, in _estimate_capacity
    self.hass2df(
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 235, in hass2df
    df = pd.DataFrame(hist[0]).set_index("last_updated")["state"]
                      ~~~~^^^
TypeError: 'NoneType' object is not subscriptable

2023-12-21 20:11:55.118805 WARNING pv_opt: ------------------------------------------------------------
2023-12-21 20:11:55.156563 WARNING AppDaemon: Excessive time spent in utility loop: 13552.0ms, 13551.0ms in check_app_updates(), 1.0ms in other

fboundy, 2023-12-21:
It’s timing out on the AppDaemon get_history() method so I think there’s some up with your install

solarjones, 2023-12-22:
Hello, I've spent hours trying to get AppDaemon to log, still not working.
Everything is upto date.

Reinstalled AppDaemon, MQTT - runs HelloWorld to screen ok, but refuses to log with appdaemon.yaml log parameters and no idea why. It is using appdaemon.yaml as the secrets.yaml is loaded.  I deinstalled PV_opt restarted multiple times, and ran HelloWorld app. This logs to file if I explicitly tell it in hello.py.  

secrets: /homeassistant/secrets.yaml
appdaemon:
  latitude: 1.0
  longitude: 1.0
  elevation: 10
  time_zone: Europe/London
  thread_duration_warning_threshold: 45
  app_dir: /homeassistant/appdaemon/apps
  plugins:
    HASS:
      type: hass
      app_init_delay: 20
    MQTT:
      type: mqtt
      namespace: mqtt
      verbose: True
      client_host: core-mosquitto
      client_port: 1883
      client_id: localad
      event_name: MQTT_MESSAGE
      client_topics: NONE
      client_user: !secret mqtt-username
      client_password: !secret mqtt-password
  logs:
    main_log:
      filename: /config/appdaemon/logs/main.log
      date_format: "%H:%M:%S"
    error_log:
      filename: /config/appdaemon/logs/error.log
      date_format: "%H:%M:%S"
    pv_opt_log:
      name: PV_Opt
      filename: /share/logs/pv_opt.log
      date_format: "%H:%M:%S"
      format: "{asctime} {levelname:>8s}: {message}"

  http:
    url: http://127.0.0.1:5050 
    admin: 
    api: 
    hadashboard: 
-----------------------------------------------------------------------------------

Installed PV_OPT 3.4.4 and then Alpha 3.4.5.  

Output from 3.4.4:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting
-----------------------------------------------------------

 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 11.2  (amd64 / qemux86-64)
 Home Assistant Core: 2023.12.3
 Home Assistant Supervisor: 2023.12.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
[... 211 lines cut ...]
2023-12-22 14:15:19.795064 WARNING pv_opt: Unexpected error initializing app: pv_opt:
2023-12-22 14:15:19.798180 WARNING pv_opt: ------------------------------------------------------------
2023-12-22 14:15:19.802105 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1035, in check_app_updates
    await self.init_object(app)
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 323, in init_object
    modname = await utils.run_in_executor(self, __import__, app_args["module"])
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 239
    self.log(f">>> {hist})
             ^
SyntaxError: unterminated string literal (detected at line 239)

2023-12-22 14:15:19.809198 WARNING pv_opt: ------------------------------------------------------------
2023-12-22 14:15:19.811399 INFO AppDaemon: Calling initialize() for hello_world
2023-12-22 14:15:20.226807 INFO hello_world: Hello from AppDaemon
2023-12-22 14:15:20.237127 INFO hello_world: You are now ready to run Apps!
2023-12-22 14:15:20.250206 WARNING AppDaemon: Unable to find module pv_opt - initialize() skipped
2023-12-22 14:15:20.252548 INFO AppDaemon: App initialization complete

fboundy, 2023-12-22:
> File "", line 1204, in _gcd_import
> File "", line 1176, in _find_and_load
> File "", line 1147, in _find_and_load_unlocked
> File "", line 690, in _load_unlocked
> File "", line 936, in exec_module
> File "", line 1074, in get_code
> File "", line 1004, in source_to_code
> File "", line 241, in _call_with_frames_removed

Ok - looks like there’s a typo in 3.4.5 alpha that I need to fix. Should have checked it! If you want to edit it manually you can just change line 239 in pv_opt.py to:
> self.log(f">>> {hist}”)

Ie close the double quotes before the )

I’ll get it fixed properly but that might be quicker.

fboundy, 2023-12-22:
Fixed in 3.4.5-beta.

solarjones, 2023-12-22:
ok, getting a bit further now...

.....
22T14:00:15.422263+00:00'}, {'entity_id': 'sensor.solis_battery_soc', 'state': '95', 'attributes': {'unit_of_measurement': '%', 'device_class': 'battery', 'friendly_name': 'Solis Battery SOC'}, 'last_changed': '2023-12-22T15:40:24.697935+00:00', 'last_updated': '2023-12-22T15:40:24.697935+00:00'}]]
2023-12-22 15:49:08.614070 INFO pv_opt: >>> last_updated
2023-12-21 15:49:04+00:00                87
2023-12-21 16:03:23.829776+00:00         86
2023-12-21 16:08:59.504421+00:00         85
2023-12-21 16:15:49.704407+00:00         84
2023-12-21 16:21:59.704229+00:00         83
                                     ...   
2023-12-22 13:16:26.427414+00:00         97
2023-12-22 13:24:21.887546+00:00         96
2023-12-22 13:59:56.604727+00:00    unknown
2023-12-22 14:00:15.422263+00:00         96
2023-12-22 15:40:24.697935+00:00         95
Name: state, Length: 218, dtype: object
2023-12-22 15:49:08.657432 INFO pv_opt: >>> last_updated
2023-12-21 15:49:04+00:00           87
2023-12-21 16:03:23.829776+00:00    86
2023-12-21 16:08:59.504421+00:00    85
2023-12-21 16:15:49.704407+00:00    84
2023-12-21 16:21:59.704229+00:00    83
                                    ..
2023-12-22 11:02:24.380767+00:00    97
2023-12-22 13:16:26.427414+00:00    97
2023-12-22 13:24:21.887546+00:00    96
2023-12-22 14:00:15.422263+00:00    96
2023-12-22 15:40:24.697935+00:00    95
Name: state, Length: 209, dtype: object
2023-12-22 15:49:08.717058 INFO pv_opt: Initial SOC: 95.5
2023-12-22 15:49:08.722782 INFO pv_opt: Calculating Base flows
2023-12-22 15:49:08.762175 WARNING pv_opt: ------------------------------------------------------------
2023-12-22 15:49:08.771139 WARNING pv_opt: Unexpected error running initialize() for pv_opt
2023-12-22 15:49:08.772163 WARNING pv_opt: ------------------------------------------------------------
2023-12-22 15:49:08.891102 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 3791, in get_loc
    return self._engine.get_loc(casted_key)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "index.pyx", line 152, in pandas._libs.index.IndexEngine.get_loc
  File "index.pyx", line 181, in pandas._libs.index.IndexEngine.get_loc
  File "pandas/_libs/hashtable_class_helper.pxi", line 7080, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas/_libs/hashtable_class_helper.pxi", line 7088, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'consumption'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 305, in initialize
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1293, in optimise
    self.base = self.pv_system.flows(
                ^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 458, in flows
    consumption = static_flows[cols["consumption"]]
                  ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/frame.py", line 3893, in __getitem__
    indexer = self.columns.get_loc(key)
              ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 3798, in get_loc
    raise KeyError(key) from err
KeyError: 'consumption'

2023-12-22 15:49:08.914489 WARNING pv_opt: ------------------------------------------------------------
2023-12-22 15:49:08.915360 INFO AppDaemon: App initialization complete

solarjones, 2023-12-22:
The 'unable to get historic data' message relating to solis_house_load again. It's definitely there...

[image]

fboundy, 2023-12-22:
It’s the “unknown” in the history that is screwing with it. Think I can fix that.

fboundy, 2023-12-22:
No it's not. That is in the SOC data which it gets succesfully. I'm wondering if there are just too many data points in your history. Can you try adding this to `config.yaml`  after `device_name: solis`:

    consumption_history_days: 2

I have another workaround which is to use the cumulative daily data instead of the instantaneous load. It's arguably a better approach anyway - I'll release another version using that but it will need a bit of testing first.

fboundy, 2023-12-22:
Try 3.4.5-beta-02 which loads consumption from the daily total not the instantaneous power

solarjones, 2023-12-22:
Brilliant, it's now working! Thankyou so much for this.  I've put it into Readonly mode to work out what wizzardy you have going on. Probably a mindset change on my behalf, as usually I simply set the inverter to start battery charging (usually last thing at night) until it's 100%. That's my winter mode, but for spring/summer, I'd setup a good, medium, bad solar forecast and adjusted the charing hours according.  Very much looking forward to seeing how this very intelligent approach operates.

Truly appreciate your fast responses. Thank you. 

(I may have more questions, hope you don't mind!)

fboundy, 2023-12-22:
Good to see it’s working. I need to do a bit of checking that this method is working ok. 3.5,0 will have a few other tweaks

closed 2023-12-27 by fboundy