stevebuk1/pv_opt #71: Missing entities

solarjones, 2023-12-21:
Hello, if you are able to provide some assistance, that would be very much appriciated. This isn't a bug I don't think, but the pvopt_dashboard.yaml has some errors for me, of which it reports these are not available:
sensor.solis_grid_import_power
sensor.solis_grid_export_power

How have you defined these? I don't seem to have them. 

I'm connecting to my Solis inverter using:
[alienatedsec](https://github.com/alienatedsec) RS485 to TCP 
[homeassistant-solax-modbus](https://github.com/wills106/homeassistant-solax-modbus)

Everything is working fine but the above sensors are not defined anywhere.

Thanks.

fboundy, 2023-12-21:
Added the necessary template sensors to the readmin in 3.4.1

closed 2023-12-21 by fboundy

solarjones, 2023-12-21:
Thanks I've added those now.

lazy-pete, 2024-01-12:
Hi - I was also missing those. I've copied / pasted them into yaml, then moved them around / added quotation marks etc so the yaml saves successfully, but no foy in dashboard. Looks like the entity names are ok, but not finding them either in dashboard or entities list.

[image]
[image]