stevebuk1/pv_opt #75: energidata service for pricing

chirs-51, 2023-12-21:
hey here in Danemark we use thes addon to get energy price https://github.com/MTrab/energidataservice
i can't find a way to use it on the addon withs is a shame love to try you'r andon  

its not clear what you mean with   
  inverter_type: "SOLIS_SOLAX_MODBUS"
  # inverter_type: "SOLIS_CORE_MODBUS"
  # inverter_type: SOLIS_SOLARMAN
  device_name: solis 

are that just names or are it a sensor from ha? 

all the unmarked ( # ) in config needs to be removed or what to do with them? 

i just tried my bedst to add what i could but its not working 

  # ========================================
  # Octopus account parameters
  # ========================================

  # octopus_auto: False # Read tariffs from the Octopus Energy integration. If successful this over-rides the following parameters
  id_import_tariff: sensor.energi_data_service
  id_export_tariff: sensor.energi_data_service_salg
 
  # modbus_hub: "{device_name}"
  # modbus_slave: 1
  battery_voltage: sensor.omcma3800g_battery_voltage
  # maximum_dod_percent: sensor.{device_name}_overdischarge_soc
  # update_cycle_seconds: 60
  
  id_consumption:
  - sensor.omcma3800g_import_from_grid

fboundy, 2023-12-21:
Hi – at the moment it’s really only set up to work with UK data. Let me have a look at what it would take to make it more general but it may take a while. I’ll try to address some of the other questions at the same time

chirs-51, 2023-12-21:
Nice otherwise you're app seems promising! And grate worke with s really good description!