stevebuk1/pv_opt #81: Charging time slots not aligned to cheapest

solarjones, 2023-12-23:
Version 3.4.5 B2.
This may be desired behaviour and me not understanding things or my setup isn't correct!

Parameter setup and output:
[image]

Todays negative Octopus Agile slots (they pay us for usage) would be the optimal charging time slots, however the Charging Plan doesn't seem to align. Similarly with SOC being 100% a lot of the time. Is the Opt Plan to run off battery and top up during cheap rates based on historic load pattern?  Wondering if this usage pattern isn't correct as our usage can be sporadic although we try to shift loads to cheap night rates (just off Eco7 tariff). 

Octopus Rates card highlights cheapest slots for a target charge pattern:
[image]

I'll keep looking at the Opt Plan outputs and tweak parameters to see how it behaves in ReadOnly mode.

Merry Christams!

fboundy, 2023-12-23:
Yes - mine was doing this. It’s because I had it always importing at max rate when prices were negative but that’s not optimal when they are negative for this long!

Will put out a fix when I get a chance

fboundy, 2023-12-25:
Also not re-loading new days rates

fboundy, 2024-02-01:
Please check release `3.7.0 beta 01` which I hope fixes this and has additional logging to debug

solarjones, 2024-02-04:
V3.7.1

Something still not quite correct. Charge plan not optimum:

[image]

Restarted AppDaemon resulted in this updated charging plan:

[image]

Not related to PVOpt but I've been unable to get AppDaemon to log to file despite trying everything I can. I'm running HA on a VM on a Synology NAS. Updated appdaemon.config to the relevant directories to log to (tried all sorts), but still not working for me. So unable to provide logs from file, only via Appdaemon console log.

SzosszeNET, 2024-02-04:
@solarjones with regards writing the logs probably the easiest what fboundy recommended -
Install the "samba share" add-on and than just add the log filters from the readme - it will drop the AppDaemon log files to the /share/log folder (I believe I had to create the log folder first) and you can easily access them.

The other way is probably to map a share from your Nas (haven't tried myself yet) under settings>system>storage  network storage?

fboundy, 2024-02-05:
Unfortunately it is very difficult for me to diagnose what is going on without the log of the optimisation loop.

solarjones, 2024-02-05:
Eureka! I don't know why, but Samba share had stopped working (even though I could browse HA folders). Anyway, I now have log files.

Think this is looking ok for now. I did just restart Appdaemon so will keep an eye on things.

[image]

[pv_opt 050224a.log](https://github.com/fboundy/pv_opt/files/14172262/pv_opt.050224a.log)

Thanks

fboundy, 2024-02-08:
Closing this as I think any ongoing issues are covered by #88

fboundy, 2024-02-08:
Closing this as I think any ongoing issues are covered by https://github.com/fboundy/pv_opt/issues/88

closed 2024-02-08 by fboundy