stevebuk1/pv_opt #84: AttributeError: 'NoneType' object has no attribute 'name'

dolce08, 2023-12-29:
Home Assistant Core: 2023.12.4
Home Assistant Supervisor: 2023.12.0
AppDaemon Add-on version: 0.16.0
Solax Modbus Integration 

PVOpt results in an initialize_error. Here is the log. Any advice is greatly appreciated.
log:

AttributeError: 'NoneType' object has no attribute 'name'
^^^^^^
self.log(f" {imp_exp.title()}: {t.name}")
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 648, in _load_contract
self._load_contract()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 288, in initialize
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
result = self.fn(*self.args, **self.kwargs)
File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
^^^^^^^^^^^^^^^

fboundy, 2023-12-29:
Hi

Looks like it’s not getting the tariff information. Can you provide the last few lines of the pv_opt.log file?

Thanks

fboundy, 2023-12-29:
It should have something like this:
    
    16:44:10     INFO: Loading Contract:
    16:44:10     INFO: -----------------
    16:44:10     INFO: Trying to auto detect Octopus tariffs:
    16:44:10     INFO:     Found import entity event.octopus_energy_electricity_19m1337498_1610032016836_current_day_rates
    16:44:10     INFO:     Found export entity event.octopus_energy_electricity_19m1337498_1650000216929_export_current_day_rates
    16:44:10     INFO: Contract tariffs loaded OK
    16:44:10     INFO:   Import: E-1R-AGILE-FLEX-22-11-25-G               Start: 2023-12-20 13:00:00+0000 End: 2023-12-30 23:00:00+0000 
    16:44:10     INFO:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-G         Start: 2019-05-15 23:00:00+0000 End: N/A 
    16:44:10     INFO: AGILE tariff detected. Rates will update at 16:00 daily

That's if you are using Bottlecap Dave's Octopus integration. If you aren't you either need to un-comment the Octopus Account lines in the`config.yaml` file:

     octopus_auto: False 
     octopus_account: !secret octopus_account # or type your account number
     octopus_api_key: !secret octopus_api_key # or type your api_key

Or you can explicitly set the tariff codes (example is Flux for the NW but you would need your own):

     octopus_auto: False 
     octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
     octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G

dolce08, 2023-12-30:
Thanks. Managed to get a bit further. I'm using Bottlecap Dave's integration but added my account details as a back up and it loaded tariffs.

`10:57:38     INFO: Loading Contract:
10:57:38     INFO: -----------------
10:57:38     INFO: Trying to load tariffs using Account: A-12345 API Key: abc123
10:57:38     INFO: Connecting to https://api.octopus.energy/v1/accounts/A-123445
10:57:38     INFO: Getting details for MPAN 123456789
10:57:38     INFO: Retrieved most recent tariff code E-1R-OUTGOING-SEG-FIX-12M-20-07-07-J
10:57:38     INFO: Getting details for MPAN 123456789
10:57:38     INFO: Retrieved most recent tariff code E-1R-AGILE-FLEX-22-11-25-J
10:57:38    ERROR: 'WARN'
10:57:38  WARNING: Unable to load Octopus Account details using API Key: 'WARN' Trying other methods.
10:57:38     INFO:   Import: E-1R-AGILE-FLEX-22-11-25-J
10:57:38     INFO:   Export: E-1R-OUTGOING-SEG-FIX-12M-20-07-07-J
10:57:38     INFO: AGILE tariff detected. Rates will update at 16:00 daily
`

It now hangs on 
`10:57:38     INFO: Finished loading contract
10:57:38     INFO: 
10:57:38     INFO: Comparing yesterday's tariffs
10:57:38     INFO:   - Getting yesterday's solar generation
`

error log: 
`10:56:30 WARNING pv_opt: ------------------------------------------------------------
10:57:38 WARNING pv_opt: ------------------------------------------------------------
10:57:38 WARNING pv_opt: Unexpected error running initialize() for pv_opt
10:57:38 WARNING pv_opt: ------------------------------------------------------------
10:57:38 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 291, in initialize
    self._compare_tariffs()
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1793, in _compare_tariffs
    solar = self._get_solar(start, end)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1875, in _get_solar
    df = self.hass2df(entity_id, days=days).astype(float).resample("30T").ffill()
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/generic.py", line 6534, in astype
    new_data = self._mgr.astype(dtype=dtype, copy=copy, errors=errors)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/internals/managers.py", line 414, in astype
    return self.apply(
           ^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/internals/managers.py", line 354, in apply
    applied = getattr(b, f)(**kwargs)
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/internals/blocks.py", line 616, in astype
    new_values = astype_array_safe(values, dtype, copy=copy, errors=errors)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/dtypes/astype.py", line 238, in astype_array_safe
    new_values = astype_array(values, dtype, copy=copy)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/dtypes/astype.py", line 183, in astype_array
    values = _astype_nansafe(values, dtype, copy=copy)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/dtypes/astype.py", line 134, in _astype_nansafe
    return arr.astype(dtype, copy=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: ''`

fboundy, 2023-12-30:
OK - so the bit it’s hanging on is a bit experimental and not critical. If you comment out the last few lines right at the end of config.yaml it should skip this bit.

I should have a new release out later today and I will turn this off by default

dolce08, 2024-01-01:
Thanks, it's working perfectly now. Thanks for all the effort you've put into this integration.

closed 2024-01-03 by dolce08