stevebuk1/pv_opt #87: Prediction / charge accuracy

solarjones, 2024-01-03:
Happy New Year! 
* This may not be a bug - probably my lack of understanding *

v 3.5.1
Yesterday I switched out from Read Only mode so early days with my understanding of how the logic works. Today at 16:00 when the new Agile import rates were published, the PV_Opt has chosen the following time slots to charge the battery:

[image]

I can't work out why it wouldn't select the cheapest time and you can see it will be charging (or holding, still charging?) until 06:00. Why not charge earlier and upto the 05:30 slot?  

I don't fully understand the Threshold per pass (currently 4.0) & per slot (1.0), guess that affects things.

Also, tomorrow's solar forecast is good (11kWh - believe it when I see it!), but perhaps the charging forecast is skewed based on consumption over Christmas. Does the optimiser work out the battery change required to clear the expensive peak rates based on consumption history and expected solar forecast? This would explain charging later into the morning but not if 11kWh PV is predicted.

[image]

Regards

fboundy, 2024-01-03:
That looks pretty optimal to me.... Mine looks pretty similar - albeit with only the 10% Solcast forecast showing that much sun.

The cheapest slots are basically 3:00 to 5:30. After that it is, like you say, holding charge at 100% for a bit so that the solar when it kicks in will get you past the Peak. If earlier slots were used I suspect the solar wouldn't get you to past Peak rate so you would be worse off overall. Everything you are buying costs less than your export tariff so it makes sense to take as much as possible. Some of you 11 kWh of solar is then exported for the period from about 10:00 to 14:00 when you are at 100% SOC.

The only slot that maybe doesn't make sense is the last one but again it may be that buy buying a little at ~20p you save having to buy half that at 40p. If you send through the full log I could diagnose a bit better.

I am finding the Xmas has skewed my consumption history quite a bit - using more days will iron this out depending on how much history your keep in HA.

Threshold per pass and per slot probably won't make much difference. They are there to weed out marginally beneficial charge/discharge slots that don't make a material difference.

solarjones, 2024-01-04:
Hit a problem. Noticing the inverter isn't being commanded correctly on occasions.
I've failed to get AppDaemon file logging working (still no idea why, very frustrating), but from the console log it shows the following:

[image]
[image]

My inverter is a 6kWh Solis and the 4x Pylontech batteries (U2000C) allow for 100amp max charge (25amp per battery), but the 6000W @ 48.6v = 123.5 amp. Could this be attempting to command the inverter charging at a greater rate than the BMS/inverter allows? Or is it something else?
Thanks.

solarjones, 2024-01-04:
[image]
Some inverter values that may help... Note I manually set the charge time and amps when the PV_Opt hit the issue.

fboundy, 2024-01-04:
Yes - I’ve had similar today. Not sure why. Raised an issue just now. What version are you using?

fboundy, 2024-01-04:
OK - didn’t look properly. Different issue to mine.

Will have a look properly tomorrow if I have time

fboundy, 2024-01-06:
I think you are right in that the BMS Limit of 100A overrides the desired 6000w. The simple fix is to set your charger to be 5000w since it can't actually charge at 6000w with your batteries. I will also try to add a check to see if the desired charge/discharge current exceeds the BMS limits.

solarjones, 2024-01-06:
[image]

Ok, I've adjusted the battery charging down accordingly. Also switched to Solcast P10 and it just started charging the batteries. I still can't understand why it would do this now and not wait until 13:30 when it's cheapest. Is it the historic load playing a part?

Thanks.

fboundy, 2024-01-06:
Yes - that’s a bit odd. If you can send through the log where it does the logic slot by slot I’ll have a look

solarjones, 2024-01-06:
2024-01-06 12:34:14.508175 INFO pv_opt: 07/01 00:00:  0.50 kWh at   8.41p. <==> 06/01 22:30: 15.61p/kWh  7.78p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 19.7% Net:  580.9
2024-01-06 12:34:14.605612 INFO pv_opt: 07/01 00:00:  0.50 kWh at   8.41p. <==> 06/01 22:30: 15.61p/kWh  7.78p  SOC:  15.0%-> 19.7% New SOC:  15.0%-> 24.5% Net:  581.3
2024-01-06 12:34:14.715828 INFO pv_opt: 07/01 00:00:  0.50 kWh at   8.41p. <==> 06/01 22:30: 15.61p/kWh  7.78p  SOC:  15.0%-> 24.5% New SOC:  15.0%-> 29.2% Net:  581.0
2024-01-06 12:34:14.858058 INFO pv_opt: 07/01 00:30:  0.43 kWh at   7.21p. <==> 06/01 22:30: 15.61p/kWh  6.67p  SOC:  15.0%-> 29.2% New SOC:  15.0%-> 33.2% Net:  580.4
2024-01-06 12:34:15.194761 INFO pv_opt: 07/01 01:30:  0.29 kWh at   4.65p. <==> 06/01 22:30: 15.61p/kWh  4.59p  SOC:  15.0%-> 33.2% New SOC:  15.0%-> 36.0% Net:  580.2
2024-01-06 12:34:15.289435 INFO pv_opt: 07/01 01:30:  0.29 kWh at   4.65p. <==> 06/01 22:30: 15.61p/kWh  4.59p  SOC:  15.0%-> 36.0% New SOC:  15.0%-> 38.8% Net:  580.2
2024-01-06 12:34:15.384014 INFO pv_opt: 07/01 01:30:  0.23 kWh at   3.66p. <==> 06/01 22:30: 15.61p/kWh  3.61p  SOC:  15.0%-> 38.8% New SOC:  15.0%-> 41.0% Net:  580.1
2024-01-06 12:34:15.507864 INFO pv_opt: 06/01 18:30:  0.11 kWh at   3.49p. <==> 06/01 12:30: 18.45p/kWh  2.08p  SOC:  88.5%-> 95.9% New SOC:  88.5%-> 97.0% Net:  578.7
2024-01-06 12:34:15.602004 INFO pv_opt: 07/01 18:30:  0.05 kWh at   1.40p. <==> 07/01 14:00: 18.50p/kWh  0.84p  SOC:  91.7%-> 83.7% New SOC:  91.7%-> 91.9% Net:  577.7
2024-01-06 12:34:15.612020 INFO pv_opt: 
2024-01-06 12:34:15.615224 INFO pv_opt: Maximum 1st pass slot delta is 1.7p
2024-01-06 12:34:15.617972 INFO pv_opt: 
2024-01-06 12:34:15.649422 INFO pv_opt: Previous message repeated 1 times
2024-01-06 12:34:15.650507 INFO pv_opt: Low Cost Charging
2024-01-06 12:34:15.653139 INFO pv_opt: ------------------
2024-01-06 12:34:15.655826 INFO pv_opt: 
2024-01-06 12:34:15.658596 INFO pv_opt: Max export price when there is no forced charge: 15.00p/kWh.
2024-01-06 12:34:15.662577 INFO pv_opt: 0 slots have an import price less than the max export price
2024-01-06 12:34:15.665830 INFO pv_opt: 
2024-01-06 12:34:15.668821 INFO pv_opt: Charge net cost delta: -0.0p: < threshold 4.0 => Excluded
2024-01-06 12:34:15.672464 INFO pv_opt: Iteration 11: Slots added:   0
2024-01-06 12:34:15.676826 INFO pv_opt: 
2024-01-06 12:34:15.681013 INFO pv_opt: Removing cyclic charge/discharge
2024-01-06 12:34:15.758992 INFO pv_opt:   Net cost revised from 577.7p to 577.9p
2024-01-06 12:34:15.816616 INFO pv_opt: 
2024-01-06 12:34:15.847715 INFO pv_opt: Checking for Hold SOC slots
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:1572: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame

See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  self.windows["hold_soc"].loc[
2024-01-06 12:34:15.853587 INFO pv_opt: 
2024-01-06 12:34:15.856168 INFO pv_opt: Optimal forced charge/discharge slots:
2024-01-06 12:34:15.859716 INFO pv_opt:    06-Jan 12:30 - 06-Jan 13:00   Power:  1800W  SOC:   88% ->   97%  
2024-01-06 12:34:15.862680 INFO pv_opt:    06-Jan 22:30 - 06-Jan 23:00   Power:  5500W  SOC:   15% ->   41%  
2024-01-06 12:34:15.865526 INFO pv_opt:    07-Jan 04:00 - 07-Jan 05:00   Power:  3700W  SOC:   15% ->   50%  
2024-01-06 12:34:15.868321 INFO pv_opt:    07-Jan 05:00 - 07-Jan 06:30   Power:  6000W  SOC:   50% ->  100%  
2024-01-06 12:34:15.871093 INFO pv_opt:    07-Jan 06:30 - 07-Jan 07:30   Power:  1200W  SOC:  100% ->  100%  <=
2024-01-06 12:34:15.873936 INFO pv_opt:    07-Jan 07:30 - 07-Jan 08:30   Power:  1200W  SOC:  100% ->  100%  <=
2024-01-06 12:34:15.876707 INFO pv_opt:    07-Jan 08:30 - 07-Jan 09:00   Power:  1100W  SOC:  100% ->  100%  <=
2024-01-06 12:34:15.879461 INFO pv_opt:    07-Jan 12:00 - 07-Jan 13:00   Power:  2900W  SOC:   73% ->  100%  
2024-01-06 12:34:15.882706 INFO pv_opt:    07-Jan 14:00 - 07-Jan 14:30   Power:     0W  SOC:   92% ->   92%  <=
2024-01-06 12:34:15.889999 INFO pv_opt: 
2024-01-06 12:34:15.893115 INFO pv_opt: Plan time: 06-Jan 12:30 - 07-Jan 23:30 Initial SOC: 88.5 Base Cost: 678.99 Opt Cost: 577.86
2024-01-06 12:34:15.896291 INFO pv_opt: 
2024-01-06 12:34:15.898757 INFO pv_opt: Optimiser elapsed time 3.9 seconds
2024-01-06 12:34:15.900961 INFO pv_opt: 
2024-01-06 12:34:17.536462 INFO pv_opt: Previous message repeated 1 times
2024-01-06 12:34:17.538951 INFO pv_opt: Output written to sensor.pvopt_optimiser_elapsed
2024-01-06 12:34:17.722991 INFO pv_opt: Average unit cost today: 23.30p/kWh
2024-01-06 12:34:17.744850 INFO pv_opt: Output written to sensor.pvopt_unit_cost_today
2024-01-06 12:34:17.971031 INFO pv_opt: Output written to sensor.pvopt_base_cost
2024-01-06 12:34:18.224020 INFO pv_opt: Output written to sensor.pvopt_opt_cost
2024-01-06 12:34:18.264465 INFO pv_opt: Output written to sensor.pvopt_charge_start
2024-01-06 12:34:18.284774 INFO pv_opt: Output written to sensor.pvopt_charge_end
2024-01-06 12:34:18.311461 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-01-06 12:34:18.347318 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-01-06 12:34:18.376873 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-01-06 12:34:18.398802 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-01-06 12:34:18.420309 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-01-06 12:34:18.458942 INFO pv_opt: 
[... cut ...]
2024-01-06 12:34:18.562513 INFO pv_opt: Updating inverter charge control:
2024-01-06 12:34:18.567432 INFO pv_opt:   start: None
2024-01-06 12:34:18.570415 INFO pv_opt:   end: 2024-01-06 13:00:00+00:00
2024-01-06 12:34:18.573121 INFO pv_opt:   power: 1800
2024-01-06 12:34:18.577471 INFO pv_opt: Inverter already at correct time settings
2024-01-06 12:34:18.583622 INFO pv_opt: Power 1800 = 35.4A at 50.9V
2024-01-06 12:34:19.098673 INFO pv_opt: Current 35.4 written to inverter
2024-01-06 12:34:19.102573 INFO pv_opt: Wating for Modbus Read cycle: 18 seconds

solarjones, 2024-01-06:
2024-01-06 12:34:15.879461 INFO pv_opt:    07-Jan 12:00 - 07-Jan 13:00   Power:  2900W  SOC:   73% ->  100%  
2024-01-06 12:34:15.882706 INFO pv_opt:    07-Jan 14:00 - 07-Jan 14:30   Power:     0W  SOC:   92% ->   92%  <=
2024-01-06 12:34:15.889999 INFO pv_opt: 
2024-01-06 12:34:15.893115 INFO pv_opt: Plan time: 06-Jan 12:30 - 07-Jan 23:30 Initial SOC: 88.5 Base Cost: 678.99 Opt Cost: 577.86
2024-01-06 12:34:15.896291 INFO pv_opt: 
2024-01-06 12:34:15.898757 INFO pv_opt: Optimiser elapsed time 3.9 seconds
2024-01-06 12:34:15.900961 INFO pv_opt: 
2024-01-06 12:34:17.536462 INFO pv_opt: Previous message repeated 1 times
2024-01-06 12:34:17.538951 INFO pv_opt: Output written to sensor.pvopt_optimiser_elapsed
2024-01-06 12:34:17.722991 INFO pv_opt: Average unit cost today: 23.30p/kWh
2024-01-06 12:34:17.744850 INFO pv_opt: Output written to sensor.pvopt_unit_cost_today
2024-01-06 12:34:17.971031 INFO pv_opt: Output written to sensor.pvopt_base_cost
2024-01-06 12:34:18.224020 INFO pv_opt: Output written to sensor.pvopt_opt_cost
2024-01-06 12:34:18.264465 INFO pv_opt: Output written to sensor.pvopt_charge_start
2024-01-06 12:34:18.284774 INFO pv_opt: Output written to sensor.pvopt_charge_end
2024-01-06 12:34:18.311461 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-01-06 12:34:18.347318 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-01-06 12:34:18.376873 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-01-06 12:34:18.398802 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-01-06 12:34:18.420309 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-01-06 12:34:18.458942 INFO pv_opt: 
2024-01-06 12:34:18.461958 INFO pv_opt: Current inverter status:
2024-01-06 12:34:18.464386 INFO pv_opt: ------------------------
2024-01-06 12:34:18.466775 INFO pv_opt:   mode              : Timed Charge/Discharge
2024-01-06 12:34:18.469175 INFO pv_opt:   code              : 35
2024-01-06 12:34:18.482072 INFO pv_opt:   switches          :
2024-01-06 12:34:18.485566 INFO pv_opt:     SelfUse         : True
2024-01-06 12:34:18.488187 INFO pv_opt:     Timed           : True
2024-01-06 12:34:18.491019 INFO pv_opt:     OffGrid         : False
2024-01-06 12:34:18.494309 INFO pv_opt:     BatteryWake     : False
2024-01-06 12:34:18.496940 INFO pv_opt:     Backup          : False
2024-01-06 12:34:18.499496 INFO pv_opt:     GridCharge      : True
2024-01-06 12:34:18.502163 INFO pv_opt:     FeedInPriority  : False
2024-01-06 12:34:18.504739 INFO pv_opt:   charge            :
2024-01-06 12:34:18.507397 INFO pv_opt:     start           : 06-Jan 12:30
2024-01-06 12:34:18.510106 INFO pv_opt:     end             : 06-Jan 13:00
2024-01-06 12:34:18.512514 INFO pv_opt:     current         : 13.9
2024-01-06 12:34:18.514943 INFO pv_opt:     active          : True
2024-01-06 12:34:18.517295 INFO pv_opt:   discharge         :
2024-01-06 12:34:18.520375 INFO pv_opt:     start           : 06-Jan 00:00
2024-01-06 12:34:18.522993 INFO pv_opt:     end             : 06-Jan 00:00
2024-01-06 12:34:18.525571 INFO pv_opt:     current         : 0.0
2024-01-06 12:34:18.528773 INFO pv_opt:     active          : False
2024-01-06 12:34:18.531897 INFO pv_opt:   hold_soc          :
2024-01-06 12:34:18.534977 INFO pv_opt:     active          : False
2024-01-06 12:34:18.537924 INFO pv_opt:     soc             : 100.0
2024-01-06 12:34:18.540879 INFO pv_opt: 
2024-01-06 12:34:18.543554 INFO pv_opt: Current charge/discharge windows ends in 25.7 minutes.
2024-01-06 12:34:18.548413 INFO pv_opt: >>> SelfUse: True
2024-01-06 12:34:18.551356 INFO pv_opt: >>> Timed: True
2024-01-06 12:34:18.554246 INFO pv_opt: >>> Backup: False
2024-01-06 12:34:18.556960 INFO pv_opt: >>> GridCharge: True
2024-01-06 12:34:18.562513 INFO pv_opt: Updating inverter charge control:
2024-01-06 12:34:18.567432 INFO pv_opt:   start: None
2024-01-06 12:34:18.570415 INFO pv_opt:   end: 2024-01-06 13:00:00+00:00
2024-01-06 12:34:18.573121 INFO pv_opt:   power: 1800
2024-01-06 12:34:18.577471 INFO pv_opt: Inverter already at correct time settings
2024-01-06 12:34:18.583622 INFO pv_opt: Power 1800 = 35.4A at 50.9V
2024-01-06 12:34:19.098673 INFO pv_opt: Current 35.4 written to inverter
2024-01-06 12:34:19.102573 INFO pv_opt: Wating for Modbus Read cycle: 18 seconds
2024-01-06 12:34:42.833208 INFO pv_opt: 
2024-01-06 12:34:42.837304 INFO pv_opt: Current inverter status:
2024-01-06 12:34:42.841028 INFO pv_opt: ------------------------
2024-01-06 12:34:42.844534 INFO pv_opt:   mode              : Timed Charge/Discharge
2024-01-06 12:34:42.848525 INFO pv_opt:   code              : 35
2024-01-06 12:34:42.851757 INFO pv_opt:   switches          :
2024-01-06 12:34:42.854157 INFO pv_opt:     SelfUse         : True
2024-01-06 12:34:42.856554 INFO pv_opt:     Timed           : True
2024-01-06 12:34:42.861337 INFO pv_opt:     OffGrid         : False
2024-01-06 12:34:42.864308 INFO pv_opt:     BatteryWake     : False
[... 5 lines cut ...]
2024-01-06 12:34:42.880613 INFO pv_opt:     end             : 06-Jan 13:00
2024-01-06 12:34:42.883036 INFO pv_opt:     current         : 35.3
2024-01-06 12:34:42.886217 INFO pv_opt:     active          : True
2024-01-06 12:34:42.888727 INFO pv_opt:   discharge         :
2024-01-06 12:34:42.891125 INFO pv_opt:     start           : 06-Jan 00:00
2024-01-06 12:34:42.893520 INFO pv_opt:     end             : 06-Jan 00:00
2024-01-06 12:34:42.895903 INFO pv_opt:     current         : 0.0
[... cut ...]
2024-01-06 12:34:42.930339 INFO pv_opt: Updating inverter charge control:
2024-01-06 12:34:42.933215 INFO pv_opt:   start: None
2024-01-06 12:34:42.937185 INFO pv_opt:   end: 2024-01-06 13:00:00+00:00
2024-01-06 12:34:42.940266 INFO pv_opt:   power: 1800
2024-01-06 12:34:42.944550 INFO pv_opt: Inverter already at correct time settings
2024-01-06 12:34:42.949926 INFO pv_opt: Power 1800 = 35.1A at 51.3V
2024-01-06 12:34:42.953498 INFO pv_opt: Inverter already at correct current
2024-01-06 12:34:42.957021 INFO pv_opt: Wating for Modbus Read cycle: 18 seconds

solarjones, 2024-01-06:
Set to read only after I reset the inverter charge times manually:

2024-01-06 12:49:39.004007 INFO pv_opt: 
2024-01-06 12:49:39.006700 INFO pv_opt: Starting Opimisation with discharge disabled
2024-01-06 12:49:39.008585 INFO pv_opt: --------------------------------------------
2024-01-06 12:49:39.012716 INFO pv_opt: 
2024-01-06 12:49:39.016574 INFO pv_opt: Checking tariffs:
2024-01-06 12:49:39.018830 INFO pv_opt: -----------------
2024-01-06 12:49:39.043495 INFO pv_opt:   Import: E-1R-AGILE-FLEX-22-11-25-A               Start: 2023-12-26 13:00:00+0000 End: 2024-01-05 23:00:00+0000 <<< ERROR: Tariff end datetime in past 
2024-01-06 12:49:39.048358 INFO pv_opt:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-A         Start: 2019-05-15 23:00:00+0000 End: N/A 
2024-01-06 12:49:39.052614 INFO pv_opt: AGILE tariff detected. Rates will update at 16:00 daily
2024-01-06 12:49:39.054899 INFO pv_opt:   - Tariffs OK
2024-01-06 12:49:39.058491 INFO pv_opt: 
2024-01-06 12:49:39.074688 INFO pv_opt: Solcast forecast loaded OK
2024-01-06 12:49:39.077481 INFO pv_opt: Getting expected consumption data
2024-01-06 12:49:39.886135 INFO pv_opt:   - Estimated consumption from sensor.solis_house_load_today loaded OK 
2024-01-06 12:49:39.937503 INFO pv_opt: Initial SOC: 88.5
2024-01-06 12:49:39.942567 INFO pv_opt: Calculating Base flows
2024-01-06 12:49:40.009757 INFO pv_opt: Base cost: 678.25p
2024-01-06 12:49:40.012855 INFO pv_opt: 
2024-01-06 12:49:40.015682 INFO pv_opt: Optimising for Solcast_p10 forecast from 06-Jan 12:30 to 07-Jan 23:30
2024-01-06 12:49:40.099203 INFO pv_opt:   Optimiser prices loaded for period 06/01 12:30 - 07/01 23:30
2024-01-06 12:49:40.192432 INFO pv_opt: 
2024-01-06 12:49:40.195375 INFO pv_opt: High Cost Usage Swaps
2024-01-06 12:49:40.197769 INFO pv_opt: ---------------------
2024-01-06 12:49:40.200093 INFO pv_opt: 
2024-01-06 12:49:40.299162 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 05:00: 15.43p/kWh 18.96p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 20.8% Net:  679.9
2024-01-06 12:49:40.386969 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 05:00: 15.43p/kWh 18.96p  SOC:  15.0%-> 26.7% New SOC:  15.0%-> 26.7% Net:  680.8
2024-01-06 12:49:40.583273 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 05:00: 15.43p/kWh 18.96p  SOC:  15.0%-> 38.3% New SOC:  15.0%-> 32.5% Net:  680.8
2024-01-06 12:49:40.682649 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 05:00: 15.43p/kWh 18.96p  SOC:  15.0%-> 50.0% New SOC:  15.0%-> 38.3% Net:  679.7
2024-01-06 12:49:40.775003 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 05:00: 15.43p/kWh 18.96p  SOC:  15.0%-> 61.6% New SOC:  15.0%-> 43.4% Net:  678.0
2024-01-06 12:49:40.859046 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 04:00: 15.44p/kWh 18.97p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 20.8% Net:  676.2
2024-01-06 12:49:40.965102 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 04:00: 15.44p/kWh 18.97p  SOC:  15.0%-> 26.7% New SOC:  15.0%-> 26.7% Net:  674.9
2024-01-06 12:49:41.055749 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 04:00: 15.44p/kWh 18.97p  SOC:  15.0%-> 38.3% New SOC:  15.0%-> 32.5% Net:  674.4
2024-01-06 12:49:41.143041 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 06:00: 15.95p/kWh 19.59p  SOC: 100.0%-> 83.3% New SOC: 100.0%->100.0% Net:  669.7
2024-01-06 12:49:41.228438 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 07:00: 17.17p/kWh 21.10p  SOC: 100.0%-> 87.3% New SOC: 100.0%->100.0% Net:  657.7
2024-01-06 12:49:41.313586 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 08:00: 17.95p/kWh 22.05p  SOC: 100.0%-> 88.6% New SOC: 100.0%->100.0% Net:  639.7
2024-01-06 12:49:41.403299 INFO pv_opt: 07/01 17:30:  1.23 kWh at  39.07p. <==> 07/01 12:00: 18.45p/kWh 22.67p  SOC:  73.0%-> 66.6% New SOC:  73.0%-> 78.8% Net:  612.9
2024-01-06 12:49:41.485884 INFO pv_opt: 07/01 18:00:  0.86 kWh at  26.50p. <==> 07/01 12:00: 18.45p/kWh 15.78p  SOC:  73.0%-> 84.7% New SOC:  73.0%-> 82.9% Net:  602.2
2024-01-06 12:49:41.588760 INFO pv_opt: 06/01 18:30:  0.78 kWh at  24.27p. <==> 06/01 12:30: 18.45p/kWh 14.45p  SOC:  88.5%-> 86.5% New SOC:  88.5%-> 95.9% Net:  589.2
2024-01-06 12:49:41.695418 INFO pv_opt: 07/01 18:30:  0.78 kWh at  24.27p. <==> 07/01 12:00: 18.45p/kWh 14.45p  SOC:  73.0%-> 92.8% New SOC:  73.0%-> 86.6% Net:  580.0
2024-01-06 12:49:41.887209 INFO pv_opt: 07/01 00:00:  0.50 kWh at   8.41p. <==> 06/01 22:30: 15.61p/kWh  7.78p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 19.7% Net:  580.4
2024-01-06 12:49:41.974688 INFO pv_opt: 07/01 00:00:  0.50 kWh at   8.41p. <==> 06/01 22:30: 15.61p/kWh  7.78p  SOC:  15.0%-> 19.7% New SOC:  15.0%-> 24.5% Net:  580.8
2024-01-06 12:49:42.067657 INFO pv_opt: 07/01 00:00:  0.50 kWh at   8.41p. <==> 06/01 22:30: 15.61p/kWh  7.78p  SOC:  15.0%-> 24.5% New SOC:  15.0%-> 29.2% Net:  580.5
2024-01-06 12:49:42.207979 INFO pv_opt: 07/01 00:30:  0.43 kWh at   7.21p. <==> 06/01 22:30: 15.61p/kWh  6.67p  SOC:  15.0%-> 29.2% New SOC:  15.0%-> 33.2% Net:  579.9
2024-01-06 12:49:42.425080 INFO pv_opt: 07/01 01:30:  0.29 kWh at   4.65p. <==> 06/01 22:30: 15.61p/kWh  4.59p  SOC:  15.0%-> 33.2% New SOC:  15.0%-> 36.0% Net:  579.7
2024-01-06 12:49:42.538472 INFO pv_opt: 07/01 01:30:  0.29 kWh at   4.65p. <==> 06/01 22:30: 15.61p/kWh  4.59p  SOC:  15.0%-> 36.0% New SOC:  15.0%-> 38.8% Net:  579.7
[... cut ...]
2024-01-06 12:49:44.414611 INFO pv_opt: Output written to sensor.pvopt_charge_end
2024-01-06 12:49:44.442527 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-01-06 12:49:44.491133 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-01-06 12:49:44.521855 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-01-06 12:49:44.560408 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-01-06 12:49:44.592790 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-01-06 12:49:44.599086 INFO pv_opt: Read only mode enabled. Not querying inverter.

solarjones, 2024-01-06:
I let time progress to see if it would pickup the next cheap slot, but it doesn't...

024-01-06 13:20:00.055401 INFO pv_opt: No upcoming Octopus Saving Events detected or joined:
2024-01-06 13:20:00.057240 INFO pv_opt: 
2024-01-06 13:20:00.058954 INFO pv_opt: Starting Opimisation with discharge disabled
2024-01-06 13:20:00.060685 INFO pv_opt: --------------------------------------------
2024-01-06 13:20:00.062459 INFO pv_opt: 
2024-01-06 13:20:00.064270 INFO pv_opt: Checking tariffs:
2024-01-06 13:20:00.067661 INFO pv_opt: -----------------
2024-01-06 13:20:00.089538 INFO pv_opt:   Import: E-1R-AGILE-FLEX-22-11-25-A               Start: 2023-12-26 13:00:00+0000 End: 2024-01-05 23:00:00+0000 <<< ERROR: Tariff end datetime in past 
2024-01-06 13:20:00.092374 INFO pv_opt:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-A         Start: 2019-05-15 23:00:00+0000 End: N/A 
2024-01-06 13:20:00.094444 INFO pv_opt: AGILE tariff detected. Rates will update at 16:00 daily
2024-01-06 13:20:00.096586 INFO pv_opt:   - Tariffs OK
2024-01-06 13:20:00.098499 INFO pv_opt: 
2024-01-06 13:20:00.111943 INFO pv_opt: Solcast forecast loaded OK
2024-01-06 13:20:00.114928 INFO pv_opt: Getting expected consumption data
2024-01-06 13:20:00.439673 INFO pv_opt:   - Estimated consumption from sensor.solis_house_load_today loaded OK 
2024-01-06 13:20:00.493271 INFO pv_opt: Initial SOC: 86.5
2024-01-06 13:20:00.495885 INFO pv_opt: Calculating Base flows
2024-01-06 13:20:00.564296 INFO pv_opt: Base cost: 673.22p
2024-01-06 13:20:00.567883 INFO pv_opt: 
2024-01-06 13:20:00.570830 INFO pv_opt: Optimising for Solcast_p10 forecast from 06-Jan 13:00 to 07-Jan 23:30
2024-01-06 13:20:00.655305 INFO pv_opt:   Optimiser prices loaded for period 06/01 13:00 - 07/01 23:30
2024-01-06 13:20:00.854342 INFO pv_opt: 
2024-01-06 13:20:00.859022 INFO pv_opt: High Cost Usage Swaps
2024-01-06 13:20:00.863034 INFO pv_opt: ---------------------
2024-01-06 13:20:00.867402 INFO pv_opt: 
2024-01-06 13:20:01.009961 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 05:00: 15.43p/kWh 15.11p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 19.6% Net:  674.6
2024-01-06 13:20:01.095200 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 05:00: 15.43p/kWh 15.11p  SOC:  15.0%-> 24.3% New SOC:  15.0%-> 24.3% Net:  675.4
2024-01-06 13:20:01.186244 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 05:00: 15.43p/kWh 15.11p  SOC:  15.0%-> 33.6% New SOC:  15.0%-> 28.9% Net:  675.6
2024-01-06 13:20:01.278823 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 05:00: 15.43p/kWh 15.11p  SOC:  15.0%-> 42.8% New SOC:  15.0%-> 33.6% Net:  675.2
2024-01-06 13:20:01.371462 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 05:00: 15.43p/kWh 15.11p  SOC:  15.0%-> 52.1% New SOC:  15.0%-> 38.2% Net:  673.9
2024-01-06 13:20:01.457793 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 05:00: 15.43p/kWh 15.11p  SOC:  15.0%-> 61.4% New SOC:  15.0%-> 42.8% Net:  672.3
2024-01-06 13:20:01.564139 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 05:00: 15.43p/kWh 15.11p  SOC:  15.0%-> 70.7% New SOC:  15.0%-> 43.4% Net:  672.1
2024-01-06 13:20:02.008844 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 04:00: 15.44p/kWh 15.12p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 19.6% Net:  670.8
2024-01-06 13:20:02.197339 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 04:00: 15.44p/kWh 15.12p  SOC:  15.0%-> 24.3% New SOC:  15.0%-> 24.3% Net:  669.8
2024-01-06 13:20:02.418597 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 04:00: 15.44p/kWh 15.12p  SOC:  15.0%-> 33.6% New SOC:  15.0%-> 28.9% Net:  668.9
2024-01-06 13:20:02.601739 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 06:00: 15.95p/kWh 15.62p  SOC:  99.7%-> 85.4% New SOC:  99.7%->100.0% Net:  664.5
2024-01-06 13:20:03.034896 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 07:00: 17.17p/kWh 16.82p  SOC: 100.0%-> 87.9% New SOC: 100.0%->100.0% Net:  655.1
2024-01-06 13:20:03.230215 INFO pv_opt: 07/01 17:00:  0.98 kWh at  31.14p. <==> 07/01 08:00: 17.95p/kWh 17.58p  SOC: 100.0%-> 89.2% New SOC: 100.0%->100.0% Net:  638.1
2024-01-06 13:20:03.320266 INFO pv_opt: 07/01 17:30:  0.98 kWh at  31.14p. <==> 07/01 12:00: 18.45p/kWh 18.07p  SOC:  74.2%-> 66.6% New SOC:  74.2%-> 78.8% Net:  612.6
2024-01-06 13:20:03.422532 INFO pv_opt: 07/01 18:00:  0.99 kWh at  30.60p. <==> 07/01 12:00: 18.45p/kWh 18.23p  SOC:  74.2%-> 83.5% New SOC:  74.2%-> 83.5% Net:  600.2
2024-01-06 13:20:03.507443 INFO pv_opt: 06/01 18:30:  0.72 kWh at  22.34p. <==> 06/01 14:00: 18.50p/kWh 13.34p  SOC:  81.1%-> 66.7% New SOC:  81.1%-> 84.5% Net:  586.0
2024-01-06 13:20:03.597883 INFO pv_opt: 07/01 18:30:  0.72 kWh at  22.34p. <==> 07/01 12:00: 18.45p/kWh 13.30p  SOC:  74.2%-> 92.8% New SOC:  74.2%-> 86.9% Net:  577.2
2024-01-06 13:20:03.707519 INFO pv_opt: 07/01 00:00:  0.56 kWh at   9.46p. <==> 06/01 22:30: 15.61p/kWh  8.76p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 20.3% Net:  577.7
2024-01-06 13:20:03.808160 INFO pv_opt: 07/01 00:00:  0.56 kWh at   9.46p. <==> 06/01 22:30: 15.61p/kWh  8.76p  SOC:  15.0%-> 20.3% New SOC:  15.0%-> 25.6% Net:  577.9
2024-01-06 13:20:04.034346 INFO pv_opt: 07/01 00:00:  0.44 kWh at   7.35p. <==> 06/01 22:30: 15.61p/kWh  6.80p  SOC:  15.0%-> 25.6% New SOC:  15.0%-> 29.8% Net:  577.3
2024-01-06 13:20:04.360948 INFO pv_opt: 07/01 00:30:  0.36 kWh at   6.16p. <==> 06/01 22:30: 15.61p/kWh  5.70p  SOC:  15.0%-> 29.8% New SOC:  15.0%-> 33.2% Net:  576.9
[... cut ...]
2024-01-06 13:20:05.655513 INFO pv_opt: Output written to sensor.pvopt_charge_end
2024-01-06 13:20:05.678843 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-01-06 13:20:05.702599 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-01-06 13:20:05.737741 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-01-06 13:20:05.759606 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-01-06 13:20:05.786469 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-01-06 13:20:05.792377 INFO pv_opt: Read only mode enabled. Not querying inverter.

[image]

fboundy, 2024-01-06:
Ok - I need to add a bit more to the logging to diagnose what’s going on.

solarjones, 2024-01-06:
> Ok - I need to add a bit more to the logging to diagnose what’s going on.
> […](#)

Thanks - guessing you are seeing similar issues?

solarjones, 2024-01-08:
A further example log showing charging time slot selected differently than expected.
Also note the Optimum Charge Current didn't seem to lower when I set the Charger Power lower from 6000 to 5000W. I'm still in ReadOnly mode for the time being.

[image]
[image]

2024-01-08 16:11:06.809402 INFO pv_opt: Average unit cost today: 21.90p/kWh
2024-01-08 16:11:06.859385 INFO pv_opt: Output written to sensor.pvopt_unit_cost_today
2024-01-08 16:11:07.031338 INFO pv_opt: Output written to sensor.pvopt_base_cost
2024-01-08 16:11:07.248266 INFO pv_opt: Output written to sensor.pvopt_opt_cost
2024-01-08 16:11:07.284546 INFO pv_opt: Output written to sensor.pvopt_charge_start
2024-01-08 16:11:07.314248 INFO pv_opt: Output written to sensor.pvopt_charge_end
2024-01-08 16:11:07.348543 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-01-08 16:11:07.375376 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-01-08 16:11:07.396676 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-01-08 16:11:07.417930 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-01-08 16:11:07.441971 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-01-08 16:11:07.451588 INFO pv_opt: Read only mode enabled. Not querying inverter.
2024-01-08 16:11:09.088366 INFO pv_opt: State change detected for number.pvopt_battery_capacity_wh [config item: battery_capacity_wh] from 10000 to 9600:
2024-01-08 16:11:09.156604 INFO pv_opt: 
2024-01-08 16:11:09.171856 INFO pv_opt: Previous message repeated 1 times
2024-01-08 16:11:09.173016 INFO pv_opt: Found Octopus Savings Events entity: event.octopus_energy_a_f1f23c9b_octoplus_saving_session_events
2024-01-08 16:11:09.177058 INFO pv_opt: 
2024-01-08 16:11:09.179679 INFO pv_opt: No upcoming Octopus Saving Events detected or joined:
2024-01-08 16:11:09.182466 INFO pv_opt: 
2024-01-08 16:11:09.185036 INFO pv_opt: Starting Opimisation with discharge enabled
2024-01-08 16:11:09.187028 INFO pv_opt: -------------------------------------------
2024-01-08 16:11:09.189039 INFO pv_opt: 
2024-01-08 16:11:09.191717 INFO pv_opt: Checking tariffs:
2024-01-08 16:11:09.193780 INFO pv_opt: -----------------
2024-01-08 16:11:09.214510 INFO pv_opt:   Import: E-1R-AGILE-FLEX-22-11-25-A               Start: 2023-12-29 13:00:00+0000 End: 2024-01-08 23:00:00+0000 
2024-01-08 16:11:09.217332 INFO pv_opt:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-A         Start: 2019-05-15 23:00:00+0000 End: N/A 
2024-01-08 16:11:09.219465 INFO pv_opt: AGILE tariff detected. Rates will update at 16:00 daily
2024-01-08 16:11:09.221456 INFO pv_opt:   - Tariffs OK
2024-01-08 16:11:09.223379 INFO pv_opt: 
2024-01-08 16:11:09.235542 INFO pv_opt: Solcast forecast loaded OK
2024-01-08 16:11:09.238565 INFO pv_opt: Getting expected consumption data
2024-01-08 16:11:09.475858 INFO pv_opt:   - Estimated consumption from sensor.solis_house_load_today loaded OK 
2024-01-08 16:11:09.542710 INFO pv_opt: Initial SOC: 55.5
2024-01-08 16:11:09.545339 INFO pv_opt: Calculating Base flows
2024-01-08 16:11:09.609234 INFO pv_opt: Base cost: 242.29p
2024-01-08 16:11:09.612528 INFO pv_opt: 
2024-01-08 16:11:09.615517 INFO pv_opt: Optimising for Solcast forecast from 08-Jan 16:00 to 09-Jan 23:30
2024-01-08 16:11:09.723313 INFO pv_opt:   Optimiser prices loaded for period 08/01 16:00 - 09/01 23:30
2024-01-08 16:11:09.831410 INFO pv_opt: 
2024-01-08 16:11:09.836739 INFO pv_opt: High Cost Usage Swaps
2024-01-08 16:11:09.842189 INFO pv_opt: ---------------------
2024-01-08 16:11:09.845289 INFO pv_opt: 
2024-01-08 16:11:09.994056 INFO pv_opt: 09/01 06:00:  0.89 kWh at  15.00p. <==> 09/01 02:00: 15.87p/kWh 14.12p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 19.2% Net:  243.9
2024-01-08 16:11:10.079766 INFO pv_opt: 09/01 06:00:  0.89 kWh at  15.00p. <==> 09/01 02:00: 15.87p/kWh 14.12p  SOC:  15.0%-> 23.4% New SOC:  15.0%-> 23.4% Net:  245.5
2024-01-08 16:11:10.168248 INFO pv_opt: 09/01 06:00:  0.89 kWh at  15.00p. <==> 09/01 02:00: 15.87p/kWh 14.12p  SOC:  15.0%-> 31.9% New SOC:  15.0%-> 27.7% Net:  247.1
2024-01-08 16:11:10.251078 INFO pv_opt: 09/01 06:00:  0.89 kWh at  15.00p. <==> 09/01 02:00: 15.87p/kWh 14.12p  SOC:  15.0%-> 40.3% New SOC:  15.0%-> 31.9% Net:  248.4
2024-01-08 16:11:10.407436 INFO pv_opt: 09/01 06:00:  0.89 kWh at  15.00p. <==> 09/01 02:00: 15.87p/kWh 14.12p  SOC:  15.0%-> 48.7% New SOC:  15.0%-> 36.1% Net:  248.4
2024-01-08 16:11:10.679294 INFO pv_opt: 09/01 07:30:  0.55 kWh at  11.30p. <==> 09/01 02:00: 15.87p/kWh  8.75p  SOC:  15.0%-> 57.2% New SOC:  15.0%-> 38.7% Net:  247.9
2024-01-08 16:11:10.782010 INFO pv_opt: 09/01 07:30:  0.55 kWh at  11.30p. <==> 09/01 03:00: 15.90p/kWh  8.77p  SOC:  62.4%-> 49.1% New SOC:  62.4%-> 65.0% Net:  237.8
2024-01-08 16:11:10.954915 INFO pv_opt: 
2024-01-08 16:11:10.958225 INFO pv_opt: Maximum 1st pass slot delta is 1.6p
2024-01-08 16:11:10.961623 INFO pv_opt: 
2024-01-08 16:11:10.986311 INFO pv_opt: Previous message repeated 1 times
2024-01-08 16:11:10.987336 INFO pv_opt: Low Cost Charging
2024-01-08 16:11:10.990023 INFO pv_opt: ------------------
2024-01-08 16:11:10.992772 INFO pv_opt: 
2024-01-08 16:11:10.995497 INFO pv_opt: Max export price when there is no forced charge: 15.00p/kWh.
2024-01-08 16:11:10.999537 INFO pv_opt: 0 slots have an import price less than the max export price
2024-01-08 16:11:11.004117 INFO pv_opt: 
2024-01-08 16:11:11.009529 INFO pv_opt: Charge net cost delta: -0.0p: < threshold 4.0 => Excluded
2024-01-08 16:11:11.013469 INFO pv_opt: 
[... cut ...]
2024-01-08 16:11:14.636870 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-01-08 16:11:14.676127 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-01-08 16:11:14.705001 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-01-08 16:11:14.727948 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-01-08 16:11:14.847136 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-01-08 16:11:14.863317 INFO pv_opt: Read only mode enabled. Not querying inverter.

Thanks again for your help with this.

solarjones, 2024-01-08:
Something odd happening with the last information I provided - it's as if the config was lost. The above had battery discharge enabled for some reason. I just restarted AppDaemon and it now has a different charging plan! See below:

2024-01-08 16:31:56.123206 INFO pv_opt: Finished loading contract
2024-01-08 16:31:56.402382 INFO pv_opt: 
2024-01-08 16:31:56.406116 INFO pv_opt: Running initial Optimisation:
2024-01-08 16:31:56.409579 INFO pv_opt: 
2024-01-08 16:31:56.464300 INFO pv_opt: Previous message repeated 1 times
2024-01-08 16:31:56.464605 INFO pv_opt: Found Octopus Savings Events entity: event.octopus_energy_a_f1f23c9b_octoplus_saving_session_events
2024-01-08 16:31:56.473122 INFO pv_opt: 
2024-01-08 16:31:56.474897 INFO pv_opt: No upcoming Octopus Saving Events detected or joined:
2024-01-08 16:31:56.483191 INFO pv_opt: 
2024-01-08 16:31:56.484707 INFO pv_opt: Starting Opimisation with discharge disabled
2024-01-08 16:31:56.490384 INFO pv_opt: --------------------------------------------
2024-01-08 16:31:56.496131 INFO pv_opt: 
2024-01-08 16:31:56.497687 INFO pv_opt: Checking tariffs:
2024-01-08 16:31:56.524349 INFO pv_opt: -----------------
2024-01-08 16:31:56.609303 INFO pv_opt:   Import: E-1R-AGILE-FLEX-22-11-25-A               Start: 2023-12-30 13:00:00+0000 End: 2024-01-09 23:00:00+0000 
2024-01-08 16:31:56.619372 INFO pv_opt:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-A         Start: 2019-05-15 23:00:00+0000 End: N/A 
2024-01-08 16:31:56.628408 INFO pv_opt: AGILE tariff detected. Rates will update at 16:00 daily
2024-01-08 16:31:56.642590 INFO pv_opt:   - Tariffs OK
2024-01-08 16:31:56.649278 INFO pv_opt: 
2024-01-08 16:31:56.662468 INFO pv_opt: Solcast forecast loaded OK
2024-01-08 16:31:56.665367 INFO pv_opt: Getting expected consumption data
2024-01-08 16:31:59.574928 INFO pv_opt:   - Estimated consumption from sensor.solis_house_load_today loaded OK 
2024-01-08 16:31:59.705615 INFO pv_opt: Initial SOC: 52.5
2024-01-08 16:31:59.709242 INFO pv_opt: Calculating Base flows
2024-01-08 16:31:59.791859 INFO pv_opt: Base cost: 224.99p
2024-01-08 16:31:59.794559 INFO pv_opt: 
2024-01-08 16:31:59.797296 INFO pv_opt: Optimising for Solcast forecast from 08-Jan 16:30 to 09-Jan 23:30
2024-01-08 16:31:59.874674 INFO pv_opt:   Optimiser prices loaded for period 08/01 16:30 - 09/01 23:30
2024-01-08 16:31:59.964225 INFO pv_opt: 
2024-01-08 16:31:59.967072 INFO pv_opt: High Cost Usage Swaps
2024-01-08 16:31:59.969647 INFO pv_opt: ---------------------
2024-01-08 16:31:59.971957 INFO pv_opt: 
2024-01-08 16:32:00.123067 INFO pv_opt: 09/01 07:30:  0.68 kWh at  13.82p. <==> 09/01 03:30: 14.90p/kWh 10.08p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 21.4% Net:  225.4
2024-01-08 16:32:00.323843 INFO pv_opt: 09/01 07:30:  0.68 kWh at  13.82p. <==> 09/01 03:30: 14.90p/kWh 10.08p  SOC:  15.0%-> 21.4% New SOC:  15.0%-> 27.8% Net:  226.2
2024-01-08 16:32:00.471151 INFO pv_opt: 09/01 07:30:  0.68 kWh at  13.82p. <==> 09/01 03:30: 14.90p/kWh 10.08p  SOC:  15.0%-> 27.8% New SOC:  15.0%-> 34.2% Net:  227.0
2024-01-08 16:32:00.546138 INFO pv_opt: 09/01 07:30:  0.68 kWh at  13.82p. <==> 09/01 03:30: 14.90p/kWh 10.08p  SOC:  15.0%-> 34.2% New SOC:  15.0%-> 38.7% Net:  227.6
2024-01-08 16:32:00.632295 INFO pv_opt: 09/01 07:30:  0.68 kWh at  13.82p. <==> 08/01 23:30: 15.21p/kWh 10.29p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 16.1% Net:  227.0
2024-01-08 16:32:00.847169 INFO pv_opt: 09/01 07:30:  0.68 kWh at  13.82p. <==> 08/01 23:30: 15.21p/kWh 10.29p  SOC:  15.0%-> 22.7% New SOC:  15.0%-> 17.1% Net:  227.0
2024-01-08 16:32:00.985159 INFO pv_opt: 09/01 07:30:  0.61 kWh at  12.48p. <==> 08/01 23:30: 15.21p/kWh  9.29p  SOC:  15.0%-> 24.8% New SOC:  15.0%-> 18.1% Net:  226.4
2024-01-08 16:32:01.101984 INFO pv_opt: 09/01 07:30:  0.41 kWh at   8.30p. <==> 08/01 23:30: 15.21p/kWh  6.18p  SOC:  15.0%-> 26.8% New SOC:  15.0%-> 18.7% Net:  226.0
2024-01-08 16:32:01.252231 INFO pv_opt: 09/01 08:00:  0.36 kWh at   6.65p. <==> 08/01 23:30: 15.21p/kWh  5.42p  SOC:  15.0%-> 28.0% New SOC:  15.0%-> 19.3% Net:  225.7
2024-01-08 16:32:01.333001 INFO pv_opt: 09/01 08:00:  0.36 kWh at   6.65p. <==> 08/01 23:30: 15.21p/kWh  5.42p  SOC:  15.0%-> 29.2% New SOC:  15.0%-> 19.9% Net:  225.4
2024-01-08 16:32:01.422067 INFO pv_opt: 09/01 08:00:  0.36 kWh at   6.65p. <==> 08/01 23:30: 15.21p/kWh  5.42p  SOC:  15.0%-> 30.3% New SOC:  15.0%-> 20.4% Net:  225.2
2024-01-08 16:32:01.535491 INFO pv_opt: 09/01 08:00:  0.27 kWh at   5.06p. <==> 08/01 23:30: 15.21p/kWh  4.12p  SOC:  15.0%-> 31.4% New SOC:  15.0%-> 20.9% Net:  225.2
2024-01-08 16:32:01.626706 INFO pv_opt: 09/01 08:00:  0.18 kWh at   3.37p. <==> 08/01 23:30: 15.21p/kWh  2.75p  SOC:  15.0%-> 32.3% New SOC:  15.0%-> 21.1% Net:  225.1
2024-01-08 16:32:01.741210 INFO pv_opt: 09/01 08:00:  0.12 kWh at   2.25p. <==> 08/01 23:30: 15.21p/kWh  1.84p  SOC:  15.0%-> 32.8% New SOC:  15.0%-> 21.3% Net:  225.1
2024-01-08 16:32:01.839511 INFO pv_opt: 09/01 08:00:  0.08 kWh at   1.51p. <==> 08/01 23:30: 15.21p/kWh  1.23p  SOC:  15.0%-> 33.2% New SOC:  15.0%-> 21.5% Net:  225.1
2024-01-08 16:32:01.926757 INFO pv_opt: 09/01 08:00:  0.05 kWh at   1.00p. <==> 08/01 23:30: 15.21p/kWh  0.82p  SOC:  15.0%-> 33.5% New SOC:  15.0%-> 21.5% Net:  225.0
2024-01-08 16:32:02.046571 INFO pv_opt: 09/01 08:00:  0.04 kWh at   0.67p. <==> 08/01 23:30: 15.21p/kWh  0.54p  SOC:  15.0%-> 33.6% New SOC:  15.0%-> 21.6% Net:  225.0
2024-01-08 16:32:02.075964 INFO pv_opt: 
[... cut ...]
2024-01-08 16:32:06.143913 INFO pv_opt: Read only mode enabled. Not querying inverter.
2024-01-08 16:32:06.187164 INFO pv_opt: Optimiser will run every 10 minutes from 16:40 or on PV_OPT Event
2024-01-08 16:32:06.193913 INFO AppDaemon: App initialization complete
2024-01-08 16:32:56.130827 INFO pv_opt: >>> Agile Callback Handler

[image]
[image]

This makes more sense with higher charging rate during the cheapest slot. 27% current prediction for battery charge starting the day, with good solar forecast.

fboundy, 2024-01-08:
Agree that looks pretty sensible it's charging as much as possible in the cheapest slot and then using the next cheapest for the rest of what's needed. When there are multiple slots at the same price it just spreads the load across them so that the charge current is lower.  

Keep me posted if is starts to misbehave.

solarjones, 2024-01-11:
What do you think of this evenings charging scheme? Unsure why its missing the cheapest slot.

[image]
[image]

2024-01-11 20:30:00.639997 INFO pv_opt: Base cost: 664.08p
2024-01-11 20:30:00.642569 INFO pv_opt: 
2024-01-11 20:30:00.648012 INFO pv_opt: Optimising for Solcast_p10 forecast from 11-Jan 20:30 to 12-Jan 23:30
2024-01-11 20:30:00.848746 INFO pv_opt:   Optimiser prices loaded for period 11/01 20:30 - 12/01 23:30
2024-01-11 20:30:00.965825 INFO pv_opt: 
2024-01-11 20:30:00.969571 INFO pv_opt: High Cost Usage Swaps
2024-01-11 20:30:00.973478 INFO pv_opt: ---------------------
2024-01-11 20:30:00.975327 INFO pv_opt: 
2024-01-11 20:30:01.086252 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 03:00: 15.41p/kWh 17.70p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 20.4% Net:  665.7
2024-01-11 20:30:01.185711 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 03:00: 15.41p/kWh 17.70p  SOC:  15.0%-> 25.9% New SOC:  15.0%-> 25.9% Net:  664.6
2024-01-11 20:30:01.297692 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 03:00: 15.41p/kWh 17.70p  SOC:  15.0%-> 36.8% New SOC:  15.0%-> 31.3% Net:  659.6
2024-01-11 20:30:01.412501 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 03:00: 15.41p/kWh 17.70p  SOC:  15.0%-> 47.6% New SOC:  15.0%-> 36.3% Net:  652.4
2024-01-11 20:30:01.511654 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 02:00: 15.45p/kWh 17.73p  SOC:  15.0%-> 15.0% New SOC:  15.0%-> 20.4% Net:  644.9
2024-01-11 20:30:01.595364 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 02:00: 15.45p/kWh 17.73p  SOC:  15.0%-> 25.9% New SOC:  15.0%-> 25.9% Net:  638.6
2024-01-11 20:30:01.684571 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 02:00: 15.45p/kWh 17.73p  SOC:  15.0%-> 36.8% New SOC:  15.0%-> 31.3% Net:  634.5
2024-01-11 20:30:01.778942 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 02:00: 15.45p/kWh 17.73p  SOC:  15.0%-> 47.6% New SOC:  15.0%-> 36.3% Net:  632.9
2024-01-11 20:30:01.867778 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 04:00: 15.73p/kWh 18.06p  SOC: 100.0%-> 91.2% New SOC: 100.0%->100.0% Net:  630.1
2024-01-11 20:30:01.959324 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 05:00: 16.33p/kWh 18.76p  SOC: 100.0%-> 92.2% New SOC: 100.0%->100.0% Net:  626.7
2024-01-11 20:30:02.066461 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 12:00: 18.45p/kWh 21.18p  SOC:  32.8%-> 27.2% New SOC:  32.8%-> 38.2% Net:  616.4
2024-01-11 20:30:02.182163 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 12:00: 18.45p/kWh 21.18p  SOC:  32.8%-> 43.7% New SOC:  32.8%-> 43.7% Net:  598.7
2024-01-11 20:30:02.290624 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 12:00: 18.45p/kWh 21.18p  SOC:  32.8%-> 54.6% New SOC:  32.8%-> 49.1% Net:  581.0
2024-01-11 20:30:02.401153 INFO pv_opt: 12/01 18:00:  1.15 kWh at  37.58p. <==> 12/01 12:00: 18.45p/kWh 21.18p  SOC:  32.8%-> 65.4% New SOC:  32.8%-> 54.1% Net:  565.9
2024-01-11 20:30:02.507229 INFO pv_opt: 12/01 18:30:  0.74 kWh at  24.18p. <==> 12/01 11:00: 18.90p/kWh 13.96p  SOC:  39.4%-> 32.8% New SOC:  39.4%-> 42.9% Net:  551.0
2024-01-11 20:30:02.627663 INFO pv_opt: 12/01 19:30:  0.68 kWh at  13.34p. <==> 12/01 11:00: 18.90p/kWh 12.78p  SOC:  39.4%-> 46.4% New SOC:  39.4%-> 46.1% Net:  550.5
2024-01-11 20:30:02.860923 INFO pv_opt: 12/01 19:30:  0.33 kWh at   6.45p. <==> 12/01 11:00: 18.90p/kWh  6.18p  SOC:  39.4%-> 52.9% New SOC:  39.4%-> 47.7% Net:  550.2
2024-01-11 20:30:02.891547 INFO pv_opt: 
2024-01-11 20:30:02.894948 INFO pv_opt: Maximum 1st pass slot delta is 1.6p
2024-01-11 20:30:02.897901 INFO pv_opt: 
2024-01-11 20:30:02.926055 INFO pv_opt: Previous message repeated 1 times
2024-01-11 20:30:02.928671 INFO pv_opt: Low Cost Charging
2024-01-11 20:30:02.933854 INFO pv_opt: ------------------
2024-01-11 20:30:02.938059 INFO pv_opt: 
2024-01-11 20:30:02.940920 INFO pv_opt: Max export price when there is no forced charge: 15.00p/kWh.
2024-01-11 20:30:02.945328 INFO pv_opt: 0 slots have an import price less than the max export price
2024-01-11 20:30:02.948832 INFO pv_opt: 
2024-01-11 20:30:02.952077 INFO pv_opt: Charge net cost delta: -0.0p: < threshold 4.0 => Excluded
2024-01-11 20:30:02.955265 INFO pv_opt: Iteration 11: Slots added:   0
2024-01-11 20:30:02.959141 INFO pv_opt: 
2024-01-11 20:30:02.962813 INFO pv_opt: Removing cyclic charge/discharge
2024-01-11 20:30:03.041623 INFO pv_opt:   Net cost revised from 550.2p to 566.9p
2024-01-11 20:30:03.124344 INFO pv_opt: 
2024-01-11 20:30:03.294925 INFO pv_opt: Checking for Hold SOC slots
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:1572: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame

See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  self.windows["hold_soc"].loc[
2024-01-11 20:30:03.307508 INFO pv_opt: 
2024-01-11 20:30:03.314884 INFO pv_opt: Optimal forced charge/discharge slots:
2024-01-11 20:30:03.320514 INFO pv_opt:    12-Jan 02:00 - 12-Jan 04:30   Power:  4500W  SOC:   15% ->  100%  
2024-01-11 20:30:03.324176 INFO pv_opt:    12-Jan 04:30 - 12-Jan 05:30   Power:   800W  SOC:  100% ->  100%  <=
2024-01-11 20:30:03.330170 INFO pv_opt:    12-Jan 05:30 - 12-Jan 06:00   Power:   800W  SOC:  100% ->  100%  <=
2024-01-11 20:30:03.335660 INFO pv_opt:    12-Jan 11:00 - 12-Jan 12:00   Power:  1700W  SOC:   39% ->   56%  
[... cut ...]
2. Perhaps my electricity usage pattern isn't consistent with 2x children. Somethings the washing machine/tumble drier has to go on at suboptimal times but we do try to run appliances over night to coincide with the cheapest rates. Same for the electric oven or induction hub, we may eat at different times of the evening or not cook etc.  Is there a way to switch on/off the predicted historic usage for the charging model? Perhaps it may help understand what it thinks my daily usage will be as a total (I can see the +15m prediction.

solarjones, 2024-01-14:
Unsure why the batteries have charged past the end SOC.

[image]

2024-01-14 22:50:04.894319 INFO pv_opt: 
2024-01-14 22:50:04.897971 INFO pv_opt: Plan time: 14-Jan 22:30 - 15-Jan 23:30 Initial SOC: 14.5 Base Cost: 464.02 Opt Cost: 354.55
2024-01-14 22:50:04.900990 INFO pv_opt: 
2024-01-14 22:50:04.903696 INFO pv_opt: Optimiser elapsed time 4.8 seconds
2024-01-14 22:50:04.906648 INFO pv_opt: 
2024-01-14 22:50:04.935567 INFO pv_opt: Previous message repeated 1 times
2024-01-14 22:50:04.946853 INFO pv_opt: Output written to sensor.pvopt_optimiser_elapsed
2024-01-14 22:50:05.192428 INFO pv_opt: Average unit cost today: 14.80p/kWh
2024-01-14 22:50:05.212716 INFO pv_opt: Output written to sensor.pvopt_unit_cost_today
2024-01-14 22:50:05.440604 INFO pv_opt: Output written to sensor.pvopt_base_cost
2024-01-14 22:50:05.661724 INFO pv_opt: Output written to sensor.pvopt_opt_cost
2024-01-14 22:50:05.712402 INFO pv_opt: Output written to sensor.pvopt_charge_start
2024-01-14 22:50:05.735580 INFO pv_opt: Output written to sensor.pvopt_charge_end
2024-01-14 22:50:05.762942 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-01-14 22:50:08.524730 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-01-14 22:50:08.554671 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-01-14 22:50:08.582445 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-01-14 22:50:08.609345 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-01-14 22:50:08.649569 INFO pv_opt: 
2024-01-14 22:50:08.652402 INFO pv_opt: Current inverter status:
2024-01-14 22:50:08.655931 INFO pv_opt: ------------------------
2024-01-14 22:50:08.658393 INFO pv_opt:   mode              : Timed Charge/Discharge
2024-01-14 22:50:08.660704 INFO pv_opt:   code              : 35
2024-01-14 22:50:08.663012 INFO pv_opt:   switches          :
2024-01-14 22:50:08.665300 INFO pv_opt:     SelfUse         : True
2024-01-14 22:50:08.667592 INFO pv_opt:     Timed           : True
2024-01-14 22:50:08.676384 INFO pv_opt:     OffGrid         : False
2024-01-14 22:50:08.680157 INFO pv_opt:     BatteryWake     : False
2024-01-14 22:50:08.682845 INFO pv_opt:     Backup          : False
2024-01-14 22:50:08.685237 INFO pv_opt:     GridCharge      : True
2024-01-14 22:50:08.688113 INFO pv_opt:     FeedInPriority  : False
2024-01-14 22:50:08.690582 INFO pv_opt:   charge            :
2024-01-14 22:50:08.692926 INFO pv_opt:     start           : 14-Jan 22:30
2024-01-14 22:50:08.697124 INFO pv_opt:     end             : 14-Jan 23:00
2024-01-14 22:50:08.700841 INFO pv_opt:     current         : 44.7
2024-01-14 22:50:08.703992 INFO pv_opt:     active          : True
2024-01-14 22:50:08.707690 INFO pv_opt:   discharge         :
2024-01-14 22:50:08.710299 INFO pv_opt:     start           : 14-Jan 22:20
2024-01-14 22:50:08.712655 INFO pv_opt:     end             : 14-Jan 22:20
2024-01-14 22:50:08.714955 INFO pv_opt:     current         : 0.0
2024-01-14 22:50:08.717253 INFO pv_opt:     active          : False
2024-01-14 22:50:08.719488 INFO pv_opt:   hold_soc          :
2024-01-14 22:50:08.722095 INFO pv_opt:     active          : False
2024-01-14 22:50:08.724418 INFO pv_opt:     soc             : 97.0
2024-01-14 22:50:08.726675 INFO pv_opt: 
2024-01-14 22:50:08.729193 INFO pv_opt: Current charge/discharge windows ends in 9.9 minutes.
2024-01-14 22:50:08.735735 INFO pv_opt: >>> SelfUse: True
2024-01-14 22:50:08.738865 INFO pv_opt: >>> Timed: True
2024-01-14 22:50:08.741571 INFO pv_opt: >>> Backup: False
2024-01-14 22:50:08.744200 INFO pv_opt: >>> GridCharge: True
2024-01-14 22:50:08.749499 INFO pv_opt: Updating inverter charge control:
2024-01-14 22:50:08.756525 INFO pv_opt:   start: None
2024-01-14 22:50:08.760146 INFO pv_opt:   end: 2024-01-14 23:00:00+00:00
2024-01-14 22:50:08.764771 INFO pv_opt:   power: 1000
2024-01-14 22:50:08.769794 INFO pv_opt: Inverter already at correct time settings
2024-01-14 22:50:08.776198 INFO pv_opt: Power 1000 = 19.5A at 51.2V
2024-01-14 22:50:09.294684 INFO pv_opt: Current 19.5 written to inverter
2024-01-14 22:50:09.327140 INFO pv_opt: Wating for Modbus Read cycle: 18 seconds
2024-01-14 22:50:32.727219 INFO pv_opt: 
2024-01-14 22:50:32.730330 INFO pv_opt: Current inverter status:
2024-01-14 22:50:32.732710 INFO pv_opt: ------------------------
2024-01-14 22:50:32.735044 INFO pv_opt:   mode              : Timed Charge/Discharge
2024-01-14 22:50:32.737333 INFO pv_opt:   code              : 35
2024-01-14 22:50:32.739578 INFO pv_opt:   switches          :
2024-01-14 22:50:32.741882 INFO pv_opt:     SelfUse         : True
2024-01-14 22:50:32.744139 INFO pv_opt:     Timed           : True
[... 118 lines cut ...]
2024-01-14 23:00:15.071496 INFO pv_opt:     SelfUse         : True
2024-01-14 23:00:15.076938 INFO pv_opt:     Timed           : True
2024-01-14 23:00:15.080599 INFO pv_opt:     OffGrid         : False
2024-01-14 23:00:15.083119 INFO pv_opt:     BatteryWake     : False
2024-01-14 23:00:15.085517 INFO pv_opt:     Backup          : False
2024-01-14 23:00:15.087897 INFO pv_opt:     GridCharge      : True
2024-01-14 23:00:15.091016 INFO pv_opt:     FeedInPriority  : False
2024-01-14 23:00:15.093524 INFO pv_opt:   charge            :
2024-01-14 23:00:15.095885 INFO pv_opt:     start           : 14-Jan 22:30
2024-01-14 23:00:15.098355 INFO pv_opt:     end             : 14-Jan 23:00
2024-01-14 23:00:15.100912 INFO pv_opt:     current         : 19.5
2024-01-14 23:00:15.103243 INFO pv_opt:     active          : False
[... cut ...]
2024-01-14 23:00:15.111553 INFO pv_opt:     end             : 14-Jan 22:20
2024-01-14 23:00:15.114775 INFO pv_opt:     current         : 0.0
2024-01-14 23:00:15.118761 INFO pv_opt:     active          : False
2024-01-14 23:00:15.122474 INFO pv_opt:   hold_soc          :
2024-01-14 23:00:15.125456 INFO pv_opt:     active          : False
2024-01-14 23:00:15.128876 INFO pv_opt:     soc             : 97.0
2024-01-14 23:00:15.131389 INFO pv_opt: 
2024-01-14 23:00:15.134097 INFO pv_opt: Next charge window starts in 59.7 minutes. Nothing to do.

Hope this helps.
Thanks.

solarjones, 2024-01-23:
Back on negative costs again with Agile :)
But PV_Opt doesn't seem to want to charge during these periods...

[image]

fboundy, 2024-01-23:
I’m away at present so I haven’t been able to look at this I afraid

solarjones, 2024-01-23:
No problem!  I was logging it for reference.  We all have lives to lead :)  Very grateful you have written this to optimise everything.

Thanks.

solarjones, 2024-01-29:
Since forcing a restart of AppDaemon periodically (and more recently a daily automation), PV_OPT now seems to be inline with the correct cheapest time slots and behaving as designed.  Unsure what would cause the app to get itself out of sync but if I don't force a restart it will pick the wrong charging slots, but once AppDaemon is restarted, the recalculations are spot on.

fboundy, 2024-01-29:
I’ve seen a bug where it fails to pick up the new Agile prices correctly and then by default uses the previous day’s until a restart but I haven’t tracked down what causes it. This sounds similar. You can tell because the BottlecapDave OE pricing doesn’t match PV_Opt’s

I haven’t seen it for a while though.

fboundy, 2024-01-30:
I think this is the same as  #88 so I am closing this one. Please post any useful logs/screenshots to that issue

fboundy, 2024-02-01:
Please check release `3.7.0 beta 01` which I hope fixes this and has additional logging to debug

closed 2024-02-05 by fboundy