stevebuk1/pv_opt #91: Check Charge Times and Mode

fboundy, 2024-01-04:
**Describe the bug**
The app isn't fully checking that the charge/discharge times and inverter mode are correct on each run.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

solarjones, 2024-01-05:
Thanks. As I've only just switched from ReadOnly to Live modes the other day, this may be a pre-existing issue to v3.5.1.
I've not looked at the code so not sure how the desired charging Amps are calculated for the relevant time slots. It didn't appear to change the Optimum Charge Current when I lowered the Charging Power. Perhaps it needs to limit based on max charging current as a new parameter? 

My current settings:
[image]

Also spotted some random force discharge times being set (they had the same start and end times) despite Optimise Discharge being disabled.

Not your issue but I'll have another attempt at getting AppDaemon logging working, just refuses to write a log file anywhere!

fboundy, 2024-01-30:
Yes, you are right it needs to check max charge current. Added to the list

closed 2024-02-23 by fboundy