stevebuk1/pv_opt #94: Discharge slots take estimated consumption into consideration ?

Bart39, 2024-01-26:
When I have a bit more time I will add some logs / screenshots but for context I am only using it in read only mode at the moment and using my own automations.
What I am seeing though for discharge slots is that it will schedule a slot during peak Import/Export costs at 4PM at full power (6000w). Because it doesn't appear to take the consumption into account this means it will drain my battery before the peak finishes at 7PM.
Is this intended ?
Thanks
Mark

fboundy, 2024-01-30:
That's not intended behaviour. Please add some screenshots/log files so I can diagnose

fboundy, 2024-02-11:
Closing this as I believe recent updates should fix this. Please re-open or raise a new issue with logs if it re-occurs

closed 2024-02-11 by fboundy