stevebuk1/pv_opt #95: Stuck on ‘Loading Tariffs’

lazy-pete, 2024-01-29:
**Describe the bug**
My setup appears to always be stuck on ‘Loading Tarrifs’ , which is presumably why some sensors are also missing. 

**To Reproduce**
Steps to reproduce the behavior:
Log File - 

T23:00
29-01-2024
28-01-2024
27-01-2024
26-01-2024
25-01-2024
24-01-2024
23-01-2024
22-01-2024
29-01-2024
28-01-2024
27-01-2024
26-01-2024
25-01-2024
24-01-2024
23-01-2024
22-01-2024
2024-01-28 19:44:08.960443 INFO pv_opt: 
2024-01-28 19:44:08.961274 INFO pv_opt: Retrieved day ahead forecast for period 21/01 23:00 - 29/01 22:00 for tariff E-1R-AGILE-FLEX-22-11-25-N
2024-01-28 19:44:09.002330 WARNING pv_opt: ------------------------------------------------------------
2024-01-28 19:44:09.002452 WARNING pv_opt: Unexpected error running initialize() for pv_opt
2024-01-28 19:44:09.002521 WARNING pv_opt: ------------------------------------------------------------
2024-01-28 19:44:09.003181 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 292, in initialize
    self._compare_tariffs()
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1923, in _compare_tariffs
    opt = self.pv_system.optimised_force(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 614, in optimised_force
    max_slot = import_cost[import_cost == max_import_cost].index[0]
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 5366, in __getitem__
    return getitem(key)
           ^^^^^^^^^^^^
IndexError: index 0 is out of bounds for axis 0 with size 0

2024-01-28 19:44:09.003267 WARNING pv_opt: ------------------------------------------------------------
2024-01-28 22:34:23.446573 WARNING AppDaemon: Excessive time spent in utility loop: 2235.0ms, 1.0ms in check_app_updates(), 2234.0ms in other

**Expected behavior**
I had presumed that setting my local Tariff codes and my API / account details etc into AppDaemon would have this running smoothly, but it seems not. Hasn’t worked yet. 

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

SzosszeNET, 2024-01-29:
@lazy-pete , this I believe is the same as https://github.com/fboundy/pv_opt/issues/93

fboundy, 2024-01-29:
Can I confirm whether you have the “alt_tariffs” parameter in your config.yaml file? If you do can you try commenting it out please?

If it’s not there then let me know and I will try to diagnose further. May need a bit more logging to debug fully.

SzosszeNET, 2024-01-29:
Thank you @fboundy YES! the alt_tariffs was the issue

fboundy, 2024-01-29:
Good - that’s a bit experimental and I should comment it out in the repo. I’ll do a patch when I get a minute

lazy-pete, 2024-01-29:
Hi - thanks forgetting back to me. I have this in my yaml - 

# Tariff comparison
  id_daily_solar: sensor.{device_name}_power_generation_today
  alt_tariffs:
    - name: Eco7_Agile
      octopus_import_tariff_code: E-2R-VAR-22-11-01-G
      octopus_export_tariff_code: E-1R-AGILE-OUTGOING-19-05-13-G

    - name: Flux
      octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
      octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G

Changed to the below - I'm hoping I've grasped 'comment out' properly. Now says 'optimising charge plan' ,although it has said that before from time to time. 

I'll try to check the solved cases in future.

# Tariff comparison
  # id_daily_solar: sensor.{device_name}_power_generation_today
  # alt_tariffs:
  #   - name: Eco7_Agile
  #     octopus_import_tariff_code: E-2R-VAR-22-11-01-G
  #     octopus_export_tariff_code: E-1R-AGILE-OUTGOING-19-05-13-G
  #
  #   - name: Flux
  #     octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
  #     octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G

SzosszeNET, 2024-01-29:
@lazy-pete , just from alt_tariffs, the id_daily_solar can/should stay.
Also I found once updated the config restarting appdaemon was quick to pickup the new config

lazy-pete, 2024-01-29:
Thanks - now as below, with Appdaemon reboot log. Did make we wonder whether I should have been going through all the other lines and putting 'solax' where it says {device_name}  , plus almost every other line above is also 'commented out' . I may have failed to grasp how much post-install config is perhaps necessary ?

Cheers,

Pete

[image]

Sitting in 'optimising charge plan' with the below results ...

[image]

Downloading pandas-2.2.0-cp311-cp311-musllinux_1_1_x86_64.whl (13.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.9/13.9 MB 9.2 MB/s eta 0:00:00
Installing collected packages: tzdata, pandas
Successfully installed pandas-2.2.0 tzdata-2023.4
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[23:21:56] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-01-29 23:21:57.867242 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-01-29 23:21:57.867342 INFO AppDaemon: Python version is 3.11.6
2024-01-29 23:21:57.867406 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-01-29 23:21:57.867464 INFO AppDaemon: Added log: AppDaemon
2024-01-29 23:21:57.867522 INFO AppDaemon: Added log: Error
2024-01-29 23:21:57.867571 INFO AppDaemon: Added log: Access
2024-01-29 23:21:57.867620 INFO AppDaemon: Added log: Diag
2024-01-29 23:21:57.944596 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-01-29 23:21:58.079692 INFO HASS: HASS Plugin Initializing
2024-01-29 23:21:58.079800 WARNING HASS: ha_url not found in HASS configuration - module not initialized
2024-01-29 23:21:58.079865 INFO HASS: HASS Plugin initialization complete
2024-01-29 23:21:58.079966 INFO AppDaemon: Loading Plugin MQTT using class MqttPlugin from module mqttplugin
2024-01-29 23:21:58.109801 INFO MQTT: MQTT Plugin Initializing
2024-01-29 23:21:58.110095 INFO MQTT: Using 'localad/status' as Will Topic
2024-01-29 23:21:58.110169 INFO MQTT: Using 'localad/status' as Birth Topic
2024-01-29 23:21:58.110377 INFO AppDaemon: HTTP is disabled
2024-01-29 23:21:59.172426 INFO HASS: Connected to Home Assistant 2024.1.5
2024-01-29 23:21:59.177945 WARNING AppDaemon: ------------------------------------------------------------
2024-01-29 23:21:59.178239 WARNING AppDaemon: Unexpected error loading config file: /homeassistant/appdaemon/apps/pv_opt/config/config.yaml
2024-01-29 23:21:59.178342 WARNING AppDaemon: ------------------------------------------------------------
2024-01-29 23:21:59.182216 WARNING AppDaemon: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 578, in read_config_file
    return utils.read_config_file(file)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 586, in read_config_file
    return read_yaml_config(path)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 722, in read_yaml_config
    config = yaml.load(config_file_contents, Loader=yaml.SafeLoader)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
[... 17 lines cut ...]
  in "<unicode string>", line 179, column 5:
        id_daily_solar: sensor.solax_pow ... 
        ^

2024-01-29 23:21:59.182791 WARNING AppDaemon: ------------------------------------------------------------
2024-01-29 23:21:59.184015 WARNING AppDaemon: File '/homeassistant/appdaemon/apps/pv_opt/config/config.yaml' invalid structure - ignoring
2024-01-29 23:21:59.184394 INFO AppDaemon: Found 0 active apps
2024-01-29 23:21:59.184515 INFO AppDaemon: Found 0 inactive apps
2024-01-29 23:21:59.184636 INFO AppDaemon: Found 0 global libraries
2024-01-29 23:21:59.184738 INFO AppDaemon: Starting Apps with 0 workers and 0 pins
2024-01-29 23:21:59.204906 INFO HASS: Evaluating startup conditions
2024-01-29 23:21:59.211989 INFO HASS: Startup condition met: hass state=RUNNING
2024-01-29 23:21:59.212119 INFO HASS: All startup conditions met
2024-01-29 23:21:59.232983 INFO AppDaemon: Got initial state from namespace default
2024-01-29 23:21:59.268726 INFO MQTT: Connected to Broker at URL core-mosquitto:1883
[... cut ...]
2024-01-29 23:22:02.192358 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pv_opt/config to module import path
2024-01-29 23:22:02.193363 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py - ignoring
2024-01-29 23:22:02.193615 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/pv_opt/pvpy.py - ignoring
2024-01-29 23:22:02.193894 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/pv_opt/solis.py - ignoring
2024-01-29 23:22:02.194202 INFO AppDaemon: App initialization complete

SzosszeNET, 2024-01-29:
@lazy-pete, line 25 should take care of the device-name as {device_name} is a variable and once you set it correctly at the beginning of the file you should be good.

Two things however. In the pv_opt config file, find the block of values corresponding to your inverter "integration specific config" and uncomment.

Also make sure that under settings/integrations/entities you can find the entries the automation is looking for - say search there for "_battery_voltage" and that should indicate what you should set as device name.

Hope this helps,

Gabor

lazy-pete, 2024-01-29:
@SzosszeNET - thanks - I'll check those points. Edited my last post with additional screenshots and after working out the line 25 bit ! I'll report back.

P

fboundy, 2024-01-30:
This should be fixed with the 3.5.3 but the `config.yaml` file isn't updated by HACS (to preserve your config) so you will need to manually compare and change

closed 2024-01-30 by fboundy

lazy-pete, 2024-02-04:
@SzosszeNET - once you sorted the alt_tarrifs issue you suggested everything just worked. I’m wondering whether that was immediate ? I’ve now upgraded the appdaemon pvopt to the most recent version and copied across the newest config.yaml sections, been through the yaml and uncommented everything that looks relevant (except maximum_dod_percent as that’s in two places and gives an error) - although as my setup is ‘standard’ I don’t think I should need to do that. I am now stuck on ‘Optimising Charge Plan’. I also cannot find any evidence of the output sensors in my entities list - everything else is there. Is the system waiting to gather 7 days of usage data before it fires up ? Does it then create the output sensors ? I can’t find them anywhere in the files to be added to either appdaemon or HA yaml. I’m rather confused at the moment.

solarjones, 2024-02-06:
Randomly, I've just hit this issue of stuck on Loading Tariffs myself.

PV Opt v3.7.1

[error.log](https://github.com/fboundy/pv_opt/files/14186690/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14186691/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14186692/pv_opt.log)

solarjones, 2024-02-07:
Hung overnight...stuck on loading tariffs. I tried switching to Solcast_p10 but it didn't work, so stopped appdaemon and manually charging for the time being.
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14190209/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14190211/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14190213/error.log)

fboundy, 2024-02-07:
This can happen when there’s an issue with the Octopus API.

My plan is to default to picking up pricing from the Octopus integration when it’s available. I think most people are using this anyway.