stevebuk1/pv_opt #96: PV_OPT starts ignoring parameters

SzosszeNET, 2024-01-30:
**Describe the bug**

In: /homeassistant/appdaemon/apps/pv_opt/config/config.yaml

I have :   consumption_history_days: 5

  # ========================================
  # Plant parameters
  # ========================================

  # All parameters can be a number or point to an entity UNLESS they start with 'id_'. All of
  # these are required but will be defaulted if not specified

  battery_capacity_wh: 5000
  inverter_efficiency_percent: 97 # Default: 97
  # charger_efficiency_percent: 91 # Default: 91
  # maximum_dod_percent: 15
  charger_power_watts: 3600
  inverter_power_watts: 3600
  # inverter_loss_watts: 100
  switch.pvopt_read_only: off
  switch.pvopt_allow_cyclic: on

However when PV_OPT starts on the dashboard I find:
  switch.pvopt_read_only: on
  switch.pvopt_allow_cyclic: off
  charger_power_watts: 3000
  battery_capacity_wh: 10000

**To Reproduce**
Steps to reproduce the behavior:
restart appdaemon / HA

**Expected behavior**
PV_OPT / AppDaemon to pick up the parameters correctly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 App and Browser too

fboundy, 2024-01-30:
This was on my to-do list. Should get addressed in 3.6.0.

fboundy, 2024-01-30:
This should now be fixed in 3.6.0 which should be available as a pre-release. Please check and revert.

SzosszeNET, 2024-01-30:
Unfortunately still not the case
[image]

SzosszeNET, 2024-01-30:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14100889/pv_opt.log)

Hopefully removed all data I didn't mean to disclose.

The first start is just appdaemon restarted
Second is HA restarted

fboundy, 2024-01-30:
Do you have `overwrite_ha_on_restart: true` in your `config.yaml` like this:
    
      inverter_type: "SOLIS_SOLAX_MODBUS"
      # inverter_type: "SOLIS_CORE_MODBUS"
      # inverter_type: SOLIS_SOLARMAN
      device_name: solis
    
      # If true the current config in HA will be over-written with that in the config.yaml. 
      overwrite_ha_on_restart: true

It doesn't look like it as the log file should say have:

    14:46:38     INFO: Reading arguments from YAML:
    14:46:38     INFO: -----------------------------------
    14:46:38     INFO: 
    14:46:38     INFO:   Over-write flag is set so YAML will over-write HA

This is in the release notes for 3.6.0 and the updated `ReadMe` but you may have missed it.

fboundy, 2024-01-30:
On another note I keep meaning to redact personal data from the log file by default. I will add a new issue to catch this.

SzosszeNET, 2024-01-30:
> Do you have `overwrite_ha_on_restart: true` in your `config.yaml` like this:
>
> ```
>   inverter_type: "SOLIS_SOLAX_MODBUS"
>   # inverter_type: "SOLIS_CORE_MODBUS"
>   # inverter_type: SOLIS_SOLARMAN [...]

Indeed missed it. Looks like that was it, thank you.

fboundy, 2024-01-30:
It should have defaulted to True so let me check that.

3.6.2 will also have sensitive details redacted from the log files