stevebuk1/pv_opt #104: Unexpected error running initialize() for pv_opt

SzosszeNET, 2024-02-01:
I'm terribly sorry, been trying to work out in the past two days what's going on, but now I'm at loss...

On version 3.6.0 and as far as loading data everything looks just fine:

20:08:33     INFO: ******************* PV Opt v3.6.0 *******************
20:08:33     INFO: 
20:08:33     INFO: Local timezone set to GB
20:08:33     INFO: Time Zone Offset: 0.0 minutes
20:08:33     INFO: Inverter type: SOLIS_SOLAX_MODBUS: inverter module: solis.py
20:08:33     INFO: Reading arguments from YAML:
20:08:33     INFO: -----------------------------------
20:08:33     INFO:     consumption_history_days           = 2                                                              2: value in YAML
20:08:33     INFO:     battery_capacity_wh                = 5000                                                        5000: value in YAML
20:08:33     INFO:     charger_power_watts                = 3600                                                        3600: value in YAML
20:08:33     INFO:     inverter_power_watts               = 3600                                                        3600: value in YAML
20:08:33     INFO:     id_solcast_today                   = sensor.solcast_pv_forecast_forecast_today                    0.0: value(s) in YAML
20:08:33     INFO:     id_solcast_tomorrow                = sensor.solcast_pv_forecast_forecast_tomorrow              1.53035: value(s) in YAML
20:08:33  WARNING:     consumption_margin                 = 10                                                            10: system default. Unable to read from HA entities listed in YAML. No default in YAML.
20:08:33     INFO:     octopus_account                    = *************                                               ******** YAML default value. No default defined.
20:08:33     INFO:     octopus_api_key                    = ************************                          s*********************: YAML default value. No default defined.
20:08:33     INFO:     battery_voltage                    = sensor.solis_battery_voltage                                52.1: value in YAML
20:08:33     INFO:     update_cycle_seconds               = 15                                                            15: value in YAML
20:08:33     INFO:     maximum_dod_percent                = number.solis_battery_minimum_soc                            15.0: value in YAML
20:08:33     INFO:     id_consumption_today               = sensor.solis_house_load_today                               22.6: value(s) in YAML
20:08:33     INFO:     id_grid_import_today               = sensor.solis_grid_import_today                              19.2: value(s) in YAML
20:08:33     INFO:     id_grid_export_today               = sensor.solis_grid_export_today                               0.3: value(s) in YAML
20:08:33     INFO:     id_battery_soc                     = sensor.solis_battery_soc                                    15.0: value(s) in YAML
20:08:33     INFO:     id_timed_charge_start_hours        = number.solis_timed_charge_start_hours                       20.0: value(s) in YAML
20:08:33     INFO:     id_timed_charge_start_minutes      = number.solis_timed_charge_start_minutes                     30.0: value(s) in YAML
20:08:33     INFO:     id_timed_charge_end_hours          = number.solis_timed_charge_end_hours                         21.0: value(s) in YAML
20:08:33     INFO:     id_timed_charge_end_minutes        = number.solis_timed_charge_end_minutes                        0.0: value(s) in YAML
20:08:33     INFO:     id_timed_charge_current            = number.solis_timed_charge_current                           30.0: value(s) in YAML
20:08:33     INFO:     id_timed_discharge_start_hours     = number.solis_timed_discharge_start_hours                     0.0: value(s) in YAML
20:08:33     INFO:     id_timed_discharge_start_minutes   = number.solis_timed_discharge_start_minutes                   0.0: value(s) in YAML
20:08:33     INFO:     id_timed_discharge_end_hours       = number.solis_timed_discharge_end_hours                       0.0: value(s) in YAML
20:08:33     INFO:     id_timed_discharge_end_minutes     = number.solis_timed_discharge_end_minutes                     0.0: value(s) in YAML
20:08:33     INFO:     id_timed_discharge_current         = number.solis_timed_discharge_current                        60.0: value(s) in YAML
20:08:33     INFO:     id_timed_charge_discharge_button   = button.solis_update_charge_discharge_times                2024-02-01T16:30:20.738670+00:00: value(s) in YAML
20:08:33     INFO:     id_inverter_mode                   = select.solis_energy_storage_control_switch                Timed Charge/Discharge: value(s) in YAML
20:08:33     INFO:     id_daily_solar                     = sensor.solis_power_generation_today                          2.9: value(s) in YAML
20:08:33     INFO: 
20:08:33     INFO: Checking config:
20:08:33     INFO: -----------------------
20:08:33  WARNING:     forced_charge                      = True                                                        True: system default. Not in YAML.
20:08:33  WARNING:     forced_discharge                   = True                                                        True: system default. Not in YAML.
20:08:33  WARNING:     read_only                          = True                                                        True: system default. Not in YAML.
20:08:33  WARNING:     allow_cyclic                       = False                                                      False: system default. Not in YAML.
20:08:33  WARNING:     optimise_frequency_minutes         = 10                                                            10: system default. Not in YAML.
20:08:33  WARNING:     slot_threshold_p                   = 1.0                                                          1.0: system default. Not in YAML.
20:08:33  WARNING:     day_of_week_weighting              = 0.5                                                          0.5: system default. Not in YAML.
20:08:33  WARNING:     pass_threshold_p                   = 4.0                                                          4.0: system default. Not in YAML.
20:08:33  WARNING:     octopus_auto                       = True                                                        True: system default. Not in YAML.
20:08:33  WARNING:     inverter_efficiency_percent        = 97                                                            97: system default. Not in YAML.
20:08:33  WARNING:     charger_efficiency_percent         = 91                                                            91: system default. Not in YAML.
20:08:33  WARNING:     inverter_loss_watts                = 100                                                          100: system default. Not in YAML.
20:08:33  WARNING:     solar_forecast                     = Solcast                                                   Solcast: system default. Not in YAML.
20:08:33  WARNING:     consumption_grouping               = mean                                                        mean: system default. Not in YAML.
20:08:33  WARNING:     forced_power_group_tolerance       = 100                                                          100: system default. Not in YAML.
20:08:33  WARNING:     id_battery_charge_power            = sensor.solis_battery_input_energy                            0.0: system default. Not in YAML.
20:08:33  WARNING:     id_inverter_ac_power               = sensor.solis_active_power                                  -90.0: system default. Not in YAML.
20:08:33  WARNING:     supports_hold_soc                  = True                                                        True: system default. Not in YAML.
20:08:33  WARNING:     id_backup_mode_soc                 = number.solis_backup_mode_soc                               100.0: system default. Not in YAML.
20:08:33     INFO: 
20:08:33     INFO: Syncing config with Home Assistant:
20:08:33     INFO: -----------------------------------
20:08:33     INFO: 
20:08:33     INFO:   Config Item                               HA Entity                                   Current State
20:08:33     INFO:   -----------                               ---------                                   -------------
20:08:33     INFO:   forced_charge                             switch.pvopt_forced_charge                  on
20:08:33     INFO:   forced_discharge                          switch.pvopt_forced_discharge               on
[... cut ...]
    result = cast("Union[Self, DTScalarOrNaT]", super().__getitem__(key))
                                                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/arrays/_mixins.py", line 284, in __getitem__
    result = self._ndarray[key]
             ~~~~~~~~~~~~~^^^^^
IndexError: index -1 is out of bounds for axis 0 with size 0

20:08:35 WARNING pv_opt: ------------------------------------------------------------

Tried deleting and reinstalling PV OPT unfortunately no difference...

fboundy, 2024-02-01:
Can you try 3.7.0 beta? You’ll need to enable pre-release in HACS

SzosszeNET, 2024-02-01:
I did already and went back to 3.6.0 just in case.

Unfortunately it's the same issue...

21:41:35     INFO: ******************* PV Opt v3.6.1 *******************
21:41:35     INFO: 
21:41:35     INFO: Local timezone set to GB
21:41:35     INFO: Time Zone Offset: 0.0 minutes
21:41:35     INFO: Inverter type: SOLIS_SOLAX_MODBUS: inverter module: solis.py
21:41:35     INFO: Reading arguments from YAML:
21:41:35     INFO: -----------------------------------
21:41:35     INFO: 
21:41:35     INFO:   Over-write flag is set so YAML will over-write HA
21:41:35     INFO:     consumption_history_days           = 2                                                              2: value in YAML
21:41:35     INFO:     redact_personal_data_from_log      = True                                                        True: value in YAML
21:41:35     INFO:     battery_capacity_wh                = 5000                                                        5000: value in YAML
21:41:35     INFO:     charger_power_watts                = 3600                                                        3600: value in YAML
21:41:35     INFO:     inverter_power_watts               = 3600                                                        3600: value in YAML
21:41:35     INFO:     id_solcast_today                   = sensor.solcast_pv_forecast_forecast_today                    0.0: value(s) in YAML
21:41:35     INFO:     id_solcast_tomorrow                = sensor.solcast_pv_forecast_forecast_tomorrow              1.53035: value(s) in YAML
21:41:35  WARNING:     consumption_margin                 = 10                                                            10: system default. Unable to read from HA entities listed in YAML. No default in YAML.
21:41:35     INFO:     octopus_account                    = ***********                                               ************: YAML default value. No default defined.
21:41:35     INFO:     octopus_api_key                    = ***********************                          *****************: YAML default value. No default defined.
21:41:35     INFO:     battery_voltage                    = sensor.solis_battery_voltage                                51.7: value in YAML
21:41:35     INFO:     update_cycle_seconds               = 15                                                            15: value in YAML
21:41:35     INFO:     maximum_dod_percent                = number.solis_battery_minimum_soc                            15.0: value in YAML
21:41:35     INFO:     id_consumption_today               = sensor.solis_house_load_today                               23.6: value(s) in YAML
21:41:35     INFO:     id_grid_import_today               = sensor.solis_grid_import_today                              20.6: value(s) in YAML
21:41:35     INFO:     id_grid_export_today               = sensor.solis_grid_export_today                               0.4: value(s) in YAML
21:41:35     INFO:     id_battery_soc                     = sensor.solis_battery_soc                                    24.0: value(s) in YAML
21:41:35     INFO:     id_timed_charge_start_hours        = number.solis_timed_charge_start_hours                       20.0: value(s) in YAML
21:41:35     INFO:     id_timed_charge_start_minutes      = number.solis_timed_charge_start_minutes                     30.0: value(s) in YAML
21:41:35     INFO:     id_timed_charge_end_hours          = number.solis_timed_charge_end_hours                         21.0: value(s) in YAML
21:41:35     INFO:     id_timed_charge_end_minutes        = number.solis_timed_charge_end_minutes                        0.0: value(s) in YAML
21:41:35     INFO:     id_timed_charge_current            = number.solis_timed_charge_current                           30.0: value(s) in YAML
21:41:35     INFO:     id_timed_discharge_start_hours     = number.solis_timed_discharge_start_hours                     0.0: value(s) in YAML
21:41:35     INFO:     id_timed_discharge_start_minutes   = number.solis_timed_discharge_start_minutes                   0.0: value(s) in YAML
21:41:35     INFO:     id_timed_discharge_end_hours       = number.solis_timed_discharge_end_hours                       0.0: value(s) in YAML
21:41:35     INFO:     id_timed_discharge_end_minutes     = number.solis_timed_discharge_end_minutes                     0.0: value(s) in YAML
21:41:35     INFO:     id_timed_discharge_current         = number.solis_timed_discharge_current                        60.0: value(s) in YAML
21:41:35     INFO:     id_timed_charge_discharge_button   = button.solis_update_charge_discharge_times                2024-02-01T16:30:20.738670+00:00: value(s) in YAML
21:41:35     INFO:     id_inverter_mode                   = select.solis_energy_storage_control_switch                Timed Charge/Discharge: value(s) in YAML
21:41:35     INFO: 
21:41:35     INFO: Checking config:
21:41:35     INFO: -----------------------
21:41:35  WARNING:     forced_charge                      = True                                                        True: system default. Not in YAML.
21:41:35  WARNING:     forced_discharge                   = True                                                        True: system default. Not in YAML.
21:41:35  WARNING:     read_only                          = True                                                        True: system default. Not in YAML.
[... cut ...]
21:41:35 INFO AppDaemon: Loading App Module: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py
21:41:35 INFO AppDaemon: Loading Global Module: /homeassistant/appdaemon/apps/pv_opt/pvpy.py
21:41:35 WARNING AppDaemon: Unable to find app solis in dependencies for pv_opt
21:41:35 WARNING AppDaemon: Ignoring app pv_opt
21:41:35 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
21:41:35 INFO AppDaemon: Calling initialize() for pv_opt
21:41:36 WARNING AppDaemon: Logged an error to /share/logs/appdaemon/error.log
21:41:36 INFO AppDaemon: App initialization complete

fboundy, 2024-02-01:
Please can you try 3.7.1 (pre-release). I'm not sure it will fix the issue but it has some additional logging to diagnose.

fboundy, 2024-02-01:
It looks like it's not loading the Agile prices properly

SzosszeNET, 2024-02-01:
Looks like just the same unfortunately.

22:59:35 WARNING pv_opt: ------------------------------------------------------------
22:59:35 WARNING pv_opt: Unexpected error running initialize() for pv_opt
22:59:35 WARNING pv_opt: ------------------------------------------------------------
22:59:35 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 324, in initialize
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1352, in optimise
    self.base_cost = self.contract.net_cost(self.base)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 446, in net_cost
    imp_df = self.imp.to_df(start, end, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 234, in to_df
    newindex = pd.date_range(x.index[0], df.index[-1], freq="30T")
                                         ~~~~~~~~^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 5385, in __getitem__
    return getitem(key)
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/arrays/datetimelike.py", line 379, in __getitem__
    result = cast("Union[Self, DTScalarOrNaT]", super().__getitem__(key))
                                                ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/arrays/_mixins.py", line 284, in __getitem__
    result = self._ndarray[key]
             ~~~~~~~~~~~~~^^^^^
IndexError: index -1 is out of bounds for axis 0 with size 0

22:59:35 WARNING pv_opt: ------------------------------------------------------------

22:59:34     INFO: AGILE tariff detected. Rates will update at 16:00 daily
22:59:34     INFO:   - Tariffs OK
22:59:34     INFO: 
22:59:34     INFO: Solcast forecast loaded OK
22:59:34     INFO: Getting expected consumption data
22:59:34     INFO:   - Estimated consumption from sensor.solis_house_load_today loaded OK 
22:59:34     INFO: Initial SOC: 15.0
22:59:34     INFO: Calculating Base flows
22:59:34     INFO: >>> Start: 01/02 22:30 End: 01/02 18:30
22:59:34     INFO: >>> 2  1
22:59:35     INFO: 
22:59:35     INFO: Retrieved day ahead forecast for period 25/01 23:00 - 02/02 22:00 for tariff E-1R-AGILE-FLEX-22-11-25-N
23:00:34     INFO: >>> Agile Callback Handler
23:00:34     INFO: >>> Contract end day:  2 Today: 1  False
23:00:34     INFO: >>> Current hour:     23           True

fboundy, 2024-02-01:
But I can see that it’s only loaded the Agile prices to 18.30 today which is odd
On 1 Feb 2024 at 23:10 +0000, fboundy/pv_opt ***@***.***>, wrote:
>
> You

fboundy, 2024-02-01:
Is the Octopus integration loading prices OK?

SzosszeNET, 2024-02-02:
I believe have no problems with that.
[image]
[image]

fboundy, 2024-02-02:
> I believe have no problems with that.
> Screenshot_20240202-075023.png (view on web)
> Screenshot_20240202-074941.png (view on web)

That looks like the Octopus app. Do you use Bottlecap Dave’s OE integration for HA and is it working?

SzosszeNET, 2024-02-02:
Sorry for the confusion, the first screengrab is indeed the octopus app I used for reference. The second screengrab was from HA the price card working with Bottlecap Dave's OE integration loading the prices just fine

SzosszeNET, 2024-02-02:
Anyhow this gave me an idea and did hard-code my tarrifs

`  #octopus_account: !secret octopus_account
  #octopus_api_key: !secret octopus_api_key

  # The following Can be omitted if either of the above options is working correctly:

  octopus_import_tariff_code: E-1R-AGILE-FLEX-22-11-25-N
  octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-N`

This did seem to do the trick so you are right it's indeed something around the octopus API integration goes wrong.

SzosszeNET, 2024-02-02:
Not sure if this is an issue or not - if you think it is I'll open an other bug for this, but wasn't expecting this charge plan to be generated as there are no prices yet for the 3rd February?

10:35:21     INFO: Iteration  2: Slots added:   0
10:35:21     INFO: >>> Start: 02/02 10:30 End: 03/02 23:30
10:35:21     INFO: >>> 2  3
10:35:21     INFO: >>> Start: 02/02 10:30 End: 03/02 23:30
10:35:22     INFO: 
10:35:22     INFO: Checking for Hold SOC slots
10:35:22     INFO: 
10:35:22     INFO: Optimal forced charge/discharge slots:
10:35:22     INFO:    02-Feb 14:30 - 02-Feb 15:00   Power:  3000W  SOC:   44% ->   72%  
10:35:22     INFO:    02-Feb 15:00 - 02-Feb 16:00   Power:  3200W  SOC:   72% ->  100%  
10:35:22     INFO:    02-Feb 20:30 - 02-Feb 21:00   Power:  1000W  SOC:   15% ->   24%  
10:35:22     INFO:    02-Feb 21:30 - 02-Feb 23:00   Power:  3200W  SOC:   15% ->  100%  
10:35:22     INFO:    02-Feb 23:00 - 03-Feb 00:00   Power: -3400W  SOC:  100% ->   30%  
10:35:22     INFO:    03-Feb 01:30 - 03-Feb 02:00   Power:  3200W  SOC:   15% ->   44%  
10:35:22     INFO:    03-Feb 02:30 - 03-Feb 03:00   Power:  3200W  SOC:   24% ->   53%  
10:35:22     INFO:    03-Feb 03:30 - 03-Feb 04:00   Power:  3200W  SOC:   35% ->   64%  
10:35:22     INFO:    03-Feb 05:00 - 03-Feb 06:30   Power:  3200W  SOC:   45% ->  100%  
10:35:22     INFO:    03-Feb 07:00 - 03-Feb 07:30   Power:  2900W  SOC:   90% ->  100%  
10:35:22     INFO:    03-Feb 15:30 - 03-Feb 16:00   Power:  1000W  SOC:  100% ->  100%  <=
10:35:22     INFO:    03-Feb 20:30 - 03-Feb 21:00   Power:   600W  SOC:   15% ->   21%  
10:35:22     INFO:    03-Feb 21:30 - 03-Feb 22:00   Power:   900W  SOC:   15% ->   23%  
10:35:22     INFO:    03-Feb 22:30 - 03-Feb 23:00   Power:  1400W  SOC:   15% ->   27%  
10:35:22     INFO: 
10:35:22     INFO: Plan time: 02-Feb 10:30 - 03-Feb 23:30 Initial SOC: 46.5 Base Cost: 377.53 Opt Cost: 266.56
10:35:22     INFO: 
10:35:22     INFO: Optimiser elapsed time 14.6 seconds
10:35:22     INFO: 

Or is this to be expected?

fboundy, 2024-02-02:
No it’s expected. It uses Day Ahead pricing to estimate before Agile is released

lazy-pete, 2024-02-04:
> Anyhow this gave me an idea and did hard-code my tarrifs
>
> ` #octopus_account: !secret octopus_account #octopus_api_key: !secret octopus_api_key
>
> # The following Can be omitted if either of the above [...]

Hi - I think I am (again) having the same issue - did you fix by just ‘uncommenting’ the tarrifs directly ? 

Cheers,

p

lazy-pete, 2024-02-04:
I'm also wondering whether appdaemon is loading things at all - where are you pulling the above logs from ? My appdaemon one is just - 

     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 17.9/17.9 MB 9.1 MB/s eta 0:00:00
Installing collected packages: numpy
Successfully installed numpy-1.25.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting pandas
  Downloading pandas-2.2.0-cp311-cp311-musllinux_1_1_x86_64.whl.metadata (19 kB)
Requirement already satisfied: numpy<2,>=1.23.2 in /usr/lib/python3.11/site-packages (from pandas) (1.25.0)
Requirement already satisfied: python-dateutil>=2.8.2 in /usr/lib/python3.11/site-packages (from pandas) (2.8.2)
Requirement already satisfied: pytz>=2020.1 in /usr/lib/python3.11/site-packages (from pandas) (2023.3.post1)
Collecting tzdata>=2022.7 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/tzdata-2023.4-py2.py3-none-any.whl (346 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 346.6/346.6 kB 5.6 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3.11/site-packages (from python-dateutil>=2.8.2->pandas) (1.16.0)
Downloading pandas-2.2.0-cp311-cp311-musllinux_1_1_x86_64.whl (13.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.9/13.9 MB 8.9 MB/s eta 0:00:00
Installing collected packages: tzdata, pandas
Successfully installed pandas-2.2.0 tzdata-2023.4
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[10:39:15] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-02-04 10:39:16.272541 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-02-04 10:39:16.272642 INFO AppDaemon: Python version is 3.11.6
2024-02-04 10:39:16.272731 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-02-04 10:39:16.272826 INFO AppDaemon: Added log: AppDaemon
2024-02-04 10:39:16.272904 INFO AppDaemon: Added log: Error
2024-02-04 10:39:16.272954 INFO AppDaemon: Added log: Access
2024-02-04 10:39:16.273003 INFO AppDaemon: Added log: Diag
2024-02-04 10:39:16.350268 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-02-04 10:39:16.487260 INFO HASS: HASS Plugin Initializing
2024-02-04 10:39:16.487380 WARNING HASS: ha_url not found in HASS configuration - module not initialized
2024-02-04 10:39:16.487468 INFO HASS: HASS Plugin initialization complete
2024-02-04 10:39:16.487593 INFO AppDaemon: Loading Plugin MQTT using class MqttPlugin from module mqttplugin
2024-02-04 10:39:16.520284 INFO MQTT: MQTT Plugin Initializing
2024-02-04 10:39:16.520623 INFO MQTT: Using 'localad/status' as Will Topic
2024-02-04 10:39:16.520708 INFO MQTT: Using 'localad/status' as Birth Topic
2024-02-04 10:39:16.520978 INFO AppDaemon: HTTP is disabled
2024-02-04 10:39:16.526313 INFO HASS: Connected to Home Assistant 2024.1.5
2024-02-04 10:39:16.540788 WARNING AppDaemon: ------------------------------------------------------------
2024-02-04 10:39:16.540912 WARNING AppDaemon: Unexpected error loading config file: /homeassistant/appdaemon/apps/pv_opt/config/config.yaml
2024-02-04 10:39:16.540979 WARNING AppDaemon: ------------------------------------------------------------
2024-02-04 10:39:16.542319 WARNING AppDaemon: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 578, in read_config_file
    return utils.read_config_file(file)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 586, in read_config_file
    return read_yaml_config(path)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 756, in read_yaml_config
    config = yaml.load(config_file_contents, Loader=yaml.SafeLoader)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/constructor.py", line 51, in get_single_data
    return self.construct_document(node)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/constructor.py", line 60, in construct_document
    for dummy in generator:
  File "/usr/lib/python3.11/site-packages/yaml/constructor.py", line 413, in construct_yaml_map
    value = self.construct_mapping(node)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/yaml/constructor.py", line 218, in construct_mapping
    return super().construct_mapping(node, deep=deep)
[... 7 lines cut ...]
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 676, in _secret_yaml
    raise ValueError("{} not found in secrets file".format(node.value))
ValueError: octopus_account not found in secrets file

[... cut ...]
2024-02-04 10:39:19.552166 INFO AppDaemon: Adding /homeassistant/appdaemon/apps/pv_opt/config to module import path
2024-02-04 10:39:19.553080 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py - ignoring
2024-02-04 10:39:19.553313 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/pv_opt/pvpy.py - ignoring
2024-02-04 10:39:19.553677 WARNING AppDaemon: No app description found for: /homeassistant/appdaemon/apps/pv_opt/solis.py - ignoring
2024-02-04 10:39:19.554183 INFO AppDaemon: App initialization complete

SzosszeNET, 2024-02-04:
@lazy-pete as the error states something wrong with your secrets?

"ValueError("{} not found in secrets file".format(node.value))
ValueError: octopus_account not found in secrets file"

Are "octopus_account" and " octopus_api_key" defined in your "secrets.yaml"? It's next to your home assistant configuration.yaml

lazy-pete, 2024-02-04:
thanks - they were in there, but I must've set it up wrong - removed indents and heading to account and api and got a bit further - now a different issue .... first from HA Logs

Logger: homeassistant.helpers.event
Source: helpers/template.py:569
First occurred: 02:07:54 (100 occurrences)
Last logged: 12:41:38

Error while processing template: Template<template=(<h3><u>Cost Summary (GBP)</u></h3> | | Today | Tomorrow | Total | |:--|---:|---:|---:| |Base | {{'%0.2f' | format(state_attr('sensor.pvopt_base_cost','cost_today')| float)}} | {{'%0.2f' | format(state_attr('sensor.pvopt_base_cost','cost_tomorrow')| float)}} | {{'%0.2f' | format(states('sensor.pvopt_base_cost')| float)}} | |Optimised | {{'%0.2f' | format(state_attr('sensor.pvopt_opt_cost','cost_today')| float)}} | {{'%0.2f' | format(state_attr('sensor.pvopt_opt_cost','cost_tomorrow')| float)}} | {{'%0.2f' | format(states('sensor.pvopt_opt_cost')| float)}} | |<b>Cost Saving</b> | <b>{{'%0.2f' | format((state_attr('sensor.pvopt_base_cost','cost_today')| float-state_attr('sensor.pvopt_opt_cost','cost_today')| float) | round(2))}}</b> |<b>{{'%0.2f' | format((state_attr('sensor.pvopt_base_cost','cost_tomorrow')| float-state_attr('sensor.pvopt_opt_cost','cost_tomorrow')| float) | round(2))}} </b> |<b> {{'%0.2f' | format((states('sensor.pvopt_base_cost')| float - states('sensor.pvopt_opt_cost')| float) | round(2)) }}</b> | <h3><u>Charge Plan</u></h3> | Start | | | End ||| Power ||| Start SOC ||| End SOC | Hold SOC | |:-------|--|--|:---------|--|--|:--------:|--|--|:--------:|--|--|:----------:|:--|{% for a in state_attr('sensor.pvopt_charge_start', 'windows') %} {% set tf = '%d-%b %H:%M'%} | {{as_local(as_datetime(a['start'])).strftime(tf)}} ||| {{as_local(as_datetime(a['end'])).strftime(tf)}} ||| {{a['forced'] | float | round(0)}}W ||| {{a['soc'] | float | round(1)}}% ||| {{a['soc_end'] | float | round(1)}}% | {{a['hold_soc']}} |{%endfor%}) renders=130>
Error while processing template: Template<template=({{(as_local(as_datetime(states('sensor.pvopt_charge_start')))|string)[:16]}}) renders=126>
Error while processing template: Template<template=({{(as_local(as_datetime(states('sensor.pvopt_charge_end')))|string)[:16]}}) renders=126>
Error while processing template: Template<template=({{(as_local(as_datetime(states('sensor.pvopt_charge_start')))|string)[:16]}}) renders=130>
Error while processing template: Template<template=({{(as_local(as_datetime(states('sensor.pvopt_charge_end')))|string)[:16]}}) renders=130>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 567, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2305, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
ZeroDivisionError: division by zero

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 693, in async_render_to_info
    render_info._result = self.async_render(
                          ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 569, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ZeroDivisionError: division by zero

**This from AppDaemon logs**

/homeassistant/appdaemon/apps/pv_opt/pvpy.py:208: FutureWarning: 'T' is deprecated and will be removed in a future version, please use 'min' instead.
  extended_index = pd.date_range(
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:230: FutureWarning: 'T' is deprecated and will be removed in a future version, please use 'min' instead.
  newindex = pd.date_range(x.index[0], df.index[-1], freq="30T")
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:202: FutureWarning: 'T' is deprecated and will be removed in a future version, please use 'min' instead.
  newindex = pd.date_range(df.index[0], end, freq="30T")
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:492: FutureWarning: 'T' is deprecated and will be removed in a future version. Please use 'min' instead of 'T'.
  freq = pd.infer_freq(static_flows.index) / pd.Timedelta("60T")
2024-02-04 12:32:00.678427 INFO pv_opt: 
2024-02-04 12:32:00.679675 INFO pv_opt: Discharge net cost delta: -0.0p: < threshold (4.0) => Excluded
2024-02-04 12:32:00.681029 INFO pv_opt: Iteration  2: Slots added:   0
2024-02-04 12:32:00.682443 INFO pv_opt: 
2024-02-04 12:32:00.683661 INFO pv_opt: Removing cyclic charge/discharge
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:492: FutureWarning: 'T' is deprecated and will be removed in a future version. Please use 'min' instead of 'T'.
  freq = pd.infer_freq(static_flows.index) / pd.Timedelta("60T")
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:209: FutureWarning: 'T' is deprecated and will be removed in a future version. Please use 'min' instead of 'T'.
  df.index[-1] + pd.Timedelta("30T"),
[... cut ...]
    self.charge_current = self.charge_power / self.get_config("battery_voltage")
                          ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: ufunc 'divide' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''

2024-02-04 12:32:00.752788 WARNING pv_opt: ------------------------------------------------------------
2024-02-04 12:32:00.752938 INFO AppDaemon: App initialization complete
2024-02-04 12:32:52.861413 INFO pv_opt: >>> Agile Callback Handler

SzosszeNET, 2024-02-04:
If I'm reading this right your battery_voltage isn't coming through?

Check your config and search for that parameter

Should map to something like 

sensor.{{device}}_battery_voltage if you are using the solax Solis integration.

You might want to check if you have that entity in HA under Settings>Devices & Services>Entities

lazy-pete, 2024-02-04:
Ah - I have sensor.solis_battery_voltage_2 
Must have failed to delete a previous version - I can’t rename _2  as there seems to be an orphan sensor.solis_battery_voltage in the registry. I’ll have to work out how to get rid of that then rename. Probably easier than finding every reference in the config and changing to _2. 

Thanks for helping with this - no idea how to interpret the log files !

SzosszeNET, 2024-02-04:
It's all right, you might want to check the other sensors, it's possible
that every one of them will have an orphaned.

Either works, but indeed the cleanest to delete the orphaned and rename the
working one.

If you have a duplicate of all, probably the fastest is to add "_2" to all
sensors in the pv-opt config file since they are in one single place.

lazy-pete, 2024-02-04:
Eureka ! 

It turns out I have many duplicate entities, but almost all the rest have been 'disabled by integration' and not forced _2 on the 'new' entities. The battery voltage one did led me rename the hidden disabled original to _3 so I could rename the active one to the correct label for PVOpt and NOW IT WORKS !

I haven't worked out how to access the entity register as yet , but I'll have to as I've a real mess of 'disabled by integration' entities I can't delete.

Anyway - the current goal of getting PVOpt running is over the line - thanks again for all your help @SzosszeNET  and @fboundy for writng it in the first place.

SzosszeNET, 2024-02-04:
@fboundy , I have since reverted back to get the tariff via the octopus API and that seems to be working fine now, might been an indentation somewhere... Closing this as it works perfectly. Thank you

fboundy, 2024-02-05:
``

closed 2024-02-05 by fboundy