stevebuk1/pv_opt #105: export tariff OUTGOING-LITE-FIX-12M-23-09-12 causes 400 error

mergwyn, 2024-02-04:
**Describe the bug**
Im currently on the above export tariff.  When I use this tariff pv_opt fails.

I suspect this is because the tariff has no end date.

**To Reproduce**

1. Select the above tariff in the config
2. Initialisation fails with 400 error

**Expected behavior**
Tariff to be treated as valid

**Logs**
The config:
```
  octopus_import_tariff_code: E-1R-GO-18-06-12-F                                                                                                                                                                                                  
  octopus_export_tariff_code: E-1R-OUTGOING-LITE-FIX-12M-23-09-12-F  
```
Detecting the config in the logs:
```
2024-02-04 15:31:46.858473 INFO pv_opt: Starting Opimisation with discharge enabled
2024-02-04 15:31:46.859185 INFO pv_opt: -------------------------------------------
2024-02-04 15:31:46.859864 INFO pv_opt: 
2024-02-04 15:31:46.860571 INFO pv_opt: Checking tariffs:
2024-02-04 15:31:46.861282 INFO pv_opt: -----------------
2024-02-04 15:31:46.866274 INFO pv_opt:   Import: E-1R-GO-18-06-12-F                       Start: 2023-05-31 23:30:00+0000 End: 2024-02-06 00:30:00+0000 
2024-02-04 15:31:46.867082 INFO pv_opt:   Export: E-1R-OUTGOING-LITE-FIX-12M-23-09-12-F    Start: 2023-09-11 23:00:00+0000 End: N/A 
2024-02-04 15:31:46.867813 INFO pv_opt:   - Tariffs OK
2024-02-04 15:31:46.868516 INFO pv_opt: 

```
The error:
```
2024-02-04 15:31:51.337563 WARNING HASS: Error setting Home Assistant state default.sensor.pvopt_opt_cost, {'state': 0.0, 'attributes': {'friendly_name': 'PV Opt Optimised Cost', 'device_class': 'monetary', 'state_class': 'measurement', 'unit_of_measurement': 'GBP', 'cost_today': 0.0, 'cost_tomorrow': 0.0, 'soc': [{'period_start': '2024-02-04T15:30:00+00:00', 'soc': 9.0}, {'period_start': '2024-02-04T16:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T16:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T17:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T17:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T18:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T18:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T19:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T19:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T20:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T20:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T21:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T21:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T22:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T22:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T23:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-04T23:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T00:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T00:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T01:00:00+00:00', 'soc': 21.35}, {'period_start': '2024-02-05T01:30:00+00:00', 'soc': 32.7}, {'period_start': '2024-02-05T02:00:00+00:00', 'soc': 44.04}, {'period_start': '2024-02-05T02:30:00+00:00', 'soc': 55.39}, {'period_start': '2024-02-05T03:00:00+00:00', 'soc': 66.74}, {'period_start': '2024-02-05T03:30:00+00:00', 'soc': 78.09}, {'period_start': '2024-02-05T04:00:00+00:00', 'soc': 89.44}, {'period_start': '2024-02-05T04:30:00+00:00', 'soc': 88.73}, {'period_start': '2024-02-05T05:00:00+00:00', 'soc': 86.46}, {'period_start': '2024-02-05T05:30:00+00:00', 'soc': 83.91}, {'period_start': '2024-02-05T06:00:00+00:00', 'soc': 81.78}, {'period_start': '2024-02-05T06:30:00+00:00', 'soc': 81.21}, {'period_start': '2024-02-05T07:00:00+00:00', 'soc': 80.65}, {'period_start': '2024-02-05T07:30:00+00:00', 'soc': 79.37}, {'period_start': '2024-02-05T08:00:00+00:00', 'soc': 76.96}, {'period_start': '2024-02-05T08:30:00+00:00', 'soc': 71.53}, {'period_start': '2024-02-05T09:00:00+00:00', 'soc': 67.04}, {'period_start': '2024-02-05T09:30:00+00:00', 'soc': 64.11}, {'period_start': '2024-02-05T10:00:00+00:00', 'soc': 62.37}, {'period_start': '2024-02-05T10:30:00+00:00', 'soc': 63.45}, {'period_start': '2024-02-05T11:00:00+00:00', 'soc': 64.35}, {'period_start': '2024-02-05T11:30:00+00:00', 'soc': 66.26}, {'period_start': '2024-02-05T12:00:00+00:00', 'soc': 67.61}, {'period_start': '2024-02-05T12:30:00+00:00', 'soc': 68.5}, {'period_start': '2024-02-05T13:00:00+00:00', 'soc': 68.85}, {'period_start': '2024-02-05T13:30:00+00:00', 'soc': 71.06}, {'period_start': '2024-02-05T14:00:00+00:00', 'soc': 71.34}, {'period_start': '2024-02-05T14:30:00+00:00', 'soc': 70.12}, {'period_start': '2024-02-05T15:00:00+00:00', 'soc': 70.63}, {'period_start': '2024-02-05T15:30:00+00:00', 'soc': 60.39}, {'period_start': '2024-02-05T16:00:00+00:00', 'soc': 48.5}, {'period_start': '2024-02-05T16:30:00+00:00', 'soc': 37.47}, {'period_start': '2024-02-05T17:00:00+00:00', 'soc': 34.28}, {'period_start': '2024-02-05T17:30:00+00:00', 'soc': 29.6}, {'period_start': '2024-02-05T18:00:00+00:00', 'soc': 22.23}, {'period_start': '2024-02-05T18:30:00+00:00', 'soc': 16.98}, {'period_start': '2024-02-05T19:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T19:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T20:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T20:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T21:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T21:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T22:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T22:30:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T23:00:00+00:00', 'soc': 10.0}, {'period_start': '2024-02-05T23:30:00+00:00', 'soc': 10.0}], 'forced': [{'period_start': '2024-02-04T15:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T16:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T16:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T17:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T17:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T18:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T18:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T19:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T19:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T20:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T20:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T21:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T21:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T22:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T22:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T23:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-04T23:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T00:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T00:30:00+00:00', 'forced': 2494}, {'period_start': '2024-02-05T01:00:00+00:00', 'forced': 2494}, {'period_start': '2024-02-05T01:30:00+00:00', 'forced': 2494}, {'period_start': '2024-02-05T02:00:00+00:00', 'forced': 2494}, {'period_start': '2024-02-05T02:30:00+00:00', 'forced': 2494}, {'period_start': '2024-02-05T03:00:00+00:00', 'forced': 2494}, {'period_start': '2024-02-05T03:30:00+00:00', 'forced': 2494}, {'period_start': '2024-02-05T04:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T04:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T05:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T05:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T06:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T06:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T07:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T07:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T08:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T08:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T09:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T09:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T10:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T10:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T11:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T11:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T12:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T12:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T13:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T13:30:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T14:00:00+00:00', 'forced': 0}, {'period_start': '2024-02-05T14:30:00+00:00' [... cut ...]
2024-02-04 15:31:51.338351 WARNING HASS: Code: 400, error: {"message":"Invalid JSON specified."}
```
**Additional context**
```
$ curl -u <<api>> https://api.octopus.energy/v1/products/OUTGOING-LITE-FIX-12M-23-09-12/electricity-tariffs/E-1R-OUTGOING-LITE-FIX-12M-23-09-12-F/standard-unit-rates/ | jq .
{
  "count": 1,
  "next": null,
  "previous": null,
  "results": [
    {
      "value_exc_vat": 8,
      "value_inc_vat": 8,
      "valid_from": "2023-09-11T23:00:00Z",
      "valid_to": null,
      "payment_method": null
    }
  ]
}

```

SzosszeNET, 2024-02-04:
Sorry to butt in but my export tarrif has no end date either and not seeing that an an issue

15:40:00     INFO: Checking tariffs:
15:40:00     INFO: -----------------
15:40:00     INFO:   Import: E-1R-AGILE-FLEX-22-11-25-N               Start: 2024-01-25 13:00:00+0000 End: 2024-02-04 23:00:00+0000 
15:40:00     INFO:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-N         Start: 2019-05-15 23:00:00+0000 End: N/A 
15:40:00     INFO: AGILE tariff detected. Rates will update at 16:00 daily
15:40:00     INFO:   - Tariffs OK
15:40:00     INFO:

mergwyn, 2024-02-04:
Interesting, querying the API produces:
```
product=OUTGOING-FIX-12M-19-05-13, tariff=E-1R-OUTGOING-FIX-12M-19-05-13-N
{
  "count": 3,
  "next": null,
  "previous": null,
  "results": [
    {
      "value_exc_vat": 15,
      "value_inc_vat": 15,
      "valid_from": "2022-09-19T23:00:00Z",
      "valid_to": null,
      "payment_method": null
    },
    {
      "value_exc_vat": 7.5,
      "value_inc_vat": 7.5,
      "valid_from": "2022-02-01T00:00:00Z",
      "valid_to": "2022-09-19T23:00:00Z",
      "payment_method": null
    },
    {
      "value_exc_vat": 5.5,
      "value_inc_vat": 5.5,
      "valid_from": "2019-05-15T23:00:00Z",
      "valid_to": "2022-02-01T00:00:00Z",
      "payment_method": null
    }
  ]
}
```
So your tariff has 3 rates, some with and one without an end date.

SzosszeNET, 2024-02-04:
Yes, but two of the tariffs end in the past hence no longer valid and the current one is used without the end date.

What else is in your appdaemon log(s) as like said I'm almost certain due to the fact that my current tariff does not have an end date that it's not your issue.

mergwyn, 2024-02-04:
Thanks for your help - here is the full log (without the `FutureWarnings`:

```
Collecting numpy==1.25
  Downloading numpy-1.25.0-cp310-cp310-musllinux_1_1_x86_64.whl (17.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 17.4/17.4 MB 17.1 MB/s eta 0:00:00
Collecting pandas
  Downloading pandas-2.2.0-cp310-cp310-musllinux_1_1_x86_64.whl (13.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.9/13.9 MB 17.2 MB/s eta 0:00:00
Collecting pvpy
  Downloading pvpy-1.4.1a0-py3-none-any.whl (64 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.0/64.0 kB 8.2 MB/s eta 0:00:00
Collecting solis
  Downloading solis-0.1.1-py3-none-any.whl (6.1 kB)
Requirement already satisfied: pytz>=2020.1 in /usr/local/lib/python3.10/site-packages (from pandas->-r /conf/requirements.txt (line 2)) (2023.3)
Collecting tzdata>=2022.7
  Downloading tzdata-2023.4-py2.py3-none-any.whl (346 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 346.6/346.6 kB 13.0 MB/s eta 0:00:00
Requirement already satisfied: python-dateutil>=2.8.2 in /usr/local/lib/python3.10/site-packages (from pandas->-r /conf/requirements.txt (line 2)) (2.8.2)
Collecting scipy
  Downloading scipy-1.12.0-cp310-cp310-musllinux_1_1_x86_64.whl (38.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 38.6/38.6 MB 12.9 MB/s eta 0:00:00
Collecting asyncclick<9.0.0.0,>=8.1.3.4
  Downloading asyncclick-8.1.7.1-py3-none-any.whl (99 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 99.2/99.2 kB 17.0 MB/s eta 0:00:00
Collecting click<9.0.0,>=8.1.3
  Downloading click-8.1.7-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.9/97.9 kB 18.6 MB/s eta 0:00:00
Collecting zem-pysolarmanv5<3.0.0,>=2.5.0rc2
  Downloading zem_pysolarmanv5-2.5.0rc2-py3-none-any.whl (13 kB)
Collecting anyio
  Downloading anyio-4.2.0-py3-none-any.whl (85 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 85.5/85.5 kB 11.8 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.10/site-packages (from python-dateutil>=2.8.2->pandas->-r /conf/requirements.txt (line 2)) (1.16.0)
Collecting umodbus
  Downloading uModbus-1.0.4-py2.py3-none-any.whl (31 kB)
Collecting typing-extensions>=4.1
  Downloading typing_extensions-4.9.0-py3-none-any.whl (32 kB)
Collecting exceptiongroup>=1.0.2
  Downloading exceptiongroup-1.2.0-py3-none-any.whl (16 kB)
Collecting sniffio>=1.1
  Downloading sniffio-1.3.0-py3-none-any.whl (10 kB)
Requirement already satisfied: idna>=2.8 in /usr/local/lib/python3.10/site-packages (from anyio->asyncclick<9.0.0.0,>=8.1.3.4->solis->-r /conf/requirements.txt (line 4)) (3.4)
Collecting pyserial~=3.4
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 17.2 MB/s eta 0:00:00
Installing collected packages: pyserial, umodbus, tzdata, typing-extensions, sniffio, numpy, exceptiongroup, click, zem-pysolarmanv5, scipy, pandas, anyio, pvpy, asyncclick, solis
Successfully installed anyio-4.2.0 asyncclick-8.1.7.1 click-8.1.7 exceptiongroup-1.2.0 numpy-1.25.0 pandas-2.2.0 pvpy-1.4.1a0 pyserial-3.5 scipy-1.12.0 sniffio-1.3.0 solis-0.1.1 typing-extensions-4.9.0 tzdata-2023.4 umodbus-1.0.4 zem-pysolarmanv5-2.5.0rc2
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

[notice] A new release of pip is available: 23.0.1 -> 24.0
[notice] To update, run: pip install --upgrade pip
2024-02-04 16:44:55.990502 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-02-04 16:44:55.990605 INFO AppDaemon: Python version is 3.10.11
2024-02-04 16:44:55.990692 INFO AppDaemon: Configuration read from: /conf/appdaemon.yaml
2024-02-04 16:44:55.990762 INFO AppDaemon: Added log: AppDaemon
2024-02-04 16:44:55.990834 INFO AppDaemon: Added log: Error
2024-02-04 16:44:55.990894 INFO AppDaemon: Added log: Access
2024-02-04 16:44:55.990956 INFO AppDaemon: Added log: Diag
2024-02-04 16:44:56.008134 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-02-04 16:44:56.187154 INFO HASS: HASS Plugin Initializing
2024-02-04 16:44:56.187265 INFO HASS: HASS Plugin initialization complete
2024-02-04 16:44:56.187395 INFO AppDaemon: Loading Plugin MQTT using class MqttPlugin from module mqttplugin
2024-02-04 16:44:56.190861 INFO MQTT: MQTT Plugin Initializing
2024-02-04 16:44:56.191136 INFO MQTT: Using 'localad/status' as Will Topic
2024-02-04 16:44:56.191204 INFO MQTT: Using 'localad/status' as Birth Topic
2024-02-04 16:44:56.191451 INFO AppDaemon: Initializing HTTP
2024-02-04 16:44:56.191603 INFO AppDaemon: Using 'ws' for event stream
2024-02-04 16:44:56.193405 INFO AppDaemon: Starting API
2024-02-04 16:44:56.194814 INFO AppDaemon: Starting Admin Interface
[... 692 lines cut ...]
  df.index[-1] + pd.Timedelta("30T"),
  df.index[-1] + pd.Timedelta("24H"),
  extended_index = pd.date_range(
  df.index[-1] + pd.Timedelta("30T"),
  df.index[-1] + pd.Timedelta("24H"),
  extended_index = pd.date_range(
  midnight = pd.Timestamp.now(tz="UTC").normalize() + pd.Timedelta("24H")
  (cost["cost"].loc[: midnight - pd.Timedelta("30T")].sum()) / 100,
2024-02-04 16:45:00.867186 WARNING HASS: Error setting Home Assistant state default.sensor.pvopt_opt_cost, {'state': 0.0, 'attributes': {'friendly_name': 'PV Opt Optimised Cost', 'device_class': 'monetary', 'state_class': 'measurement', 'unit_of_measurement': 'GBP', 'cost_today': 0.0, 'cost_tomorrow': 0.0,  [... cut ...]
2024-02-04 16:45:00.883835 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-02-04 16:45:00.887897 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-02-04 16:45:00.892167 INFO pv_opt: Output written to sensor.pvopt_soc_h12
2024-02-04 16:45:00.892605 INFO pv_opt: Read only mode enabled. Not querying inverter.
  .ceil(f"{self.get_config('optimise_frequency_minutes')}T")
2024-02-04 16:45:00.896813 INFO pv_opt: Optimiser will run every 10 minutes from 16:50 or on PV_OPT Event
2024-02-04 16:45:00.897390 INFO AppDaemon: App initialization complete
```

SzosszeNET, 2024-02-04:
That looks like it's working to me

"2024-02-04 16:45:00.896813 INFO pv_opt: Optimiser will run every 10 minutes from 16:50 or on PV_OPT Event
2024-02-04 16:45:00.897390 INFO AppDaemon: App initialization complete"

Did you add the dashboard and checked what it does?

SzosszeNET, 2024-02-04:
Were the sensors created?

assuming you added the logging filters it's in the main log

20:46:33 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
20:46:33 INFO AppDaemon: Calling initialize() for pv_opt
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_optimiser_elapsed created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_unit_cost_today created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_base_cost created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_opt_cost created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_charge_start created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_charge_end created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_charge_current created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_soc_h1 created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_soc_h4 created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_soc_h8 created in namespace: default
20:46:45 INFO AppDaemon: pv_opt: Entity sensor.pvopt_soc_h12 created in namespace: default

mergwyn, 2024-02-04:
I'm confused by the logs as well.  I don't see any created messaged in the logs, but I do see this:
```
2024-02-04 16:45:00.670913 INFO pv_opt: Output written to sensor.pvopt_optimiser_elapsed
2024-02-04 16:45:00.728390 INFO pv_opt: Output written to sensor.pvopt_unit_cost_today
2024-02-04 16:45:00.802500 INFO pv_opt: Output written to sensor.pvopt_base_cost
2024-02-04 16:45:00.869726 INFO pv_opt: Output written to sensor.pvopt_opt_cost
2024-02-04 16:45:00.872653 INFO pv_opt: Output written to sensor.pvopt_charge_start
2024-02-04 16:45:00.874917 INFO pv_opt: Output written to sensor.pvopt_charge_end
2024-02-04 16:45:00.877569 INFO pv_opt: Output written to sensor.pvopt_charge_current
2024-02-04 16:45:00.880875 INFO pv_opt: Output written to sensor.pvopt_soc_h1
2024-02-04 16:45:00.883835 INFO pv_opt: Output written to sensor.pvopt_soc_h4
2024-02-04 16:45:00.887897 INFO pv_opt: Output written to sensor.pvopt_soc_h8
2024-02-04 16:45:00.892167 INFO pv_opt: Output written to sensor.pvopt_soc_h12
```
These are the sensors in HA - pvopt_base_cost and pvopt_cost are missing:
[image]

And this is what the dashboard shows:
[image]

SzosszeNET, 2024-02-04:
"pv_opt: Output written to sensor.pvopt_soc_h12" these entries are from "pv_opt.log" the create messages I shared are from "main.log"

Also might worth looking into error.log there since the HASS entry is warning only

mergwyn, 2024-02-04:
Those lines are in main.log after a reboot of HA, but the entities are not created in HA as per the screenshots. error.log is empty.

If I use a different export tariff (eg. E-1R-AGILE-OUTGOING-19-05-13-F) it works - so confusing!

SzosszeNET, 2024-02-04:
Well, must be something with the tariff that case.

fboundy, 2024-02-05:
That's not a tariff that I've looked at when testing so I'll need to see how Octopus format the output from the API.

fboundy, 2024-02-08:
Hi 

I've not been able to replicate this simply by switching to this tariff. 

The problem is that there are NaN's in the calculated cumulative cost which HA won't accept as an attribute. I've put a fix in 3.7.2 to replace the NaN with 0 but the cumulative shouldn't be zero or NaN so it suggests there's something else going on. 

Please can you try 3.7.2 and let me know what you find?

mergwyn, 2024-02-09:
Thanks for this =- the fix helped in so much that pv_opt no longer throws an error and the charge plan is produced. However, the costs are not passed through and do not appear in the cost summary. For example:
tariff = E-1R-AGILE-OUTGOING-19-05-13-F
[image]

tariff = E-1R-OUTGOING-LITE-FIX-12M-23-09-12-F
[image]

Let me know if you want any logs.

fboundy, 2024-02-09:
That’s what I expected. Odd that it works fine on my system.

Can you provide a copy of a section of pv_opt.log? I may have to add some more detailed looking just for this tariff to get to the bottom of what’s going on.

mergwyn, 2024-02-09:
Thanks - I have a workaround for now manually selecting a different outgoing tariff and not using discharge optimisation, so not an urgent fix for me.

fboundy, 2024-02-10:
This was because there was only 1 entry for this in the OE API JSON response. Should be fixed in 3.7.4

fboundy, 2024-02-11:
Closing this as I believe it to be fixed. Please re-open if needed

closed 2024-02-11 by fboundy