stevebuk1/pv_opt #106: PV_opt costs not displaying

spetrie13, 2024-02-06:
AppDaemon Current version: 0.16.4
PV_OPT VERSION = "3.6.2"

- PVOPT Results don't display on dashboard
- sensor.pvopt_base_cost  gives errors to display and errors in logs
- FutureWarning: 'T' is deprecated and will be removed in a future version
- FutureWarning: 'H' is deprecated and will be removed in a future version. 

Any help appreciated. Do you need any more log files?
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14179408/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14179411/main.log)

# User configuration ---  EDIT AWAY! ---

  inverter_type: "SOLIS_SOLAX_MODBUS"
  # inverter_type: "SOLIS_CORE_MODBUS"
  # inverter_type: SOLIS_SOLARMAN
  device_name: solis_modbus
  consumption_history_days: 7

 # If true the current config in HA will be over-written with that in the config.yaml. 
  overwrite_ha_on_restart: true
  redact_personal_data_from_log: true

Dashboard screenshot:
[image]

PV_OPT log:
I have attached the log files
01:00:02     INFO: Output written to sensor.pvopt_optimiser_elapsed
01:00:02     INFO: Average unit cost today: 0.00p/kWh
01:00:02     INFO: Output written to sensor.pvopt_unit_cost_today
01:00:02  WARNING: pv_opt: Entity sensor.pvopt_base_cost not found in namespace default
01:00:02     INFO: Output written to sensor.pvopt_base_cost
01:00:03  WARNING: pv_opt: Entity sensor.pvopt_opt_cost not found in namespace default
01:00:03     INFO: Output written to sensor.pvopt_opt_cost
01:00:03     INFO: Output written to sensor.pvopt_charge_start
01:00:03     INFO: Output written to sensor.pvopt_charge_end
01:00:03     INFO: Output written to sensor.pvopt_charge_current
01:00:03     INFO: Output written to sensor.pvopt_soc_h1
01:00:03     INFO: Output written to sensor.pvopt_soc_h4
01:00:03     INFO: Output written to sensor.pvopt_soc_h8
01:00:03     INFO: Output written to sensor.pvopt_soc_h12
01:00:03     INFO: Read only mode enabled. Not querying inverter.
01:10:00     INFO: Optimiser triggered by Scheduler 

Main log
11:50:03 WARNING HASS: Error setting Home Assistant state default.sensor.pvopt_opt_cost, {'state': 0.0, 'attributes': {'friendly_name': 'PV Opt Optimised Cost', 'device_class': 'monetary', 'state_class': 'measurement', 'unit_of_measurement': 'GBP', 'cost_today': 0.0, 'cost_tomorrow': 0.0, 'soc': [{'period_start': '2024-02-06T11:30:00+00:00', 'soc': 68.5}, {'period_start': '2024-02-06T12:00:00+00:00', 'soc': .....
11:50:03 WARNING HASS: Code: 400, error: {"message":"Invalid JSON specified."}

Error log:
<html>
<body>

10:00:10 WARNING pv_opt: ------------------------------------------------------------
--
 
TypeError: 'NoneType' object is not subscriptable
~~~~^^^
df = pd.DataFrame(hist[0]).set_index("last_updated")["state"]
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 251, in hass2df
^^^^^^^^^^^^^
df = self.hass2df(
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1892, in _get_hass_power_from_daily_kwh
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
df = self._get_hass_power_from_daily_kwh(
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1922, in load_consumption
^^^^^^^^^^^^^^^^^^^^^^
consumption = self.load_consumption(
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1316, in optimise
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
self.optimise()
File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1273, in optimise_time
^^^^^^^^^^^^^^
return f(*args, **kw)
File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
10:00:10 WARNING pv_opt: Traceback (most recent call last):
10:00:10 WARNING pv_opt: ------------------------------------------------------------
10:00:10 WARNING pv_opt: Worker Ags: {'id': '32579a1f996144f98f30b695d2032c1f', 'name': 'pv_opt', 'objectid': '5e948e0aa1314043a069275605dad6af', 'type': 'scheduler', 'function': <bound method PVOpt.optimise_time of <pv_opt.PVOpt object at 0x7f4aa65499d0>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 600, '__thread_id': 'thread-0'}}
10:00:10 WARNING pv_opt: Unexpected error in worker for App pv_opt:
10:00:10 WARNING pv_opt: ------------------------------------------------------------

<!--EndFragment-->
</body>
</html>

APPDemon log:

/homeassistant/appdaemon/apps/pv_opt/pvpy.py:209: FutureWarning: 'T' is deprecated and will be removed in a future version. Please use 'min' instead of 'T'.
  df.index[-1] + pd.Timedelta("30T"),
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:210: FutureWarning: 'H' is deprecated and will be removed in a future version. Please use 'h' instead of 'H'.
  df.index[-1] + pd.Timedelta("24H"),
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:208: FutureWarning: 'T' is
 deprecated and will be removed in a future version, please use 'min' instead.
  extended_index = pd.date_range(
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:209: FutureWarning: 'T' is deprecated and will be removed in a future version. Please use 'min' instead of 'T'.
  df.index[-1] + pd.Timedelta("30T"),
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:210: FutureWarning: 'H' is deprecated and will be removed in a future version. Please use 'h' instead of 'H'.
  df.index[-1] + pd.Timedelta("24H"),
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:208: FutureWarning: 'T' is deprecated and will be removed in a future version, please use 'min' instead.
  extended_index = pd.date_range(
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:209: FutureWarning: 'T' is deprecated and will be removed in a future version. Please use 'min' instead of 'T'.
  df.index[-1] + pd.Timedelta("30T"),
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:210: FutureWarning: 'H' is deprecated and will be removed in a future version. Please use 'h' instead of 'H'.
  df.index[-1] + pd.Timedelta("24H"),

SzosszeNET, 2024-02-07:
By the looks exact same issue as #105 

From the log 

11:50:00     INFO: Checking tariffs:
11:50:00     INFO: -----------------
11:50:00     INFO:   Import: E-1R-GO-VAR-22-10-14-N                   Start: 2023-06-02 03:30:00+0000 End: 2024-02-07 00:30:00+0000 
11:50:00     INFO:   Export: E-1R-OUTGOING-LITE-FIX-12M-23-09-12-N    Start: 2023-09-11 23:00:00+0000 End: N/A 
11:50:00     INFO:   - Tariffs OK
11:50:00     INFO: 

Might want to close this and start watching https://github.com/fboundy/pv_opt/issues/105

fboundy, 2024-02-07:
Looks like it.

I should get on to this at the end of the week.

fboundy, 2024-02-08:
Closing this because it is the same as #105.

closed 2024-02-08 by fboundy