stevebuk1/pv_opt #115: Igonre past consumption feature

SzosszeNET, 2024-02-11:
Just wondering if it would make sense or posible to add a switch to ignore past consumption? I'd think this would be useful for:

1. troubleshooting purposes
2.  help with users not having the required past consumption data or issues with unreliable adapter

Not something that I think is huge one, but I think could potentially help with those two issues.

fboundy, 2024-02-11:
Yes, thins could be done. You’d need to set an estimate of daily consumption with it. I’d probably also have an option to use a constant load or to scale it to typical usage profile. I did have something like this in an earlier version.

SzosszeNET, 2024-02-11:
Thank you, you are a rockstart. Wish I could help you with this effort, like your project a lot

fboundy, 2024-02-11:
Implemented in #117

closed 2024-02-11 by fboundy

SzosszeNET, 2024-02-11:
Thank you, that's a very quick turnaround. Sadly I don't think it works as intended.

Updated the new dashboard the config.yaml and the module however sadly I think it's still wanting to load data?

16:12:59     INFO: ******************* PV Opt v3.8.0 *******************
16:12:59     INFO: 
16:12:59     INFO: Local timezone set to GB
16:12:59     INFO: Time Zone Offset: 0.0 minutes
16:12:59     INFO: Inverter type: SOLIS_SOLAX_MODBUS: inverter module: solis.py
16:12:59     INFO: Reading arguments from YAML:
16:12:59     INFO: -----------------------------------
16:12:59     INFO: 
16:12:59     INFO:   Over-write flag is set so YAML will over-write HA
16:12:59     INFO:     redact_personal_data_from_log      = True                                                        True: value in YAML
16:12:59     INFO:     read_only                          = False                                                      False: value in YAML
16:12:59     INFO:     allow_cyclic                       = True                                                        True: value in YAML
16:12:59     INFO:     battery_capacity_wh                = 5000                                                        5000: value in YAML
16:12:59     INFO:     charger_power_watts                = 3200                                                        3200: value in YAML
16:12:59     INFO:     inverter_power_watts               = 3200                                                        3200: value in YAML
16:12:59     INFO:     id_solcast_today                   = sensor.solcast_pv_forecast_forecast_today                 3.71985: value(s) in YAML
16:12:59     INFO:     id_solcast_tomorrow                = sensor.solcast_pv_forecast_forecast_tomorrow              11.7548: value(s) in YAML
16:12:59     INFO:     use_consumption_history            = False                                                      False: value in YAML
16:12:59  WARNING:     consumption_margin                 = 10                                                            10: system default. Unable to read from HA entities listed in YAML. No default in YAML.
16:12:59     INFO:     octopus_account                    = XXXXXXXXX                                                XXXXXXXXX: YAML default value. No default defined.
16:12:59     INFO:     octopus_api_key                    = API_KEY                        API_KEY: YAML default value. No default defined.
16:12:59     INFO:     battery_voltage                    = sensor.solis_battery_voltage                                53.5: value in YAML
16:12:59     INFO:     update_cycle_seconds               = 15                                                            15: value in YAML
16:12:59     INFO:     maximum_dod_percent                = number.solis_battery_minimum_soc                            15.0: value in YAML
16:12:59     INFO:     id_consumption_today               = sensor.solis_house_load_today                               14.1: value(s) in YAML
16:12:59     INFO:     id_grid_import_today               = sensor.solis_grid_import_today                              16.4: value(s) in YAML
16:12:59     INFO:     id_grid_export_today               = sensor.solis_grid_export_today                               0.2: value(s) in YAML
16:12:59     INFO:     id_battery_soc                     = sensor.solis_battery_soc                                    98.0: value(s) in YAML
16:12:59     INFO:     id_timed_charge_start_hours        = number.solis_timed_charge_start_hours                       13.0: value(s) in YAML
16:12:59     INFO:     id_timed_charge_start_minutes      = number.solis_timed_charge_start_minutes                     30.0: value(s) in YAML
16:12:59     INFO:     id_timed_charge_end_hours          = number.solis_timed_charge_end_hours                         14.0: value(s) in YAML
16:12:59     INFO:     id_timed_charge_end_minutes        = number.solis_timed_charge_end_minutes                        0.0: value(s) in YAML
16:12:59     INFO:     id_timed_charge_current            = number.solis_timed_charge_current                           60.0: value(s) in YAML
16:12:59     INFO:     id_timed_discharge_start_hours     = number.solis_timed_discharge_start_hours                     0.0: value(s) in YAML
16:12:59     INFO:     id_timed_discharge_start_minutes   = number.solis_timed_discharge_start_minutes                   0.0: value(s) in YAML
16:12:59     INFO:     id_timed_discharge_end_hours       = number.solis_timed_discharge_end_hours                       0.0: value(s) in YAML
16:12:59     INFO:     id_timed_discharge_end_minutes     = number.solis_timed_discharge_end_minutes                     0.0: value(s) in YAML
16:12:59     INFO:     id_timed_discharge_current         = number.solis_timed_discharge_current                        61.0: value(s) in YAML
16:12:59     INFO:     id_timed_charge_discharge_button   = button.solis_update_charge_discharge_times                2024-02-11T13:01:26.545320+00:00: value(s) in YAML
16:12:59     INFO:     id_inverter_mode                   = select.solis_energy_storage_control_switch                Timed Charge/Discharge: value(s) in YAML
16:12:59     INFO: 
16:12:59     INFO: Checking config:
16:12:59     INFO: -----------------------
16:12:59  WARNING:     forced_charge                      = True                                                        True: system default. Not in YAML.
[... cut ...]
  File "/usr/lib/python3.11/site-packages/pandas/core/dtypes/astype.py", line 182, in astype_array
    values = _astype_nansafe(values, dtype, copy=copy)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/dtypes/astype.py", line 133, in _astype_nansafe
    return arr.astype(dtype, copy=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: ''

16:13:00 WARNING pv_opt: ------------------------------------------------------------

fboundy, 2024-02-11:
Odd - it's failing on getting the SOC not the consumption. Try 3.8.1 which is a pre-release version

SzosszeNET, 2024-02-11:
Yes, but looking for a days worth of SOC data?

This is waht I see on 3.8.1

19:47:33     INFO: >>> Getting 1 days' history for sensor.solis_battery_soc
19:47:33     INFO: >>> Entity exits: True
19:47:33     INFO: >>> soc_now: 25.0
19:47:33     INFO: >>> x: last_updated
2024-02-11 12:13:09.679393+00:00    33
2024-02-11 12:13:19.882428+00:00    32
2024-02-11 12:17:09.670724+00:00    31
2024-02-11 12:21:13.768453+00:00    30
2024-02-11 12:26:21.471081+00:00    30
                                    ..
2024-02-11 19:38:20.876831+00:00    29
2024-02-11 19:40:35.377827+00:00    28
2024-02-11 19:42:35.381992+00:00    27
2024-02-11 19:45:07.081359+00:00    26
2024-02-11 19:47:22.087304+00:00    25
Name: state, Length: 190, dtype: object
19:48:33     INFO: >>> Agile Callback Handler
19:48:33     INFO: >>> Contract end day: 12 Today:11  False
19:48:33     INFO: >>> Current hour:     19           True

fboundy, 2024-02-11:
Try 3.8.2 - also pre-release

SzosszeNET, 2024-02-11:
20:25:32     INFO: >>> Getting 1 days' history for sensor.solis_battery_soc
20:25:32     INFO: >>> Entity exits: True
20:25:32     INFO: >>> soc_now: 15.0
20:25:32     INFO: >>> x: last_updated
2024-02-11 12:13:09.679393+00:00    33
2024-02-11 12:13:19.882428+00:00    32
2024-02-11 12:17:09.670724+00:00    31
2024-02-11 12:21:13.768453+00:00    30
2024-02-11 12:26:21.471081+00:00    30
                                    ..
2024-02-11 20:02:07.078033+00:00    19
2024-02-11 20:04:26.877589+00:00    18
2024-02-11 20:07:22.283664+00:00    17
2024-02-11 20:10:01.678380+00:00    16
2024-02-11 20:12:22.181466+00:00    15
Name: state, Length: 200, dtype: object
20:25:32     INFO: >>> Original: last_updated
2024-02-11 19:59:26.878510+00:00    20
2024-02-11 20:02:07.078033+00:00    19
2024-02-11 20:04:26.877589+00:00    18
2024-02-11 20:07:22.283664+00:00    17
2024-02-11 20:10:01.678380+00:00    16
2024-02-11 20:12:22.181466+00:00    15
Name: state, dtype: object

20:25:32 WARNING pv_opt: Unexpected error running initialize() for pv_opt
20:25:32 WARNING pv_opt: ------------------------------------------------------------
20:25:32 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 342, in initialize
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1358, in optimise
    x = pd.to_numeic(x, errors="coerce").interpolate()
        ^^^^^^^^^^^^
AttributeError: module 'pandas' has no attribute 'to_numeic'

20:25:32 WARNING pv_opt: ------------------------------------------------------------

fboundy, 2024-02-11:
Typo! Should be “to_numeric”. That’s what happens when you rush…

SzosszeNET, 2024-02-11:
Yes, that did the trick! Appears to be working

21:07:28     INFO: >>> Start: 11/02 21:00 End: 12/02 23:30
21:07:28     INFO: 
21:07:28     INFO: Checking for Hold SOC slots
21:07:28     INFO: 
21:07:28     INFO: Optimal forced charge/discharge slots:
21:07:28     INFO:    11-Feb 21:00 - 11-Feb 21:30   Power: -2800W  SOC:   48% ->   20%  
21:07:28     INFO:    11-Feb 22:30 - 11-Feb 23:00   Power:  3200W  SOC:   15% ->   44%  
21:07:28     INFO:    12-Feb 00:00 - 12-Feb 00:30   Power:  3200W  SOC:   29% ->   59%  
21:07:28     INFO:    12-Feb 03:30 - 12-Feb 04:00   Power:  3200W  SOC:   41% ->   70%  
21:07:28     INFO:    12-Feb 04:30 - 12-Feb 05:00   Power:  3200W  SOC:   67% ->   97%  
21:07:28     INFO:    12-Feb 09:30 - 12-Feb 10:00   Power:  3000W  SOC:   65% ->   92%  
21:07:28     INFO:    12-Feb 20:00 - 12-Feb 20:30   Power:  2600W  SOC:   15% ->   38%  
21:07:28     INFO:    12-Feb 21:30 - 12-Feb 22:00   Power:  1300W  SOC:   15% ->   27%  
21:07:28     INFO:    12-Feb 22:30 - 12-Feb 23:00   Power:  1600W  SOC:   15% ->   30%  
21:07:28     INFO: 
21:07:28     INFO: Plan time: 11-Feb 21:00 - 12-Feb 23:30 Initial SOC: 48.5 Base Cost: 146.33 Opt Cost: 115.26
21:07:28     INFO: 
21:07:28     INFO: Optimiser elapsed time 11.6 seconds
21:07:28     INFO: 
21:07:28     INFO: 
21:07:28  WARNING: pv_opt: Entity sensor.pvopt_optimiser_elapsed not found in namespace default
21:07:28     INFO: Output written to sensor.pvopt_optimiser_elapsed