stevebuk1/pv_opt #124: Froze/not updating charge plan correctly

solarjones, 2024-02-14:
VERSION = "3.8.5"

This morning I checked up on the automation and found the charing plan appearing to be stuck in the past and wanting to change at a more expensive rate. Also noticed its start SOC wasn't correct (much lower than the actual battery SOC). This would explain why to wanted to charge when it wasn't necessary. (10:47 time stamp) I tried toggling to ReadOnly and live mode but it didn't seem to make any new calculations. Then tried switching from predicted load to the new profile load modes. No change. Then I restarted AppDaemon which seem to do the trick.

Not sure if these logs contain enough, I can always switch debug=true if needed.

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14285614/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14285615/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14285616/error.log)

Thanks for all your work in the recent days. I could hardly keep up with the fixes and versions that came out, brilliant stuff.

Regards

fboundy, 2024-02-14:
Think this was probably caused by two weird Octopus Savings Events that had zero value and only lasted 3 minutes each. Mys system did the same. 3.8.6 should fix this.

fboundy, 2024-02-14:
I initially though this was something else but looking at your error log it was caused by an empty list. Should be fixed with 3.8.7

solarjones, 2024-02-14:
Got some errors on startup now...

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14289404/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14289405/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14289406/error.log)

fboundy, 2024-02-15:
Did these persist? I think there may have been some issues with the Octopus API last night.

solarjones, 2024-02-15:
Yes, still not working this morning.
[error.log](https://github.com/fboundy/pv_opt/files/14293668/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14293665/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14293667/main.log)

Anything I can do?

fboundy, 2024-02-15:
Hmmm - it’s reporting an error 602 from the octopus API query. Unfortunately I’ve no idea what this means! I’ll add some additional logging and get something out later today.

fboundy, 2024-02-15:
There are some issues around OE tariff changes. Try restarting HA and AppDaemon.

solarjones, 2024-02-15:
> There are some issues around OE tariff changes. Try restarting HA and AppDaemon.
> […](#)

No change after several HA & AppDaemon restarts.

SzosszeNET, 2024-02-15:
@solarjones as usual butting in. Can't see your tariff mentioned in your pv_opt log.
Assume you have the octopus energy integration, can you check if the integration mentions what's your tariff?
Can you check if the entities are written?
In your pv_opt config I assume you have your tariff baked in and not your octopus API deeds?

If the integration doesn't write the data (see attached image. Disable it. Wait a little. Re-enable. Check pv_opt and the sensors. Just discovered this, however my charge was working by the looks of it.

[image]
[image]

solarjones, 2024-02-15:
The PV_OPT config I've left as default so not changed it.

I am using Octopus integration in HA
[image]

It doesn't mention my tariff from what I can see. Where would I find this in HA?

And this was from a previous PV_OPT log when things were working:

22:10:00     INFO: Checking tariffs:
22:10:00     INFO: -----------------
22:10:00     INFO:   Import: E-1R-AGILE-FLEX-22-11-25-A               Start: 2024-02-04 13:00:00+0000 End: 2024-02-14 23:00:00+0000 
22:10:00     INFO:   Export: E-1R-OUTGOING-FIX-12M-19-05-13-A         Start: 2019-05-15 23:00:00+0000 End: N/A 
22:10:00     INFO: 
22:10:00     INFO: AGILE tariff detected. Rates will update at 16:00 daily
22:10:00     INFO:   - Tariffs OK

Just noticed the Tariff end date above was yesterday?! Even though I'm still on Agile, its not ended as per my OE account:
[image]

Thanks for any other suggestions.

solarjones, 2024-02-15:
[image]
[image]

SzosszeNET, 2024-02-15:
Ah, so assume you restarted HA at 8 in the morning and that kicked back your integration into life.

I have the octopus API in my PV_OPT config, not sure now I said it does anything as it also looks at and finds the OE integration sensors. Guess would not hurt trying to add it? And removing if doesn't do anything.

octopus_account: !secret octopus_account
octopus_api_key: !secret octopus_api_key

Yes agile "ended" yesterday inline with the email octopus sent out about a month ago.
[image]

solarjones, 2024-02-16:
As suggested, I added in Octopus API & Account, restarted HA and AppDaemon.  Different error now.

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14308860/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14308861/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14308863/error.log)

solarjones, 2024-02-16:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14308979/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14308980/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14308981/error.log)

SzosszeNET, 2024-02-16:
Magic 🤣

fboundy, 2024-02-16:
OK - so it was  detecting two octopus import entities - probably because of the changes to the Agile tariff and defaulted to the 1st one which I suspect is empty. I've added some logging which should confirm this if you can send me the `pv_opt.log`.

The issue with consumption seems to have been just the once. I've added a check that the history. 

Hopefully 3.8.3 addresses both.

solarjones, 2024-02-16:
v 3.8.8 installed.

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14312157/pv_opt.log)

solarjones, 2024-02-16:
v 3.8.8 installed.
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14312220/pv_opt.log)
Note, the MPAN was in plain text and obfuscated.

solarjones, 2024-02-16:
Noticed this too:

[image]

fboundy, 2024-02-16:
OK - think I understand now on the tariff_code. The attribute doesn't exist for the old entity. Hence the error.

Fixed in 3.8.9

MPAN redaction should also be fixed.

solarjones, 2024-02-16:
Installed 3.8.9

MPAN redaction working thanks.
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14314092/pv_opt.log)

[image]

Do I need this in the config still?
octopus_account: !secret octopus_account
octopus_api_key: !secret octopus_api_key

Thanks.

SzosszeNET, 2024-02-16:
It was just a hunch from me, feel free to remove.

fboundy, 2024-02-16:
Looks like you do for now. There’s another tweak I need to make.

SzosszeNET, 2024-02-17:
Looks like they is an OA integration update out addressing this issue

solarjones, 2024-02-17:
Is this expected behaviour?
My SOC will be depleted well before its next planned charge slot.
[image]

[image]

And there are cheaper slots prior to 03:00.
[image]

Thanks

solarjones, 2024-02-17:
I restarted AppDaemon and its stuck on Loading Tariffs now.
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14320273/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14320274/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14320275/error.log)

Thanks.

spetrie13, 2024-02-19:
My charge plan is also not updating. It would appear that this is due to the tariff prices not being pulled from the Octopus API. The future price seems to be updated around 10:30 GMT in API but pv_opt does not seem to load the information. If I restart the App Demon then charge plan updates again. 
V3.8.8
Pre AppDemon restart:
[image]
[image]
[image]

[error.log](https://github.com/fboundy/pv_opt/files/14331214/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14331215/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14331216/pv_opt.log)

Post AppDemon restart
[image]
[image]

[main (1).log](https://github.com/fboundy/pv_opt/files/14331264/main.1.log)
[pv_opt (1).log](https://github.com/fboundy/pv_opt/files/14331266/pv_opt.1.log)

solarjones, 2024-02-19:
There was a new release of Octopus Energy integration this morning. I downloaded it, restarted HA + AppDaemon. It appears to be working for the time being. Will see if it auto updates at 16:00 this afternoon.

fboundy, 2024-02-19:
Ok that’s interesting. I have been busy since Saturday morning so not had a chance to look at anything yet

fboundy, 2024-02-19:
I think I have found the source of these bugs and fixed in 3.8.10.

fboundy, 2024-02-22:
Closing this as #132 supersedes it

spetrie13, 2024-02-22:
Still not working for me with latest revision 3.8.10. On OE Go tarrif.
Optimises if I restart AppDemon. Tried modification the load margin and
toggle read to see if it would update, but no joy.

fboundy, 2024-02-23:
Can you please provide the logs and then I can try to work our what it's doing?

spetrie13, 2024-02-24:
Will have to download logs when I get home tomorrow. Can’t remotely
download the logs.
Will apply the 3.9.0 update tonight.

fboundy, 2024-02-24:
3.9.2 now which fixes a few more bugs - mainly if you don't have export

spetrie13, 2024-02-26:
Seems to be loading the tariff info today with V3.9.0 and providing an
optimised charge schedule.

Do you still want the logs?
Back home now

closed 2024-02-26 by fboundy