stevebuk1/pv_opt #128: Trying to set inverter current value higher than threshold

ThomasPrior, 2024-02-18:
**Describe the bug**
PV_OPT keeps trying to set my inverter import/export current above the 100A limit of my inverter.

**Expected behavior**
I would expect PV_OPT to either retrieve the maximum value from the inverter or the config file and ensure that the value passed for export does not exceed that value.

**Screenshots**
[image]
![image](https://github.com/fboundy/pv_opt/assets/34111848/a004ffd0-aa38-48e
[image]

**Additional context**
pv_opt.log attached
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14323828/pv_opt.log)
9-a92f-b33ee85ddebf)

HA version:
Core 2024.2.2
Frontend 20240207.1

PV_Opt version:
3.8.9

fboundy, 2024-02-19:
Closing this issue as it is the same as #91

closed 2024-02-19 by fboundy

fboundy, 2024-02-19:
It looks like this is at least partially because the discharge power is being rounded up from 4840w to 5000w. In 3.8.10 try setting the Power Resolution to 0 (it will probably cause a divide by zero error in previous versions) and let me know if that helps.  The trouble with the Charge Current Limit is that it is set dynamically by the BMS.

ThomasPrior, 2024-04-06:
I'm afraid this is still happening. 

My next charge is calculated at 102A, which cannot be written to the inverter as the limit is 100A.

Can a limit be set that caps the optimum charge current value?

[image]

[image]

[image]