stevebuk1/pv_opt #129: Support for Sunsynk inverter via Solarsynk2 add-on for Home Assistant

ElectricMolehill, 2024-02-18:
Hi fboundy 
- this is fantastic, just what I was looking for. Unfortunately I bought a solar system before coming across pv_opt and it's Sunsynk. Sorry I'm not qualified to write the inverter class would you consider doing it? I have written up what I know in the attached document and also completed the template below. There are one or two gaps, maybe we can fill them by testing or asking the developer of the solarsynk2 add-on martinville

regards
Mark

PV Opt can be updated to support additional models and brands of inverter. If you want to request this, please supply as much of the following information as possible:

<h2>Home Assistant Entities to Read</h2>
PV Opt reads data from a number of entities in Home Assistant. These will vary by inverter. Please document what your inverter provides. Examples are provided from the Solis Solax Modbus integration:

    Battery minimum SOC: number.solis_battery_minimum_soc
    Battery SOC:         sensor.solis_battery_soc
    Either:
      Load power:        [sensor.solis_house_load, sensor.solis_bypass_load], or
      Load today:        sensor.solis_house_load_today
    Either:
      Grid import power: sensor.solis_grid_import_power and
      Grid export power: sensor.solis_grid_export_power
    Or:
      Grid import today: sensor.solis_grid_import_today and
      Grid export today: sensor.solis_grid_export_today

<h2>Inverter Control</h2>
PV Opt also needs to control your inverter. How this is done and what can be done varies hugely by brand, by model and by the integration you use with Home Assistant. Please provide as much detail under each heading. Examples are provided for the Solis inverter using the both the Solax Modbus Integration and the HA Core Modbus functionality. Please add your details below:

<h3>Examples</h3>

| | Solis Solax Modbus | Solis Core HA Modbus |
|:--|:--:|:--:|
| How do you enable forced charging? | Only done via timed charge | Only done via timed charge |
| How do you enable forced discharging? | Only done via timed discharge | Only done via distimed charge |
| Are forced charge and discharge controlled using Power or Current?| Current | Current |
| Does your inverter support timed charging slots? | Yes | Yes |      
| Does your inverter support timed discharging slots? | Yes | Yes |   
| How do you enabled timed charge? | Set the following: `number.timed_charge_start_hour`, `number.timed_charge_start_minute`, `number.timed_charge_end_hour`, `number.timed_charge_start_minute`. Then press '`button.solis_update_charge_discharge_times`. Set the required current to `number.solis_timed_charge_current`. Ensure `select.solis_energy_storage_control_switch` has Timed Grid Charging enabled. | Use the `modbus\write_register` service to write start and end hours and minutes to the correct registers. Ensure the `energy_storage_control_switch` has Bits 2 and 5 enabled. Set the required current using the `modbus\write_register` service |      
| How do you enabled timed charge? | As for charging but all entities are `discharge` rather than `charge`| As for charging but registers are different. |
| Can you set a target SOC for a timed charge period? | No | No |
| Can you set a target SOC for a timed discharge period? | No | No |
| What modes does you inverter have and how are they controlled?| One register on the inverter `Energy Control Switch` which controls mode in a bit-wise sense. Only some modes can be selected vai the entity `select.solis_energy_storage_control_switch` | One register on the inverter `Energy Control Switch` which controls mode in a bit-wise sense. The required number is set using the `modbus\write_register` service |      

<h3>Your Setup</h3>

<h4>What is the inverter brand?</h4>
Sunsynk

<h4>What is the inverter model?</h4>
ECCO 3.6kW hybrid inverter SYNK-3.6K-SG04LP1

<h4>What integration do you use in Home Assistant</h4>
Solarsynk2 add-on. There is another one Deye/Sunsynk Home Assistant add-on by kellerza but this requires a direct wired connection to the inverter
<h4>How do you enable forced charging?</h4>
Only done via a timer. Set a time slot x, tick grid time x to allow charging from grid, target SOC x (above current SOC) and power x. 6 slots are available

<h4>How do you enable forced discharging?</h4>
Only done via a timer. Set a time slot x, tick grid time x to allow charging from grid, target SOC x (below current SOC) and power x

<h4>Are forced charge and discharge controlled using Power or Current?</h4>
Power (W) and target SOC whichever is the smaller

<h4> Does your inverter support timed charging slots?</h4>
6 slots used for charge and discharge. Should be contiguous, sequential and cover the whole 24 hrs

<h4> Does your inverter support timed discharging slots?</h4>
Same timeslots as for charging but with a different target SOC

<h4>How do you enabled timed charge?</h4>
Set Use Timer = true To do this, populate helper input_text.solarsynk_{inverter serial no}_settings with json to update Sunsynk cloud api:
{"peakAndVallery": "1"} (just possible this is a misspelling)
Also required for slot x - 
Start time ("sellTime(x)": "02:00")
Grid timer x ("time(x)on": true)
Target SOC ("cap(x)": "80")  if this is higher value than current SOC, batteries wil charge up to this value from the grid
Power x ("sellTime(x)Pac": "3600")
End time is defined implicitly by the start time of the next slot. An appdaemon process runs every 5 min to post the contents of this helper to the Sunsunk cloud api. This is not ideal because it's not online. Perhaps I could get the Solarsynk developer martinville to add a realtime update method

<h4>How do you enabled timed discharge?</h4>
As for charging but set target SOC low

<h4>Can you set a target SOC for a timed charge period?</h4>
Yes - this would be the usual manual method

<h4>Can you set a target SOC for a timed discharge period?</h4>
Yes

<h4>What modes does you inverter have and how are they controlled?</h4>
There are some settings which limit export to load or grid. Details to be confirmed but probably should be set manually

<h4>Any other useful information</h4>
See attached doc
[sunsynk inverter class for pv_opt.odt](https://github.com/fboundy/pv_opt/files/14324224/sunsynk.inverter.class.for.pv_opt.odt)

ElectricMolehill, 2024-02-18:
Forgot to add HA entities to read:

    Battery minimum SOC: sensor.solarsynk_{inverter sn}_battery_shutdown_cap (%)
    Battery SOC:         sensor.solarsynk_{inverter sn}_battery_soc (%)
    Either:
      Load power:        sensor.solarsynk_{inverter sn}_load_power (kW), or
      Load today:        sensor.solarsynk_{inverter sn}_day_load_energy (i.e kWh)
    Either:
      Grid import power: sensor.solarsynk_{inverter sn}_grid_power (this is probably total import and export)
      Grid export power: 
    Or:
      Grid import today: sensor.solarsynk_{inverter sn}_day_grid_import (kWh) and
      Grid export today: sensor.solarsynk_{inverter sn}_day_grid_export (kWh)

ElectricMolehill, 2024-02-19:
That's great, thankyou. The installation's only been in for a week and we're not certified to export yet. Then I'll need to familiarise myself with programming it manually. What do you think about the update settings method provided by the Solarsynk add-on? It takes up to 5 min to send to the cloud, then maybe another 5 min until the results are readable from HA sensors. Is that fast enough?

regards
Mark

ElectricMolehill, 2024-02-23:
Thanks I'll give it a go! I'll try to clarify the working modes for the inverter tomorrow - they control whether the inverter gives priority to essential load / non-essential load / battery / grid export. I've also discovered another check we should do - the inverter will support battery types which use voltage instead of SOC to control charge / discharge so we should check that the battery is compatible with SOC. Also, I think my documentation is wrong about ticking 'Grid' for a discharge, it looks like it should be unticked. I'll update the documentation

ElectricMolehill, 2024-02-23:
Sorry my mistake - {inverter sn} should be {sunsynk_serial} as defined in solarsynk addon. Also - cosmetic change - could you change the spelling of solarsync2 to solarsynk2 please. I don't know how far I can get with testing because we're not certified for export yet so we haven't changed tariff from the fixed rate Flexible Octopus to some thing smart such as Agile or Flux. Unless there's a way to bypass it for testing purposes?

ElectricMolehill, 2024-02-24:
I think there are 2 typos in sunsynk.py:
[image]

fboundy, 2024-02-24:
Is there any way of extracting {sunsynk_serial} from the addon? If not I might prefer to keep {inverter_sn} as a generic parameter across brands. Will fix the typos now and re-release as alpha-2

fboundy, 2024-02-24:
alpha-2 available. You will need to manually copy `config-sunsynk.yaml`

ElectricMolehill, 2024-02-24:
[sunsynk_serial} is in the config for the Solarsynk add-on but I can't see a sensor for it

fboundy, 2024-02-24:
OK - then it shouldn't matter if we call it something else in the config file. I suspect other brand/integrations may need something similar so I'd rather keep it generic where possible

ElectricMolehill, 2024-02-24:
Sorry but I think there's another typo:
[image]

ElectricMolehill, 2024-02-24:
Also I think the underline character is missing in {inverter_sn} on all of the sensors above

ElectricMolehill, 2024-02-24:
Updated my documentation for the inverter. There are a couple of problems, such as

- No method in Solarsynk integration to check max power of inverter
- No method in Solarsynk integration to check whether an Auxiliary power source (e.g. generator / wind turbine) is connected which I imagine would invalidate a pv_opt plan
- The Solarsynk update method is based on a batch process and takes 5 - 10 minutes to return, plus the return codes are only visible in the Solarsynk log

Apart from that I think it should work!
[sunsynk inverter class for pv_opt v2.odt](https://github.com/fboundy/pv_opt/files/14394269/sunsynk.inverter.class.for.pv_opt.v2.odt)

fboundy, 2024-02-25:
That's great - I will have a proper read of this now and have a think about how to implement it

ElectricMolehill, 2024-03-04:
Here are the logs from running v4.0.0 alpha2
[error.log](https://github.com/fboundy/pv_opt/files/14483328/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14483329/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14483330/pv_opt.log)

fboundy, 2024-03-06:
Cheers

Away on holiday this week. Will have a look when I’m back.

ElectricMolehill, 2024-03-07:
Hope you're having a nice holiday. I made some progress by hard coding the inverter serial number in config.yaml (inverter serial number redacted manually - 10 digits, also my Octopus account has an e instead of an m):

[image]
[image]

Now pv_opt gets further:
[error.log](https://github.com/fboundy/pv_opt/files/14526077/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14526078/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14526079/pv_opt.log)

You'll see I'm on the Octopus Flux tariff for import but don't yet have an export tariff. I have submitted the MCS certificate and permission from the DNO but I think it can take weeks to get an export MPAN number

alz41, 2024-03-09:
following developments

fboundy, 2024-03-16:
I've synced this branch up with the main branch so it should now handle the lack of an export contract OK which was the point of failure before. Also added the inverter_sn to the redaction list. Released as 4.0.0-alpha3

ElectricMolehill, 2024-03-16:
Hope you had a good holiday! We got our export MPAN this morning  but Octopus don't seem to have taken any readings yet. Perhaps overnight. I reloaded the Octopus integration in HA and the export sensors appeared. I installed pv_opt 4.0.0-alpha4 and ran again. I reverted to the config_sunsynk.yaml from GitHub and used this to overwrite my config.yaml because pv_opt doesn't seem to be picking up some values correctly. I edited it to add the correct value of {inverter_sn}. From the logs it looks like pv_opt isn't picking up {device_name} or {inverter_sn}. Also I think there are a couple of typos in config.yaml:
 - inverter_type: "**SUNSYNK_SOLARSYNC2**" should be inverter_type: "**SUNSYNK_SOLARSYNK2**"
 - "maximum_dod_percent":  "sensor.{device_name}_{**inverter sn**}_battery_shutdown_cap"  # presumably should be {**inverter_sn**}
 - "id_battery_soc": " sensor.{device_name}_{**inverter sn**}_battery_soc" # as above underline character missing
 - "id_consumption_today": "sensor.{device_name}_{**inverter sn**}_day_load_energy" # as above, underline character missing
 - "id_grid_import_today": "sensor.{device_name}_{**inverter sn**}_day_grid_import" # as above, underline character missing
 - "id_grid_export_today": "sensor.{device_name}_{**inverter sn**}_day_grid_import" # as above, underline character missing, also I think **import** should be **export**
[error.log](https://github.com/fboundy/pv_opt/files/14625401/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14625402/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14625403/pv_opt.log)
Also I think redaction list needs another line for my MPAN number which is of the form 99e9999999_9999999999999 instead of 99m9999999_9999999999999 (perhaps e and m are region markers?)

fboundy, 2024-03-17:
Thanks for this. In 4.0.0-alpha-5 I have removed `config-sunsynk.yaml` and merged all the changes in to the main `config.yaml`. You should just need to un-comment the final block of parameters in this file.

I've also updated the regex pattern for the MPAN.

Hopefully with the correction of SOLARSYNC2 to SOLARSYNK2 it should get a bit further. Let me know how you get on.

ElectricMolehill, 2024-03-17:
Thanks - got a lot further this time but still seems to be ending with an error: 
ValueError: cannot reindex on an axis with duplicate labels
Redaction mostly working but there are a couple of lines in pv_opt.log still containing inverter serial number and account_MPAN number
One of the defaults in config.yaml still looks wrong :
[image]
Think this should read 
id_grid_export_today: sensor.{device_name}_{inverter_sn}_day_grid_export
instead of 
id_grid_export_today: sensor.{device_name}_{inverter_sn}_day_grid_import
[error.log](https://github.com/fboundy/pv_opt/files/14628889/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14628890/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14628891/pv_opt.log)

alz41, 2024-03-25:
would really like to get this working

fboundy, 2024-03-27:
OK - apologies for the delay. Please can you try pre-release `4.0.0-sunsynk-alpha-7` which should address the issues above and also includes various patches made to the production release.

ElectricMolehill, 2024-03-27:
Thanks I'll give it a go. Actually I wasn't ready until today - SSD failed and it's taken me a week to get a replacement and recover Home Assistant from backup. We've also got some persistent connectivity problems with Solarsynk as a result of which my updates from a couple of HA automations don't seem to be working. It seems to be affecting all users of Solarsynk so it might be the cloud app that is at fault. On the plus side I'm now exporting to the grid and getting paid for it. As a result I've found out more about the Sunsynk inverter's eccentricities. For example, charge current is not controlled by the timers. Also, to discharge the batteries to the grid I need to change the Work Mode from 'Limited to Home' = 2 to 'Selling First' = 0 in real time because otherwise I can't discharge the battery to Home at other times without exporting as well. I've also found a different Sunsynk HA integration by MorneSaunders360 which also works via the cloud API but has an HA service to update the inverter which might be more appropriate for pv_opt. I'll have a look at that. In the meantime I have updated my documentation - sorry if I've misled you so far!
[sunsynk inverter class for pv_opt v3.odt](https://github.com/fboundy/pv_opt/files/14781090/sunsynk.inverter.class.for.pv_opt.v3.odt)

alz41, 2024-03-27:
> OK - apologies for the delay. Please can you try pre-release `4.0.0-sunsynk-alpha-7` which should address the issues above and also includes various patches made to the production release.

Sunsynk.py still has references to "SUNSYNK_SOLARSYNC2" should be "SUNSYNK_SOLARSYNK2" (Lines 72, 86, 117)

[error.log](https://github.com/fboundy/pv_opt/files/14781704/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14781705/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14781706/pv_opt.log)

fboundy, 2024-03-28:
`4.0.0-sunsynk-alpha-8` should fix the typos and also bring the underlying code up to date with the latest production release

ElectricMolehill, 2024-03-28:
I think I've got the latest version but it's reporting 
 - PV Opt v4.0.0-4.0.0-sunsynk-alpha-7 
 in the pv_opt.log

When I start the application it is stuck at 'Initialising PV model'. There is a warning in error.log:
 - AttributeError: 'InverterController' object has no attribute 'is_online'
[error.log](https://github.com/fboundy/pv_opt/files/14793281/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14793282/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14793283/pv_opt.log)

fboundy, 2024-03-28:
Sorry about that. Fixed both of those so the latest is still `4.0.0-sunsynk-alpha-8`

ElectricMolehill, 2024-03-28:
Got a lot further this time - stopped at Optimising Charge Plan with an error
[error.log](https://github.com/fboundy/pv_opt/files/14794979/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14794980/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14794981/pv_opt.log)

fboundy, 2024-03-28:
The error is because the SOC is nan at 20:02:22.

It should just be interpolating any non-numeric states to prevent this. I did fix this code in the production version so maybe that didn’t get merged into this branch properly. I’ll check it tomorrow and if necessary add a bit more logging to diagnose further.

alz41, 2024-03-28:
sunsynk.py - lines;- 7,34,35,36,37,38 require changing from {inverter sn} to {inverter_sn}

fboundy, 2024-03-29:
> sunsynk.py - lines;- 7,34,35,36,37,38 require changing from {inverter sn} to {inverter_sn}

Fixed this and added some checks to the SOC in `4.0.0-sunsynk-alpha-9`

ElectricMolehill, 2024-03-29:
Error during initialize() for pv_opt:
- AttributeError: 'PVOpt' object has no attribute 'redact_patterns'

[error.log](https://github.com/fboundy/pv_opt/files/14804051/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14804053/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14804054/pv_opt.log)

fboundy, 2024-03-29:
fixed in `4.0.0-sunsynk-alpha-10`

ElectricMolehill, 2024-03-29:
I think it's published as alpha-19
It completes a plan!

It probably doesn't help that I lost a week's history when my SSD failed, and also the Solarsynk add-on is still suffering intermittent (but frequent) connectivity problems. Several users of Solarsynk have reported the same symptoms so probably cause is with the add-on or the Sunsynk cloud. This not only causes gaps in the history but any inverter updates are missed. This is fatal at the moment because it is hard to check whether a command has been successful. There's an open issue for this.

One problem I can see with the inverter class for pv_opt is that we are using the Battery Shutdown Capacity (set by the installer to 15 on mine) to define the minimum DOD. Ideally it would be the 'Low Batt' setting (set by my installer to 20) to avoid the battery actually shutting down. Unfortunately this is not mapped to a sensor in Solarsynk. Can I set a hard value in config.yaml?

fboundy, 2024-03-29:
Yes hard value will work

ElectricMolehill, 2024-03-29:
Just discovered another eccentricity of the Sunsynk cloud api - in order to enable Grid charging in a timeslot you also have to specify Gen charge ({"time2on": true, "genTime2on": false} instead of {"time2on": true}. If you only specify one, the api responds 'Success' but doesn't do anything. I'll update my document but trying to avoid too many versions in a day!
By the way, thanks for all the time you've put into this so far. It must be more or less impossible to test without a Sunsynk inverter.  I was delighted when my dashboard graphs sprang into life!

fboundy, 2024-03-29:
When you've got a minute can you upload a copy of the logs just to make sure there are no funnies. 

It is tricky without the same inverter. What's even more confusing is that I'm trying to get a Solax inverter going at the same time. Fortunately the way I've set it up with separate plugins by brand makes it easier to manage.

ElectricMolehill, 2024-03-29:
[error.log](https://github.com/fboundy/pv_opt/files/14808307/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14808308/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14808309/pv_opt.log)
[pv_opt.log.1.txt](https://github.com/fboundy/pv_opt/files/14808318/pv_opt.log.1.txt)
[sunsynk inverter class for pv_opt v4.odt](https://github.com/fboundy/pv_opt/files/14808319/sunsynk.inverter.class.for.pv_opt.v4.odt)

fboundy, 2024-03-30:
There's clearly an issue where data is periodically unavailable. You can see this if you search the log for `soc_now`.  At 16:20 it is 92.0 but at 16:30 it is blank. I guess this is down to the issues with the Solarsynk add-on that you've highlighted. I'll add some code to get it to re-try if it fails to get a number.

fboundy, 2024-03-30:
Try `4.0.0-sunsynk-alpha-11`

fboundy, 2024-03-30:
Just had to patch to `4.0.0-sunsynk-alpha-12` to fix a bug associated with the clock change

ElectricMolehill, 2024-03-30:
Here are the logs from alpha 12
The data coming from Solarsynk is pretty ropey at the moment I'm sure it wasn't like this to start with so perhaps fixable. Difficult to make much progress until Solarsynk is fixed though:
[image]
[error.log](https://github.com/fboundy/pv_opt/files/14812044/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14812045/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14812046/pv_opt.log)

fboundy, 2024-03-30:
That's pretty poor. For history it's OK as PV_Opt interpolates any gaps but the problem comes when reading current states. Fortunately the optimiser only really needs the current SOC. In `alpha-13` I'm setting it to use the last value from the history if it can't get the current state from HA.

Controlling the inverter needs reliable access to the current states and also the ability to write reliably. This, combined with the 5 minute interval you mentioned before could make things tricky. One option would be to control the inverter directly through the sunsynk API rather than solarsynk. What do you think?

fboundy, 2024-03-30:
Hold off on alpha-13 it has a bug

alz41, 2024-03-30:
Mark

I have been looking at this
https://smarthomeintegrations.co.za/product/smartdeyedonglev4-intl-shipment/

It gives direct access to the inverter for read/write

ElectricMolehill, 2024-03-30:
I agree Solarsynk is not looking too promising at the moment. I imagine the connectivity problem will get fixed but the 5 min delay and lack of response to commands are serious weaknesses which can't easily be fixed because the whole add-on runs as a 5 minute batch process. I'm a bit reluctant to connect directly to modbus though. The only easily accessible port on the inverter is already occupied by an e-Linter data logger which is used by Sunsynk and my installer for support as well as feeding the Sunsynk app. I'm not sure it's a good idea to have 2 parallel sources of commands connected at the same time. Also I think using the Sunsynk API might be somewhat safer (though I can believe it would be possible to ruin an inverter by sending a wrong API call), 
@alz41 I'm wondering whether the MorneSaunders360 integration would be a better bet. Sounds like you might have tried it so what do you think?

ElectricMolehill, 2024-03-30:
@fboundy just re-read your post and realised you're talking about directly connecting to the Sunsynk API, not a wired modbus connection. I think that could work but I don't have any documentation of it apart from unofficial lists of json keys. I guess you could copy one of the integrations already published in Github

alz41, 2024-03-30:
@ElectricMolehill I'm having the same sort of problems with that integration. Regarding modbus connection iv been running solar assistant connected via splitter on the battery input for six months with no problems.

fboundy, 2024-03-30:
All the necessary API calls appear to be in the solarsync add-on bash script but it would be rather duplicating effort.

`alpha-13` should be working now I think

ElectricMolehill, 2024-03-31:
alpha-13 - there's a type error:
TypeError: object of type 'bool' has no len()
[error.log](https://github.com/fboundy/pv_opt/files/14815325/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14815326/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14815327/pv_opt.log)
Connecting direct to Sunsynk API - yes let's give it a go. It's a shame to duplicate effort but the Solarsynk add-on doesn't have the features we need and it's not really feasible to add them. We could try the MorneSaunders integration but it sounds like that also has a connectivity issue and it would be quite a bit of work before we know whether it's suitable. It's mostly your time so I'm happy to go with whichever you think has the best prospects

fboundy, 2024-03-31:
FIxed in `4.0.0-sunsynk-alpha-14`

Once let's see if we can consistently get charge plans in read-only and then we'll make a call on which route to go down for control

ElectricMolehill, 2024-04-01:
That's looking good so far. Perhaps I'd better leave it running for a few days to give it a proper run with better data. I missed a day's inverter history in HA yesterday because the Sunsynk dongle lost contact with my local network when the clocks changed and I didn't discover a fix until yesterday evening
Good news is, Solarsynk has been patched to handle connectivity failures more gracefully. Also it sounds like the connectivity issues may have been something to do with verification of SSL certificates since Sunsynk.net changed certificates. The cause of this is still not known so unfortunately the workaround is to disable verification. Still, that might close up the gaps in the data
[error.log](https://github.com/fboundy/pv_opt/files/14822983/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14822984/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14822985/pv_opt.log)

fboundy, 2024-04-01:
> @fboundy should the pvopt entities be created? With current release

Should be. If they aren’t it suggests an error. What does error.log say?

alz41, 2024-04-01:
> Should be. If they aren’t it suggests an error. What does error.log say?
> […](#)

[main.log](https://github.com/fboundy/pv_opt/files/14826806/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14826807/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14826808/pv_opt.log)

alz41, 2024-04-02:
logs for alpha15

entities still not created for dashboard

[error.log](https://github.com/fboundy/pv_opt/files/14844369/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14844370/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14844371/pv_opt.log)

fboundy, 2024-04-03:
The last error.log was empty. Could you please re-upload?

alz41, 2024-04-03:
It was empty at the time of posting

fboundy, 2024-04-03:
Ok - maybe that it had just rolled over into a new file. Is there anything in error.log.1?

ElectricMolehill, 2024-04-03:
@alz41 is Mosquitto running OK?

[25 more replies]

closed 2024-07-09 by github-actions[bot]