stevebuk1/pv_opt #131: sensor.pv_opt* failing to start

pepetheprawn, 2024-02-20:
[appdaemon_error.log](https://github.com/fboundy/pv_opt/files/14348201/appdaemon_error.log)
[appdaemon_pv_opt.log](https://github.com/fboundy/pv_opt/files/14348202/appdaemon_pv_opt.log)
Hi, 

I've recently done a complete new build on a HA green appliance and everything appears to be working as expected. PV_OPT installation has all gone ok, and the dashboard is looking good. The only issue I have is that for some reason the sensor.pv_opt* sensors are not coming into service and I can't figure out why:

[image]

Any help to point me in the right direction would be massively appreciated as it feels like I'm 99% there now! 

Thanks,

fboundy, 2024-02-20:
1st thing I suggest is that you upgrade to the latest version of PV-Opt (3.8.10). This should be straightforward enough via HACS. This won't necessarily fix the problem but it will at least put you on the same revision as me!

If things still aren't there then please update this issue with the latest logs and I'll have another look

image.png (view on web)<https://github.com/fboundy/pv_opt/assets/104442544/3c4f58d3-68ec-4da3-965f-b780b9142c4e>

pepetheprawn, 2024-02-20:
Thanks for the suggestion, I've just upgraded pv_opt, and here are the logs now
[AppDaemon_error_log.txt](https://github.com/fboundy/pv_opt/files/14349163/AppDaemon_error_log.txt)
[AppDaemon_pv_opt_log.txt](https://github.com/fboundy/pv_opt/files/14349164/AppDaemon_pv_opt_log.txt)

I've now got this appliance disconnected from the inverter so the solis sensors are down which you'll see in the logs, but the situation is the same regarding all the pv_opt* sensors regardless of whether the inverter is connected or not.

fboundy, 2024-02-20:
It won't run with so many required sensors unavailable - but it doesn't behave very nicely when it fails. I've added some additional error traps to 3.8.11 as a pre-release which you should be able to get through HACS if you allow Beta releases. 

Can you re-send the logs when the various sensors are available?

pepetheprawn, 2024-02-21:
[AppDaemon_error_log.txt](https://github.com/fboundy/pv_opt/files/14356443/AppDaemon_error_log.txt)
[AppDaemon_pv_opt_log.txt](https://github.com/fboundy/pv_opt/files/14356444/AppDaemon_pv_opt_log.txt)
Hi Francis, 

I've done the upgrade to 3.8.11 this morning and have attached to the inverter again. The Solis sensors are now up, but I'm still seeing the pv_opt sensors as unavailable. Interestingly pv_opt is also saying that the Solcast sensors are unavailable, although they are actually ok:

[image]

[image]

Thanks again for all your help with this. 

-Gordon

fboundy, 2024-02-21:
Ok - I’ll have at your logs and get back to you.

fboundy, 2024-02-21:
The issues you have highlighted aren't the problem. It is because the button that is used to send times to the inverter in the Solax integration has never been pressed so it is returning an "unknown" state. I've update 3.8.12_pre to allow for this. Let me know how you get on.

pepetheprawn, 2024-02-22:
[AppDaemon_error_log.txt](https://github.com/fboundy/pv_opt/files/14371298/AppDaemon_error_log.txt)
[AppDaemon_pv_opt_log.txt](https://github.com/fboundy/pv_opt/files/14371299/AppDaemon_pv_opt_log.txt)
Morning Francis, 

Please find new log files attached following the upgrade to 3.8.12_pre

Let me know if you need any more information or anything. 

-Gordon

fboundy, 2024-02-22:
Hmm - weird. Same error as before when I thought I’d fixed that. Will have another look

pepetheprawn, 2024-02-22:
Brilliant, thanks for your help with this.

fboundy, 2024-02-22:
OK - found what I'd done wrong. 3.8.14 should be better. Just waiting on a response to a different issue before I release it.

pepetheprawn, 2024-02-23:
Brilliant, thanks for investigating! Let me know when the new version is available and I'll give it a test :-)

davidcornwell1969, 2024-02-23:
Just to add I was getting exactly the same error as Gordon. More importantly though I'd like to say a massive thank you to the author for all his hard work and skill.

pepetheprawn, 2024-02-23:
@davidcornwell1969 can I ask what the resolution was to your issue, if it's something I can fix myself and save hassling the developers that would be amazing!

davidcornwell1969, 2024-02-23:
Hi Gordon, sorry for misleading message, when I say I was getting the same error, I should say I AM getting the same error. Was loath to raise a bug myself in case me not reading the setup instructions correctly. I am a developer but very new to HA, so thought it likely to be me missing a step.

fboundy, 2024-02-23:
Please try 3.8.14 and let me know how you get on

davidcornwell1969, 2024-02-23:
Hi I just got the latest version 
PV Opt v3.8.14 as appears in the log file,

Really sorry to report I don't see any change,

 
I have attached log file. I really hope I'm not wasting your time and the bug isn't caused by me not following instruction correctly. BTW all Solax Modbus entities appear to be functioning fine.

Kind Regards
 
Dave

>  
>  
> 
>  
> 
> Please try 3.8.14 and let me know how you get on

2024-02-23 14:53:41.413305 WARNING AppDaemon: Unable to find app solis in dependencies for pv_opt
2024-02-23 14:53:41.413436 WARNING AppDaemon: Ignoring app pv_opt
2024-02-23 14:53:41.413708 INFO AppDaemon: Terminating pv_opt
2024-02-23 14:53:41.415802 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/pvpy.py
2024-02-23 14:53:41.453216 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py
2024-02-23 14:53:41.505038 WARNING AppDaemon: Unable to find app solis in dependencies for pv_opt
2024-02-23 14:53:41.505614 WARNING AppDaemon: Ignoring app pv_opt
2024-02-23 14:53:41.506758 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
2024-02-23 14:53:41.511641 ERROR pv_opt: User defined log pv_opt_log not found
2024-02-23 14:53:41.513481 INFO AppDaemon: Calling initialize() for pv_opt
2024-02-23 14:53:41.516876 INFO pv_opt: 
2024-02-23 14:53:41.518240 INFO pv_opt: ******************* PV Opt v3.8.14 *******************
2024-02-23 14:53:41.520765 INFO pv_opt: 
2024-02-23 14:53:41.522698 ERROR pv_opt: User defined log pv_opt_log not found
2024-02-23 14:53:41.523876 WARNING AppDaemon: Unknown Plugin Configuration in get_plugin_api()
2024-02-23 14:53:41.527421 INFO pv_opt: Local timezone set to GB
2024-02-23 14:53:41.530292 INFO pv_opt: Time Zone Offset: 0.0 minutes
2024-02-23 14:53:41.532594 INFO pv_opt: Inverter type: SOLIS_SOLAX_MODBUS: inverter module: solis.py
2024-02-23 14:53:41.536069 WARNING pv_opt: pv_opt: Entity sensor.pvopt_status not found in namespace default
2024-02-23 14:53:41.538037 INFO AppDaemon: pv_opt: Entity sensor.pvopt_status created in namespace: default
2024-02-23 14:53:41.578717 INFO pv_opt: Reading arguments from YAML:
2024-02-23 14:53:41.580553 INFO pv_opt: -----------------------------------
2024-02-23 14:53:41.582179 INFO pv_opt: 
2024-02-23 14:53:41.584091 INFO pv_opt:   Over-write flag is set so YAML will over-write HA
2024-02-23 14:53:41.587792 INFO pv_opt:     consumption_history_days           = 7                                                              7: value in YAML
2024-02-23 14:53:41.597176 INFO pv_opt: 
2024-02-23 14:53:41.600291 INFO pv_opt: Checking config:
2024-02-23 14:53:41.604016 INFO pv_opt: -----------------------
2024-02-23 14:53:41.607310 WARNING pv_opt:     forced_charge                      = True                                                        True: system default. Not in YAML.
2024-02-23 14:53:41.613794 WARNING pv_opt:     forced_discharge                   = True                                                        True: system default. Not in YAML.
2024-02-23 14:53:41.617125 WARNING pv_opt:     read_only                          = True                                                        True: system default. Not in YAML.
2024-02-23 14:53:41.620549 WARNING pv_opt:     allow_cyclic                       = False                                                      False: system default. Not in YAML.
2024-02-23 14:53:41.625235 WARNING pv_opt:     optimise_frequency_minutes         = 10                                                            10: system default. Not in YAML.
2024-02-23 14:53:41.632609 WARNING pv_opt:     slot_threshold_p                   = 1.0                                                          1.0: system default. Not in YAML.
2024-02-23 14:53:41.637126 WARNING pv_opt:     day_of_week_weighting              = 0.5                                                          0.5: system default. Not in YAML.
2024-02-23 14:53:41.642767 WARNING pv_opt:     pass_threshold_p                   = 4.0                                                          4.0: system default. Not in YAML.
2024-02-23 14:53:41.650961 WARNING pv_opt:     octopus_auto                       = True                                                        True: system default. Not in YAML.
2024-02-23 14:53:41.654627 WARNING pv_opt:     battery_capacity_wh                = 10000                                                      10000: system default. Not in YAML.
2024-02-23 14:53:41.657996 WARNING pv_opt:     inverter_efficiency_percent        = 97                                                            97: system default. Not in YAML.
2024-02-23 14:53:41.661421 WARNING pv_opt:     charger_efficiency_percent         = 91                                                            91: system default. Not in YAML.
2024-02-23 14:53:41.665223 WARNING pv_opt:     charger_power_watts                = 3000                                                        3000: system default. Not in YAML.
2024-02-23 14:53:41.668797 WARNING pv_opt:     inverter_power_watts               = 3600                                                        3600: system default. Not in YAML.
2024-02-23 14:53:41.672183 WARNING pv_opt:     inverter_loss_watts                = 100                                                          100: system default. Not in YAML.
[... cut ...]
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1121, in _load_args
    self._expose_configs(over_write)
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1182, in _expose_configs
    self.mqtt.mqtt_publish(conf_topic, dumps(conf), retain=True)
    ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'mqtt_publish'

2024-02-23 14:53:41.884794 WARNING pv_opt: ------------------------------------------------------------
2024-02-23 14:53:45.129136 INFO AppDaemon: New client Admin Client connected

fboundy, 2024-02-23:
It looks like this is a problem with the MQTT setup during AppDaemon configuration.
```
2024-02-23 14:53:41.522698 ERROR pv_opt: User defined log pv_opt_log not found
2024-02-23 14:53:41.523876 WARNING AppDaemon: Unknown Plugin Configuration in get_plugin_api()

self.mqtt.mqtt_publish(conf_topic, dumps(conf), retain=True)
```

Can I suggest you check that your `appdaemon.yaml` file looks like the one specified in section 11 of the README?

davidcornwell1969, 2024-02-23:
Thanks for the above, I was having difficulting getting the log file not to issue an error even though I thought its should have created in the SAMBA share error. Will go and try to get this working.
I did dump some more debug out for the parameters passed to the self.mqtt.mqtt_publish call (hope you don't mind me changing your code? on my local copy)
This is what gets passed into to the method call,
2024-02-23 18:17:59.284712 INFO pv_opt:   - DC about to make call mqtt_publish conf-> homeassistant/switch/pvopt_forced_charge/config dumps(conf) value is {"state_topic": "homeassistant/switch/pvopt_forced_charge/state", "command_topic": "homeassistant/switch/pvopt_forced_charge/set", "name": "Pvopt Forced Charge", "optimistic": true, "object_id": "pvopt_forced_charge", "unique_id": "pvopt_forced_charge", "payload_on": "ON", "payload_off": "OFF", "state_on": "ON", "state_off": "OFF"} 

But will take your suggestion and try and gett appdaemon matching yours exactly.
Thanks again
David

fboundy, 2024-02-23:
The first warning is when the `initialize` method tries to load the MQTT plugin with `self.mqtt = self.get_plugin_api("MQTT")` so that suggests to me that there is something wrong with this section of `appdaemon.yaml`:
```
  appdaemon:
      MQTT:
        type: mqtt
        namespace: mqtt #
        verbose: True
        client_host: core-mosquitto
        client_port: 1883
        client_id: localad
        event_name: MQTT_MESSAGE 
        client_topics: NONE
        client_user: !secret mqtt-user
        client_password: !secret mqtt-password
```

You should then get something like this when AppDaemon starts up:

```
    13:16:24 INFO AppDaemon: AppDaemon Version 4.4.2 starting
    13:16:24 INFO AppDaemon: Python version is 3.11.6
    13:16:24 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
    13:16:24 INFO AppDaemon: Added log: AppDaemon
    13:16:24 INFO AppDaemon: Added log: Error
    13:16:24 INFO AppDaemon: Added log: Access
    13:16:24 INFO AppDaemon: Added log: Diag
    13:16:24 INFO AppDaemon: Added log: PV_Opt
    13:16:25 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
    13:16:25 INFO HASS: HASS Plugin Initializing
    13:16:25 WARNING HASS: ha_url not found in HASS configuration - module not initialized
    13:16:25 INFO HASS: HASS Plugin initialization complete
    13:16:25 INFO AppDaemon: Loading Plugin MQTT using class MqttPlugin from module mqttplugin
    13:16:26 INFO MQTT: MQTT Plugin Initializing
    13:16:26 INFO MQTT: Using 'localad/status' as Will Topic
    13:16:26 INFO MQTT: Using 'localad/status' as Birth Topic
```

davidcornwell1969, 2024-02-23:
Thank you so much Francis, I think I had tabs rather than spaces to indent appdaemon.yaml. @pepetheprawn Wonder if you maybe have the same issue? Happy to help in any way I can?

Thanks again Francis for all your hard work.

pepetheprawn, 2024-02-23:
Great to hear that you've got yours working @davidcornwell1969 . That appdaemon.yaml I literally copied/pasted from the readme, and I used the file editor within HA that gives you the green tick if the formatting is correct: 
[image]
and the in secrets.yaml I have: 
[image]

I don't see anything obviously wrong, but if you guys spot something or have any suggestions I'd be happy to receive them! 

Thanks again for all your hard work. 

-Gordon

pepetheprawn, 2024-02-23:
I've just sanity checked my mqtt setup with this handy tool (https://mqtt-explorer.com/) and the mqtt side of things seems to be behaving as I'd expect:

[image]
 so it does seem to be related to the integration between mqtt and pv_opt, but I can't see what's wrong! 

-Gordon

fboundy, 2024-02-23:
@pepetheprawn - I've not seen any evidence your issues are MQTT related. When you get a chance send me the latest logs that you have and I'll have another look

@davidcornwell1969 - if the MQTT is fixed now you should be getting a bit further so please also send me log files when you get a chance

davidcornwell1969, 2024-02-24:
Francis, seems to get a lot further now, loading tarrifs. I get an error further into the code now hopefully not my bad config.
[pv_opt_error.txt.txt](https://github.com/fboundy/pv_opt/files/14392070/pv_opt_error.txt.txt)
Thx again

fboundy, 2024-02-24:
Do you have an export tariff? It looks like you don’t and I need to allow for this. Should be a quick fix.

davidcornwell1969, 2024-02-24:
No export tariff (currently), harder to get for a self install !
:)

fboundy, 2024-02-24:
Ok - should get an update to correct for this later today

davidcornwell1969, 2024-02-24:
Whenever it comes, greatly appreciated. :)

fboundy, 2024-02-24:
Try 3.9.2 - works ok with mine when i explicity say no export tariff. Hopefully should play noce with Octopus Auto but let me know.

davidcornwell1969, 2024-02-24:
Really sorry, I think the error is the same as before. Could you guide me as to where in config I can explicitly say no export tariff. I could try that for now and then comment out when I retest a fix?
[pv_opt_error.txt.txt](https://github.com/fboundy/pv_opt/files/14392462/pv_opt_error.txt.txt)
Thanks
Dave :)

fboundy, 2024-02-24:
Hi Dave

Can you send the "pv_opt.log" file as well please so I can see what it has done successfully?

You shouldn't need to specify no export with this version.

Thanks,
Francis

davidcornwell1969, 2024-02-24:
Hey Francis,
was pulling my hair out last week, trying to get to the log files if they are indeed being generated. Did a fair bit of googling and thought they should get created in my Samba share/logs folder but nothing in their. Will google some more and send once I have them. I shouldn't have to create the file manually right?
Thanks

Dave

fboundy, 2024-02-24:
No - if you've set up the share correctly and put it in the `appdaemon.yaml` file it should be there. You do have to restart AppDaemon addon IIRC. If you have the File editor Add On you can check the samba shared folder is there so long as "Enforce Basepath" is turned OFF in the Configuration for it:

[image]

[image]

fboundy, 2024-02-24:
> Exciting news from the Plymouth office (and it's not about an unexploded WW2 bomb)! I do believe that it's now working! I just upgraded to 3.9.2 and the dashboard all looks great:
> image.png (view on web)
> AppDaemon_error_log.txt
> AppDaemon_pv_opt_log.txt
> Latest log files attached for your review just to check that everything looks good from your perspective as well.
> Thanks so much for all your hard work @fboundy, This really is an amazing bit of software.
> -Gordon

If I’d known I was dealing with a Janner… Exeter boy (originally) here…

I’ll have a look at the logs etc tomorrow. Glad it seems to be working.

fboundy, 2024-02-25:
It looks like you don't have any load history yet so you may be best switching to using a fixed estimate of consumption till you have some history data. Otherwise it looks fine to me.

fboundy, 2024-02-25:
It looks like you don't have any load history yet so you may be best switching to using a fixed estimate of consumption till you have some history data. Otherwise it looks fine to me. I'll close this issue now.

closed 2024-02-25 by fboundy

pepetheprawn, 2024-02-25:
> If I’d known I was dealing with a Janner… Exeter boy (originally) here… I’ll have a look at the logs etc tomorrow. Glad it seems to be working.
> […](#)

I wasn't born here so I'm not a true janner! Thanks for taking a look at the logs and confirming all ok. This is running on a HA green appliance that I'm going to ship off to a friend, so I'll get them to run it 'read only' for a week or so to collect load history, and then they can kick it into production. 

Thanks again for all your hard work with this, it's massively appreciated. 

-Gordon