stevebuk1/pv_opt #132: Unexpected error in worker for App pv_opt - charge slots not optimal

solarjones, 2024-02-22:
PV Opt v3.8.12 

Think there's still a bug lurking somewhere as the optimised charge plan wasn't optimal:
[image]

I reloaded OE integration, tried to nudge PV_OPT by toggling between readonly and normal modes but the plan remained the same. Next I restarted AppDaemon which aligned to cheapest slots.
[image]

I can see some errors, but that was from last night. Perhaps it caused the above issue?

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14377086/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14377087/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14377088/error.log)

Thanks.

fboundy, 2024-02-22:
Do you have the pv_opt.log from 22:30 when the error occurred? 

AppDaemon failed to return recent SOC data which is what caused the error in error.log and I suspect this may have stopped it loading correct tariffs after that.

solarjones, 2024-02-22:
[1pv_opt.log](https://github.com/fboundy/pv_opt/files/14378510/1pv_opt.log)

solarjones, 2024-02-23:
v3.8.14 downloaded at about 15:30, restarted AP.

[image]

Doesn't look like it picked up the OE slots correctly. No errors.

Restarted AD.

[image]

Looks ok. However, (and this isn't specific to this version as I noticed this morning) I don't think the PV forecast has been taken into consideration, 26 kWh tomorrow! :)  My projected consumption is lowish.

Also toggled between consumption history and fixed. It changed the charge plan by always resulted with 100% SOC around 6am.

[image]

[image]

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14387751/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14387752/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14387753/error.log)

Thanks

fboundy, 2024-02-23:
In 3.9.0 I've added a feature to compare the pv_opt tariffs with those from the Octopus Energy automation (so long as you have `octopus_auto` set to `true`). At the moment all it does is flag if they are different and also populate a sensor called `sensor.pvopt_tariff_import_OK` with `True` or `False'. This should at least help identify when there is a problem and, hopefully, eventually track down the source.

solarjones, 2024-02-23:
> In 3.9.0 I've added a feature to compare the pv_opt tariffs with those from the Octopus Energy automation (so long as you have `octopus_auto` set to `true`). At the moment all it does is flag if [...]

Ok, I'll try 3.9.0.  Is this correct?
[image]

What do you think about the PV forecast & SOC with the charge plan?

Thanks

fboundy, 2024-02-23:
> > In 3.9.0 I've added a feature to compare the pv_opt tariffs with those from the Octopus Energy automation (so long as you have `octopus_auto` set to `true`). At the moment all it does is flag if they are different and also populate a sensor called `sensor.pvopt_tariff_import_OK` with `True` or `False'. This should at least help identify when there is a problem and, hopefully, eventually track down the source.
> 
> Ok, I'll try 3.9.0. Is this correct? <img alt="image" width="414" src="https://private-user-images.githubusercontent.com/100320468/307391904-6ec2c22b-ff21-481a-9877-0e07727491cc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3MDc2ODAsIm5iZiI6MTcwODcwNzM4MCwicGF0aCI6Ii8xMDAzMjA0NjgvMzA3MzkxOTA0LTZlYzJjMjJiLWZmMjEtNDgxYS05ODc3LTBlMDc3Mjc0OTFjYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyM1QxNjU2MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MjY5ZGEzOWI0MTcyNzEyMzAyOTMzYmU4YjM0ODRjNjA2YjczYWIwM2IxM2Q3MjA5ZjVmNWQ2ZjVmNmZhNDZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7T35uzw-_V2rs6dUFUuFnamrY3Wyd28iRNsGoDn4K4c">
> 
> What do you think about the PV forecast & SOC with the charge plan?
> 
> Thanks

You should be able to comment it all out as `octopus_auto: true` is the default

fboundy, 2024-02-23:
> What do you think about the PV forecast & SOC with the charge plan?
> Thanks

I think that's correct. Mine is similar. There are some marginal benefits from charging at <15p and exporting at 15p. My cost saving for tomorrow is only 20p though. If you increase the two thresholds (per slot and per pass) it will probably reduce the amount of forced charging.

fboundy, 2024-02-23:
> v3.8.14 downloaded at about 15:30, restarted AP.
> 
> Doesn't look like it picked up the OE slots correctly. No errors.
> 

One thing I don't understand in your comparison is that Octopus don't release the next day's tariffs until ~16.00 so when I look at the log file for 15.30 PV_Opt doesn't have them - it is using a guesstimate based on the published wholesale day-ahead prices which is only hourly so it won't be 100% correct. Once the Octopus prices are available (after 16.30) it all looks OK.

fboundy, 2024-02-23:
> I'd been using the Solcast default (p50?), so have now switched to P10.
> Thanks.

This will be more conservative in that it's likely you will charge your battery more than you needed to but you are very unlikely to charge too little.

fboundy, 2024-02-23:
3.9.1 tidies up a few things related to these marginal charging slots and makes it less likely to charge unnecessarily,

solarjones, 2024-02-25:
3.9.2
I've got some more errors and just checked the charging plan, which doesn't look like its updated or got he OE slots correct.

[image]

Restarting AppD, and the charging plan looks correct now. 
[image]

Logs:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14397236/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14397240/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14397242/error.log)
[1pv_opt.log](https://github.com/fboundy/pv_opt/files/14397244/1pv_opt.log)

Just noticed you have version 3.9.3, I'll update to this.  Does this have any changes in this area?

Thanks.

fboundy, 2024-02-25:
Yes 3.9.3 should fix this

solarjones, 2024-02-25:
v3.9.3
Unexpected error in worker for App pv_opt:

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14397906/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14397907/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14397908/error.log)

fboundy, 2024-02-26:
Added a check on this in 3.9.4

fboundy, 2024-02-28:
Think I may have finally found the root cause! Please check out 3.9.5

solarjones, 2024-03-02:
Happy to report a lovely stable version for me, zero errors and good time slots!  Truly very grateful for you being so responsive and what a brilliant optimiser!

closed 2024-03-16 by fboundy