stevebuk1/pv_opt #139: Backup mode in final IOG slot not being cleared

stevebuk1, 2024-02-25:
If Pv_opt has scheduled an "SOC hold" in the final slot allocated in an Intelligent Octopus Go cheap rate, the inverter is left in Backup/Reserve (Mode 49) and is never clears back to Timed Charge/Discharge mode (Mode 35) or other mode that allows the inverter to resume using the house batteries to supply the house load. 

The issue appears to happen when the last slot in the IOG cheap rate is allocated to an 'SOC hold' but there are no more charging slots allocated after that. I would expect this never/rarely happens on the Agile Tariff, but is almost certain for IOG as the next cheap period is too far away for the Optimiser to consider).

This leads to the time to next slot start always being negative as per attached log, so the else statement on line 1572 (Github line number) in pv_opt.py never gets triggered and backup mode is never disabled (on line 1677). If this was an actual charging slot then the inverter would automatically finish the slot and revert back to selfuse -  but as its a holding slot intervention by Pv_opt is needed. 

Attached log covers the cheap IOG period (23:30 to 05:30) as well as including the addtional 2 hours afterwards where the inverter was held in Backup mode. 

[pv_opt1.log](https://github.com/fboundy/pv_opt/files/14396446/pv_opt1.log)

fboundy, 2024-02-26:
I've updated the logic in 3.9.4 that should ensure that the inverter reverts to Time Of Use mode in this situiation.

stevebuk1, 2024-02-26:
Thanks, I've pulled in 3.9.4 and will run overnight soon (not tonight, as have an EV to charge). I pulled in 3.9.3 yesterday which I think ended last nights charging with a backup slot and it functioned ok, but will double check with 3.9.4.

closed 2024-02-26 by stevebuk1

fboundy, 2024-02-26:
Yes -I think 3.9.3 might have been ok but I made it a bit more explicit in 3.9.4