stevebuk1/pv_opt #144: ERROR: tariff inconsistency

mergwyn, 2024-03-02:
**Describe the bug**
When looking at the pv_opt dashboard, I occasionally see the above message flash up in the status line

[image]

This seems to be related to the code here: https://github.com/fboundy/pv_opt/blob/d59befed04a97062b8320d118de315887162046e/apps/pv_opt/pv_opt.py#L2298

There are two log message produced:
```
15:10:28 INFO: Export: Average Prices - PV_OPT: 18.65 p/kWh OE Integration: 15.00 p/kWh Mean difference: 3.65 p/kWh <<< ERROR
15:10:28 INFO: Import: Average Prices - PV_OPT: 20.10 p/kWh OE Integration: 16.29 p/kWh Mean difference: 3.80 p/kWh <<< ERROR
```
Is this an error or a warning, and what if anything should I do about it?

**Logs**

**Context**
pv_opt: v 3.9.5
HA: 2024.2.4
Appdaemon: 4.4.2

fboundy, 2024-03-02:
It *should* be a warning that there is an inconsistency between the Octopus Energy tariffs and the PV Opt tariffs so worth checking if that is the case.

I think it’s because of the upcoming Saving Session where I bump up the rates to the equivalent value of the Octopoints. When I gen a chance I will update it to filter out Savings Sessions from the comparison so please leave this Issue open

fboundy, 2024-03-16:
This should be fixed now but we need another saving session to test fully

closed 2024-03-18 by fboundy