stevebuk1/pv_opt #146: Defaults not preserved

lazy-pete, 2024-03-09:
**Describe the bug**
A clear and concise description of what the bug is.

I fairly regularly find that my PV-opt setup ‘loses’ its default settings. I have edited the pv_opt.py to have read only as false etc so a HA / Appdaemon restart doesn’t turn things off, however this doesn’t seem to work. I’m trying to get the logs, however I must’ve set that up wrong as there nothing in my Share/logs folder. 

Meantime a screenshot to illustrate what I found this morning (same as often happens) in case I’ve just edited the file incorrectly. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mergwyn, 2024-03-09:
I also set `readonly` to `false` but rather than modify `pv_opt.py` I change the config file `config.yaml` eg.
```
pv_opt:
  module: pv_opt
  class: PVOpt
  dependencies:
    - pvpy
    - solis

  log: pv_opt_log
  prefix: pvopt
  #debug: True

# User configuration ---  EDIT AWAY! ---

  inverter_type: "SOLIS_SOLAX_MODBUS"
  # inverter_type: "SOLIS_CORE_MODBUS"
  # inverter_type: SOLIS_SOLARMAN
  device_name: solax

  # If true the current config in HA will be over-written with that in the config.yaml.
  overwrite_ha_on_restart: true
  redact_personal_data_from_log: true

  forced_charge: true
  forced_discharge: true
  allow_cyclic: false
  read_only: false
```
This works well for me - it might be worth trying this method.

fboundy, 2024-03-09:
The setting to change is overwite_ha_on_restart. Set this to false and settings in config.yaml will be ignored in favour of whatever you have set in HA

lazy-pete, 2024-03-09:
Thanks both - I’ll try one / other / both of those suggestions.

Cheers,

Pete

lazy-pete, 2024-03-12:
Hi - I tried both those options, but still not holding my custom defaults. 

2024-03-12 20:50:00.306604 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
2024-03-12 20:50:00.307206 ERROR pv_opt: User defined log pv_opt_log not found
2024-03-12 20:50:00.307738 INFO AppDaemon: Calling initialize() for pv_opt
2024-03-12 20:50:00.308301 INFO pv_opt: 
2024-03-12 20:50:00.308742 INFO pv_opt: ******************* PV Opt v3.9.5 *******************
2024-03-12 20:50:00.309369 INFO pv_opt: 
2024-03-12 20:50:00.309564 ERROR pv_opt: User defined log pv_opt_log not found
2024-03-12 20:50:00.309947 ERROR pv_opt: User defined log pv_opt_log not found
2024-03-12 20:50:00.310527 INFO pv_opt: Local timezone set to GB
2024-03-12 20:50:00.311220 INFO pv_opt: Time Zone Offset: 0.0 minutes
2024-03-12 20:50:00.311877 INFO pv_opt: Inverter type: SOLIS_SOLAX_MODBUS: inverter module: solis.py
2024-03-12 20:50:00.312401 WARNING pv_opt: pv_opt: Entity sensor.pvopt_status not found in namespace default
2024-03-12 20:50:00.312783 INFO AppDaemon: pv_opt: Entity sensor.pvopt_status created in namespace: default
2024-03-12 20:50:00.363916 INFO pv_opt: Reading arguments from YAML:
2024-03-12 20:50:00.364584 INFO pv_opt: -----------------------------------
2024-03-12 20:50:00.365577 INFO pv_opt:     consumption_history_days           = 7                                                              7: value in YAML
2024-03-12 20:50:00.366792 INFO pv_opt:     forced_charge                      = True                                                        True: value in YAML
2024-03-12 20:50:00.367785 INFO pv_opt:     forced_discharge                   = False                                                      False: value in YAML
2024-03-12 20:50:00.368730 INFO pv_opt:     allow_cyclic                       = False                                                      False: value in YAML
2024-03-12 20:50:00.369684 INFO pv_opt:     read_only                          = False                                                      False: value in YAML
2024-03-12 20:50:00.370598 INFO pv_opt:     battery_capacity_wh                = 10000                                                      10000: value in YAML
2024-03-12 20:50:00.371575 INFO pv_opt:     inverter_efficiency_percent        = 97                                                            97: value in YAML
2024-03-12 20:50:00.372476 INFO pv_opt:     charger_efficiency_percent         = 91                                                            91: value in YAML
2024-03-12 20:50:00.373353 INFO pv_opt:     charger_power_watts                = 3500                                                        3500: value in YAML
2024-03-12 20:50:00.374248 INFO pv_opt:     inverter_power_watts               = 3600                                                        3600: value in YAML
2024-03-12 20:50:00.375121 INFO pv_opt:     inverter_loss_watts                = 100                                                          100: value in YAML
2024-03-12 20:50:00.376665 INFO pv_opt:     id_solcast_today                   = sensor.solcast_pv_forecast_forecast_today                 4.0415: value(s) in YAML
2024-03-12 20:50:00.377988 INFO pv_opt:     id_solcast_tomorrow                = sensor.solcast_pv_forecast_forecast_tomorrow              7.68225: value(s) in YAML
2024-03-12 20:50:00.378886 INFO pv_opt:     octopus_auto                       = False                                                      False: value in YAML
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:978: FutureWarning: errors='ignore' is deprecated and will raise in a future version. Use to_datetime without passing `errors` and catch exceptions explicitly instead
  pd.to_datetime(values[0], errors="ignore", format="%H:%M")
2024-03-12 20:50:00.380455 INFO pv_opt:     octopus_account                    = A63A07001                                                 A63A07001: YAML default value. No default defined.
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:978: FutureWarning: errors='ignore' is deprecated and will raise in a future version. Use to_datetime without passing `errors` and catch exceptions explicitly instead
  pd.to_datetime(values[0], errors="ignore", format="%H:%M")
2024-03-12 20:50:00.382454 INFO pv_opt:     octopus_api_key                    = ********************************                          ********************************: YAML default value. No default defined.
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:978: FutureWarning: errors='ignore' is deprecated and will raise in a future version. Use to_datetime without passing `errors` and catch exceptions explicitly instead
  pd.to_datetime(values[0], errors="ignore", format="%H:%M")
2024-03-12 20:50:00.384317 INFO pv_opt:     octopus_import_tariff_code         = E-1R-AGILE-FLEX-22-11-25-N                                E-1R-AGILE-FLEX-22-11-25-N: YAML default value. No default defined.
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:978: FutureWarning: errors='ignore' is deprecated and will raise in a future version. Use to_datetime without passing `errors` and catch exceptions explicitly instead
  pd.to_datetime(values[0], errors="ignore", format="%H:%M")
2024-03-12 20:50:00.385999 INFO pv_opt:     octopus_export_tariff_code         = E-1R-OUTGOING-FIX-12M-19-05-13-N                          E-1R-OUTGOING-FIX-12M-19-05-13-N: YAML default value. No default defined.
[... cut ...]
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 538, in get_config
    x = self.get_ha_value(self.config[item])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 853, in get_ha_value
    raise ValueError(e)
ValueError: HA returned unknown for state of sensor.solis_battery_soc

2024-03-12 20:50:01.144581 WARNING pv_opt: ------------------------------------------------------------

fboundy, 2024-03-16:
@mergwyn's `config.yaml` should be the right way to go. However, I think the problem is dues to this error which is crashing the load_args method. I'm updating the code to handle this more nicely but you might want to check it's set correctly:

    ValueError: HA returned unknown for state of sensor.solis_battery_voltage

lazy-pete, 2024-03-16:
Ah - thanks.

I suspect what’s happening is that the Solax integration is interrupted
briefly due to using both Solax and the Solis cloud through a 485 hub
simultaneously (I had to put the SolisCloud dongle back on to troubleshoot
a battery issue). I’ve detached it again and things seem more stable.

IIRC there is another method that allows two masters that doesn’t have this
issue - I just can’t remember what it is !

I just need the solis cloud intermittently for firmware updates etc - I
find their support service really good.

Cheers,

Pete

lazy-pete, 2024-04-01:
Still seeing this oddity - I’m using the yaml to set defaults, and have overwrite as TRUE, however I still see a dashboard with disparity between the yaml and HA. 

[image]
[image]

[main.log](https://github.com/fboundy/pv_opt/files/14822266/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14822267/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14822268/pv_opt.log)

fboundy, 2024-04-02:
OK - on my `to do` list

lazy-pete, 2024-04-02:
I tried editing and resaving the yaml just in case there was a glitch there somehow. Last HA restart seemed to load correctly (restarting at least daily at present as the Solax integration is still dropping out).

fboundy, 2024-04-02:
Which version of Solax are you on? It's been stable for me for the last week

lazy-pete, 2024-04-28:
Hi - sorry - I missed your reply. 

Been on various versions of the Solax Integration - currently 2024.04.4b1 

This is still dropping the Modbus connection a few times a day, although the errors seem to point to a loss of modbus connection to the inverter ip address. I don't really understand what's going on there as if it were an issue with my wifi to Elfin EW11 I wouldn't expect a HA restart to resolve it. I've created an automation that triggers a restart if the coinverter RTC doesn't update fot 15mins. Most of the time that is successful, though occasionally the Solax interface doesn't update and I have to manually reload it - think I may try increasing the delay to restarting PV-opt after a restart in case it's too quick for Solax. 

I've spent some of today putting in the various files for your own Modbus integration, however I cannot see how to actually use it - there is no sign of an innate Modbus integration within my HA instance. Nothing on ~ pip show pymodbus , which is wierd as PV-Opt works fine when the Solax connection is up. 

In terms of my original bug I think it's not an issue - I think I was not waiting long enough after a restart to allow it to setup properly, which it doesn't do until it can pull all the solax entities. Can close this ticket I think - apologies again for the delay. I'll search for other Modbus related issues and see if I can get to the bottom of that.

Cheers,

Pete

fboundy, 2024-04-28:
ModBus is part of core HA so with mine you just need the YAML file and a HA restart no need to install pymodbus or any integration

lazy-pete, 2024-04-28:
Thanks - does that then create the various entities I would need to remap
for a second dashboard (would probably disable Solax but leave the
dashboard initially).

That all seemed to be missing when I gave it a try …

Cheers,

Pete

fboundy, 2024-04-28:
Yes, should do.

I have both running in parallel - just use a different name than ‘Solis’ for one of them

lazy-pete, 2024-04-29:
Thanks - I’ll try that out. 

Meantime I seem to have lost a couple of dashboard templates, but I can’t work out why. Both the templates yaml and the main HA config yaml seem to have them in correctly (I did have to tweak a few copy / paste indent issues to get the green tick) but the dashboard doesn’t seem to be picking them up. Have I done anything obvious incorrectly ?

cheers,

pete

[image]
[image]
[image]
[image]

lazy-pete, 2024-04-29:
> Yes, should do. I have both running in parallel - just use a different name than ‘Solis’ for one of them
> […](#)

Forgive my ignorance, but do you mean you’ve installed the Appdaemon app twice and referenced different yaml files ? I thought I perhaps just uncommented both inverter models, however that creates issues.

fboundy, 2024-04-29:
The core HA integration doesn’t use AppDaemon. It is part of HA’s configuration.yaml file.

lazy-pete, 2024-04-29:
So if I just pop all the yaml files etc in the correct places / with
correct links / ip etc and create a new dashboard it will ‘just work’ ?

Still struggling to get the import / export bits available as well. The
base entity data is there - it’s just the templates that aren’t reporting.

Cheers,

Pete

fboundy, 2024-04-29:
Yes should be that simple

lazy-pete, 2024-05-01:
Hi - really sorry, but still struggling to get my head around your Modbus integration. 

The templates etc seem to work in that I have a bunch of Modbus entities that seem relevant. A bit unclear whether I also need to set up all the dashboard details etc for your own Modbus , and how that interacts with just selecting that inverter in PV-opt and it’s own dashboard thereafter. 

What I can’t get my head around is using both in parallel with PV-opt - that requires commenting the inverter used in the config.yaml , and to run those in parallel I can’t activate two ‘inverters’ with different setups. When you say you run both Solax and native Modbus ‘in parallel’ do you mean on the same HA instance, or are you running two separate installs ? 

I’ve also tried completely removing and reinstalling Solax Modbus, and now have an interesting collection of entities with _2 etc 

Sorry if I’m being really dim !

fboundy, 2024-05-01:
They will both work in parallel in HA but you can’t use both in parallel in pv_opt. You either select SOLIS_SOLAX_MODBUS or SOLIS_CORE_MODBUS as the inverter type and set the parameters accordingly.

lazy-pete, 2024-05-01:
That makes more sense - will ‘complete’ the Modbus version install and then
try both alternately with PV-opt.

Ps - I had to add a ‘state’ to the first template sensor in Modbus to get
it all working - was throwing a HA error and I only got the rest of the
templates available after doing that - maybe something changed. Will put a
ticket on the Modbus GitHub.

Cheers,

Pete

closed 2024-05-07 by fboundy