stevebuk1/pv_opt #147: Saving Sessions

stjohncanning, 2024-03-15:
Enjoying the functions of this excellent program.  Would it be possible to have an option to enter all the saving sessions but NOT activate the 'discharge' option for background day-to-day discharging?   
Reason is to avoid cycling the batteries for the (normally marginal) day to day savings, but do want to take advantage of the saving sessions which pay significantly better.   
Currently have to change the settings ('discharge' and 'cycle') in advance of the session, and reset afterwards.  A switch to enter or not enter saving sessions, which then dealt with the discharge and cycling for all future sessions, would be good.  
Obviously would have to be activated in advance of each session to set up the charging schedule, probably when the saving session was announced, and cancelled after that session.

fboundy, 2024-03-15:
Yes - that should be do-able

fboundy, 2024-03-16:
Rather than set this as a switch I have added a separate theshold for discharging: `number.pvopt_discharge_threshold_p`. You should be able to set this at an appropriate level than only saver sessions are included. It should also work for the situation where a lot of PV generated charge is worth dumping at the end of the day.

stjohncanning, 2024-03-16:
That's a much better more generic solution than my proposal - thanks!
Do you have a 'buy me a beer' link anywhere?

fboundy, 2024-03-16:
I don’t but maybe I’ll set one up…

lazy-pete, 2024-03-16:
Good idea !

closed 2024-03-18 by fboundy

stjohncanning, 2024-03-21:
donation made to Mountain Rescue team - great idea.