stevebuk1/pv_opt #148: Incorrect change slots?

solarjones, 2024-03-15:
v3.9.5
No errors reported:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14618517/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14618518/main.log)
[error.log](https://github.com/fboundy/pv_opt/files/14618519/error.log)

Restarted AppD just in case.

[image]

Also, 27kWh predicted tomorrow :)

I had updated HA and Solax to the latest versions earlier today but can't see how that could affect PV_Opt.

Core:  2024.3.1
Supervisor:  2024.03.0
Operating System: 12.1
Frontend: 20240307.0
Solax Modbus: 2024.03.1
OE: v10.1.4

Regards

fboundy, 2024-03-15:
This is an odd one. The log is saying that swapping the high cost slots for the cheapest ones isn’t actually saving any money but charging during the later ones is. I’m not convinced that hangs together but I’ll have to add some more detail to the log file to diagnose what’s going on.

solarjones, 2024-03-15:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14620789/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14620790/main.log)

[image]

It now has no charging plan! Very odd.

solarjones, 2024-03-15:
Not touched anything and it now has one charging entry...

[image]

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14620924/pv_opt.log)

fboundy, 2024-03-16:
I'm guessing that this is because you are using the 90% Solcast on a sunny day and the forecast just keeps getting better so there is less and less need for forced charging. Difficult for me to diagnose further at present. If we ever get a properly sunny day in Cumbria I may get to check it on my system...

solarjones, 2024-03-23:
We had a lovely sunny morning today and OE Agile had lots of zero cost slots and some paid too. However, the PV_Opt didn't seem to align to any of them and wanted to charge on cost slots. No matter what I did, it refused to switch to free times. Maybe I've missed something with the turning parameters?  I switched to Read Only mode today to control the inverter manually. 

Would there be benefit in having some logic for when the tariff is zero or being paid to use electricity? In these situations set the charging plan to prioritise charging when getting paid and then free? And simply charge to 100% SOC. Any excess solar exports etc.

Thanks

fboundy, 2024-03-23:
If it's sunny and solar is charging the battery at full power then there may be no benefit to charging from the grid even it it's free. There may be a small benefit from charging from negative pricing but it may have fallen below your threshold. Can you send me your pv_opt.log file so I can have a look?

solarjones, 2024-03-23:
But if the forecast changes (like today), the solar wouldn't charge the batteries fully. Also, when there's free/paid for electricity, we tend to do all our washing/drying which is much higher load, using battery when solar doesn't cover it. Forcing charge would definitely be beneficial.    

Similarly, the myenergi eddi has a schedule to force heating water when the cost drops below a threshold, which I've set to be 0p. That way it heats water for free/paid, but retains the core function of diverting excess solar to water heating as normal.

I don't have any logs from this morning as I switched to read only to control the inverter charging manually. I think today was unusual as there were many free daytime slots.

fboundy, 2024-03-23:
OK - that makes sense if your usage changes when there is a plunge. I can put in a switch to force charge when <0. 

This morning was screwed up for me because my Solax integration was playing up but tomorrow looks sensible. Charging on free electric so that the battery is at 100% when the price picks up

[image]

What does yours look like for tomorrow?

solarjones, 2024-03-23:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14732958/pv_opt.log)
[main.log](https://github.com/fboundy/pv_opt/files/14732959/main.log)

solarjones, 2024-03-23:
Ah, you hit the Solax bug too?

If you could add a < or = 0p threshold, that would be great!

fboundy, 2024-03-23:
I’m testing a variable “plunge threshold” of -5 to 10p/kwh (default of 2) below which it will just charge at max power. -5p is effectively “off”.