stevebuk1/pv_opt #149: Solax X1 AC Inverter compatibility

punkymuzzle, 2024-03-16:
Hi
This is an amazing addon for HA.  Thanks very much for the effort being put into this.
Is there any possibility of making this compatible with the Solax X1 AC inverter?  I believe that this is the gen 1 version of the current G4 inverter (although it's not Hybrid, so I have two Growatt string inverters feeding the PV charge).
I have managed to get my X1 AC working with HA, Modbus etc and have all the entities in place.

I went through the process of installing and configuring this, and tried to remap all the "solis" entities in the code to my corresponding solax entities.  However, nothing seemed to work.

If this could be made to work for my Inverter this would be great (I know there are others out there as well with the same setup as me)
I'll try and provide as much information below as I can, but please ask if you need anything else.
Thank you very much
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

PV Opt can be updated to support additional models and brands of inverter. If you want to request this, please supply as much of the following information as possible:

<h2>Home Assistant Entities to Read</h2>
PV Opt reads data from a number of entities in Home Assistant. These will vary by inverter. Please document what your inverter provides. Examples are provided from the Solis Solax Modbus integration:

    Battery minimum SOC: number.solax_battery_minimum_capacity
    Battery SOC:         sensor.solax_battery_capacity
    Either:
      Load power:        sensor.solax_house_load, or
      Load today:        Not Known
    Either:
      Grid import power: sensor.solax_grid_import and
      Grid export power: sensor.solax_grid_export
    Or:
      Grid import today: sensor.solax_today_s_import_energy and
      Grid export today: sensor.solax_today_s_export_energy

<h2>Inverter Control</h2>
PV Opt also needs to control your inverter. How this is done and what can be done varies hugely by brand, by model and by the integration you use with Home Assistant. Please provide as much detail under each heading. Examples are provided for the Solis inverter using the both the Solax Modbus Integration and the HA Core Modbus functionality. Please add your details below:

<h3>Examples</h3>

<h3>Your Setup</h3>

<h4>What is the inverter brand?</h4>
Answer
Solax

<h4>What is the inverter model?</h4>
Answer
X1 AC

<h4>What integration do you use in Home Assistant</h4>
Answer

Solaxx Inverter:
Solax Modbus inverter integration:
URL = https://github.com/wills106/homeassistant-solax-modbus

Also have the Solax Cloud api integration to capture stats, but trying to move away from this (5 minute lag)

Growatt Inverters:
Growatt integration to capture PV generation (captured via api) - 5 minute lag on this so will ultimately look to install GROTT to access the inverter data in realtime.  Still a wip at the moment

<h4>How do you enable forced charging?</h4>
Answer 
As below selection process to set the charger use mode

<h4>How do you enable forced discharging?</h4>
Answer 
As below selection process to set the charger use mode

<h4>Are forced charge and discharge controlled using Power or Current?<h4>
Answer
Not sure 

<h4> Does your inverter support timed charging slots?</h4>
Answer
Yes

<h4> Does your inverter support timed discharging slots?</h4>
Answer
Yes

<h4>How do you enabled timed charge?</h4>
Answer
Set the following:  
select `select.solax_charger_use_mode - options under this are:
 -Self use mode (power the house and charge from surplus solar)
 -Force time use (as above but with two time slots for charging from the grid)
 -Backup Mode ( Will only charge from grid and solar, won't discharge)
 -Feedin Priority (supplies the house first, then exports the rest)
then select the time slots:
"select.solax_charger_start_time_1"
"select.solax_charger_end_time_1", 
"select.solax_charger_start_time_2"
"select.solax_charger_end_time_2", 

<h4>How do you enabled timed discharge?</h4>
Answer
Think it's the "Feed in Priority" option above

<h4>Can you set a target SOC for a timed charge period?</h4>
Answer
Yes

<h4>Can you set a target SOC for a timed discharge period?</h4>
Answer
Yes

<h4>What modes does you inverter have and how are they controlled?</h4>
Answer
As per the above answer

<h4>Any other useful information</h4>
Answer

Happy to provide a list of entities extracted through the Modbus Interface

fboundy, 2024-03-16:
I've added this as an inverter type in pre-release v4.0.0-alpha-4 along with the Sunsynk. You will need to install this as a Beta version through HACS and also manually copy the `config_solax_x1.yaml` file from this repo as your `config.yaml`. I'm sure it won't work but if you can send me the `pvopt,log` and `error.log` files I will see what I can do....

punkymuzzle, 2024-03-16:
Thank you very much for this, really appreciate it.
As my last installation didn't work at all, I restored my HA back to it's previous state.  So I need to install everything from scratch again - so please bear with me whilst I do that and get back to you
Many thanks again

punkymuzzle, 2024-03-17:
Not sure where the two files you requested are.  If you can please let me know I can then go and hunt them down for you.

However, here's an extract from the AppDaemon error Log:

`2024-03-16 23:25:07.918772 INFO AppDaemon: /homeassistant/appdaemon/apps/pv_opt/config/config.yaml added or modified
2024-03-16 23:25:07.923485 INFO AppDaemon: Found 1 active apps
2024-03-16 23:25:07.924140 INFO AppDaemon: Found 0 inactive apps
2024-03-16 23:25:07.924770 INFO AppDaemon: Found 2 global libraries
2024-03-16 23:44:01.685757 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:06.705206 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:11.724479 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:16.741749 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:21.758088 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:28.095563 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:34.395394 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:39.422313 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:44.446984 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:49.470716 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:54.493225 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:44:59.517220 WARNING HASS: Disconnected from Home Assistant, retrying in 5 seconds
2024-03-16 23:45:04.567295 INFO HASS: Connected to Home Assistant 2024.3.1
2024-03-16 23:45:04.905961 INFO HASS: Evaluating startup conditions
2024-03-16 23:45:05.068373 INFO HASS: Startup condition met: hass state=RUNNING
2024-03-16 23:45:05.069084 INFO HASS: All startup conditions met
2024-03-16 23:45:05.308763 INFO AppDaemon: Processing restart for HASS
2024-03-16 23:45:05.310308 WARNING AppDaemon: Unable to find app solis in dependencies for pv_opt
2024-03-16 23:45:05.311055 WARNING AppDaemon: Ignoring app pv_opt
2024-03-16 23:45:05.312445 WARNING AppDaemon: Unable to find app sunsynk in dependencies for pv_opt
2024-03-16 23:45:05.313315 WARNING AppDaemon: Ignoring app pv_opt
2024-03-16 23:45:05.315413 INFO AppDaemon: Terminating pv_opt
2024-03-16 23:45:05.322654 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/pvpy.py
2024-03-16 23:45:05.365448 INFO AppDaemon: Reloading Module: /homeassistant/appdaemon/apps/pv_opt/pv_opt.py
2024-03-16 23:45:05.456279 WARNING AppDaemon: Unable to find app solis in dependencies for pv_opt
2024-03-16 23:45:05.456865 WARNING AppDaemon: Ignoring app pv_opt
2024-03-16 23:45:05.457373 WARNING AppDaemon: Unable to find app sunsynk in dependencies for pv_opt
2024-03-16 23:45:05.457841 WARNING AppDaemon: Ignoring app pv_opt
2024-03-16 23:45:05.460644 INFO AppDaemon: Loading app pv_opt using class PVOpt from module pv_opt
2024-03-16 23:45:05.465357 ERROR pv_opt: User defined log pv_opt_log not found
2024-03-16 23:45:05.470079 INFO AppDaemon: Calling initialize() for pv_opt
2024-03-16 23:45:05.474507 INFO pv_opt: 
2024-03-16 23:45:05.478570 INFO pv_opt: ******************* PV Opt v4.0.0-alpha-4 *******************
2024-03-16 23:45:05.482438 INFO pv_opt: 
2024-03-16 23:45:05.486360 INFO pv_opt: Pre-release version. Enabling debug logging
2024-03-16 23:45:05.488321 ERROR pv_opt: User defined log pv_opt_log not found
2024-03-16 23:45:05.491696 ERROR pv_opt: User defined log pv_opt_log not found
2024-03-16 23:45:05.495918 INFO pv_opt: Local timezone set to GB
2024-03-16 23:45:05.501696 INFO pv_opt: Time Zone Offset: 0.0 minutes
2024-03-16 23:45:05.511211 ERROR pv_opt: Inverter type SUNSYNK_SOLARSYNC2 is not yet supported. Only read-only mode with explicit config from the YAML will work.
2024-03-16 23:45:05.513431 INFO pv_opt: {'module': 'pv_opt', 'class': 'PVOpt', 'dependencies': ['pvpy', 'solis', 'sunsynk'], 'log': 'pv_opt_log', 'prefix': 'pvopt', 'overwrite_ha_on_restart': True, 'consumption_history_days': 7, 'maximum_dod_percent': 'sensor.{device_name}_{inverter sn}_battery_shutdown_cap', 'id_battery_soc': ' sensor.{device_name}_{inverter sn}_battery_soc', 'id_consumption_today': 'sensor.{device_name}_{inverter sn}_day_load_energy', 'id_grid_import_today': 'sensor.{device_name}_{inverter sn}_day_grid_import', 'id_grid_export_today': 'sensor.{device_name}_{inverter sn}_day_grid_import', 'supports_hold_soc': False, 'update_cycle_seconds': 300}
2024-03-16 23:45:05.517298 WARNING pv_opt: pv_opt: Entity sensor.pvopt_status not found in namespace default
2024-03-16 23:45:05.520066 INFO AppDaemon: pv_opt: Entity sensor.pvopt_status created in namespace: default
2024-03-16 23:45:05.949380 INFO pv_opt: Reading arguments from YAML:
2024-03-16 23:45:05.954555 INFO pv_opt: -----------------------------------
2024-03-16 23:45:05.958122 INFO pv_opt: 
2024-03-16 23:45:05.961885 INFO pv_opt:   Over-write flag is set so YAML will over-write HA
2024-03-16 23:45:05.965874 INFO pv_opt: consumption_history_days:
2024-03-16 23:45:05.969346 INFO pv_opt: 	7
2024-03-16 23:45:05.973576 INFO pv_opt: 	Found a single default value
2024-03-16 23:45:05.978998 INFO pv_opt:     consumption_history_days           = 7                                                              7: value in YAML
2024-03-16 23:45:05.982111 INFO pv_opt: maximum_dod_percent:
[... cut ...]
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 939, in _load_args
    if isinstance(self.get_default_config(item), str) and len(values) == 1:
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 833, in get_default_config
    if item in self.inverter.config:
               ^^^^^^^^^^^^^
AttributeError: 'PVOpt' object has no attribute 'inverter'

2024-03-16 23:45:05.999301 WARNING pv_opt: ------------------------------------------------------------`

punkymuzzle, 2024-03-17:
Here's a screenshot of my dashboard:

[image]

fboundy, 2024-03-17:
If you set up `appdaemon.yaml` as described in step 11.3 of the Readme you will have a separate logs as described there. I suggest you install the Samba Share addon to make accessing these files easier.

In v4.0.0-alpha-5 I have merged all the `config.yaml` into one file. You will need to dowwnload this from the repo and un-comment the 2nd last block as the `config.yaml` doesn't get updated by HACS:

      # ===============================================================================================================
      # Brand / Integration Specific Config: SOLAX_X1: 
      # ===============================================================================================================
      #
      #
      # These are the default entities used with the Solax X1 integration. You can change them here and over-ride the defaults
    
      inverter_type: SOLAX_X1  
      device_name: solax
    
      maximum_dod_percent:  number.{device_name}_battery_minimum_capacity
      id_battery_soc:  sensor.{device_name}_battery_capacity
      id_consumption: sensor.{device_name}_house_load
      id_grid_import_today: sensor.{device_name}_today_s_import_energy
      id_grid_export_today: sensor.{device_name}_today_s_export_energy
      supports_hold_soc: false
      update_cycle_seconds: 300

punkymuzzle, 2024-03-18:
Thanks for this.
I must have misunderstood the bit about samba, as I thought it was optional, and as I have ssh access to my device I can access the files. So I've installed and configured samba and can browse to the shares with no issues.  However, I still can't see any error.log or pvopt.log anywhere.

I've copied the config.yaml contents, replaced the info in my local version as requested and uncommented and amended the various sections.

AppDaemon is also working now (following your instructions it was "starting" and the log said it was running.  However I had a red dot against it and it didn't appear to be running.  I removed it, reinstalled and then set it up as default settings and it ran OK.  I then amended the config and rebooted my RasPi, and it's working now. 

so, I think it's running, there are no log files being generated that I can see, and the charts are still as my earlier screenshot.

Thanks

fboundy, 2024-03-18:
In the absence of the log files are you able to copy the contents of the logs from the AppDaemoin web UI?

punkymuzzle, 2024-03-18:
When I look at the AppDaemon Web Gui, there are no logs to look at.  None whatsoever.
[image]

Not sure if I'm missing something to get this working??

Here's the AppDaemon Logfile:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.4
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.1  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.3.1
 Home Assistant Supervisor: 2024.03.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting pandas
  Downloading pandas-2.2.1-cp311-cp311-musllinux_1_1_aarch64.whl.metadata (19 kB)
Collecting numpy<2,>=1.23.2 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/numpy-1.26.4-cp311-cp311-musllinux_1_2_aarch64.whl (11.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.3/11.3 MB 1.1 MB/s eta 0:00:00
Requirement already satisfied: python-dateutil>=2.8.2 in /usr/lib/python3.11/site-packages (from pandas) (2.8.2)
Requirement already satisfied: pytz>=2020.1 in /usr/lib/python3.11/site-packages (from pandas) (2023.3.post1)
Collecting tzdata>=2022.7 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/tzdata-2024.1-py2.py3-none-any.whl (345 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 345.4/345.4 kB 1.2 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3.11/site-packages (from python-dateutil>=2.8.2->pandas) (1.16.0)
Downloading pandas-2.2.1-cp311-cp311-musllinux_1_1_aarch64.whl (16.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.2/16.2 MB 1.2 MB/s eta 0:00:00
Installing collected packages: tzdata, numpy, pandas
Successfully installed numpy-1.26.4 pandas-2.2.1 tzdata-2024.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[09:08:33] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-03-18 09:08:39.727470 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-03-18 09:08:39.728910 INFO AppDaemon: Python version is 3.11.6
2024-03-18 09:08:39.729857 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-03-18 09:08:39.730353 INFO AppDaemon: Added log: AppDaemon
2024-03-18 09:08:39.731018 INFO AppDaemon: Added log: Error
2024-03-18 09:08:39.731453 INFO AppDaemon: Added log: Access
2024-03-18 09:08:39.732009 INFO AppDaemon: Added log: Diag
2024-03-18 09:08:40.124249 INFO AppDaemon: Loading Plugin HASS using class HasPlugin from module hasplugin
2024-03-18 09:08:40.129078 WARNING AppDaemon: error loading plugin: HASS - ignoring
2024-03-18 09:08:40.129539 WARNING AppDaemon: ------------------------------------------------------------
2024-03-18 09:08:40.131678 WARNING AppDaemon: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/plugin_management.py", line 140, in __init__
    mod = __import__(full_module_name, globals(), locals(), [module_name], 0)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'hasplugin'

[... 4 lines cut ...]
2024-03-18 09:08:40.286690 INFO MQTT: Using 'localad/status' as Birth Topic
2024-03-18 09:08:40.288142 INFO AppDaemon: Initializing HTTP
2024-03-18 09:08:40.289063 INFO AppDaemon: Using 'ws' for event stream
2024-03-18 09:08:40.297596 INFO AppDaemon: Starting API
2024-03-18 09:08:40.302046 INFO AppDaemon: Starting Admin Interface
2024-03-18 09:08:40.303012 INFO AppDaemon: Starting Dashboards
2024-03-18 09:08:40.373708 INFO MQTT: Connected to Broker at URL core-mosquitto:1883
2024-03-18 09:08:40.378661 INFO AppDaemon: App 'hello_world' added
2024-03-18 09:08:40.380623 INFO AppDaemon: Found 1 active apps
2024-03-18 09:08:40.381645 INFO AppDaemon: Found 0 inactive apps
2024-03-18 09:08:40.382304 INFO AppDaemon: Found 0 global libraries
2024-03-18 09:08:40.383125 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-03-18 09:08:40.385163 INFO AppDaemon: Running on port 5050
2024-03-18 09:08:40.389199 INFO AppDaemon: Got initial state from namespace mqtt
2024-03-18 09:08:40.390268 INFO MQTT: MQTT Plugin initialization complete
[... cut ...]
2024-03-18 09:08:41.484064 INFO AppDaemon: Loading app hello_world using class HelloWorld from module hello
2024-03-18 09:08:41.487055 INFO AppDaemon: Calling initialize() for hello_world
2024-03-18 09:08:41.594703 INFO hello_world: Hello from AppDaemon
2024-03-18 09:08:41.596768 INFO hello_world: You are now ready to run Apps!
2024-03-18 09:08:41.599423 INFO AppDaemon: App initialization complete
2024-03-18 09:24:53.631308 INFO AppDaemon: New client Admin Client connected
2024-03-18 09:25:16.477802 INFO AppDaemon: Client disconnection from Admin Client`

fboundy, 2024-03-18:
Could you share you appdaemon.yaml file please?

punkymuzzle, 2024-03-18:
`---
appdaemon:
  latitude: xx.xxxxx
  
  longitude: -xx.xxxxxx
  elevation: 2
  time_zone: Europe/London
  plugins:
    HASS:
      type: has
    MQTT:
       type: mqtt
       namespace: mqtt #
       verbose: True
       client_host: core-mosquitto
       client_port: 1883
       client_id: localad
       event_name: MQTT_MESSAGE 
       client_topics: NONE
       client_user: xxxxxx
       client_password: xxxxxx
http:
  url: http://127.0.0.1:5050
admin:
api:
hadashboard:

 logs:
   main_log:
     filename: /share/logs/main.log
     date_format: '%H:%M:%S'
   error_log:
     filename: /share/logs/error.log
     date_format: '%H:%M:%S'    
   pv_opt_log:
     name: PV_Opt
     filename: /share/logs/pv_opt.log
     date_format: '%H:%M:%S'      
     format: '{asctime} {levelname:>8s}: {message}'`
     
     (Name, password and location details removed)
     
     Thanks very much

fboundy, 2024-03-18:
There is something wrong with your AppDaemon installation. It is not loading the HASS plugin correctly:

    2024-03-18 09:08:40.124249 INFO AppDaemon: Loading Plugin HASS using class HasPlugin from module hasplugin
    2024-03-18 09:08:40.129078 WARNING AppDaemon: error loading plugin: HASS - ignoring
    2024-03-18 09:08:40.129539 WARNING AppDaemon: ------------------------------------------------------------
    2024-03-18 09:08:40.131678 WARNING AppDaemon: Traceback (most recent call last):
    File "/usr/lib/python3.11/site-packages/appdaemon/plugin_management.py", line 140, in init
    mod = import(full_module_name, globals(), locals(), [module_name], 0)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ModuleNotFoundError: No module named 'hasplugin'

The module it is trying to load is wrong. It should be `hassplugin` as per my log:

    09:56:43 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
    09:56:43 INFO HASS: HASS Plugin Initializing
    09:56:43 INFO HASS: HASS Plugin initialization complete

fboundy, 2024-03-18:
OK - you need to correct:

    HASS:
    type: has

to
  
    HASS:
    type: hass

punkymuzzle, 2024-03-18:
Thanks for spotting that....
OK updated the config and no error in the log file below.

However, there is still one warning ref  a missing URL, but not sure if that one matters (or if it does, where to put the URL)

Logfile:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.4
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.1  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.3.1
 Home Assistant Supervisor: 2024.03.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting pandas
  Downloading pandas-2.2.1-cp311-cp311-musllinux_1_1_aarch64.whl.metadata (19 kB)
Collecting numpy<2,>=1.23.2 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/numpy-1.26.4-cp311-cp311-musllinux_1_2_aarch64.whl (11.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.3/11.3 MB 1.1 MB/s eta 0:00:00
Requirement already satisfied: python-dateutil>=2.8.2 in /usr/lib/python3.11/site-packages (from pandas) (2.8.2)
Requirement already satisfied: pytz>=2020.1 in /usr/lib/python3.11/site-packages (from pandas) (2023.3.post1)
Collecting tzdata>=2022.7 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/tzdata-2024.1-py2.py3-none-any.whl (345 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 345.4/345.4 kB 1.1 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3.11/site-packages (from python-dateutil>=2.8.2->pandas) (1.16.0)
Downloading pandas-2.2.1-cp311-cp311-musllinux_1_1_aarch64.whl (16.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.2/16.2 MB 1.1 MB/s eta 0:00:00
Installing collected packages: tzdata, numpy, pandas
Successfully installed numpy-1.26.4 pandas-2.2.1 tzdata-2024.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
s6-rc: info: service init-appdaemon successfully started
s6-rc: info: service appdaemon: starting
s6-rc: info: service appdaemon successfully started
s6-rc: info: service legacy-services: starting
[11:10:45] INFO: Starting AppDaemon...
s6-rc: info: service legacy-services successfully started
2024-03-18 11:10:51.846474 INFO AppDaemon: AppDaemon Version 4.4.2 starting
2024-03-18 11:10:51.847180 INFO AppDaemon: Python version is 3.11.6
2024-03-18 11:10:51.847533 INFO AppDaemon: Configuration read from: /config/appdaemon.yaml
2024-03-18 11:10:51.847795 INFO AppDaemon: Added log: AppDaemon
2024-03-18 11:10:51.848633 INFO AppDaemon: Added log: Error
2024-03-18 11:10:51.849310 INFO AppDaemon: Added log: Access
2024-03-18 11:10:51.849750 INFO AppDaemon: Added log: Diag
2024-03-18 11:10:52.237612 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-03-18 11:10:52.859844 INFO HASS: HASS Plugin Initializing
2024-03-18 11:10:52.860678 WARNING HASS: ha_url not found in HASS configuration - module not initialized
2024-03-18 11:10:52.861019 INFO HASS: HASS Plugin initialization complete
2024-03-18 11:10:52.861503 INFO AppDaemon: Loading Plugin MQTT using class MqttPlugin from module mqttplugin
2024-03-18 11:10:53.005063 INFO MQTT: MQTT Plugin Initializing
2024-03-18 11:10:53.006512 INFO MQTT: Using 'localad/status' as Will Topic
2024-03-18 11:10:53.007023 INFO MQTT: Using 'localad/status' as Birth Topic
2024-03-18 11:10:53.008228 INFO AppDaemon: Initializing HTTP
[... 4 lines cut ...]
2024-03-18 11:10:53.050480 INFO HASS: Connected to Home Assistant 2024.3.1
2024-03-18 11:10:53.065976 INFO MQTT: Connected to Broker at URL core-mosquitto:1883
2024-03-18 11:10:53.077799 INFO AppDaemon: App 'hello_world' added
2024-03-18 11:10:53.080287 INFO AppDaemon: Found 1 active apps
2024-03-18 11:10:53.080870 INFO AppDaemon: Found 0 inactive apps
2024-03-18 11:10:53.081331 INFO AppDaemon: Found 0 global libraries
2024-03-18 11:10:53.081854 INFO AppDaemon: Starting Apps with 1 workers and 1 pins
2024-03-18 11:10:53.083365 INFO AppDaemon: Running on port 5050
2024-03-18 11:10:53.088417 INFO AppDaemon: Got initial state from namespace mqtt
2024-03-18 11:10:53.088997 INFO MQTT: MQTT Plugin initialization complete
2024-03-18 11:10:53.166897 INFO HASS: Evaluating startup conditions
2024-03-18 11:10:53.183822 INFO HASS: Startup condition met: hass state=RUNNING
2024-03-18 11:10:53.185013 INFO HASS: All startup conditions met
[... cut ...]
2024-03-18 11:10:55.181929 INFO AppDaemon: Calling initialize() for hello_world
2024-03-18 11:10:55.308294 INFO hello_world: Hello from AppDaemon
2024-03-18 11:10:55.310704 INFO hello_world: You are now ready to run Apps!
2024-03-18 11:10:55.313004 INFO AppDaemon: App initialization complete
2024-03-18 11:10:58.458437 INFO AppDaemon: New client Admin Client connected
2024-03-18 11:16:28.708377 INFO AppDaemon: Client disconnection from Admin Client
2024-03-18 11:16:29.883540 INFO AppDaemon: New client Admin Client connected`

punkymuzzle, 2024-03-18:
I've tailored some of the chart sensors in the screenshot below (this still needs reworking as some of the numbers aren't right), but it still loos like the pvopt entities aren't being created:
[image]
[image]
[image]
[image]

fboundy, 2024-03-18:
Do you have the expected log files now? They should be in the Web UI and the samba share

punkymuzzle, 2024-03-18:
No, nothing in any of the logs when looking via the AppDaemon webgui

fboundy, 2024-03-18:
I think @solarjones had a similar issue with logging but I can't find his solution

punkymuzzle, 2024-03-18:
Just checked and i can access the samba share, create, amend, and delete files here.

[image]

OK thanks for your help.  Not really sure what else to do but I'll have another look around to see if I can stumble across them.

Thank you

solarjones, 2024-03-18:
The AppDaemon Logs visible via Refresh button are different to those written to Pv_opt.log, main.log and error.log. I was expecting the logs to be written within the HA folder structures when viewed from FileEditor or Visual Studio, but they aren't. That was my misunderstanding and struggled trying all sorts. In the end I simply followed the PV_Opt instructions to the letter for the config.yaml (I had different paths which weren't valid but didn't error) and samba share. It then sprang into logging life!

punkymuzzle, 2024-03-18:
@solarjones OK thanks very much, I'll revisit the instructions and double check.
Ta

solarjones, 2024-03-18:
@punkymuzzle 
Correction: 
It's the AppDaemon.yaml (not config.yaml)  you need to pay attention too. Here's mine (obviously swap out your own long & lat)... the indents had to be correct too.

secrets: /homeassistant/secrets.yaml
appdaemon:
  latitude: {nnn}
  longitude: {nnn}
  elevation: 10
  time_zone: Europe/London
  thread_duration_warning_threshold: 45
  app_dir: /homeassistant/appdaemon/apps
  plugins:
    HASS:
      type: hass
    MQTT:
      type: mqtt
      namespace: mqtt
      verbose: True
      client_host: core-mosquitto
      client_port: 1883
      client_id: localad
      event_name: MQTT_MESSAGE
      client_topics: NONE
      client_user: !secret mqtt-username
      client_password: !secret mqtt-password
  http:
    url: http://192.168.1.235:5050
    hadashboard:
      dash_url: http://127.0.0.1:5050
      dash_dir: /config/appdaemon/dashboards
logs:
  main_log:
    filename: /share/logs/main.log
    date_format: "%H:%M:%S"
  error_log:
    filename: /share/logs/error.log
    date_format: "%H:%M:%S"
  pv_opt_log:
    name: PV_Opt
    filename: /share/logs/pv_opt.log
    date_format: "%H:%M:%S"
    format: "{asctime} {levelname:>8s}: {message}"

punkymuzzle, 2024-03-18:
OK so I've gone through the AppDaemon config file again, restarted numerous times.

There still seem to be no log files generated.  I've:
1. Searched through the Samba Share for them
2. SSH to my RasPi and searched from the root folder for them.  I only have one log file (home-assistant.log) 

AppDaemon logfile:
There seems to be more detail in this now relating to pv_opt - please see below:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service base-addon-banner: starting

-----------------------------------------------------------
 Add-on: AppDaemon
 Python Apps and Dashboard using AppDaemon 4.x for Home Assistant
-----------------------------------------------------------
 Add-on version: 0.16.4
 You are running the latest version of this add-on.
 System: Home Assistant OS 12.1  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.3.1
 Home Assistant Supervisor: 2024.03.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service init-appdaemon: starting
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting pandas
  Downloading pandas-2.2.1-cp311-cp311-musllinux_1_1_aarch64.whl.metadata (19 kB)
Collecting numpy<2,>=1.23.2 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/numpy-1.26.4-cp311-cp311-musllinux_1_2_aarch64.whl (11.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 11.3/11.3 MB 1.1 MB/s eta 0:00:00
Requirement already satisfied: python-dateutil>=2.8.2 in /usr/lib/python3.11/site-packages (from pandas) (2.8.2)
Requirement already satisfied: pytz>=2020.1 in /usr/lib/python3.11/site-packages (from pandas) (2023.3.post1)
Collecting tzdata>=2022.7 (from pandas)
  Downloading https://wheels.home-assistant.io/musllinux-index/tzdata-2024.1-py2.py3-none-any.whl (345 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 345.4/345.4 kB 1.2 MB/s eta 0:00:00
Requirement already satisfied: six>=1.5 in /usr/lib/python3.11/site-packages (from python-dateutil>=2.8.2->pandas) (1.16.0)

2024-03-18 15:23:10.450585 INFO AppDaemon: Added log: AppDaemon
2024-03-18 15:23:10.451266 INFO AppDaemon: Added log: Error
2024-03-18 15:23:10.451883 INFO AppDaemon: Added log: Access
2024-03-18 15:23:10.452564 INFO AppDaemon: Added log: Diag
2024-03-18 15:23:10.834704 INFO AppDaemon: Loading Plugin HASS using class HassPlugin from module hassplugin
2024-03-18 15:23:11.455939 INFO HASS: HASS Plugin Initializing
2024-03-18 15:23:11.456472 WARNING HASS: ha_url not found in HASS configuration - module not initialized
2024-03-18 15:23:11.458177 INFO HASS: HASS Plugin initialization complete
2024-03-18 15:23:11.459043 INFO AppDaemon: Loading Plugin MQTT using class MqttPlugin from module mqttplugin
2024-03-18 15:23:11.603327 INFO MQTT: MQTT Plugin Initializing
2024-03-18 15:23:11.605099 INFO MQTT: Using 'localad/status' as Will Topic
2024-03-18 15:23:11.605562 INFO MQTT: Using 'localad/status' as Birth Topic
2024-03-18 15:23:11.607213 INFO AppDaemon: HTTP is disabled
2024-03-18 15:23:11.628534 INFO HASS: Connected to Home Assistant 2024.3.1
2024-03-18 15:23:11.640983 INFO MQTT: Connected to Broker at URL core-mosquitto:1883
2024-03-18 15:23:11.658818 INFO AppDaemon: Got initial state from namespace mqtt
2024-03-18 15:23:11.664658 INFO MQTT: MQTT Plugin initialization complete
2024-03-18 15:23:11.792084 INFO HASS: Evaluating startup conditions
2024-03-18 15:23:11.821933 INFO HASS: Startup condition met: hass state=RUNNING
2024-03-18 15:23:11.822668 INFO HASS: All startup conditions met
2024-03-18 15:23:11.825538 WARNING AppDaemon: File '/homeassistant/appdaemon/apps/pv_opt/config/config.yaml' duplicate app: pvpy - ignoring
2024-03-18 15:23:11.826262 WARNING AppDaemon: File '/homeassistant/appdaemon/apps/pv_opt/config/config.yaml' duplicate app: inverters - ignoring
2024-03-18 15:23:11.827352 WARNING AppDaemon: File '/homeassistant/appdaemon/apps/pv_opt/config/config.yaml' duplicate app: pv_opt - ignoring
[... 294 lines cut ...]
  id_grid_export_today: sensor.{device_name}_today_s_export_energy
  supports_hold_soc: false
  update_cycle_seconds: 300

  # ===============================================================================================================
  # Brand / Integration Specific Config: SUNSYNK_SOLARSYNK2: 
  # ===============================================================================================================
  #
  #
  # These are the default entities used with the Sunsynk Solarsynk2 integration. You can change them here and over-ride the defaults

  # inverter_type: SUNSYNK_SOLARSYNK2
  # device_name: solarsynk
  # inverter_sn: 000000  # enter {sunsynk_serial} from solarsynk2 addon

  # maximum_dod_percent:  sensor.{device_name}_{inverter_sn}_battery_shutdown_cap
  # id_battery_soc:  sensor.{device_name}_{inverter_sn}_battery_soc
  # id_consumption_today: sensor.{device_name}_{inverter_sn}_day_load_energy
  # id_grid_import_today: sensor.{device_name}_{inverter_sn}_day_grid_import
  # id_grid_export_today: sensor.{device_name}_{inverter_sn}_day_grid_import
  # supports_hold_soc: false
  # update_cycle_seconds: 300
`

So unless anyone else has any more pointers, I'm all out of ideas.

punkymuzzle, 2024-03-18:
I've also checked the config.yaml file but can't see any errors in this either.  File as follows:

# Internal configuration --- DO NOT EDIT ---
pvpy:
  module: pvpy
  global: true

inverters:
  module: inverters
  global: true

pv_opt:
  module: pv_opt
  class: PVOpt
  dependencies:
    - pvpy
    - solis
    - sunsynk
    - solax
    
  log: pv_opt_log
  prefix: pvopt
  # debug: True

# User configuration ---  EDIT AWAY! ---
  # ========================================
  # System parameters
  # ========================================

  # If true the current config in HA will be over-written with that in the config.yaml. 
  overwrite_ha_on_restart: true
  # redact_personal_data_from_log: false

  read_only: false

  # ========================================
  # Plant parameters
  # ========================================

  # All parameters can be a number or point to an entity UNLESS they start with 'id_'. All of
  # these are required but will be defaulted if not specified

  battery_capacity_wh: 4500
  inverter_efficiency_percent: 97 # Default: 97
  charger_efficiency_percent: 91 # Default: 91
  # maximum_dod_percent: 15
  charger_power_watts: 3500
  inverter_power_watts: 3600
  inverter_loss_watts: 100
  
  # ========================================
  # Solcast configuration
  # ========================================
  #
  id_solcast_today: sensor.solcast_pv_forecast_forecast_today
  id_solcast_tomorrow: sensor.solcast_pv_forecast_forecast_tomorrow

  # ========================================
  # Solar and consumption forecast parameters
  # ========================================
  #
  # Valid options are:
  #
  # Solcast         - the Solcast mid-case forecast [Default]
  # Solcast_p90     - the Solcast high estimate
  # Solcast_p10     - the Solcast high estimate
  #
  # Set this using an "input_select" helper and it can be varied on the fly

  solar_forecast:
[... 145 lines cut ...]
  id_grid_import_today: sensor.{device_name}_today_s_import_energy
  id_grid_export_today: sensor.{device_name}_today_s_export_energy
  supports_hold_soc: false
  update_cycle_seconds: 300

  # ===============================================================================================================
  # Brand / Integration Specific Config: SUNSYNK_SOLARSYNK2: 
  # ===============================================================================================================
  #
  #
  # These are the default entities used with the Sunsynk Solarsynk2 integration. You can change them here and over-ride the defaults

  # inverter_type: SUNSYNK_SOLARSYNK2
  # device_name: solarsynk
  # inverter_sn: 000000  # enter {sunsynk_serial} from solarsynk2 addon

  # maximum_dod_percent:  sensor.{device_name}_{inverter_sn}_battery_shutdown_cap
  # id_battery_soc:  sensor.{device_name}_{inverter_sn}_battery_soc
  # id_consumption_today: sensor.{device_name}_{inverter_sn}_day_load_energy
  # id_grid_import_today: sensor.{device_name}_{inverter_sn}_day_grid_import
  # id_grid_export_today: sensor.{device_name}_{inverter_sn}_day_grid_import
  # supports_hold_soc: false
  # update_cycle_seconds: 300

So unless anyone else has any more pointers, I'm all out of ideas.

punkymuzzle, 2024-03-18:
@solarjones  Thanks very much for this.
I've already done the very thing you suggested - when I first configured samba, there was no logs folder under "share", so I created it:
[image]

I've also SSH into the RasPi and looked at folder permissions in case it was unable to write there.  I've changed permissions on the folder so any account can write to it, and then rebooted.
Still no log files :(

fboundy, 2024-03-18:
The log files are set up by `appdaemon.yaml` rather than `config.yaml`.

You might want to check the indentation in it to make sure it is correct. This is mine:

    secrets: //homeassistant/secrets.yaml
    appdaemon:
      latitude: 54.729
      longitude: -2.991
      elevation: 175
      time_zone: Europe/London
      thread_duration_warning_threshold: 45
      app_dir: /homeassistant/appdaemon/apps
      plugins:
        HASS:
          type: hass
    
        MQTT:
          type: mqtt
          namespace: mqtt # needed
          verbose: True
          client_host: core-mosquitto
          client_port: 1883
          client_id: localad
          event_name: MQTT_MESSAGE # the event you will listen in your module
          client_topics: NONE
          client_user: !secret mqtt-user
          client_password: !secret mqtt-password
    
    http:
      url: http://127.0.0.1:5050
    admin:
    api:
    hadashboard:
    logs:
      main_log:
        filename: /share/logs/main.log
        date_format: '%H:%M:%S'
      error_log:
        filename: /share/logs/error.log
        date_format: '%H:%M:%S'    
      pv_opt_log:
        name: PV_Opt
        filename: /share/logs/pv_opt.log
        date_format: '%H:%M:%S'      
        format: '{asctime} {levelname:>8s}: {message}'

Also please delete all `config*.yaml` files other than `config.yaml` as AppDaemon will try to read them all and you will end up with duplication.

punkymuzzle, 2024-03-18:
@fboundy / @solarjones 
I've been through the config.yaml and double checked / amended the indents etc, so it matches yours.
Rebooted and its started working a lot better !!!!!
That's weird, as the file editor wasn't showing any errors in my indentation, so I thought it was OK.
I now have access to the logfiles as requested.
[error.log](https://github.com/fboundy/pv_opt/files/14643545/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14643547/pv_opt.log)

The cards are also looking a lot better as well:
[image]
[image]
[image]
[image]
[image]

Thanks very much to you both for getting me this far, really appreciate it.

Let me know if you need any more information.  I'm working away tomorrow so won't have access to HA, but I'll catch up later in the evening.
Thanks again

lazy-pete, 2024-03-19:
Thanks folks - I had the same indent issues with the log file setup in
appdaemon.yaml so hopefully I’ll be able to find those logs this evening !

🤞🏻

fboundy, 2024-03-22:
> Hi all
> Is there any other information that I can provide to help?
> Thanks

I’d missed your last update. Will have a look and let you know.

fboundy, 2024-03-22:
OK - all looking good so far. Much easier to diagnose now that the log files are sorted.

The error that is stopping it is because you only have instantaneous load rather than a daily total. Rather than re-code a lot of stuff I'm going to suggest a simple workaround (which also means you can use the Energy Dashboard more easily in HA). 

Can you please make a "Reimann Sum" helper in HA like this from your `sensor.solax_house_load` entity. Suggest you call yours `solax_house_load_today`:

[image]

Then modify your `config.yaml` to replace:

    id_consumption: sensor.solax_house_load

with:

    id_consumption: sensor.solax_house_load_today

Please also set 'read_only' to 'true' in the 'config.yaml' as we haven't addressed any of the inverter-specific control issues yet.

Saving the new `config.yaml` should automatically restart PV_Opt. Hopefully with this change the optimiser will run fully in Read Only mode We then just have to get it talking properly to your inverter. Whatever happens please send me the new log files.

punkymuzzle, 2024-03-22:
Thank you very much for this, really appreciate it.
Just to let you know that the "House Load" entity actually shows negative numbers as well as positive (I have no idea why), so previously I have created a sensor that only records Positive numbers, as follows:
`{{ [ states('sensor.solax_house_load')|float(0), 0.0 ] | max }}`
It's Called "Positive House Load" and is in W

So I've created the above helper and called it (for the time being) - "Total Solax House Load Today for pvopt"

So this will capture the daily total load on the house.
Can you please double check my rationale above?  In the meantime I'll amend the necessary in config.yaml and retest.

Thanks once again, really appreciate the support

punkymuzzle, 2024-03-22:
OK, config updated and for good measure I have restarted AppDaemon.
In the AppDaemon log I noticed the following appearing 6 times 
`/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:927: FutureWarning: errors='ignore' is deprecated and will raise in a future version. Use to_datetime without passing `errors` and catch exceptions explicitly instead`

I've also uploaded the logs
Many thanks once again
 [pv_opt.log](https://github.com/fboundy/pv_opt/files/14722815/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14722816/error.log)

fboundy, 2024-03-22:
I’ve fixed that warning in the main branch but haven’t merged it back into this one. Will have a look at your logs when I get a minute.

fboundy, 2024-03-22:
In your `config.yaml` you need:

    id_consumption_today: sensor.total_solax_house_load_today_for_pvopt

I think you have:

    id_consumption: sensor.total_solax_house_load_today_for_pvopt

Can you please update and re-send the logs?

punkymuzzle, 2024-03-22:
Sorry this is a long post, couldn't work out how to exctact these into a file...

Not sure if this helps, but here's a list of all my entities (enabled and disabled) for the SolaX Modbus integration connected to my Inverter:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

punkymuzzle, 2024-03-22:
OK done.
Logs uploaded
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14726458/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14726459/error.log)

fboundy, 2024-03-22:
OK - that looks like it is basically working. I've merged a few patches from the latest production version into this branch so please could you:

- Update to 4.0.0-alpha-6 over HACS
- Edit your `config.yaml` so it has `debug: false` which will make the log output quite a bit cleaner.

Then I suggest you check that the dashboard looks OK and have a play with a few of the settings in Read Only mode then send me latest logs and a screenshot of your dashboard. Then we can start to get controlling the inverter.

punkymuzzle, 2024-03-22:
What is this referring to:
[image]
Is it the total PV Generated today?

punkymuzzle, 2024-03-22:
I'm assuming it does relate to PV Generated today so I changed it.
This looks great!  I've added screenshots of the dashboard (I had to change some entity names to get it to work but will send you the .yaml once it's all working)
[image]
[image]
[image]
[image]
[image]

Plus Logfiles:

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14727346/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14727347/error.log)

Thank you so much for this, really appreciate it.

fboundy, 2024-03-22:
It's definitely getting there - with it being quite sunny and with no export contract it is currently saying "Do nothing" which isn't very interesting. Can you update the dashboard  to the latest version  which allows you to select a confidence level and select 10% (or turn off Solar completely) and see what that looks like?

punkymuzzle, 2024-03-22:
Thought I'd put my export rate in the cinfig. Yaml
I'll double check later and get back to you
Thanks

punkymuzzle, 2024-03-22:
OK, managed to find the export code and added it.
Changed the forecast to P10?
[image]

[error.log](https://github.com/fboundy/pv_opt/files/14727681/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14727682/pv_opt.log)

punkymuzzle, 2024-03-22:
Sorry, ran out of time tonight to add the updated dash - will do it either later or tomorrow early
Thank you

punkymuzzle, 2024-03-23:
Here's the Dashboard now I've updated the specific entities:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

What do you think?
Thanks

punkymuzzle, 2024-03-23:
OK done that, Logs attached.
Also, restarted AppDaemon and saw this again in the log - not sure if it matters:
[image]

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14731651/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14731652/error.log)

fboundy, 2024-03-23:
The warning needs dealing with but is just a warning so OK for now.

It looks like it's running OK now. What does your dashboard look like?

fboundy, 2024-03-23:
The one error I'm seeing is that there's no battery voltage. Is there a sensor that can provide that? If not I will default it to 50V

punkymuzzle, 2024-03-23:
Here's the dashboard

And I think this is the entity for the battery
sensor.solax_battery_voltage_charge

fboundy, 2024-03-23:
Ok - have you checked the history for that entity to make sure it looks sensible?

punkymuzzle, 2024-03-23:
Here's the graph

punkymuzzle, 2024-03-23:
Sorry. Just realised the graphics and files didn't upload
Dashboard:
[image]

Entity:
[image]

Graph:
[image]

And latest logs attached:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14733863/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14733864/error.log)

Thanks very much

fboundy, 2024-03-24:
100V? Didn't expect that...

What is happening during the periods where the history is null?

punkymuzzle, 2024-03-24:
Not even sure if that's the right thing that you're after, this is one of the entities that got created when I added the Modbus integration.

Nut sure about the "null" times. I do have times (overnight) where the inverter / battery are classed as "idle" and I can't seem to control them via Modbus - but controlling it through the app is fine.
But apart from that I have no idea.

Here's an updated graph for the week.  Looks like it does that pretty regularly:
[image]

punkymuzzle, 2024-03-24:
Found another one - "Solax Inverter Voltage" - but looks like it's doing something similar:
[image]

punkymuzzle, 2024-03-24:
Still not sure why there are errors in the card .yaml file, as this was working:
[image]

fboundy, 2024-03-25:
It's because of the voltage being missing. I'm suspecting the 100V is a bit of a red herring - what batteries do you have? It may actually not matter if your inverter is controlled by power rather than current. I'm going to set it as a fixed number which can be over-written if needed by `config.yaml` if we need to and we can deal with it later.

`4.0.0-solax-x1-8` should now be on HACS and hopefully this will fix this.

punkymuzzle, 2024-03-25:
My battery is a Triple Power T-45.

I've updated everything and rebooted.
Is there an updated version of pv_opt that I need to update to? I'm on
Alpha 6

"4.0.0-solax-x1-8"
What is this? My SolaX integration version is 2024.03.7b4.  Am I missing
something?

punkymuzzle, 2024-03-25:
Added the Inverter charge entity back in, so slightly improved chart:
[image]
But still stuff missing.

And here:
[image]

fboundy, 2024-03-25:
That’s the latest PV Opt pre-release. I’m separating out the various inverters into separate branches and pre-releases.

Solax integration is the latest so that’s good as there have been some ModBus issues

Just found the data sheet for your battery and it is indeed 100.8v nominal.

fboundy, 2024-03-25:
Can you please upgrade to 4.0.0-solax-x1-8 and send me the logs. I don't need screenshots for now

punkymuzzle, 2024-03-25:
OK understood.
I'll wait to hear from you.
Let me know if you need anything further from me.
Thanks

[38 more replies]

closed 2024-03-28 by fboundy