stevebuk1/pv_opt #152: Unexpected error running initialize() for pv_opt

solarjones, 2024-03-17:
3.10.0

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14628869/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14628867/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14628868/main.log)

  File "/homeassistant/appdaemon/apps/pv_opt/solis.py", line 493, in _solis_solax_solarman_mode_switch
    code = codes[inverter_mode]
           ~~~~~^^^^^^^^^^^^^^^
KeyError: 'Self-Use'

I got this after doing several updates this evening on HA, one of which was the Solax Modbus integration. Even thought I've reverted, still get the above. I'll see if I can down grade the elements.

fboundy, 2024-03-17:
Ok - let me know the Solax version numbers if you get anywhere.

fboundy, 2024-03-17:
Just had a look at the Solax integration and he has expanded and changed the select entity for the inverter modes. Should be easy enough to fix but I’ll need to try to handle both versions which is a pain.

fboundy, 2024-03-17:
I’ve just bumped mine to 24.03.3 and it looks like the names have been reverted but more options have been added. I think this shouldn’t cause a problem

solarjones, 2024-03-17:
I've found (for me anyway) PV_Opt works with Solax Modbus 2024.03.1, but not after upgrading to 2024.3.3.

https://github.com/wills106/homeassistant-solax-modbus/releases 
Release notes under 2024.3.2 
Solis:
Added "Sync RTC Offset"
Adjusted names for "Energy Storage Control Switch"
Corrected "Battery Charge Current", "Battery ChargeDischarge Current" and "Battery Discharge Current" rounding. 

I got errors trying to RTC sync on both PV_Opt and Solis Helper. So looks like something has changed with that integration.

solarjones, 2024-03-17:
> I’ve just bumped mine to 24.03.3 and it looks like the names have been reverted but more options have been added. I think this shouldn’t cause a problem
> […](#)

Did you totally deinstall Solax Modbus as per the suggestion? I did from integration and HACs, reloaded, restarted and then hit problems.

fboundy, 2024-03-17:
Yep - I thought 03.3 fixed it but it doesn’t. Will have to have a look tomorrow. Rolling back to pre 24.03 might fix it

fboundy, 2024-03-17:
Quick fix is to edit solis.py and on lines 8 and 11 replace SelfUse with Self-Use. Restart AD and all should be well.

Will patch properly tomorrow.

fboundy, 2024-03-18:
This should be fixed in 3.11.0

solarjones, 2024-03-19:
Just updated to 3.11.0 and 24.03.06 and when I tried Solis Sync RTC I got this:

[image]

SzosszeNET, 2024-03-19:
That's a solax issue

solarjones, 2024-03-19:
Ah ok, you have that too?

fboundy, 2024-03-19:
Agreed.

fboundy, 2024-03-19:
I do with 24.03.4

SzosszeNET, 2024-03-19:
Actually I had it with 24.03.3

SzosszeNET, 2024-03-19:
Opened a bug report for this against solax

solarjones, 2024-03-19:
Perhaps it's me being paranoid but I don't recall what mode my inverter usually operate in. Is this correct? Does the PV_Opt control this?

[image]

On the SolaX integration, it seems to list a few more than I remember:
[image]

Thanks.

SzosszeNET, 2024-03-19:
Check history.
Tap on the icon in front of it to see when was what.
But otherwise self-use is correct when pv_opt isn't charging/discharging.

solarjones, 2024-03-20:
Thanks for the History tip and highlighted a terminology change.

The SolaX Modbus integration has renamed the Energy_storage_control_switch mode names. What used to be mode enumeration 35 'Timed Charge/Discharge', is now 'Self-Use'. 

This explains what I noticed after installing 24.03.6 yesterday.

fboundy, 2024-03-20:
Not quite. SelfUse has been renamed Self-Use and three Feed In Priority modes have been added. The indexes remain the same because the modes are controlled but-wise

fboundy, 2024-03-20:
*bit-wise

fboundy, 2024-03-20:
Something has got broken in the Solax/PV_Opt interface. I will need to investigate

fboundy, 2024-03-20:
OK so here's the issue. Pre-24.03 the modes were:

    1: "Self-Use - No Grid Charging",
    3: "Timed Charge/Discharge - No Grid Charging",
    17: "Backup/Reserve - No Grid Charging",
    33: "Self-Use",
    35: "Timed Charge/Discharge",
    37: "Off-Grid Mode",
    41: "Battery Awaken",
    43: "Battery Awaken + Timed Charge/Discharge",
    49: "Backup/Reserve - No Timed Charge/Discharge",
    51: "Backup/Reserve",

24.03 onwards they are:

    1: "Self-Use - No Grid Charging",
    3: "Timed Charge/Discharge - No Grid Charging",
    17: "Backup/Reserve - No Grid Charging",
    33: "Self-Use - No Timed Charge/Discharge",
    35: "Self-Use",
    37: "Off-Grid Mode",
    41: "Battery Awaken",
    43: "Battery Awaken + Timed Charge/Discharge",
    49: "Backup/Reserve - No Timed Charge/Discharge",
    51: "Backup/Reserve",
    64: "Feed-in priority - No Grid Charging",
    96: "Feed-in priority - No Timed Charge/Discharge",
    98: "Feed-in priority",

The issue is 33 and 35. Fortunately he's also changed SelfUse to Self-Use so I can incorporate both versions. I'm testing a patch to 3.11 now which should resolve things I hope.

fboundy, 2024-03-20:
Should be fixed by #156 in 3.11.1

lazy-pete, 2024-03-20:
Hi - just updated Solax and PV_opt after it failed overnight (the RTC sync times were hours out so it was forcing the charge scheduled for middle of the night late). Still have an errror on RTC sync and pv_opt doesn’t seem to find number.solis_battery_minimum_soc  although I’ve confirmed it is present and set at 15% 

[image]
[image]
[image]

lazy-pete, 2024-03-20:
Actually- different error for RTC after updates - 

[image]

lazy-pete, 2024-03-20:
Seems to have sorted itself out apart from the second RTC sync error. 

Only thing I did was check the config.yaml - I unconnected the default dod line, which caused a duplication error, recommended it and things seemed to work after that.

lazy-pete, 2024-03-20:
PS - I owe you some beers !

fboundy, 2024-03-21:
Latest release has a "Don't buy me a beer or a coffee" section...

closed 2024-03-21 by fboundy