stevebuk1/pv_opt #154: Option to configure without solar

DrJohnM61, 2024-03-18:
**Is your feature request related to a problem? Please describe.**
We do not have solar (low priority as East/West roof and high latitudes in Scotland providing limited benefit in the winter months and poor ROI), but we do have a givenergy all-in-one battery/inverter. This integration looks to be perfect for scheduling charge and discharge cycles based on the Octopus Agile tariff.

**Describe the solution you'd like**
An ability to configure without solar, ie, just charging and discharging using grid supply.

**Describe alternatives you've considered**

**Additional context**

fboundy, 2024-03-18:
That should be possible. It will also require a GivEnergy inverter controller to be written

DrJohnM61, 2024-03-18:
Thank you for the quick response and consideration.

As a fyi, there is the GivTCP integration that exposes all of the givenergy inverter information/controls (https://github.com/britkat1980/giv_tcp)

fboundy, 2024-03-18:
Do you use the Rest API or MQTT with GivTCP?

DrJohnM61, 2024-03-18:
MQTT.

I just followed the process as written in this tutorial https://www.speaktothegeek.co.uk/2022/12/givtcp-v2-givenergy-and-home-assistant-with-solar-forecast-automation/#overview

fboundy, 2024-03-19:
OK - I will have a look at that. The option to have no solar will be in the next release (3.12.0) so you should be able to configure your setup in read-only mode using the parameters in `config.yaml`. Just leave the inverter type as `solis` for now. Once you have it running read-only we can look at getting the inverter controller plug-in built.

DrJohnM61, 2024-03-19:
Thank you. I will take a look when I am back from Canada.

fboundy, 2024-03-27:
OK let me know when you've had a chance to look at it

closed 2024-06-03 by github-actions[bot]

philicibine, 2024-10-12:
Sorry to bring up a stale issue here, but I have a friend that will soon be getting this inverter / battery setup.  Will be glad to help with getting it setup for this.  Hopefully you have time?