stevebuk1/pv_opt #157: Errors/Warnings in AppDaemon log

swests, 2024-03-20:
**Describe the bug**
Getting a stream of warnings in the AppDaemon log

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings -> Add-Ons -> AppDaemon
2. Click on Log
3. See error

**Expected behavior**
Suspect these are just warnings but would but do indicate a potential issue

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
N/A

**Smartphone (please complete the following information):**
N/A

**Additional context**
Getting the following in the AppDaemon log:
```
  x = df[available][df["import"] < max_export_price][df["forced"] < self.inverter.charger_power][
/homeassistant/appdaemon/apps/pv_opt/pvpy.py:807: UserWarning: Boolean Series key will be reindexed to match DataFrame index.
  x = df[available][df["import"] < max_export_price][df["forced"] < self.inverter.charger_power][
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:1684: FutureWarning: ChainedAssignmentError: behaviour will change in pandas 3.0!
You are setting values through chained assignment. Currently this works in certain cases, but when using Copy-on-Write (which will become the default behaviour in pandas 3.0) this will never work to update the original DataFrame or Series, because the intermediate object on which we are setting values will behave as a copy.
A typical example is when you are setting values in a column of a DataFrame, like:

df["col"][row_indexer] = value

Use `df.loc[row_indexer, "col"] = values` instead, to perform the assignment in a single step and ensure this keeps updating the original `df`.

See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy

  self.windows["hold_soc"].loc[
/homeassistant/appdaemon/apps/pv_opt/pv_opt.py:1684: SettingWithCopyWarning: 
A value is trying to be set on a copy of a slice from a DataFrame

See the caveats in the documentation: https://pandas.pydata.org/pandas-docs/stable/user_guide/indexing.html#returning-a-view-versus-a-copy
  self.windows["hold_soc"].loc[
```

fboundy, 2024-03-20:
Yes - these are mainly related to future changes in pandas. They are warnings for now but I do need to address them in due course.

fboundy, 2024-03-20:
Fixed with #163

PR #163 (2024-03-20): Fix pandas warnings
Fix #157
closed 2024-03-20 by fboundy