stevebuk1/pv_opt #158: Errors in share/logs/error.log

swests, 2024-03-20:
**Describe the bug**
Getting a bunch of errors, typically when I restart HA - App seems to get stuck on loading tarrifs.  Perhaps a timing issue with Octopus Energy? 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
No errors

**Screenshots**
NA

**Desktop (please complete the following information):**
NA

**Smartphone (please complete the following information):**
NA

**Additional context**
Below sample from start today:

```
09:36:45 WARNING pv_opt: ------------------------------------------------------------
09:36:45 WARNING pv_opt: Unexpected error in worker for App pv_opt:
09:36:45 WARNING pv_opt: Worker Ags: {'id': '7dd98a284c9644808f6420b29c85218d', 'name': 'pv_opt', 'objectid': '91a346ac959844aab910752294ad9dce', 'type': 'state', 'function': <bound method PVOpt.optimise_state_change of <pv_opt.PVOpt object at 0x7eff2ea50610>>, 'attribute': 'state', 'entity': 'number.pvopt_discharge_threshold_p', 'new_state': 'unknown', 'old_state': None, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'__thread_id': 'thread-0'}}
09:36:45 WARNING pv_opt: ------------------------------------------------------------
09:36:45 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1045, in worker
    funcref(
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1227, in optimise_state_change
    self.optimise()
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1286, in optimise
    if self._check_tariffs():
       ^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 734, in _check_tariffs
    for direction in self.contract.tariffs:
                     ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'tariffs'

09:36:45 WARNING pv_opt: ------------------------------------------------------------
09:36:45 WARNING pv_opt: ------------------------------------------------------------
09:36:45 WARNING pv_opt: Unexpected error in worker for App pv_opt:
09:36:45 WARNING pv_opt: Worker Ags: {}
09:36:45 WARNING pv_opt: ------------------------------------------------------------
09:36:45 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 358, in initialize
    self._load_pv_system_model()
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 461, in _load_pv_system_model
    max_dod=self.get_config("maximum_dod_percent") / 100,
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
TypeError: unsupported operand type(s) for /: 'str' and 'int'

[... 147 lines cut ...]
12:03:35 WARNING pv_opt: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app
    await utils.run_in_executor(self, init)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 362, in initialize
    self._compare_tariffs()
  File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2098, in _compare_tariffs
    opt = self.pv_system.optimised_force(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 607, in optimised_force
    max_slot = import_cost[import_cost == max_import_cost].index[0]
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^
  File "/usr/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 5389, in __getitem__
    return getitem(key)
           ^^^^^^^^^^^^
IndexError: index 0 is out of bounds for axis 0 with size 0

12:03:35 WARNING pv_opt: ------------------------------------------------------------

```

fboundy, 2024-03-20:
This is expected. Restarting HA doesn’t restart AppDaemon / PV_Opt so PV_Opt can crash due to unavailable entities.

That said I might be able to detect an HA restart and wait for it to settle down so worth looking into.

fboundy, 2024-03-20:
Perfect. @SzosszeNET  Feel free to do yourself and send a PR or I will get to it in due course.

fboundy, 2024-03-21:
The Uptime integration isn't required for this automation to work. I've slightly tweaked it to stop AD, wait a minute, check that key sensors are reading numbers and the start AD. This will be is addressed by #164 and will be in the next patch release.

    alias: Restart AppDaemon on HA Restart
    description: ""
    trigger:
      - event: start
        platform: homeassistant
    condition: []
    action:
      - service: hassio.addon_stop
        data:
          addon: a0d7b954_appdaemon
      - delay:
          hours: 0
          minutes: 1
          seconds: 0
          milliseconds: 0
      - wait_template: >
          {{(states('sensor.solcast_pv_forecast_forecast_today')| float(-1)>0) and
          (states('sensor.solis_battery_soc')| float(-1)>0)}}
        continue_on_timeout: true
      - service: hassio.addon_start
        data:
          addon: a0d7b954_appdaemon
    mode: single

closed 2024-03-21 by fboundy