stevebuk1/pv_opt #159: HASS log - recorder issue - Content length

swests, 2024-03-20:
**Describe the bug**
Warning in log from recorder - attribute length for some entiries greater than 16k and not being recorded.

**To Reproduce**
Steps to reproduce the behavior:
Check the HA log

**Expected behavior**
Content to be recorded (perhaps in segment) 

**Screenshots**
[image]

**Desktop (please complete the following information):**
HA latest and pv_opt latest

**Additional context**
Message in HA log:

```
Logger: homeassistant.components.recorder.db_schema
Source: components/recorder/db_schema.py:584
Integration: Recorder (documentation, issues)
First occurred: 09:51:46 (4 occurrences)
Last logged: 12:05:05

State attributes for sensor.pvopt_base_cost exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for sensor.pvopt_opt_cost exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
```

fboundy, 2024-03-20:
You probably don’t need this saved to the database so I don’t see it as a major issue. I’m not seeing this myself but that may be because I use MariaDB and InfluxDB rather than the default database.

swests, 2024-03-20:
I’m also using the same DB combo. 

Ok will ignore for now.

swests, 2024-03-25:
Just looking to understand this a little more.  Checking the attributes, sensor.pvopt_opt_cost_current is holding 'period start' dates from 2024-03-22.  Am I ok to clear out the soc, forced, import, export, grid, and solar data

[image]

fboundy, 2024-03-25:
Do you have entries under the `alt_tariffs` section in your `config.yaml` file? If so I suggest you comment out or delete them and then all the `sensor.pvopt_opt_cost_*` entities will be redundant (except for `sensor.pvopt_opt_cost`). If you restart HA and AppDaemon they should then disappear.

closed 2024-03-27 by fboundy