stevebuk1/pv_opt #167: Stuck in Backup/Reserve - "No Timed Charge/Discharge"

solarjones, 2024-03-22:
v3.11.0

This morning I noticed the SOC was down at 13% (inverter is set to 15% DOD, beyond that it normally forces a charge) and spotted PV_opt was stuck in Backup/Reserve - No Timed Charge/Discharge mode. 

[image]

Stopped AppDaemon, then via SolaX set to Self-Use mode and the inverter then started charging to bring the batteries up to the low DOD limit.

Nothing in error.log

[pv_opt.log.1.log](https://github.com/fboundy/pv_opt/files/14718632/pv_opt.log.1.log)
[pv_opt.log.2.log](https://github.com/fboundy/pv_opt/files/14718633/pv_opt.log.2.log)
[main.log](https://github.com/fboundy/pv_opt/files/14718630/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14718631/pv_opt.log)

Thanks

fboundy, 2024-03-22:
I think this is a known issue with 3.11.0 related to the Solax mode changes. Suggest you try 3.11.1 which should fix it.

fboundy, 2024-03-22:
Hmmm - I'm also seeing this now. Investigating further

fboundy, 2024-03-22:
A couple of potential things to check:

1. Please upgrade to 3.11.3 
2. If you have upgraded the Solax integration to 24.03.x please make sure that you have done as requested by the dev and fully removed it prior to installing the upgrade.

Let me know if this persists. As far as I can tell on my system it was an issue with `pymodbus` which I think was because I hadn't done (2) above.

lazy-pete, 2024-03-23:
Think I’m having similar issues - getting a RTC sync error which seems to be stalling for some reason. I disconnected soliscloud so only using Solax integration over Modbus. Uninstalling / reinstalling that now to see if it fixes things. 

[error.log](https://github.com/fboundy/pv_opt/files/14729668/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14729669/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14729670/pv_opt.log)
[image]
[image]

lazy-pete, 2024-03-23:
Still getting the same sync RTc error after reinstalling Solax Modbus. 

Plus my default settings still not picked up after a reboot of HA and Appdaemon. 

Plus status says charging, but isn’t actually charging. 

[error.log](https://github.com/fboundy/pv_opt/files/14729882/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14729883/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14729884/pv_opt.log)
[image]
[image]

fboundy, 2024-03-23:
@lazy-pete - Sorry but I'm a bit confused by a number of things:

1. When are you getting the RTC sync error? There is a known bug in the Solax integration that @SzosszeNET has reported. PV_Opt doesn't use RTC sync so whilst it's annoying it won't impact performance.

2. Can you please explain in more detail what you mean by "my default settings are not being picked up"? The underlying defaults are set withing `pv_opy.py` which you shouldn't change. You can change settings two ways - using the entries in `config.yaml` or via the dashboard. Settings which are changed via the dashboard will persist on re-boot unless `overwrite_ha_on_restart` is set to `true` in which cases the values set in `config.yaml` will be used.  In the screenshots above you have set `forced_discharge` to `false` in the `config.yaml` but `overwrite_ha_on_restart` is set to `false` so when pv_opt restarts it picks up the previous setting from HA which is `on`.  If you are wanting to use what is in the  `config.yaml` then set  `overwrite_ha_on_restart` is set to `true`. At no point will the coded defaults within `pv_opt.py` be changed.

3. Looking at you `pv_opt.log` I notice a few things:

- You have defined you octopus_account/api_key and the tariff codes but it appears that you have the Octopus Energy integration installed so you should need neither and can just use `octopus_auto: true`.
- Your octopus account code should start with `A-6`  not `A6`
- I also suspect that your Agile tariff code is out of date
- The entity you have supplied under `id_consumption_today` does appear to exists

4. AppDaemon is trying to set the charge current to 54.7A but this is causing an error when it calls HA's set_value service. I had a similar problem before I completely removed the Solax integration are re-installed it but you say you have already done this so I'm at a bit of a loss. What happens it you try to set the timed charge current directly using the slider for entity `number.solis_timed_charge_current`

lazy-pete, 2024-03-23:
Thanks - comments below …

> @lazy-pete <https://github.com/lazy-pete> - Sorry but I'm a bit confused
> by a number of things:
>
>    1.
>
>    When are you getting the RTC sync error? There is a known bug in the
>    Solax integration [...]
That’s good to know - error seen if I press the sync button, which I do
when I see a huge difference between the two times (like the screenshot).
I’m not sure why this occurs - they seem to drift over time slowly then
sometimes the inverter time seems way off, like it stalled or something.

>
>    1.
>
>    2.
>
>    Can you please explain in more detail what you mean by "my default
>    settings are not being picked up"? The underlying defaults are set withing
>    pv_opy.py which you shouldn't change. You can change settings two ways
>    - using the entries in config.yaml or via the dashboard. Settings
>    which are changed via the dashboard will persist on re-boot unless
>    overwrite_ha_on_restart is set to true in which cases the values set
>    in config.yaml will be used. In the screenshots above you have set
>    forced_discharge to false in the config.yaml but
>    overwrite_ha_on_restart is set to false so when pv_opt restarts it
>    picks up the previous setting from HA which is on. If you are wanting
>    to use what is in the config.yaml then set overwrite_ha_on_restart is
>    set to true. At no point will the coded defaults within pv_opt.py be
>    changed.
I changed that pv_opt.py  and config.yaml after a previous thread exchange.
I am perhaps misunderstanding - if I change the config.yaml and have
overwrite as true should I expect to see that mirrored in the dashboard on
start-up ?

What’s confusing me is that I often see the dashboard settings reflecting
neither what’s in the yaml or the previous manual dashboard settings. I
can’t quite work out what triggers a change, but I think mostly after a
restart of some sort.

>
>    1.
>
>    2.
>
>    Looking at you pv_opt.log I notice a few things:
>
>
>    - You have defined you octopus_account/api_key and the tariff codes
>    but it appears that you have the Octopus Energy integration installed so
>    you should need neither and can just use octopus_auto: true.
>    - Your octopus account code should start with A-6 not A6
>    - I also suspect that your Agile tariff code is out of date
>    - The entity you have supplied under id_consumption_today does appear
>    to exists
Thanks - that was from some fiddling around to try and get it working
initially. I’ll adjust again.

>
>    -
>
>
>    1. AppDaemon is trying to set the charge current to 54.7A but this is
>    causing an error when it calls HA's set_value service. I had a similar
>    problem before I completely removed the Solax integration are re-installed
>    it but you say you have already done this so I'm at a bit of a loss. What
>    happens it you try to set the timed charge current directly using the
>    slider for entity number.solis_timed_charge_current
I had updated without removing first. Last night I deleted the integration
(had both a solis and Solax device from trying a previous effort) and
started again.  I’ll try manually adjusting the slider.

Currently have a dashboard error of
ERROR: HA returned unknown for state of number.solis_battery_minimum_soc

and the pvopt sensors unavailable

fboundy, 2024-03-23:
I think you’ve got some issues with the Solax integration. I found mine had locked up last night and had to reload it. Maybe check its all working as you expect manually and then restart PV Opt

fboundy, 2024-03-24:
Mine locked up again this morning. I'm going to raise an issue on GitHub and for the time being I'm going back to using the Core Modbus integration

solarjones, 2024-03-24:
Is there a way of reverting to the last good version of the SolaxModbus code base? HACs doesn't show the pre 24.03.x.

fboundy, 2024-03-24:
You'd have to do it manually by selecting the appropriate Tag directly from GitHub

fboundy, 2024-03-24:
If you're not wedded to the Solax integration just use this instead: https://github.com/fboundy/ha_solis_modbus

I have both set up and I call this one `solmod` so I don't get duplicate entities. In PV Opt you switch to the `SOLIS_CORE_MODBUS` inverter type in the `config.yaml`.

fboundy, 2024-03-24:
Apparently you can show more than 5 previous versions: https://hacs.xyz/docs/configuration/options/

fboundy, 2024-03-25:
I think some of the Solax issues may be resolved by HA Core v2024.3.2 onwards which bumps `pymodbus` to 3.6.6

closed 2024-03-26 by fboundy