stevebuk1/pv_opt #177: Entity not picking up after restart number.solis_battery_minimum_soc

lazy-pete, 2024-03-25:
**Describe the bug**
A clear and concise description of what the bug is.

updated Pv_opt to most recent version via HACS. 
still noting that following reboot I get a dashboard error regarding failure to detect number.solis_battery_minimum_soc. 

This entity is present and visible in entities with a number present via slider. 

Also still seeing different setting in dashboard from config.yaml despite ‘overwrite on restart’ set to True. 

RTC sync error also persists. 

**To Reproduce**
Steps to reproduce the behavior:

Restart HA 

**Expected behavior**
Restart retains config.yaml settings in dashboard, and allows pv_opt to run by detecting all required entities. 

**Screenshots**

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

fboundy, 2024-03-25:
Ah - that release was a pre-Release. I tagged it wrong. Please revert to 3.12.1

What versions of the Solax integration and Home Assistant Core are you running?

It also looks to me like the Solax integration hadn't properly started when you restarted AppDaemon. All the `solis` entites are showing 0 or `unknown` at 18:29:11.

lazy-pete, 2024-03-25:
Ok - not quite sure I know how to revert ! Shall Google. 

Most recent versions of HAOSS / core / Solax etc (no updates available. 

Will await Solax boot and try again.

lazy-pete, 2024-03-25:
Actually just after that I found there was another Solax update with your rtc_sync fix. Installed now and seems to have sorted that error. 

Also re downloaded 3.12.1 and things seemed fine after I restarted AppDaemon. 

[error.log](https://github.com/fboundy/pv_opt/files/14748515/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14748517/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14748518/pv_opt.log)

lazy-pete, 2024-03-25:
Sent a couple of beers to the MRT - excellent cause. Not needed them yet myself, but you never know when it’s going to go wrong back there !

closed 2024-03-26 by fboundy