stevebuk1/pv_opt #181: IOG tariff goes stale, meaning optimed plan same as intial plan

stevebuk1, 2024-03-26:
My import tariff is Octopus Intelligent Go, with no export tariff (as my install is DIY with no MCS)

I find that every few days, the IOG tariff goes "stale", in that instead of the cheap rate lasting for 6 hours and then reverting to peak, it stays at cheap rate for at least the next 24 hours. 

Tariffs are loaded via BottlecapDave integration, however  setting import tariff directly in config.yaml("octopus_import_tariff_code: E-1R-INTELLI-BB-VAR-23-03-01-N") would emulate the problem. 

Screenshots below and pv_opt.log attached. 

[image]
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14764321/pv_opt.log)

fboundy, 2024-03-27:
Hmmm. I'll add some more logging to the EV- charging pre-release to try to track this down. I can see it's flagging the discrepancy with Bottlecap Dave's tariffs so when that happens I'll get it to dump some more info.

fboundy, 2024-03-27:
Is there anything in `error.log`?

stevebuk1, 2024-03-27:
> Is there anything in `error.log`?

There wasn't when I posted, but I've recently moved Pv_opt to a new Pi so the issue may not have been caught when it actually occurred. 

I have just checked it this morning next day and I found a few entries at 23.35 that might be related: 
[error.log](https://github.com/fboundy/pv_opt/files/14771996/error.log)

Note : I can clear the issue by stopping AppDaemon and then restarting.

fboundy, 2024-03-27:
Could you upload a snip of `pv_opt.log` for 23:35 too so I can see the context of the error?

stevebuk1, 2024-03-27:
> Could you upload a snip of `pv_opt.log` for 23:35 too so I can see the context of the error?

Attached, from 11pm to Midnight (I think I've got the right night):
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14772526/pv_opt.log)

fboundy, 2024-03-27:
Nothing that gives any clues there. Let's see what the latest release logs.

stevebuk1, 2024-04-05:
'3.14.0-beta-6' used.

Logs are here:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14889954/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14889983/error.log)

Current image for reference (same as in first entry above):
[image]

fboundy, 2024-04-27:
Closing. Thanks for the PR

closed 2024-04-27 by fboundy