stevebuk1/pv_opt #182: Solcast forecast optimisation

solarjones, 2024-03-26:
3.12.2

Very good forecast today and tomorrow, however the optimiser wants to change to 100% over night. I've switched to a fixed load to eradicate any historic loading, and tried various Solcast confidence levels. Always showing 100% by 5am. Today for example, I generated 22kWh and exported 14 kWh so would think the battery for tomorrow shouldn't be charging from grid to anywhere near 100%. Am I missing something or have set it up incorrectly?

[image]

[main.log](https://github.com/fboundy/pv_opt/files/14764779/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14764781/pv_opt.log)

Thanks!

fboundy, 2024-03-26:
Simply put you are buying electric at 10p overnight and selling it at 15p so the optimiser will always want to buy overnight. It’s fundamentally how the optimiser works.

solarjones, 2024-03-26:
Ah obvious when you put it like! :)  I'd made (incorrectly) the assumption the optimiser would only charge the batteries to a sufficient level to meet projected/historic demand.  Does this mean it will always charge to 100% regardless of demand/projection where the import is less than the export?

And the optimise discharge option is if on the agile outgoing tariff so it would pick the highest paying export slots?

Sorry for the silly questions.

fboundy, 2024-03-26:
Pretty much. I’m writing an explanation that will get released when I can find time to finish it!

solarjones, 2024-03-26:
Ok, perhaps a feature request (maybe it's possible now with the parameters available) would be an option to run in self sufficient mode/override. Eg. sunny months, long sunny days where I may want to utilise all the generation (eddi + air source heat pumped pool), batteries get topped up from PV to 100%, etc. Rather than charging overnight to 100% at any cost since there may not be any export. In this mode the optimiser is only be looking at getting the batteries charged over night for the minimum daily consumption figure mixing with solcast.  

Last summer, before finding your Optimiser, I'd had a threshold of solcast to trigger the heat pump on sunny days where i knew I'd comfortably have the batteries charge and plenty of excess. I know how much the heat pump consumes and would have it on a timer to trigger and cut off if the SOC dropped as I wanted to retain battery to run over peak.  That was before switching to Agile + PV_opt.

Perhaps totally out of scope or intended us. Either way, I'm very grateful for your work  as it's saving plenty over the Eco7 tariff I was on now it's controlling the inverter with Agile.

fboundy, 2024-03-27:
Ok - I get it. Easiest way would be to have an "Export multiplier" (0 - 100%)

fboundy, 2024-03-27:
Not necessarily. Discharging means forced discharging but in this case you are forced charging to export solar generation.

fboundy, 2024-04-02:
In `3.14.0` I am adding an `include export` switch. If you turn this off export won't be included in the optimisation.

solarjones, 2024-04-04:
Long overdue, I've just donated to your charity, very well done for volunteering!

fboundy, 2024-04-04:
Many thanks

solarjones, 2024-04-04:
I've installed 3.14.4 and switched Include Export = OFF
Still seems to want to charge to 100% even when setting the consumption to a very low amount. Is there something I've missed? 

[image]
[main.log](https://github.com/fboundy/pv_opt/files/14876162/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14876163/pv_opt.log)

Thanks.

fboundy, 2024-04-04:
Yep - that's not working right. Will fix and get a new beta out

fboundy, 2024-04-04:
Should be fixed in `3.14.0-beta-5`

solarjones, 2024-04-04:
3.14.0-5
Errors:
[error.log](https://github.com/fboundy/pv_opt/files/14876859/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/14876860/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14876861/pv_opt.log)

fboundy, 2024-04-04:
Should be fixed in `3.14.0-beta-6`

solarjones, 2024-04-04:
Looking good! Thanks will see how much sun we get tomorrow now :)

closed 2024-04-05 by fboundy

solarjones, 2024-04-05:
hmmm, I may have spoken too soon. Todays plan was spot on, but just looked at tomorrow...

3.14.0-6

Tomorrows tariff slots look very good, being paid for a few too (plenty of sun + windy day).  PV_Opt has selected cost slots instead and wants to charge to 100%.

I've got the Include Export = off.  Also assume that any export during negative slots results on 0p export rate anyway.

[image]

[main.log](https://github.com/fboundy/pv_opt/files/14890997/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14890999/pv_opt.log)

fboundy, 2024-04-05:
Set the plunge threshold to 10p and it will make more sense. Some people want to charge as soon as the cost is negative because they change their usage.

fboundy, 2024-04-05:
Sorry - that should have been the other way! Set it to -5p.

solarjones, 2024-04-06:
I've tried -5, -2.5, -1 and it doesn't seem to align to the plunge prices.

-5 & -2.5 show no optimised slots for charging, and -1  has these:

[image]
[image]

[main.log](https://github.com/fboundy/pv_opt/files/14894302/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14894304/pv_opt.log)

Thanks.

fboundy, 2024-04-08:
Difficult to diagnose completely but my guess is that you had so much sun that there was little that could be done with the battery.  With the plunge threshold set to 1p it was using the 2:30 and 6:00 slots but by 13:00 your battery was full so nothing that it could do. If you had optimised for export or enabled discharging it would have dumped the battery and re-filled it which is why the optimisation summary shows this:
    
    18:11:03     INFO:   Base cost:                               -152.4p
    18:11:03     INFO:   Optimised cost (Optimised Charging):     -152.4p <=== Current Setup
    18:11:03     INFO:   Optimised cost (Optimised PV Export):    -300.1p
    18:11:03     INFO:   Optimised cost (Forced Discharge):       -425.4p