stevebuk1/pv_opt #188: Solax X1 Inverter supprt

fboundy, 2024-03-28:
Continuation of #149 @punkymuzzle

fboundy, 2024-03-28:
The critical thing will be to see whether the inverter kicks in at 22:30

punkymuzzle, 2024-03-28:
Well I've left it for two charge cycles, and nothing has happened.
I've checked the error log and pv_opt log and it has the same timestamp as earlier, so not worth uploading.....

fboundy, 2024-03-29:
Sorry I think I missed the logs. Did you upload them?

punkymuzzle, 2024-03-29:
I didn't upload them as they had the same date / time stamp s the previous upload, so there was no additional information in them.
I'll upload them again anyway:
[error.log](https://github.com/fboundy/pv_opt/files/14802360/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14802361/pv_opt.log)

I've also noticed that the charging times aren't recalculating and are still showing yesterdays times to charge:
[image]
.
.
To me it sort of looks like it's just stopped and needs something to kickoff the process again, so I'll reboot HA and then upload the logs again.

punkymuzzle, 2024-03-29:
Rebooted and uploaded the logs:
[error.log](https://github.com/fboundy/pv_opt/files/14802624/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14802625/pv_opt.log)
.
.
Also noted that the dashboard is showing errors again, so assume it's working stuff out before it displays it:
[image]

punkymuzzle, 2024-03-29:
OK, installed Beta 11 and logs attached:
[error.log](https://github.com/fboundy/pv_opt/files/14803283/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14803284/pv_opt.log)

fboundy, 2024-03-29:
Has your `config.yam` been over-written? It shouldn't be but that looks like its trying to load solis settings

punkymuzzle, 2024-03-29:
No, it's still intact

fboundy, 2024-03-29:
Can you upload the latest `pv_opt,.log` then as the entries for 11:18 don't make sense to me

fboundy, 2024-03-29:
Maybe re-start AppDaemon as it looks like it might not be loading `config.yaml` at all

punkymuzzle, 2024-03-29:
Restarted AppDaemon and then re-saved config.yaml so it re-ran.
Logs as follows:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14803742/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14803743/error.log)

punkymuzzle, 2024-03-29:
Dashboard has come back to life
[image]

punkymuzzle, 2024-03-29:
Latest logs if you need them:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14804230/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14804231/error.log)

fboundy, 2024-03-29:
Looks OK in read-only. What if you turn read-only off? You could also set `debug: false` in the `config.yaml` for now which makes the log a lot more readable.

fboundy, 2024-03-29:
> OK will do it and see what happens.
> Next scheduled charge isn't until 1:00 tomorrow morning. So will give you
> an update when it's done

Would be good to see what it’s doing as soon as it’s off read only. Should be reporting inverter status even though there’s nothing to do till then

punkymuzzle, 2024-03-29:
Understood - here's the logs
[error.log](https://github.com/fboundy/pv_opt/files/14804600/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14804601/pv_opt.log)

fboundy, 2024-03-29:
As I suspected a couple of bugs once we start to use the inverter controller module. `4.0.0-solx-x1-beta-12` fixes these. If you could do the same again please. We should get to the point where it checks the inverter status and says that there is nothing to do.

fboundy, 2024-03-29:
If you have this in your `config.yaml` can you comment it out (though it's not fatal) then swicth Read-Only off
    
      id_daily_solar: sensor.{device_name}_power_generation_today
      alt_tariffs:
        - name: Agile_Fix
          octopus_import_tariff_code: E-1R-AGILE-23-12-06-G
          octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-G
    
        - name: Eco7_Fix
          octopus_import_tariff_code: E-2R-VAR-22-11-01-G
          octopus_export_tariff_code: E-1R-OUTGOING-FIX-12M-19-05-13-G
        - name: Flux
          octopus_import_tariff_code: E-1R-FLUX-IMPORT-23-02-14-G
          octopus_export_tariff_code: E-1R-FLUX-EXPORT-23-02-14-G

punkymuzzle, 2024-03-29:
I don't have those in my config.yaml.
The only things I have enabled regarding tariffs are:

`octopus_account: x-xxxxxxx
  octopus_api_key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

  octopus_import_tariff_code: x-xx-xxxxx-xxxx-xx-xx-xx-x
  octopus_export_tariff_code: x-xx-xxxxxxxx-xxx-xxx-xx-xx-xx-x`
(Removed the detail from this)

The dashboard is using the above to lookup my tariffs?

Do you want me to disable these as well?

Logs:
,
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14807133/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14807135/error.log)

fboundy, 2024-03-29:
No. They can stay

punkymuzzle, 2024-03-29:
OK ta

fboundy, 2024-03-29:
Though you only need the 1st pair or the 2nd pair or octopus_auto: true

fboundy, 2024-03-29:
Looking at the log:

Octopus autodetection is working OK. This is the default so you can actually comment out all the octopus account into in `config.yaml` but not critical.

We are nearly there! I think we can get around the latest error by adding this to the `config.yaml` (2 leading spaces):

     id_backup_grid_charge: select.solax_allow_grid_charge

punkymuzzle, 2024-03-29:
Great news.
OK, added the line above, saved and copied logs:
[error.log](https://github.com/fboundy/pv_opt/files/14807673/error.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14807674/pv_opt.log)

fboundy, 2024-03-29:
Still not quite there... 

`4.0.0-solax-x1-beta-13` fixes this one.

punkymuzzle, 2024-03-29:
Done
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14807927/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14807928/error.log)

fboundy, 2024-03-29:
Progress- we now end with the inverter status. It’s not quite as expected so I suspect the charge session won’t run. Will sort that tomorrow

punkymuzzle, 2024-03-29:
Awesome, thank you very much

fboundy, 2024-03-30:
Fixed in `4.0.0-solax-x1-beta-14`

It may not want to do any charging at present with it being quite sunny. So if there are no errors then leave it running for a while until it does

punkymuzzle, 2024-03-30:
I think it was scheduled to charge overnight, so there may be something in these logs (Also rebooted earlier so there may be some other stuff in there as well)
I'll upload the beta 14 logs after a while to let them collect stuff for you.
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14811476/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14811477/error.log)

punkymuzzle, 2024-03-30:
Here's the beta 14 logs:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14811490/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14811491/error.log)

Noticed this:
[image]

This is looking at my manually entered charging times. I default to 14:00 - 16:00 in the event of little / no solar and change it to backup mode; then change it to self use mode at 16:00 so it uses the battery during peak.
hth

fboundy, 2024-03-30:
> We are nearly there! I think we can get around the latest error by adding this to the `config.yaml` (2 leading spaces):
> 
> ```
> id_backup_grid_charge: select.solax_allow_grid_charge        
> ```

The above can now be removed.

I've made a couple of minor tweaks in `4.0.0-solax-x1-beta-15`. It should now over-write your manual charge times and then charge tonight. Finger crossed the clocks changing doesn't mess anything up...

fboundy, 2024-03-30:
Just had to patch to `4.0.0-solax-beta-16` to fix a bug associated with the clock change!

punkymuzzle, 2024-03-30:
OK great thanks.
Just installed beta 15 so will update again.

punkymuzzle, 2024-03-30:
Just noticed the charge times.
They all occur overnight. They don't seem to be taking any solar PV into
account?
My battery didn't charge at all last night but is now fully charged from
PV, that won't happen tomorrow if this schedule is followed?

fboundy, 2024-03-30:
Yes but you will buy electricity at say 10p overniyamd sell the solar electricity at 25p during the day

punkymuzzle, 2024-03-30:
But not sure if my inverter can discharge to the grid?
Have you managed to get that to work?

fboundy, 2024-03-30:
If you fill your battery overnight and then there is solar it will discharge to the grid when in Self-Use mode. That is what it is showing.

punkymuzzle, 2024-03-30:
I would have thought that was the excess PV, not the charge in the battery?
It will be interesting to see how it works.

punkymuzzle, 2024-03-30:
Looking at the charging plan and it's not selecting the cheapest?
Plan and rates attached.
Is there any way to force a recalculation?

fboundy, 2024-03-30:
You really need to look at these plots together to make sense of the plan:

[image]

punkymuzzle, 2024-03-30:
Yeah understand, but I don't understand why it would charge at a time when
theres a cheaper slot literally next door to it.

fboundy, 2024-03-30:
I see what you mean. Can you upload the log?

punkymuzzle, 2024-03-30:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14813320/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14813321/error.log)

punkymuzzle, 2024-03-31:
Decided to reboot HA and pv_opt recalculated the slots.
But something's happening to my charts.  the pv_opt chart say's the battery is charging:
[image]

My own dashboard says not (but it's using 1.6kW)
[image]
.
My smart meter says it's only drawing a few hundred watts 
And my SolaX app is showing 0 Watts anywhere....
[image]

fboundy, 2024-03-31:
There are errors yesterday afternoon which were related to an issue with Agile and the clocks going forward. I fixed this in the current production version but clearly that patch didn't make it into this version. It was a one-off related to the change rather than an ongoing problem so it should be OK today. It probably expains why the timings looked off yesterday.

What you have called the `PV Opt chart` isn't reporting anything from PV_Opt. All these sensors come direct from the solax integration so that is the source of any inconsistencies. I have noticed that your system seems to take a long time to stabilise after a restart: maybe something else is causing things to hang up. Worth having a look at the Log under Settings to see if there are any errors there.

punkymuzzle, 2024-03-31:
Thanks very much for the help.
Spent some time last night and this morning trying to fix the problem.
Several reboots and resets later an Di think I have it back up and running
(although my SolaX Cloud app is showing zeros across the board, bu I'll fix
that later).
Still two issues I think:
1. The battery didn't appear to charge last night so not sure if the
controls are there yet? I did put t in read only mode whilst I tried to fix
my issue but it's back in normal mode now.
2. The charging times don't seem to recalculate unless I reboot the
whole thing. Should there be some sort of trigger to get it to recalc the
times?
It's showing the past times it was due to charge, plus it's set to try and
charge at 16:00 today, which is exactly when I don't want it to as
that'speak time for me (26p as opposed to 11p just before)
So I'll probably leave it until we get the calculations for cheap times
(plus solar prediction) sort out before I let it charge, else it will cost
me a lot extra.
Thanks

fboundy, 2024-03-31:
The bug I though was temporary turns out to be persistent so please upgrade to `4.0.0-solax-x1-beta-17`

punkymuzzle, 2024-03-31:
Awesome thank you very much.
Installed and looking good from a charging slot point of view.
[image]
.
.
However, I notice that a discharge is due to happen from 15:30 - 16:30, leaving my battery at 18%.  I don't want this to happen as 16:00 - 19:00 is peak time (between 26 and 30p / kWh).
[image]

.
So I want to use my battery to power my house during this time.  As my export rate is a constant 15p / kWh, it doesn't make sense to discharge everything and then run the house on peak cost electricity.
No idea if this is possible, but is there a way for me to exclude any forced discharge to empty the battery prior to this?  I want to ringfence my full battery to supply the house during this time (which I manually do now).  Thanks in advance.

I'll take it out of read only and see what happens.
In the meantime, here's the initial logs:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14817790/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14817791/error.log)
.
.
One other thing to mention, when I manually program a charge I always set charging slot 1 and 2 to the same time (I was told to do this, can't remember by whom).
Not sure if this will make any difference, but if charging does not happen tonight it might be worth bearing this in mind.

Thanks very much for this, will let you know how it goes.

fboundy, 2024-04-01:
It's still got a couple of errors:

- one typo in the inverter controller is stopping it actually controlling anything
- since you don't have a forced discharge mode this should be automatically disabled (which deals with your concern about peak discharge). 

Will try to get these 2 fixed

punkymuzzle, 2024-04-01:
Awesome, thank you

fboundy, 2024-04-01:
OK try `4.0.0-solax-x1-beta-18` it *should* disable forced charging. If it doesn't just toggle it off on the dashboard or set `forced_discharge: false` in `config.yaml`

punkymuzzle, 2024-04-01:
OK, thanks
installed, added the line in config.yaml.
Errors / warnings still in the logs?
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14823277/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14823278/error.log)

punkymuzzle, 2024-04-01:
Still doesn't look like it's calculating the correct times to charge.
My Octopus rates for tonight / tomorrow:
[image]

.
.
The charging plan from the dashboard:
[image]
.
So not sure why it isn't picking up the cheapest slots to charge?

punkymuzzle, 2024-04-01:
It's looking better:
[image]

punkymuzzle, 2024-04-01:
Still choosing the wrong charging slots:

fboundy, 2024-04-02:
First thing is to check whether the optimiser is running fully. Can I see the logs please?

punkymuzzle, 2024-04-02:
Sure
[pv_opt.log](https://github.com/fboundy/pv_opt/files/14833660/pv_opt.log)
[error.log](https://github.com/fboundy/pv_opt/files/14833662/error.log)

fboundy, 2024-04-02:
Still has the error I thought I'd fixed yetserday

punkymuzzle, 2024-04-02:
Plus, the charging plan hasn`t refreshed:
[image]

[102 more replies]

closed 2024-04-27 by fboundy