stevebuk1/pv_opt #197: PV Opt discharging battery where it will cause expensive import

jacekowski, 2024-04-04:
**Describe the bug**
I'm running slightly modified version (i have added sunsynk inverter support - but i doubt this causing the issue).

**To Reproduce**
Run addon with my configuration (consumption history disabled, 35kWh/day) - everything is visible in the logs anyway.

**Expected behavior**
No discharge on day like today. Or tomorrow/

**Screenshots**
[image]

As you can see, the plan it generates is to discharge (at 15p/kWh) and then it would cause me to import at higher rate (and that is ignoring any efficiency losses). Similiar thing tomorrow.

[pv_opt.txt](https://github.com/fboundy/pv_opt/files/14873585/pv_opt.txt)

fboundy, 2024-04-04:
Are you sure you would be importing at a higher rate? Agile prices are pretty low today and tomorrow. It's basically saying buy overnight at <10p and sell at 15p which is what's driving logic. 

One possible issue is that there is "value" in having a final SOC of 30% vs 10% which it is ignoring because it knows nothing about Saturday yet. At 11:00 tomorrow when it get's the day-ahead forecast it will update and may not discharge as much. You can always turn off discharging or (in 3.14) give zero value to exporting.

fboundy, 2024-04-04:
PS - are you able to share your sunsynk mods?

jacekowski, 2024-04-04:
It does need some cleanup and my implementation relies on node-red to do final control, but it is here https://github.com/jacekowski/pv_opt/commit/de90248631a82adce9d505a189674e52b3177b5c

Because sunsynk inverter does not have charge/discharge time slots like other inverters, but instead just has a schedule, i'm using node red to check which slot is the current one, and adjusting settings for that slot at the right time.

[image]

fboundy, 2024-04-04:
> It does need some cleanup and my implementation relies on node-red to do final control, but it is here [jacekowski@de90248](https://github.com/jacekowski/pv_opt/commit/de90248631a82adce9d505a189674e52b3177b5c)
> 
> Because sunsynk inverter does not have charge/discharge time slots like other inverters, but instead just has a schedule, i'm using node red to check which slot is the current one, and adjusting settings for that slot at the right time.
> 

Thanks - I'm afraid Node Red is a mystery to me so I will leave you to it!. I am adding support via the Solarsynk Add-on: #129

jacekowski, 2024-04-04:
Unfortunately, solarsynk relies on sunsynk cloud that doesn't work very frequently so i don't consider it a viable option. I've done parts of it node red because i was developing my own solution in node red and then i've found this addon.

I'm going to migrate the whole thing to python in near future and then do a PR.

fboundy, 2024-04-04:
I presume you have a hard-wired connection then?

jacekowski, 2024-04-04:
Yes, I'm allergic to cloud, especially if it is not a reliable one.

fboundy, 2024-04-04:
Same here. At some point I will add cloud support for Solis inverters but controlling via the cloud is barely supported

jacekowski, 2024-04-05:
What is the intention of this line 

https://github.com/fboundy/pv_opt/blob/483bee34b063ffaaad67d0bffac3765f85a21472/apps/pv_opt/pv_opt.py#L1545

rather than using self.charge_start_datetime further down?

fboundy, 2024-04-06:
A quick answer without properly digging:

If the inverter is already active and we want it to stay active we don’t need to set the start time.

In general I am trying to minimise the number of writes to the inverter unless strictly needed. On the Solis the inverter registers are allegedly EEPROMs with a limited number of total writes.

jacekowski, 2024-04-06:
I'm on 3.14.0, and plan for today/tomorrow is definitely not optimal. Charging started today at 5:30 at 0p/kWh, rather than waiting for later in the day to charge at negative rates.

Current schedule discharges the battery until 18:30, which will then cause import from the grid at 20p/kWh (delaying export by 30 minutes would be the solution here), then full charge is scheduled for 23:30 rather than later at negative rates.

[image]

fboundy, 2024-04-06:
Try dropping the plunge threshold to -5p

closed 2024-04-27 by fboundy