stevebuk1/pv_opt #198: PV Opt generating negative discharge windows that don't match end time shown in UI

ThomasPrior, 2024-04-04:
**Describe the bug**
Since the DST change I've seen a few odd charging windows.

pv_opt log output shows:

17:08:08 INFO: end : 04-Apr 17:00 BST
--
17:08:08 INFO: start : 04-Apr 17:07 BST

This is displayed in HomeAssistant as finishing an hour later:

[image]

The inverter receives the end time of 17:00 instead of the expected 18:00, making the discharge window -8 minutes.

[image]

Time on inverter seems to be set correctly.

[image]

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14874079/pv_opt.log)

ThomasPrior, 2024-04-04:
Adding second log with the same behaviour. 

As shown in HomeAssistant:
Start 17:20
End: 18:00

As shown in logs and on inverter:
Start: 17:20
End: 17:00

[pv_opt.log](https://github.com/fboundy/pv_opt/files/14874148/pv_opt.log)

jacekowski, 2024-04-04:
It could be the same thing i'm seeing in #197

ThomasPrior, 2024-04-04:
Interesting, I saw what I thought to be a very poor plan for today (discharge everything, all day) and that's what led me to finding this. 

Also on fixed 15p/kWh of export.

fboundy, 2024-04-04:
> It could be the same thing I'm seeing in #197

Not sure these are the same. There is a "time zone" issue which I think I've fixed but I think at least part of what is happening in #197 is that there is a cost benefit to forced discharging but it's not intuitive.

ThomasPrior, 2024-04-04:
Thank you again for your prompt attention on this and previous issues.

closed 2024-04-04 by ThomasPrior