stevebuk1/pv_opt #200: Referenced entities switch.pvopt_charge_active are missing or not currently available

swests, 2024-04-06:
**Describe the bug**
Getting the following show up multiple 1000's of times in the system log

```
Logger: homeassistant.helpers.service
Source: helpers/service.py:251
First occurred: 2 April 2024 at 09:55:14 (1620 occurrences)
Last logged: 10:41:01

Referenced entities switch.pvopt_charge_active are missing or not currently available
Referenced entities switch.pvopt_discharge_active are missing or not currently available
Referenced entities switch.pvopt_hold_soc_active are missing or not currently available
```

[image]

**Additional context**
On latest version of PV_OPT.  HA 2024.3.3
Is there a way to recreate these? Or are they not needed?

fboundy, 2024-04-06:
Noted. Should get this fixed in 3.14.0

fboundy, 2024-04-08:
Should now be sorted in the 3.14.0. release. Let me know if it's still a problem

closed 2024-04-27 by fboundy