stevebuk1/pv_opt #201: Solis inverter S5-EH1P6K invalid charge current

mergwyn, 2024-04-07:
Using `3.14.0-beta-6` I noticed this morning that pv_opt was trying to write a value of > 100A to the my Solis S5-EH1P6K inverter. AFAIK the inverter has a maximum charge/discharge current of 100A.

Reducing `charger_power_watts/inverter_power_watts` allowed me to workaround the issue:
```
07:39:31     INFO: Power 5000 = 107.3A at 46.6V
07:39:33     INFO: Failed to write 107.3 to inverter
--
07:39:58     INFO: Power 3900 = 83.7A at 46.6V
07:39:58     INFO: Current 83.7 written to inverter
```

I've not changed the `charger_power_watts/inverter_power_watts` since I installed pv_opt.

Weirdly, all is now working correctly with the original setting:
```
09:50:27     INFO: Power 5000 = 96.0A at 52.1V
09:50:27     INFO: Inverter already at correct current
```

Here are the logs (02:10 is an example of where this happened):
[pv_opt.1.log](https://github.com/fboundy/pv_opt/files/14896712/pv_opt.1.log)
[error.1.log](https://github.com/fboundy/pv_opt/files/14896713/error.1.log)

ThomasPrior, 2024-04-07:
I've been seeing the same in #128

fboundy, 2024-04-07:
Higher voltage means lower current for the same power

fboundy, 2024-04-08:
I'm not convinced that the solis accurately reports the current limit (mine shows 50A but is happy for me to exceed it). I can add it as optional input in the config.yaml.

fboundy, 2024-04-08:
In `3.14.0` I've added `battery_current_limit_amps` to the config. This can ether be read from a sensor or explicitly set with a value in the `config.yaml` or dashboard.

spetrie13, 2024-04-26:
Battery current limit working for me with update

closed 2024-04-27 by fboundy