stevebuk1/pv_opt #203: PV_OPT randomly setting discharging in charge period

spetrie13, 2024-04-08:
I have noted over the last few days that cheap rate charging (Octopus Go 00:30 - 04:30) that is sets correct charge rates and period, last night around 23:30 SOC was around 81%, schedule was 0030 to 0430 @ 300W but during charging it will randomly set very large discharge slots that basically fully discharge my battery.  
Not been able to capture a screenshot last night but have attached logs. I noted this behaviour two nights ago when I was up at 1200 and noted the very large export slots during period it should have been charging.  

I also noted that the inverter discharge settings were still set this morning, so pv_opt did not reset them. Discharge continued during cheap rate and peak rate until battery discharged. 
I even had Optimise discharging disable last night, but it still discharged.
This has been working fine for sometime, but over last few days had issues.

**Expected behavior**
I expected cheap rate charging and not to have large discharge slots set during this period.

**Screenshots**
[image]

[logs.zip](https://github.com/fboundy/pv_opt/files/14902832/logs.zip)

fboundy, 2024-04-08:
Which version are you using? It looks like it's Time Zone related.

spetrie13, 2024-04-08:
HA timezone set to London
Core 2024.4.1

fboundy, 2024-04-08:
Which version of PV Opt?

spetrie13, 2024-04-08:
3.13.2

spetrie13, 2024-04-08:
Grabbed a screenshot

fboundy, 2024-04-09:
3.14.0 has some bug fixes related to time zones and BST. Let me know if that fixes it

spetrie13, 2024-04-09:
Will try 3.14.0 tonight
Not sure why new Optimisation Breakdown (GBP) gives Forced Discharge as Current Setup
Base | 6.74 |  
Optimised Charging | 4.49 |  
Optimised PV Export | 4.49 |  
Forced Discharge | 4.49 | <=== Current Setup

[image]

fboundy, 2024-04-09:
It *should* be because you have use_export and forced_discharge switches on. If they aren’t then please send me the logs and I’ll have a look at what’s going on.

spetrie13, 2024-04-10:
V3.14.0 still producing same results randomly scheduling forced discharge during the cheap import period.

Some screenshots of the dashboard every few minutes:
Discharge rates are way above the inverter / battery rating of 5000W

00:18
[image]
00:20
[image]
00:23
[image]
00:25
[image]
00:30
[image]
00:45
[image]
00:51
[image]
00:58
[image]

[pv_opt(1).log](https://github.com/fboundy/pv_opt/files/14931048/pv_opt.1.log)

stevebuk1, 2024-04-10:
I'm on IOG and am having the same issue. It occurs with use_export and forced_discharge switches both off. 
Log excerpt below. 

23:35:18 INFO: Base cost: 104.5p
23:35:18 INFO: Optimised cost (Optimised Charging): 80.6p <=== Current Setup
23:35:18 INFO: Optimised cost (Optimised PV Export): 80.6p
23:35:18 INFO: Optimised cost (Forced Discharge): 80.6p
23:35:18 INFO:
23:35:18 INFO: Checking for Hold SOC slots
23:35:18 INFO:
23:35:18 INFO: Optimal forced charge/discharge slots:
23:35:18 INFO: 09-Apr 23:30 BST - 10-Apr 00:00 BST Power: -1300W SOC: 64% -> 58%
23:35:18 INFO: 10-Apr 00:00 BST - 10-Apr 00:30 BST Power: -200W SOC: 58% -> 57% <=
23:35:18 INFO: 10-Apr 00:30 BST - 10-Apr 01:00 BST Power: 100W SOC: 57% -> 57% <=
23:35:18 INFO: 10-Apr 01:00 BST - 10-Apr 05:30 BST Power: 100W SOC: 57% -> 61%

I have no export tariff (no MCS), Pv_opt correctly loads this as 0p, so for me I can't see where there would be any cost benefit in doing an export at any point.

fboundy, 2024-04-11:
I know which but of code is causing this but I'm not sure why it's happening. I've added some additional and some additional logging to `3.14.1-beta-1` if you could try that please. Please also set `debug: false` in `config.yaml`

stevebuk1, 2024-04-11:
No probs, will install tonight. If you can give me a rough area of the code
(GitHub line number)  I will take a look at it as well. For mine it seemed
to be related to high cost usage swaps coming into play for 24 hours in the
future but that didn't seem to be the issue for the other poster.

spetrie13, 2024-04-12:
3.14.1-beta-1: 
Applied this last night and the charge schedule was back to normal. No discharges scheduled during the cheap import period. Charge schedule had normal charging values.

00:30:23     INFO: Optimisation Summary
00:30:23     INFO: --------------------
00:30:23     INFO: 
00:30:23     INFO:   Base cost:                                436.9p
00:30:23     INFO:   Optimised cost (Optimised Charging):      239.7p
00:30:23     INFO:   Optimised cost (Optimised PV Export):     239.7p
00:30:23     INFO:   Optimised cost (Forced Discharge):        239.8p <=== Current Setup
00:30:23     INFO: 
00:30:23     INFO: Checking for Hold SOC slots
00:30:23     INFO: 
00:30:23     INFO: Optimal forced charge/discharge slots:
00:30:23     INFO:   12-Apr 01:30 BST - 12-Apr 02:00 BST  Power:  2900W  SOC:   58% ->   63%  
00:30:23     INFO:   12-Apr 02:00 BST - 12-Apr 04:30 BST  Power:  2900W  SOC:   63% ->   90%  
00:30:23     INFO: 
00:30:23     INFO: Plan time: 11-Apr 23:30 - 13-Apr 22:30 Initial SOC: 64.0 Base Cost: 436.9 Opt Cost: 239.8
00:30:23     INFO: 
00:30:23     INFO: Optimiser elapsed time 23.1 seconds

stevebuk1, 2024-04-12:
> Good to know. You should also have a higher battery capacity

I've installed the Beta-1 and ran it overnight, but it's been pretty sunny
in Scotland so no charging slots allocated last night and I think it will
be the same for tonight as well. I'll post logs this evening anyway.

fboundy, 2024-04-12:
I’m away for the next couple of days so maybe wait till Sunday.

closed 2024-04-15 by fboundy

stevebuk1, 2024-05-08:
I think this issue was introduced by the time change to BST, and is in this portion of code which reduces the factor if already in a half hour slot. 

`                        for slot in window:
                            if pd.Timestamp.now() > slot.tz_localize(None):
                                factors.append(
                                    (
                                        (slot.tz_localize(None) + pd.Timedelta(30, "minutes")) - pd.Timestamp.now()
                                    ).total_seconds()
                                    / 1800
                                )
                            else:
                                factors.append(1)`

This works fine in GMT but doesn't in BST. 

Times recorded in 'slot' are naive UTC, tz.localize(None) does no actual conversion to local time. Therefore, timenow needs to also be naive UTC. For this, use pd.Timestamp.utcnow().tz_localize(None), as follows:

`                        for slot in window:
                            if pd.Timestamp.utcnow().tz_localize(None) > slot.tz_localize(None):
                                factors.append(
                                    (
                                        (slot.tz_localize(None) + pd.Timedelta(30, "minutes")) - pd.Timestamp.utcnow().tz_localize(None)
                                    ).total_seconds()
                                    / 1800
                                )
                            else:
                                factors.append(1)`

Note:  There is a second problem in that I think this factoring is needed so .flows works correctly, but it is done by reducing power as recorded in "forced". This has the side effect of reducing charge power sent to the inverter unless what is requested of the slot is at least 3 times higher than the power available in that slot. I would suspect in Agile it makes little difference but is very noticeable in IOG. I'm working on a fix for this which I'll include in the EV charging work.