stevebuk1/pv_opt #212: Hold lower than 100%

SzosszeNET, 2024-04-18:
Not sure how simple or complicated it would be, but was wondering if it would be possible to add a parameter to lower the maximum battery charge level.

I do have 4.5kW solar panels but only a 3.6kW inverter. Normally it's not an issue but on really sunny days I could loose about 0.5 - 0.8 even 1+kW when the battery is full. However when the battery is not completely full but 60-70% charged only it allows the excess solar to be directed into the battery instead of being lost due to the inverter unable to export.

If possible would be great to be able to configure this headroom and not to have to set pv_opt read only and manually discharge (and set a manual discharge for very sunny days).

Would be a bonus if the "headroom" would be adjusted and only used when solcast is forecasting the production that requires this measure in place

fboundy, 2024-04-27:
Should be possible. Will add it to my to-do list.

stjohncanning, 2024-05-06:
Second this request.  Thanks.

lazy-pete, 2024-05-19:
+1 will soon have 6kw array but limited to G98 3.6kw. Inverter can charger panels and export to grid to a total of 6kw I think, so requested behaviour might help.

fboundy, 2024-05-20:
Just to clarify - this would limit the SOC that any forced charging would get to but the SOC could obviously exceed this due to solar charging

fboundy, 2024-06-02:
I had hoped to get this into 3.15.0 but I haven't had time yet. It is next on the list of things to add.

closed 2024-08-09 by github-actions[bot]