stevebuk1/pv_opt #217: Question about charging optimisation algorithm

mergwyn, 2024-04-26:
I've just swapped from agile to IOG and am trying to understand some of the optimisations now that I have got a simpler set of rates. My IOG overnight rate is 7.5p between 2330 and 0530. My export rate is 15p.

The optimisation for tonight is going to charge the battery between 0300 and 0600 ie after the end of the cheap rate.
[image]

Why does the charging period end at 0600 and not 0530?

Here is the log for one of the opmistation runs.
[pv_opt_extract.log](https://github.com/fboundy/pv_opt/files/15130880/pv_opt_extract.log)

fboundy, 2024-04-27:
The log is showing your low price as being 02:00-05:00 UTC or 03:00-06:00 BST. Let me try using your tariff code on my system to see what I get

fboundy, 2024-04-27:
Seems to load OK for me but I think I need to add some logic to Intelligent so that it repeats the following day.

fboundy, 2024-04-27:
Might be the same issue as #181. I'm about to review the PR that hopefully fixes this.

mergwyn, 2024-04-28:
I've installed 3.4.6 and I remain confused and am completely open to the possibility this is likely to be my lack off understanding!

The main question is why is a charge scheduled to start at 0600 after the cheap rate has ended? As I said above, give the disparity between the cheaper import rates and export rates I would expect the goal to have the batter fully charged at the end of the cheap rate so that all surplus including any battery capacity not needed to meet the consumption forecast can be exported during the day (ideally as late as possible before the cheap rate starts again).

I guess this could be a timing issue about when and how often the tarriff information is updated and discrepancy between the pv_opt dashboard and the octopus-energy-rates-card and this might all change closer to the time the cheap rates start.

This what I can see at the moment:
[image]

And the rates card shows:
[image]

Here's the log from the last few optimisation runs:
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15141982/pv_opt.log)

mergwyn, 2024-04-28:
Ah - more clarity!.  At about 1730 I noticed that the optimisation plan had been updated such that the std cheap rate window was covered (discharge optimisation was turned off at this point).
[image]

I then plugged the car in and was allocated a charging window of 2330-0530 and 0530-0600 though so far the optimisation plan has not been updated. I think I now understand @stevebuk1  comments in #121 about tariff refresh for IOG need to happen more frequently. The definitive truth is in the dispatching sensor from the octopus integration. Is it possible to do something when that changes. I would'nt mind of that was an automation outside of pv_opt.

The dispatching entity entries in the last log are:
```
19:20:09     INFO:  entity_id            binary_sensor.octopus_energy_**********_intelligent_dispatching
19:20:09     INFO:  state                off
19:20:09     INFO:  attributes           {'planned_dispatches': [{'start': '2024-04-28T22:00:00+00:00', 'end': '2024-04-29T04:30:00+00:00', 'charge_in_kwh': -48.1, 'source': 'smart-charge', 'location': None}, {'start': '2024-04-29T04:30:00+00:00', 'end': '2024-04-29T05:00:00+00:00', 'charge_in_kwh': -3.46, 'source': 'smart-charge', 'location': None}], 'completed_dispatches': [{'start': '2024-04-26T00:00:00+00:00', 'end': '2024-04-26T00:30:00+00:00', 'charge_in_kwh': -2.9, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-26T02:00:00+00:00', 'end': '2024-04-26T02:30:00+00:00', 'charge_in_kwh': -1.74, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-26T01:30:00+00:00', 'end': '2024-04-26T02:00:00+00:00', 'charge_in_kwh': -2.9, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-27T07:30:00+00:00', 'end': '2024-04-27T08:00:00+00:00', 'charge_in_kwh': -3.48, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-27T07:00:00+00:00', 'end': '2024-04-27T07:30:00+00:00', 'charge_in_kwh': -0.97, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-27T20:00:00+00:00', 'end': '2024-04-27T20:30:00+00:00', 'charge_in_kwh': -1.62, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-27T19:30:00+00:00', 'end': '2024-04-27T20:00:00+00:00', 'charge_in_kwh': -3.04, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-27T20:30:00+00:00', 'end': '2024-04-27T21:00:00+00:00', 'charge_in_kwh': -4.76, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-28T03:00:00+00:00', 'end': '2024-04-28T03:30:00+00:00', 'charge_in_kwh': -2.9, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-28T04:30:00+00:00', 'end': '2024-04-28T05:00:00+00:00', 'charge_in_kwh': -3.51, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-28T04:00:00+00:00', 'end': '2024-04-28T04:30:00+00:00', 'charge_in_kwh': -6.9, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-28T05:00:00+00:00', 'end': '2024-04-28T05:30:00+00:00', 'charge_in_kwh': -1.19, 'source': None, 'location': 'AT_HOME'}, {'start': '2024-04-28T16:30:00+00:00', 'end': '2024-04-28T17:00:00+00:00', 'charge_in_kwh': -0.01, 'source': None, 'location': 'AT_HOME'}], 'data_last_retrieved': '2024-04-28T18:18:19.739442+00:00', 'last_evaluated': '2024-04-28T18:19:19.739592+00:00', 'current_start': None, 'current_end': None, 'next_start': '2024-04-28T22:00:00+00:00', 'next_end': '2024-04-29T05:00:00+00:00', 'icon': 'mdi:power-plug-battery', 'friendly_name': 'Octopus Energy ********** Intelligent Dispatching'}
19:20:09     INFO:  last_changed         2024-04-28T05:30:23.739606+00:00
19:20:09     INFO:  last_updated         2024-04-28T18:19:19.740342+00:00
```

I've pretty printed this below. What is interesting is the 30 minute completed dispatches in the evenings which I don't quite understand, but I guess it is the panned dispatches which are important.
```
{
   "planned_dispatches":[
      {
         "start":"2024-04-28T22:00:00+00:00",
         "end":"2024-04-29T04:30:00+00:00",
         "charge_in_kwh":-48.1,
         "source":"smart-charge",
         "location":"None"
      },
      {
         "start":"2024-04-29T04:30:00+00:00",
         "end":"2024-04-29T05:00:00+00:00",
         "charge_in_kwh":-3.46,
         "source":"smart-charge",
         "location":"None"
      }
   ],
   "completed_dispatches":[
      {
         "start":"2024-04-26T00:00:00+00:00",
         "end":"2024-04-26T00:30:00+00:00",
         "charge_in_kwh":-2.9,
         "source":"None",
         "location":"AT_HOME"
      },
      {
         "start":"2024-04-26T02:00:00+00:00",
         "end":"2024-04-26T02:30:00+00:00",
         "charge_in_kwh":-1.74,
         "source":"None",
         "location":"AT_HOME"
      },
      {
         "start":"2024-04-26T01:30:00+00:00",
         "end":"2024-04-26T02:00:00+00:00",
         "charge_in_kwh":-2.9,
         "source":"None",
         "location":"AT_HOME"
      },
      {
         "start":"2024-04-27T07:30:00+00:00",
         "end":"2024-04-27T08:00:00+00:00",
         "charge_in_kwh":-3.48,
         "source":"None",
         "location":"AT_HOME"
      },
      {
         "start":"2024-04-27T07:00:00+00:00",
         "end":"2024-04-27T07:30:00+00:00",
         "charge_in_kwh":-0.97,
         "source":"None",
         "location":"AT_HOME"
      },
      {
[... 41 lines cut ...]
      {
         "start":"2024-04-28T05:00:00+00:00",
         "end":"2024-04-28T05:30:00+00:00",
         "charge_in_kwh":-1.19,
         "source":"None",
         "location":"AT_HOME"
      },
      {
         "start":"2024-04-28T16:30:00+00:00",
         "end":"2024-04-28T17:00:00+00:00",
         "charge_in_kwh":-0.01,
         "source":"None",
         "location":"AT_HOME"
      }
   ],
   "data_last_retrieved":"2024-04-28T18:18:19.739442+00:00",
   "last_evaluated":"2024-04-28T18:19:19.739592+00:00",
   "current_start":"None",
   "current_end":"None",
   "next_start":"2024-04-28T22:00:00+00:00",
   "next_end":"2024-04-29T05:00:00+00:00",
   "icon":"mdi:power-plug-battery",
   "friendly_name":"Octopus Energy ********** Intelligent Dispatching"
}
```

mergwyn, 2024-04-28:
I think IOG specific questions might best be covered in #121 

However, I'm still interested in understanding the strategy about when best to charge. As I said above, given the disparity between the cheaper import rates and export rates I would expect the goal to have the battery fully charged at the end of the cheap rate so that all surplus, including any battery capacity not needed to meet the consumption forecast can be exported during the day ideally as late as possible before the cheap rate starts again.

stevebuk1, 2024-04-28:
> Might be the same issue as #181. I'm about to review the PR that hopefully fixes this.

Before the PR, the tariff reloaded at midnight and loaded 27 hours of data, so by the time the first cheap charging period rolls around 23hours later there is no expensive rate, so the algorithm doesn't schedule a charge. 

The PR changes the load point to 4.30pm which means there is a peak rate period still valid at the critical point of 23:30 hours, but as per my PR notes I wasn't sure it was optimum because as we approach the end of the cheap period around 5am I hadn't checked whether the next cheap rate kicks in, and I've yet to look whether the load at 4.30pm loads 27 hours worth or something else.  As I have a zero p export rate I only use Pv_opt in Charging mode but I'd imagine for a discharge plan it will be necessary to to have a valid 24 hour lookahead for general stability and for the high cost swaps to work. 

I also recognise that on EV plugin the actual cheap/peak times are likely to change (with occasional 11pm slots at 6am slots) and so we will need a tariff reload at that point too, from something that is aware of these slots. I do note that Pv_opt does currently assign a cheap rate to when the Zappi is charging with diverted excess solar and that comes from the BottlecapDave integration which will have to be from the Electricity Meter part of BottlecapDave and not the Zappi part, noting also this is a definite bug in the BottlecapDave integration.

stevebuk1, 2024-05-14:
> I also recognise that on EV plugin the actual cheap/peak times are likely to change (with occasional 11pm slots and 6am slots) and so we will need a tariff reload at that point too, from something that is aware of these slots.

Code has been added to my Dev fork (https://github.com/stevebuk1/pv_opt/tree/dev/apps/pv_opt)  that reloads the tariff on car plugin.

closed 2024-07-22 by github-actions[bot]