stevebuk1/pv_opt #221: PV sun up pre-charging

solarjones, 2024-04-29:
Hi,
Since you kindly added the Use Solar option, its worked pretty well but think there's another step to enhance things.

Example: today was a good solar day (33kWh) however my wife decided to put the washing machine on, induction hob and double oven (cooking dinner and baking a cake) this evening which drained the batteries.  Nothing can predict this :) 

However, there are times when the tariff shows it would be beneficial to charge a bit to cover some more expensive slots before the sun is up.  Tomorrow with a good solar prediction of 29 kWh and my consumption figure set at 22 kWh has no charging plan since the PV will cover my usage. But there are some cheap and expensive slots early in the morning before the sun is up. So I've set my consumption to a greater figure (35) than the predicted PV to force a charge plan.

[image]

Tomorrows rates show this is correct, but how can PV_Opt be setup without manual intervention to cover the above scenario? 

[image]

Is there a way to take the solar prediction based on when it will start generating?

Thanks!

fboundy, 2024-05-07:
Without your log files I can't see exactly what it is calculating but my guess is that the charge to 26% is enough to get you through to when the solar kicks in. If you want to have a bit more slack in the system I would suggest using a lower confidence level for the solar forecast so it doesn't expect so much sun on variable days. I normally run mine at 30% because the downside of too little is worse that the impact of too much

solarjones, 2024-05-08:
It isn't the forecast thats not working as it's pretty good currently so 50% seems ok and dropping to 30% confidence wasn't producing the charge needed as the last few days PV output have been high anyway (nice position to be in I know!). Increasing the consumption artificially isn't great either, and would have to be so large currently as I've getting great PV output.  

It's the overlap where the battery has depleted overnight and the PV output hasn't started. Together with the Agile slots pre-dawn being cheaper than morning peak (and before sufficient PV output), think there's scope for further optimisation.  I've attempted to shape the consumption accordingly this evening but it doesn't seem to force a small charge yet, maybe I need to tweak it further. 

I've attached logs to help.  My consumption is more than covered with the PV forecast, just not for the 24 hour period as my battery storage isn't sufficient. If there's a configuration parameter option to use in conjunction with  "Use Solar", something like "Use PV window" option, the PV_Opt could then calculate the charging plan with and without the PV window option. 

[image]
[main.log](https://github.com/fboundy/pv_opt/files/15255375/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15255376/pv_opt.log)

Thanks.

solarjones, 2024-05-08:
This is the charge plan with 30% forecast:

[image]

[main.log](https://github.com/fboundy/pv_opt/files/15255451/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15255452/pv_opt.log)

closed 2024-06-19 by fboundy