stevebuk1/pv_opt #222: Optimiser frequency setting not working as expected

ElectricMolehill, 2024-05-02:
**Describe the bug**
When I set optimiser frequency to 30min the optimiser seems to be running one cycle immediately after another instead of once every 30mins as I expected

**To Reproduce**
Steps to reproduce the behavior:
Set optimiser frequency to 30mins

**Expected behavior**
Optimiser runs once every 30mins

**Additional context**
Version 3.4.6

stevebuk1, 2024-05-08:
Most (if not all) changes to any of the settings in HA will cause PV_opt to perform an optimiser run immediately. After that though, the optimiser frequency time setting should work. I've just tested mine and it appears to work as intended, noting that it is aligned to full hour or 1/2 hour boundaries. 
E.g. if you change the optimiser frequency at 21:10 you get the following:
- An immediate run at 21:10
- Another at 21:30 (as its a 1/2 hour boundary)
- Another one at 22:00 (30 mins later)
- and then every 30 minutes thereafter.

ElectricMolehill, 2024-05-09:
I tried it again and the optimiser is working OK when set to 30mins now. Not sure what I was doing to make it cycle continuously like that but it looks like user error

closed 2024-05-25 by fboundy