stevebuk1/pv_opt #223: No charging schedule set in pv_opt

punkymuzzle, 2024-05-06:
**Describe the bug**
pv_opt hasn't scheduled a charge in a couple of weeks.

**To Reproduce**
pv_opyo normal schedule

**Expected behavior**
Pv_opt should schedule charges more accurtaely.
Charging schedule:

This has been for about a week now,
Logs attached:
[error.log](https://github.com/fboundy/pv_opt/files/15227232/error.log)

punkymuzzle, 2024-05-07:
Just rechecked after the new hourly rates have landed.
Still no charging slots allocated.
Here's the tariff: 
[image]

Charging Schedule:
[image]

Here's the entry from the log:
[image]

I would expect it to charge to ate the very least cover the expensive slots between 16:00 and 19:00

Log attached fyi.
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15237864/pv_opt.log)

SzosszeNET, 2024-05-07:
@punkymuzzle , no expert on this, but I belive what you seeing is all in the logs

16:34:55     INFO: High Cost Usage Swaps
16:34:55     INFO: ---------------------
16:34:55     INFO: 
16:34:56     INFO: 08/05 06:00 BST:  0.26 kWh at   4.16p. <==> 08/05 03:00 BST: 13.63p/kWh  3.55p  SOC:  10.0%-> 10.0% New SOC:  10.0%-> 12.6% Net: -134.4
16:34:56     INFO: 08/05 06:00 BST:  0.26 kWh at   4.16p. <==> 08/05 03:00 BST: 13.63p/kWh  3.55p  SOC:  10.0%-> 12.6% New SOC:  10.0%-> 15.3% Net: -134.0
16:34:56     INFO: 08/05 06:00 BST:  0.26 kWh at   4.16p. <==> 08/05 03:00 BST: 13.63p/kWh  3.55p  SOC:  10.0%-> 15.3% New SOC:  10.0%-> 17.9% Net: -133.7
16:34:57     INFO: 08/05 06:00 BST:  0.26 kWh at   4.16p. <==> 08/05 03:00 BST: 13.63p/kWh  3.55p  SOC:  10.0%-> 17.9% New SOC:  10.0%-> 20.6% Net: -133.3
16:34:57     INFO: 08/05 06:00 BST:  0.26 kWh at   4.16p. <==> 08/05 03:00 BST: 13.63p/kWh  3.55p  SOC:  10.0%-> 20.6% New SOC:  10.0%-> 23.2% Net: -133.0
16:34:57     INFO: 08/05 06:00 BST:  0.26 kWh at   4.16p. <==> 08/05 03:00 BST: 13.63p/kWh  3.55p  SOC:  10.0%-> 23.2% New SOC:  10.0%-> 25.8% Net: -133.2
16:34:57     INFO: 08/05 06:00 BST:  0.09 kWh at   1.43p. <==> 08/05 03:00 BST: 13.63p/kWh  1.24p  SOC:  10.0%-> 27.3% New SOC:  10.0%-> 26.8% Net: -133.5
16:34:57     INFO: 08/05 06:30 BST:  0.06 kWh at   0.89p. <==> 08/05 03:00 BST: 13.63p/kWh  0.77p  SOC:  10.0%-> 29.2% New SOC:  10.0%-> 27.3% Net: -133.6
16:34:58     INFO: Charge net cost delta:  -1.3p: < Pass Threshold (4.0p) => Slots Excluded

If you would to change your "Pass threshold" to 1 it would help, but maybe not costs-wise. The other thing I'd look at that potentially solcast is also playing a role of predicting enough generation to not trigger charge - if you flick "Use Solar" off, I'm sure there will be charging scheduled for the cheap period - otherwise I'd expect that due to your solar forcast the expectation would be that it will be charged from the panels as not cheaper to import?

SzosszeNET, 2024-05-07:
Besides your cost summary is negative due to your exports?

FO: Optimisation Summary
12:00:31     INFO: --------------------
12:00:31     INFO: 
12:00:31     INFO:   Base cost:                               -136.8p
12:00:32     INFO:   Optimised cost (Optimised Charging):     -136.8p
12:00:32     INFO:   Optimised cost (Optimised PV Export):    -136.8p <=== Current Setup
12:00:32     INFO:   Optimised cost (Forced Discharge):       -136.8p
12:00:32     INFO: 
12:00:32     INFO: No charging slots
12:00:32     INFO: 
12:00:32     INFO: Plan time: 07-May 11:00 - 08-May 23:30 Initial SOC: 26.0 Base Cost: -136.8 Opt Cost: -136.8

So yeah, maybe play with the pass treshold that that might help here but not sure if it would make a massive difference?

punkymuzzle, 2024-05-07:
Thanks both, much appreciated.
If it's not just me that's experiencing this then that gives me a little more confidence that it's working as it should.

Does pv_opt keep recalculating based on solar prediction as well as SOC?
The reason I'm asking is that it was quite cloudy here for most of the day so by 13:00 my SOC was still only at about 30%, so I had to intervene and set a charge before the peak period of 16:00 - 19:00.
Probably my misunderstanding but I thought that pv_opt would have calculated that I needed a charge to get up to 100% SoC?

Thanks

punkymuzzle, 2024-05-07:
> If you would to change your "Pass threshold" to 1 it would help, but maybe not costs-wise. The other thing I'd look at that potentially solcast is also playing a role of predicting enough generation [...]

Thanks for this.  I did turn off the Use Solar" option and yes it did recalculate and provide me with charging slots, so it looks like it's actually working.
[image]

However, it's recommending charging slots overnight to get the SoC high enough to cover the more expensive 0600 - 1000 slot (which I feel is right).  But I would have though that it would do this anyway, even if the "Use Solar" was enabled?
Maybe when you take the cost to charge and the inefficiencies for the round trip in the electricity, it might not be worth it if there is some solar at 0600.  Not sure....

It's a bit of a moot point at the moment though I guess, as I still have an issue with my inverter going idle at 10%, so pv_opt can't write any charging schedule to it until the inverter comes out of idle (SoC > 10% due to solar charging or me setting a charge via the app).

Thanks

SzosszeNET, 2024-05-07:
Hmm, okay - I played with the parameters a bit - looks like that the solcast confidence is producing something unexpted.

with 40% confidence it appeared have scheduled charge and discharge slots

23:12:03     INFO: Initial SOC: 77.5
23:12:03     INFO: Calculating Base flows:
23:12:03     INFO: 
23:12:03     INFO: Optimising for Solcast 40% confidence level forecast from 07-May 22:00 UTC to 08-May 23:30 UTC
23:12:03     INFO: Ignoring export pricing because Use Export is turned off
23:12:05     INFO: Optimiser prices loaded for period 07/05 22:00 UTC - 08/05 23:30 UTC
23:12:05     INFO: Base cost:  44.0
23:12:05     INFO: 
23:12:05     INFO: Agile Plunge Pricing
23:12:05     INFO: --------------------
23:12:05     INFO: 
23:12:05     INFO: Plunge cost: 44.0
23:12:05     INFO: 
23:12:05     INFO: High Cost Usage Swaps
23:12:05     INFO: ---------------------
23:12:05     INFO: 
23:12:05     INFO: Charge net cost delta:  0.0p: < Pass Threshold (4.0p) => Slots Excluded
23:12:05     INFO: 
23:12:05     INFO: Low Cost Charging
23:12:05     INFO: ------------------
23:12:05     INFO: 
23:12:05     INFO: Max export price when there is no forced charge: 15.00p/kWh.
23:12:05     INFO: 24 slots have an import price less than the max export price
23:12:05     INFO: 23 Min import price 13.01p/kWh at 08/05 13:30 UTC    0W   SOC:  85.9%-> 89.6% Net:  42.8 New SOC:  85.9%-> 95.3% Max export: 967W 
23:12:05     INFO: 14 Min import price 14.31p/kWh at 08/05 03:30 UTC    0W   SOC:  41.2%-> 37.2% Net:  41.7 New SOC:  41.2%-> 57.6% Max export: 1377W 
23:12:05     INFO: 13 Min import price 14.35p/kWh at 08/05 00:30 UTC    0W   SOC:  63.4%-> 59.0% Net:  39.5 New SOC:  63.4%-> 79.7% Max export: 1477W 
23:12:06     INFO: 12 Min import price 14.35p/kWh at 08/05 01:00 UTC    0W   SOC:  79.7%-> 75.8% Net:  37.3 New SOC:  79.7%-> 96.1% Max export: 1552W 
23:12:06     INFO:  9 Min import price 14.35p/kWh at 08/05 04:00 UTC    0W   SOC:  98.6%-> 94.7% Net:  36.2 New SOC:  98.6%->100.0% Max export: 1552W 
23:12:06     INFO: 
23:12:06     INFO: Charge net cost delta:  7.8p: > Pass Threshold 4.0p => Slots Included
23:12:06     INFO: 
23:12:06     INFO: Forced Discharging
23:12:06     INFO: ------------------
23:12:06     INFO: 
23:12:06     INFO: 47 slots have an export price greater than the min import price
23:12:07     INFO: 19 Max export price 15.00p/kWh at 08/05 14:00 UTC   SOC: 100.0%->100.0% Net:  32.1 New SOC: 100.0%-> 81.4% Max export: 4958W 
23:12:07     INFO: 18 Max export price 15.00p/kWh at 08/05 14:30 UTC   SOC:  81.4%-> 85.8% Net:  -2.2 New SOC:  81.4%-> 62.9% Max export: 4958W 
23:12:07     INFO: 17 Max export price 15.00p/kWh at 08/05 15:00 UTC   SOC:  62.9%-> 67.2% Net: -36.2 New SOC:  62.9%-> 44.3% Max export: 4958W 
23:12:08     INFO:  2 Max export price 15.00p/kWh at 08/05 22:30 UTC   SOC:  23.8%-> 21.9% Net: -37.2 New SOC:  23.8%-> 15.0% Max export: 4958W 
23:12:08     INFO: 
23:12:08     INFO: Discharge net cost delta: 73.4p: > Discharge Threshold (5.0p) => Slots included
23:12:08     INFO: Iteration  1: Slots added:   9

But at 30% it's only dischargning?

23:13:07     INFO: Initial SOC: 77.5
23:13:07     INFO: Calculating Base flows:
23:13:07     INFO: 
23:13:07     INFO: Optimising for Solcast 30% confidence level forecast from 07-May 22:00 UTC to 08-May 23:30 UTC
23:13:07     INFO: Ignoring export pricing because Use Export is turned off
23:13:08     INFO: Optimiser prices loaded for period 07/05 22:00 UTC - 08/05 23:30 UTC
23:13:08     INFO: Base cost:  60.0
23:13:08     INFO: 
23:13:08     INFO: Agile Plunge Pricing
23:13:08     INFO: --------------------
23:13:08     INFO: 
23:13:08     INFO: Plunge cost: 60.0
23:13:08     INFO: 
23:13:08     INFO: High Cost Usage Swaps
23:13:08     INFO: ---------------------
23:13:08     INFO: 
23:13:08     INFO: Charge net cost delta:  0.0p: < Pass Threshold (4.0p) => Slots Excluded
23:13:08     INFO: 
23:13:08     INFO: Low Cost Charging
23:13:08     INFO: ------------------
23:13:08     INFO: 
[... 29 lines cut ...]
23:13:12     INFO: 12 Min import price 14.31p/kWh at 08/05 02:00 UTC    0W   SOC:  35.5%-> 32.3% Net: -14.6 New SOC:  35.5%-> 51.9% Max export: 3100W 
23:13:13     INFO: 
23:13:13     INFO: Charge net cost delta:  1.6p: < Pass Threshold 4.0p => Slots Excluded
23:13:13     INFO: 
23:13:13     INFO: Forced Discharging
23:13:13     INFO: ------------------
23:13:13     INFO: 
23:13:13     INFO: 40 slots have an export price greater than the min import price
23:13:15     INFO: 
23:13:15     INFO: Discharge net cost delta: -0.0p: < Discharge threshold (5.0p) => Slots excluded
23:13:15     INFO: Iteration  2: Slots added:   0
23:13:15     INFO: 
23:13:15     INFO: Removing cyclic charge/discharge
23:13:15     INFO:   Net cost revised from -13.0p to -13.0p
23:13:15     INFO: 
23:13:15     INFO: Optimisation Summary
23:13:15     INFO: --------------------
23:13:15     INFO: 
23:13:15     INFO:   Base cost:                                 60.0p
23:13:15     INFO:   Optimised cost (Optimised Charging):       60.0p
23:13:15     INFO:   Optimised cost (Optimised PV Export):      60.0p
23:13:15     INFO:   Optimised cost (Forced Discharge):        -13.0p <=== Current Setup
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15241615/pv_opt.log)

But again messing with the load history 7 vs 14 days does seem to produce the same charge / discharge only scenarios.

punkymuzzle, 2024-05-08:
Thanks for the input. Think this can be closed now I know it's working OK

closed 2024-05-08 by punkymuzzle