stevebuk1/pv_opt #224: Static Solcast data

lazy-pete, 2024-05-08:
**Describe the bug**
A clear and concise description of what the bug is.

The Solcast prediction seems to be ‘stuck’ in my dashboard since approx 0900 today. On the Solcast site the API’s are all running fine. 

Can’t work out how to download the log files from the Mobile HA app, so will grab them this evening. 

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

fboundy, 2024-05-08:
Have you tried re-starting the Solcast integration? The Solcast data comes direct from the Solcast integration entities. Do the attributes there look correct?

punkymuzzle, 2024-05-08:
@lazy-pete , just checking - is your automation to update the forecast still running?
Might be worth checking.

Also, if you also use the "Energy" tab in HA with forecast enabled, that's also a good sign it's working.
The dotted line in this picture is the solcast information:
[image]

lazy-pete, 2024-05-08:
Thanks - tried re-loading it to no avail. Been getting odd / static values
from Solcast since about 0815.

Also says the API was last polled on 1/5 , which seems odd since there’s
data from this morning and it was fine up until today.

I’ll maybe try deleting/ reading the integration.

Cheers,

Pete

fboundy, 2024-05-08:
PV_Opt reads directly from the Solcast sensors’ detailedForecast attributes so if these are frozen there is nothing else it can do

lazy-pete, 2024-05-08:
Thanks - the update automation has vanished !

Must’ve deleted it accidentally when doing some housekeeping and this
morning was when it ran out of data.

Will get the instructor re-add from the Solcast docs.

Thanks !

Can close this one

P

punkymuzzle, 2024-05-08:
Have you checked the last time the automation ran to update the Solcast data?
Suggest checking this first.
Here's mine if it helps:

> 
> alias: Solcast Forecast Update at 16:00
> description: Solcast Forecast Update at 16:00
> trigger:
>   - platform: time
>     at: "16:00:00"
> condition: []
> action:
>   - service: solcast_solar.update_forecasts
>     metadata: {}
>     data: {}
> mode: single

lazy-pete, 2024-05-08:
Thanks - I must’ve completely deleted the Automation as it was not there (I
did delete a few other redundant ones recently so will have got mixed up).

I recreated it and ran it manually to start - seems to be working normally
now. I’ll keep an eye on the automation and check it does automate !

By the by, I also seem to have solved the Solax disconnection errors - the
various things added to my garage wiring (inverter etc) seem to have
drastically reduced the speed of the power line backhaul WiFi extender in
the garage. I moved the socket which doubled the speed (though still
terrible) but seems to have been enough to stop the Elfin dropping off the
network. Either I plugged something in  that killed the network or the
Solax integration / HA Core changed its tolerance to disconnections. Solid
for a few days now.

Thanks again,

Pete

punkymuzzle, 2024-05-08:
Awesome, glad you got it sorted.

closed 2024-05-20 by fboundy