stevebuk1/pv_opt #226: Timezone error stopping pv_opt from starting

swests, 2024-05-15:
**Describe the bug**
Upgraded to newest version (HA, PV_OPT, Octopus Tarriff, and pv_opt won't start - getting a timezone inconsistency error

**Additional context**
output from AppDaemon error.log after restart

`13:49:26 WARNING pv_opt: Traceback (most recent call last): File "/usr/lib/python3.11/site-packages/pandas/core/arrays/datetimes.py", line 2646, in _infer_tz_from_endpoints inferred_tz = timezones.infer_tzinfo(start, end) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "timezones.pyx", line 369, in pandas._libs.tslibs.timezones.infer_tzinfo AssertionError: Inputs must both have the same timezone, UTC != UTC+01:00 The above exception was the direct cause of the following exception: Traceback (most recent call last): File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 162, in initialize_app await utils.run_in_executor(self, init) File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor response = future.result() ^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run result = self.fn(*self.args, **self.kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 503, in initialize self.optimise() File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock return f(*args, **kw) ^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 1534, in optimise if self._check_tariffs(): ^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 927, in _check_tariffs self._check_tariffs_vs_bottlecap() File "/homeassistant/appdaemon/apps/pv_opt/pv_opt.py", line 2607, in _check_tariffs_vs_bottlecap self.contract.tariffs[direction].to_df(start=df.index[0], end=df.index[-1], day_ahead=False)[ ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/appdaemon/apps/pv_opt/pvpy.py", line 246, in to_df newindex = pd.date_range(df.index[0], end, freq="30min") ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/site-packages/pandas/core/indexes/datetimes.py", line 1008, in date_range dtarr = DatetimeArray._generate_range( ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/site-packages/pandas/core/arrays/datetimes.py", line 445, in _generate_range tz = _infer_tz_from_endpoints(start, end, tz) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/site-packages/pandas/core/arrays/datetimes.py", line 2649, in _infer_tz_from_endpoints raise TypeError( TypeError: Start and end cannot both be tz-aware with different timezones`

swests, 2024-05-15:
From pv_opt.log

```
13:49:13 INFO: Time Zone Offset: 60.0 minutes
--
13:49:13 INFO: Local timezone set to GB
13:49:13 INFO:
13:49:13 INFO: ******************* PV Opt v3.14.0 *******************

```

swests, 2024-05-15:
Checked Octopus integration, and it's returning as BST too...

`state_class: total
mpan: "REMOVED"
serial_number: REMOVED
is_export: false
is_smart_meter: true
tariff: E-1R-AGILE-23-12-06-H
start: "2024-05-15T16:00:00+01:00"
end: "2024-05-15T16:30:00+01:00"
is_capped: false
is_intelligent_adjusted: false
current_day_min_rate: 0.130725
current_day_max_rate: 0.31941
current_day_average_rate: 0.175464
data_last_retrieved: "2024-05-15T16:06:23.857247+01:00"
last_evaluated: "2024-05-15T16:16:23.857499+01:00"
unit_of_measurement: GBP/kWh
device_class: monetary
icon: mdi:currency-gbp
friendly_name: Octopus Electric Current Rate`

swests, 2024-05-15:
Rolled back to:
HA: 2024.4.4
PV_OPT: 3.14.0
Octopus: 10.3.1

And back working.

fboundy, 2024-05-16:
This is possibly because the OE integration now reports in GB tx rather than UTC. I’ve just updated all mine to the latest versions so will inverstigate.

fboundy, 2024-05-25:
Hi - afraid I haven't been able to replicate this. Are you able to reproduce it and send me all the logs please?

fboundy, 2024-06-02:
Hi - any update? Is this still happening or can I close this issue? Thanks

swests, 2024-06-03:
Confirmed working with
PV Opt: 3.14.9
OE: 11.0.2
HA: 2024.5.5

Thanks

closed 2024-06-03 by swests