stevebuk1/pv_opt #227: Unexpected error running initialize() for pv_opt

solarjones, 2024-05-21:
v3.14.7

Hi, I updated to this above version and  getting an error after starting: Unexpected error running initialize() for pv_opt

[error.log](https://github.com/fboundy/pv_opt/files/15395712/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/15395713/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15395714/pv_opt.log)

Thanks.

fboundy, 2024-05-22:
Should be fixed in 3.14.8

solarjones, 2024-05-23:
Thanks, that fixed it.

closed 2024-05-25 by fboundy

solarjones, 2024-05-26:
3.14.9
Error came back...

[error.log](https://github.com/fboundy/pv_opt/files/15446825/error.log)
[main.log](https://github.com/fboundy/pv_opt/files/15446826/main.log)
[pv_opt.log](https://github.com/fboundy/pv_opt/files/15446827/pv_opt.log)